networkupstools/nut #2235: Modbus driver on Proxmox

Madfox8, 2023-12-14:
Hello, 

I've followed this tutorial (https://www.kreaweb.be/diy-home-server-2021-software-proxmox-ups/) to install NUT server on my Proxmox machine running 8.1.3. However with this method UPS load is not reported. Load is reported if I use apcupsd modbus, but I would like to use the nut network for staggered shutdown.

My UPS is a ACP smart-ups 750 230v. It can report load over power via modbus, but not with the suggested driver in NUT-server.
I found there is a modbus driver via: https://networkupstools.org/docs/man/apc_modbus.html

The server works without the load reporting when `driver = usbhid-ups ` is set. However, if I want to use `driver = usbhid-ups` the following error is given: 
``root@pve:/lib/nut# upsdrvctl start
Network UPS Tools - UPS driver controller 2.8.0
Can't start /lib/nut/generic_modbus: No such file or directory``

How can I get the right driver to load in proxmox?

jimklimov, 2023-12-14:
Looks like either your setup or their packaging (inherited from Debian, CC @bigon ?) is mixed up: it may not have updated the recipes to install the drivers build against `libmodbus`, or you needed to request it as a package dependency.

Also, with `driver=usbhid-ups` (if the second mention is not a typo) it should not be loading a `generic_modbus`.

Anyhow, `generic_modbus` is not `apc_modbus`, which in turn is part of NUT since v2.8.1. However, for USB-capable builds you would currently need a custom build of `libmodbus` and then of NUT against it: https://github.com/networkupstools/nut/wiki/APC-UPS-with-Modbus-protocol

Madfox8, 2023-12-15:
I was testing various things, including the generic_modbus driver. my mistake. The resulting output however was the same.
thanks for the instructions, i'm going to try this, maybe in a vm instead of proxmox then.

jimklimov, 2023-12-15:
At some point I did test USB pass-through for UPS connections, it seemed to work :) You may have to ensure the host OS does not barge into it however (e.g. does not have its copy of NUT driver running).

Also note that it can help for testing; but for ultimate production use, with integration to turn off the UPS late in emergency shutdown, a VM might not be the right spot to send such a command - its host may need some more minutes to complete other shutdown rituals.

Madfox8, 2023-12-15:
ah thanks for pointing that out... in that case i'm going the route of another system altogether (I think a raspberry pi running raspbian lite). I'll check in sometime later when proxmox might have a more workable update available.

jimklimov, 2023-12-15:
As a Linux, it is perfectly capable of building and running a custom NUT, got one such setup "dogfeeding" in the NUT CI farm with a master-branch build protecting the PVE host.

TehH4rRy, 2025-06-26:
Did you get this working? I've just got into Proxmox, homeassistant and NUT. I have the same UPS, can only seem to install 2.8.0 through apt, nor get it to use the apc_modbus driver. I know it's an old issue but I'd kill for some help

TehH4rRy, 2025-07-02:
> Modbus drivers are, in debian, in an other package. You need to install `nut-modbus`

Do I still need to recompile NUT and my own libmodbus? I installed nut-modbus, tried changing the driver but I still get Can't start /lib/nut/apc_modbus: No such file or directory. I knew it wouldn't be that simple. Or could it be because the version of NUT on apt is only 2.8.0?

dohdoh64, 2026-10-15:
@TehH4rRy @Madfox8 I know this is a bit late, but I've just figured out how to get NUT built with custom libmodbus to work on proxmox, and I've made some scripts that automate building it in a VM. Feel free to check it out [here](https://github.com/dohdoh64/nut-custom-setup-proxmox), use at your own peril though, I'm not very experienced with scripting in bash/shell.

jimklimov, 2026-10-17:
@dohdoh64 : actually, looks cool, especially the part in https://github.com/dohdoh64/nut-custom-setup-proxmox/blob/main/setup-custom-nut.sh which looks like a Dockerfile build with a proxmox temporary container.

Regarding direct use of `upsdrvctl`, `upsd` etc. programs - you may be better served by systemd units shipped by NUT.

Regarding `/etc/udev/rules.d/99-usb-permissions.rules` manipulation, maybe copying/removing the file built along with NUT could be better (although it involves all other known USB IDs, to become owned by `nut:nut` in your case).