networkupstools/nut #2237: After firmware upgrade snmp-ups unable to get values from APC SMART UPS network card AOS AP9617

guega2GitHub, 2023-12-18:
I am new to the UPS and SNMP
Old firmware for APC network card worked with snmp v1 but snmp v3 was unavailable (timeout)
New firmware provides snmp v3 support and works with snmpwalk (see below) but oids look different from examples seen on github

ups.conf:

[apcv3]
     driver = snmp-ups
     mibs = apcc
     port = 192.168.100.220
     snmp_version = v3
     community = v3private
     secLevel = authPriv
     secName = v3private2
     authProtocol = MD5
     authPassword = <pw>
     privProtocol = DES
     privPassword = <pw>
     desc = "APC SNMP v3 device"

---

snmp-ups -DD -a apcv3 

Network UPS Tools - Generic SNMP UPS driver 1.30 (2.8.1)
   0.000000     [D1] Network UPS Tools version 2.8.1 (release/snapshot of 2.8.1) built with gcc (Debian 10.2.1-6) 10.2.1 20210110 and configured with flags: --prefix=/usrbin/nut --with-user=nut --with-group=nut --with-all --sysconfdir=/etc/nut --prefix=/usrbin/nut --with-user=nut --with-group=nut --with-all --with-net-snmp-config --prefix=/usrbin/nut --with-user=nut --with-group=nut --with-all --enable-inplace-runtime='2.8.1-256-gec3df0ef4 (release/snapshot of 2.8.1.1) built with gcc (Debian 10.2.1-6) 10.2.1 20210110 and'
   0.000036     [D1] debug level is '2'
   0.000465     [D1] Succeeded to become_user(nut): now UID=124 GID=130
   0.000526     [D1] SNMP UPS driver: entering upsdrv_initups()
   0.000552     [D2] SNMP UPS driver: entering nut_snmp_init(snmp-ups)
   0.001877     [D2] Setting SNMP retries to 5
   0.001887     [D2] Setting SNMP timeout to 1 second(s)
   0.022379     [D1] SNMP UPS driver: entering load_mib2nut(apcc) to detect proper MIB for device [apcv3] (host 192.168.100.220:161)
   0.022432     [D2] load_mib2nut: skip the "apc_ats" entry from the mapping table which is not "apcc" (and which in turn is not "auto")
   0.022445     [D2] load_mib2nut: skip the "apc_pdu" entry from the mapping table which is not "apcc" (and which in turn is not "auto")
   0.022456     [D2] load_mib2nut: skip the "apc_pdu" entry from the mapping table which is not "apcc" (and which in turn is not "auto")
   0.022466     [D2] load_mib2nut: skip the "apc_pdu" entry from the mapping table which is not "apcc" (and which in turn is not "auto")
   0.022477     [D2] load_mib2nut: skip the "apc" entry from the mapping table which is not "apcc" (and which in turn is not "auto")
   0.022488     [D2] load_mib2nut: trying classic sysOID matching method with 'apcc' mib
   0.022503     [D2] Found entry, not a template .1.3.6.1.4.1.318.1.1.1.1.1.1.0
   0.022512     [D2] Testing ups.model using OID .1.3.6.1.4.1.318.1.1.1.1.1.1.0
   0.316059     [apcv3] Warning: type error exception (OID = .1.3.6.1.4.1.318.1.1.1.1.1.1.0)
   0.316112     [D2] load_mib2nut: skip the "baytech" entry from the mapping table which is not "apcc" (and which in turn is not "auto")
   0.316122     [D2] load_mib2nut: skip the "bestpower" entry from the mapping table which is not "apcc" (and which in turn is not "auto")
   0.316130     [D2] load_mib2nut: skip the "cpqpower" entry from the mapping table which is not "apcc" (and which in turn is not "auto")
   0.316137     [D2] load_mib2nut: skip the "cyberpower" entry from the mapping table which is not "apcc" (and which in turn is not "auto")
   0.316145     [D2] load_mib2nut: skip the "cyberpower" entry from the mapping table which is not "apcc" (and which in turn is not "auto")
   0.316153     [D2] load_mib2nut: skip the "delta_ups" entry from the mapping table which is not "apcc" (and which in turn is not "auto")
   0.316160     [D2] load_mib2nut: skip the "eaton_ats16_nmc" entry from the mapping table which is not "apcc" (and which in turn is not "auto")
   0.316168     [D2] load_mib2nut: skip the "eaton_ats16_nm2" entry from the mapping table which is not "apcc" (and which in turn is not "auto")
   0.316176     [D2] load_mib2nut: skip the "eaton_ats30" entry from the mapping table which is not "apcc" (and which in turn is not "auto")
   0.316183     [D2] load_mib2nut: skip the "eaton_epdu" entry from the mapping table which is not "apcc" (and which in turn is not "auto")
   0.316191     [D2] load_mib2nut: skip the "eaton_pdu_nlogic" entry from the mapping table which is not "apcc" (and which in turn is not "auto")
   0.316199     [D2] load_mib2nut: skip the "eaton_pxg_ups" entry from the mapping table which is not "apcc" (and which in turn is not "auto")
   0.316207     [D2] load_mib2nut: skip the "eaton_pw_nm2" entry from the mapping table which is not "apcc" (and which in turn is not "auto")
   0.316215     [D2] load_mib2nut: skip the "emerson_avocent_pdu" entry from the mapping table which is not "apcc" (and which in turn is not "auto")
   0.316222     [D2] load_mib2nut: skip the "aphel_revelation" entry from the mapping table which is not "apcc" (and which in turn is not "auto")
   0.316230     [D2] load_mib2nut: skip the "aphel_genesisII" entry from the mapping table which is not "apcc" (and which in turn is not "auto")
   0.316237     [D2] load_mib2nut: skip the "pulizzi_switched1" entry from the mapping table which is not "apcc" (and which in turn is not "auto")
   0.316245     [D2] load_mib2nut: skip the "pulizzi_switched2" entry from the mapping table which is not "apcc" (and which in turn is not "auto")
   0.316253     [D2] load_mib2nut: skip the "hpe_epdu" entry from the mapping table which is not "apcc" (and which in turn is not "auto")
   0.316261     [D2] load_mib2nut: skip the "hpe_pdu3_cis" entry from the mapping table which is not "apcc" (and which in turn is not "auto")
   0.316269     [D2] load_mib2nut: skip the "huawei" entry from the mapping table which is not "apcc" (and which in turn is not "auto")
   0.316276     [D2] load_mib2nut: skip the "mge" entry from the mapping table which is not "apcc" (and which in turn is not "auto")
   0.316284     [D2] load_mib2nut: skip the "netvision" entry from the mapping table which is not "apcc" (and which in turn is not "auto")
   0.316304     [D2] load_mib2nut: skip the "raritan" entry from the mapping table which is not "apcc" (and which in turn is not "auto")
   0.316313     [D2] load_mib2nut: skip the "raritan-px2" entry from the mapping table which is not "apcc" (and which in turn is not "auto")
   0.316321     [D2] load_mib2nut: skip the "xppc" entry from the mapping table which is not "apcc" (and which in turn is not "auto")
   0.316329     [D2] load_mib2nut: skip the "tripplite" entry from the mapping table which is not "apcc" (and which in turn is not "auto")
   0.316337     [D2] load_mib2nut: skip the "ietf" entry from the mapping table which is not "apcc" (and which in turn is not "auto")
[... 537 lines cut ...]
iso.3.6.1.6.3.16.1.4.1.9.7.103.114.111.117.112.32.49.0.3.1 = INTEGER: 1
iso.3.6.1.6.3.16.1.4.1.9.7.103.114.111.117.112.32.50.0.1.1 = INTEGER: 1
iso.3.6.1.6.3.16.1.4.1.9.7.103.114.111.117.112.32.50.0.2.1 = INTEGER: 1
iso.3.6.1.6.3.16.1.4.1.9.7.103.114.111.117.112.32.50.0.3.1 = INTEGER: 1
iso.3.6.1.6.3.16.1.5.1.0 = INTEGER: -1837363664
iso.3.6.1.6.3.16.1.5.2.1.3.6.97.108.108.109.105.98.4.1.3.6.1 = Hex-STRING: FC 00 
iso.3.6.1.6.3.16.1.5.2.1.4.6.97.108.108.109.105.98.4.1.3.6.1 = INTEGER: 1
iso.3.6.1.6.3.16.1.5.2.1.5.6.97.108.108.109.105.98.4.1.3.6.1 = INTEGER: 4
iso.3.6.1.6.3.16.1.5.2.1.6.6.97.108.108.109.105.98.4.1.3.6.1 = INTEGER: 1
iso.3.6.1.6.3.18.1.1.1.2.112.114.105.118.97.116.101.99.111.109 = STRING: "private"
iso.3.6.1.6.3.18.1.1.1.2.112.117.98.108.105.99.99.111.109 = STRING: "public"
iso.3.6.1.6.3.18.1.1.1.3.112.114.105.118.97.116.101.99.111.109 = STRING: "privateuser"
iso.3.6.1.6.3.18.1.1.1.3.112.117.98.108.105.99.99.111.109 = STRING: "publicuser"
iso.3.6.1.6.3.18.1.1.1.4.112.114.105.118.97.116.101.99.111.109 = Hex-STRING: 80 00 01 3E 03 00 C0 B7 7F 30 9B 
iso.3.6.1.6.3.18.1.1.1.4.112.117.98.108.105.99.99.111.109 = Hex-STRING: 80 00 01 3E 03 00 C0 B7 7F 30 9B 
iso.3.6.1.6.3.18.1.1.1.5.112.114.105.118.97.116.101.99.111.109 = ""
iso.3.6.1.6.3.18.1.1.1.5.112.117.98.108.105.99.99.111.109 = ""
iso.3.6.1.6.3.18.1.1.1.6.112.114.105.118.97.116.101.99.111.109 = ""
iso.3.6.1.6.3.18.1.1.1.6.112.117.98.108.105.99.99.111.109 = ""
iso.3.6.1.6.3.18.1.1.1.7.112.114.105.118.97.116.101.99.111.109 = INTEGER: 3
iso.3.6.1.6.3.18.1.1.1.7.112.117.98.108.105.99.99.111.109 = INTEGER: 3
iso.3.6.1.6.3.18.1.1.1.8.112.114.105.118.97.116.101.99.111.109 = INTEGER: 1
iso.3.6.1.6.3.18.1.1.1.8.112.117.98.108.105.99.99.111.109 = INTEGER: 1
iso.3.6.1.6.3.18.1.1.1.8.112.117.98.108.105.99.99.111.109 = No more variables left in this MIB View (It is past the end of the MIB tree)
----------------------------------------------------------------------

jimklimov, 2023-12-19:
Seems you have a typo:

````
0.022477 [D2] load_mib2nut: skip the "apc" entry from the mapping table which is not "apcc" (and which in turn is not "auto")
````

guega2GitHub, 2023-12-19:
Hm, this is from snmp-ups -DD output using my [apcv3] config. Nothing *I* wrote ... (hence whose typo is it)

jimklimov, 2023-12-19:
Try `mibs=apc` (not `apcc`).

guega2GitHub, 2023-12-19:
with mibs = apc the result is still:

root@db3:/download/nut# snmp-ups -DD -a apcv3 
Network UPS Tools - Generic SNMP UPS driver 1.30 (2.8.1)
...
0.323917     Requested 'mibs' value 'apc' did not match this device [apcv3] (host 192.168.100.220)
0.324026     upsnotify: failed to notify about state 4: no notification tech defined, will not spam more about it

---------------------

root@db3:/etc/nut# systemctl status nut-driver
* nut-driver.service - Network UPS Tools - power device driver controller
     Loaded: loaded (/lib/systemd/system/nut-driver.service; static)
     Active: failed (Result: exit-code) since Tue 2023-12-19 20:34:57 CET; 1min 28s ago
    Process: 2710515 ExecStart=/sbin/upsdrvctl start (code=exited, status=1/FAILURE)
        CPU: 27ms

Dec 19 20:34:57 db3 systemd[1]: Starting Network UPS Tools - power device driver controller...
Dec 19 20:34:57 db3 upsdrvctl[2710516]: [apcv3] Warning: type error exception (OID = .1.1.30.2.1.1.4.1)
Dec 19 20:34:57 db3 upsdrvctl[2710516]: Requested 'mibs' value 'apc' did not match this device [apcv3] (host 192.168.100.220)
Dec 19 20:34:57 db3 upsdrvctl[2710516]: upsnotify: failed to notify about state 4: no notification tech defined, will not spam more about it
Dec 19 20:34:57 db3 upsdrvctl[2710516]: Network UPS Tools - Generic SNMP UPS driver 1.30 (2.8.1)
Dec 19 20:34:57 db3 upsdrvctl[2710515]: Driver failed to start (exit status=1)
Dec 19 20:34:57 db3 upsdrvctl[2710515]: Network UPS Tools - UPS driver controller 2.8.1
Dec 19 20:34:57 db3 systemd[1]: nut-driver.service: Control process exited, code=exited, status=1/FAILURE
Dec 19 20:34:57 db3 systemd[1]: nut-driver.service: Failed with result 'exit-code'.
Dec 19 20:34:57 db3 systemd[1]: Failed to start Network UPS Tools - power device driver controller.

---------------------

The problem is still that an base OID of **.iso.3.6.1.4.1.318** still fails on the device.

snmpv3walk   **iso.3.6.1.2.1.1.5.0** = STRING: "APC_SMART_UPS"

jimklimov, 2023-12-20:
Oh, so naybe we do have both `apc` and `apcc` mappings, per earlier "screenshot". I'm in mountains now so not reading too well.

Anyhow, you can try to remove (comment away) the `mibs=...` line and so have the driver pick each mapping it already knows. As a fallback, `ietf` would likely match when tried last.

If no vendor-specific subdriver fits, docs (and scripts in NUT sources) elaborate on walking the device to generate sources for a new one.

guega2GitHub, 2023-12-20:
That's what I did anyway (commenting the mibs line). Didn't work either.
I suspect the APC USP controller card AP9617 firmware update was somehow incomplete (left me without https gui and APC MIB bits). It's old hw (2006). Right now I'm trying to contact my German supplier to get some info on that firmware update.
Thank you for your comments. Have a nice stay in the mountains and keep away from the wolves.

jimklimov, 2023-12-20:
Would an explicit `mibs=ietf` help?

Also just in case try snmp v1/v2c with it - maybe the earlier "lack of
response" was because it tried a sysOID not supported by this firmware?
With UDP (no TCP-like session to establish and break off) it is nearly
impossible to differentiate lack of answer from error from bad credentials
etc.

jimklimov, 2023-12-20:
CC @aquette - maybe similar to #734 ?

guega2GitHub, 2023-12-20:
Found the solution (I hope). The firmware upgrade utility didn't do its job.
The USP firmware part was not installed -  for whatever reason?
When old hardware meets a self made RS232 cable?
I renamed apc_hw02_aos394_sumx393.exe to apc_hw02_aos394_sumx393.zip.
I Compared the content of the driver card FTP listing to the files available in this EXEC/ZIP file. 
The I put the the non existent file apc_hw02_sumx_393.bin from the ZIP file to the controller and voila iso.3.6.1.4.1.318  became a readable OID.
Now snmp-ups mibs=apcc does what it should.
Saved me from trying to create a new sub-driver ...
Thank you and everybody on the NUT team for your work and support!

aquette, 2023-12-21:
I indeed remember that APC was using specific serial wiring and selling not cheap their cables...
Glad to hear it's fixed