networkupstools/nut #2241: Communication lost after self test. Worked well some weeks ago...

sblantipodi, 2023-12-22:
Hi all...
I build that nut upsd from the sources using the latest master branch.

I have a Cyberpower_PR1500ELCD and it worked well with latest sources...

Recently I updated the sources and rebuilted the daemon and now I have a strange behaviour.

It works well until the UPS executes the self test.
Once the UPS executes the self test, I can see that the status of the UPS changes from Online, to Battery Discharching,
but then I loose the communication with the UPS.

Here the logs before the self test:
```
Network UPS Tools - Generic HID driver 0.52 (2.8.1-256-gec3df0ef4)
USB communication driver (libusb 0.1) 0.45
Using subdriver: CyberPower HID 0.8
Network UPS Tools - UPS driver controller 2.8.1-256-gec3df0ef4
kill: No such process
listening on 0.0.0.0 port 3493
/var/run is world readable
Connected to UPS [Cyberpower_PR1500ELCD]: usbhid-ups-Cyberpower_PR1500ELCD
Found 1 UPS defined in ups.conf
kill: No such process
   0.000000	[D1] Just failed to send signal, no daemon was running
   0.000214	Using power down flag file /etc/killpower
   0.000353	UPS: Cyberpower_PR1500ELCD@localhost (primary) (power value 1)
   0.000372	[D1] debug level is '1'
   0.001159	[D1] Saving PID 13 into /run/upsmon.pid
   0.001552	[D1] Succeeded to become_user(nut): now UID=1000 GID=1000
   0.005439	upsnotify: failed to notify about state 2: no notification tech defined, will not spam more about it
   0.005456	[D1] Trying to connect to UPS [Cyberpower_PR1500ELCD@localhost]
   0.006282	[D1] Logged into UPS Cyberpower_PR1500ELCD@localhost
```

and here the logs after the self test once the UPS lost the communication:
```
Using subdriver: CyberPower HID 0.8
Network UPS Tools - Generic HID driver 0.52 (2.8.1-256-gec3df0ef4)
USB communication driver (libusb 0.1) 0.45
Network UPS Tools - UPS driver controller 2.8.1-256-gec3df0ef4
kill: No such process
listening on 0.0.0.0 port 3493
/var/run is world readable
Connected to UPS [Cyberpower_PR1500ELCD]: usbhid-ups-Cyberpower_PR1500ELCD
Found 1 UPS defined in ups.conf
kill: No such process
   0.000000	[D1] Just failed to send signal, no daemon was running
   0.000240	Using power down flag file /etc/killpower
   0.000388	UPS: Cyberpower_PR1500ELCD@localhost (primary) (power value 1)
   0.000409	[D1] debug level is '1'
   0.001217	[D1] Saving PID 13 into /run/upsmon.pid
   0.001622	[D1] Succeeded to become_user(nut): now UID=1000 GID=1000
   0.005743	upsnotify: failed to notify about state 2: no notification tech defined, will not spam more about it
   0.005762	[D1] Trying to connect to UPS [Cyberpower_PR1500ELCD@localhost]
   0.006650	[D1] Logged into UPS Cyberpower_PR1500ELCD@localhost
 355.024942	Poll UPS [Cyberpower_PR1500ELCD@localhost] failed - Data stale
 355.024968	Communications with UPS Cyberpower_PR1500ELCD@localhost lost
Network UPS Tools upsmon 2.8.1-256-gec3df0ef4
 360.025701	Poll UPS [Cyberpower_PR1500ELCD@localhost] failed - Data stale
 365.026000	Poll UPS [Cyberpower_PR1500ELCD@localhost] failed - Data stale
 370.026280	Poll UPS [Cyberpower_PR1500ELCD@localhost] failed - Data stale
 375.026544	Poll UPS [Cyberpower_PR1500ELCD@localhost] failed - Data stale
 380.026818	Poll UPS [Cyberpower_PR1500ELCD@localhost] failed - Data stale
 385.027083	Poll UPS [Cyberpower_PR1500ELCD@localhost] failed - Data stale
....
```

this problem does not happen if I cut the power and then put the UPS back online.
this problem only happens if I do the self test.

is this a new bug?
any help would be much appreciated and thanks for your great work here!

sblantipodi, 2023-12-22:
After further testing I correct my previous post.
Problem happens even if I cut the power for 10 seconds only.
If I cut the power for 30 seconds, the problem does not happen.

Cutting the power for 10 seconds only, is like doing a self test since it lasts 10 seconds.

So to recap.
I the UPS switch on battery for less than 10 seconds, the problem happen, 
if the UPS switch on battery for more than 10 seconds, the problem does not happen.

This is very weird...

jimklimov, 2023-12-22:
>    0.000353	UPS: ***@***.*** (primary) (power value 1)
>    0.000388	UPS: ***@***.*** (primary) (power value 1)
>  355.024942	Poll UPS ***@***.*** failed - Data stale
>  355.024968	Communications with UPS ***@***.*** lost
>  360.025701	Poll UPS ***@***.*** failed - Data stale
>  365.026000	Poll UPS ***@***.*** failed - Data stale
>  370.026280	Poll UPS ***@***.*** failed - Data stale
>  375.026544	Poll UPS ***@***.*** failed - Data stale
>  380.026818	Poll UPS ***@***.*** failed - Data stale
>  385.027083	Poll UPS ***@***.*** failed - Data stale

Is the log a mix of several NUT daemons' messages? The `logged into` and
probably `data stale` messages are from `upsmon` (the latter may be `upsd`
also).

The practical issue is that likely the driver daemon gets stuck in
something (own code/regression? waiting on device I/O?) and does not
refresh state info regularly for `upsd` and beyond.

Can you try running with higher debug verbosity (practical maximum is
around 6) to see where it chirps last? As you build NUT from source, you
can also easily add `upsdebugx(...)` logging in places that look
interesting for this code path.

CPS are known for resetting chips at various times and losing connections.
Then `udev` or similar OS framework re-grabs the "newly inserted" USB
device. NUT drivers try to reconnect upon some comms errors during polling
- maybe this one does not get caught (e.g. yields "some other" libusb error
code than we check for)?..

Jim

sblantipodi, 2024-01-02:
hi @jimklimov, sorry for the late reply and Happy New Year!!! :)

where should I add the `upsdebugx(...)` logging?

this is how I build it:
```
git clone https://github.com/networkupstools/nut.git \
    && cd nut \
    && ./autogen.sh \
    && ./configure \
            --prefix=/usr \
            --libexecdir=/usr/lib/nut \
            --without-wrap \
            --with-user=root \
            --with-group=root \
            --disable-static \
            --with-serial \
            --with-usb \
            --without-avahi \
            --with-snmp \
            --with-neon \
            --without-powerman \
            --without-ipmi \
            --without-freeipmi \
            --with-libltdl \
            --without-cgi \
            --with-drvpath=/usr/lib/nut \
            --datadir=/usr/share/nut \
            --sysconfdir=/etc/nut \
            --with-statepath=/var/run \
            --with-altpidpath=/var/run \
    && make DESTDIR=/app install \
    && make DESTDIR=/app install-conf
```

should I add it as a param in the ./configure?

jimklimov, 2024-01-02:
@sblantipodi : no, that suggestion would be about changing the C code for the subdriver (check `drivers/cps-hid.c` for anything of interest that is not debug-logged now, to add messages that would help see what it does around the calibration state changes), and re-running the `make` parts.

Note that you can in this case iterate with testing right from the build workspace, not necessarily installing the new build all the time, like this:
````
:; make && ./drivers/usbhid-ups -s Cyberpower_PR1500ELCD -DDDDDD
````

sblantipodi, 2024-01-16:
hi @jimklimov thanks for the answer...
I am trying to debug this but I haven't found anything useful currently...

a very weird thing happened recently...
now when there is a power outage, the UPS stops sending infos to the NUT server and there is no way to restore the communication.
restarting the server does not help until a system reboot. 

how this can be possible?

this is a new problem. I don't know if the latest Ubuntu 23.10 introduced it.

jimklimov, 2024-01-16:
Sounds odd. Is there anything in `dmesg` e.g. about USB disconnections or re-connections on the link level? It may be possible that due to power surge or some internal logic, during the switch to battery the UPS controller re-establishes the connections, and maybe drivers do not expect that. Some of them should find libusb errors related to link loss, and try to reconnect, but it may be that not all of them do so yet.

Also, `udev.rules` and similar settings for the OS/kernel should take care of re-chown'ing the device filesystem nodes for the vendorid:productid for the NUT runtime user account (in case of finding a "new" connection), but it is worth checking if this does in fact happen.

sblantipodi, 2024-01-16:
If I move the UPS USB from USB3 to USB2 of my Raspberry Pi5, when an event occurs, other USB devices continue working, 
no matter if I use port auto or fixed port, but the `/dev/bus/usb/001/XXX` number changes and I need to restart nut to make it working again...

is this normal?

sblantipodi, 2024-01-18:
after some more trials I see that the dmesg contains this error from time to time:
[  109.117430] usb usb3-port1: disabled by hub (EMI?), re-enabling...

sblantipodi, 2024-01-18:
and another error contained in the dmesg.
xHCI host controller not responding, assume dead

all this errors comes after an ups event

sblantipodi, 2024-01-18:
it is a Raspberry Pi5 problem because on Pi4 the problem does not happen.
the OS is installed on a USB SSD so I can boot the exact same OS with the exact same configuration on both pi4 and pi5.

Pi4 works like a charm with no disconnections on USB,
Pi5 gives all the errors I explained above.

jimklimov, 2024-01-18:
Thanks for the tests and updates. While I'm not sure there's much that NUT can do about hardware issues like this (maybe an option to let the driver die and reconnect from scratch by restarting via systemd), this may be worth bringing up with the Raspberry org. Maybe some firmware update can address this.

If you look at NUT issues, there were quite a few mentions of Pi (3rd gen mostly, I think), but I am not sure if it is indicative more of their flaws or popularity :)

sblantipodi, 2024-01-19:
thanks for the answer jimklimov, I really appreciate it.
I workarounded the problem using a simple USB Hub.
It seems that Pi3 was plagued by this problem and the Pi4 completely solved this problem.

Now with Pi5 the problem is back.

this is obviously something not related to NUT so closing the issue.

thanks again for the answer, for the support and for your great work here ;)