networkupstools/nut #2244: nut-scanner network scans should provide a "-m auto" option

aquette, 2023-12-22:
when scanning network devices (SNMP, XML, NUT), nut-scanner should provide a "-m auto" option to automatically provide the local network information.

Example: when your IP is 192.168.1.1 on a class C network, "-m auto" should be equivalent to "-m 192.168.1.1/24"

jimklimov, 2023-12-22:
I think this logic could be already done in 42ity fork? port-back is still piecemeal and pending for the bigger features.

jimklimov, 2024-01-20:
BTW, what is wrong with libcidr? :) Not that it's hard to parse a few string patterns and do bit maths ourselves...

jimklimov, 2024-07-01:
FWIW, I've recently had another pass at porting stuff from the FTY fork and checking the remaining differences (by now two big things are packaging recipes and DMF, and also `docs/nut-names.txt`, see #1316 for the detailed list of that effort).

There were no mentions of `libcidr` there, and no different code about CIDR handling, so I actually looked deeper and found `nutscan-ip.c` which does the related magic (including CIDR conversion to a start/end IP address pair) since your and Frederic's time. I guess we dealt with libcidr elsewhere in the 42ITy project, in some other components.

So here your issue is indeed new and pristine after all, with the challenging part being about *portably* detecting what non-loopback IP address(es) we have at disposal on the current system, with what netmasks, and checking perhaps all of the resulting ranges (if there's more than one).

closed 2024-07-08 by jimklimov