networkupstools/nut #2252: NJOY Horus Plus 600 360W

lukapaunovic, 2023-12-28:
Hey guys,

Can you implement driver for: NJOY Horus Plus 600 360W 

I am willing to donate or give payment to a developer that does this.

jimklimov, 2024-01-20:
Please try asking on the nut-upsdev mailing list, more developers see it regularly...

Also, more details about the NUT version(s) you've tried, debug messages from those drivers that failed to work, etc. would be helpful. Maybe it is a configuration problem (permissions etc.) and not so much a lack of code ability to talk to that device?

fctpn, 2025-07-02:
@lukapaunovic did you manage to configure NJOY Horus Plus 600 ?

I'm using it on Unraid with nutdrv_qx but i receive only partial  info.

[image]