networkupstools/nut #2254: Riello SEP series USB even in ECO mode identifies as in bypas stating UPS needs attention on pfSense

ha11oga11o, 2023-12-29:
Hello all,

i have couple of Riello SEP2200 UPS units and have strange issue. Namely, at pfSense all works fine with rs232 connection but it insists that UPS needs attention when is set on ECO mode. At [manual](https://www.yumpu.com/en/document/read/33909952/user-manual-pdf-riello-ups) i found info which states 
"In order to optimize performance, in ECO mode, the load is normally powered by the bypass. If the mains goes out of the permitted tolerance range, the UPS switches to ON LINE operation. If the mains returns within the permitted tolerance range for at least five minutes, the UPS goes back to powering the load from the bypass."

Indeed UPS is in bypass, but thats ECO feature (page 15). Does nut interpret that "feature" in riello driver, does it UPS send it at all as status, can it be added if not? UPS happily switches ON LINE on power outage, or any other disturbance on power grid. It simply works fine. 

Theres also my questioning about it on [pfSense forum](https://forum.netgate.com/topic/183961/nut-package-2-8-1-and-above/116?_=1703702947925). Forum username ha11oga11o. 

Problem occurring at

2.7.2-RELEASE (amd64)
built on Fri Dec 8 21:55:00 CET 2023
FreeBSD 14.0-CURRENT

nut 2.8.2 

System log of successful nut service start:

Dec 29 13:50:22 	upsmon 	59353 	UPS RielloSEP2200 on line power
Dec 29 13:50:17 	upsmon 	59353 	UPS RielloSEP2200: on bypass (powered, not protecting)
Dec 29 13:50:17 	upsmon 	59353 	Communications with UPS RielloSEP2200 established
Dec 29 13:50:16 	upsd 	60905 	Connected to UPS [RielloSEP2200]: riello_ser-RielloSEP2200
Dec 29 13:50:15 	riello_ser 	87467 	Startup successful

From UPS detail page:

ups.status | OL BYPASS CHRG

[image]

Is there any way for nut to understand ECO mode for these models, and get it out from alerted state?

Many thnx in advance.

jimklimov, 2023-12-30:
Hello, can you please clarify the NUT versions involved? 2.7.2 is antique (in IT terms), and 2.8.2 is only planned.

The description of this ECO mode seems to be what was called "line-interactive" in the early days of UPSes (as opposed to "online", which kept wall and load on separate circuits witha battery and inverter in the middle), maybe except the smart part of waiting for 5 minutes after a hiccup.

As for fixing the driver behavior - sure it is possible, albeit with a coding effort :)

On the one hand, at least a toggle option can be added to "treat situation X as Y on this device", generally speaking. Quite a few precedents in usbhid-ups about "OL+DISCHRG" handling, for example.

Hopefully at some stage criteria would be found (by device model/id or better) to set the default values of such toggles.

On another, I think riello driver authors can still be contacted by mail for insights...

ha11oga11o, 2023-12-30:
Hello,

thanks for fast response. All i can see on pfSense about nut that latest version is 2.8.2. Its package which is installed on demand. 

[image]

Highlighted links are

[nut version](https://github.com/pfsense/FreeBSD-ports/commits/devel/sysutils/pfSense-pkg-nut)

[Package Dependencies](https://freshports.org/sysutils/nut-devel)

If any other data needed i can provide. 

Thanks

P.S. version 2.7.2 is pfSense actually, i stated below its nut version 2.8.2. I think i have latest which pfSense offers.

jimklimov, 2023-12-30:
Hm, so it comes from a devel stream, so versions may be a bit off. NUT v2.8.1 was released 2023-10-31, so by the numbers in screenshot they may have an even earlier snapshot. But maybe just the naming is off and they included an errata patch... 

CC @gdt : any insights on this one?

jimklimov, 2024-06-27:
Sorry about the delay with "some answer".

Surely, "something" can be done - but it would take a developer and a device, and hopefully protocol information all coming together (also assuming the HW interface chips actually report correct states and do not raise an alarm as such). I do have hope  @mzampieri70 or their colleagues can take a look at this, as best-positioned to have the tools and knowledge for a proper fix.

masterwishx, 2024-09-28:
> Hello,
> 
> i hope i can bump this again? Can it be done anything about this? I dont wnat to be rude, but any answe would be ok. Thank you

Hi, I'm trying to make ECO mode for nut for Eaton models, will be cool if you can print debug log when your ups in eco mode to see what variables path it using I hope it can help to include eco mode for nut for other UPS models too

VincentKueszter, 2025-06-14:
> Hi, I'm trying to make ECO mode for nut for Eaton models, will be cool if you can print debug log when your ups in eco mode to see what variables path it using I hope it can help to include eco mode [...]

I have a Riello USV that has a similar issue, and I'd like to help.
I'd like to help, how can I print a debug log you need?

My Riello (Sentinel Dual) has three modes:
ONLINE (for which NUT correctly shows OL)
ECO (which is line-interactive, for which NUT shows OL BYPASS)
SMART ACTIVE (which is line-interactive, switching to online only if necessary, for which NUT also shows OL BYPASS when input power is fine, which makes sense, since it's basically just line-interactive in that case)

I don't mind NUT not being able to detect the smart mode, I just don't want it to show bypass when in Line-Interactive.
The UPS also has it's own actual bypass mode by the way, which correctly switches from OL to OL BYPASS when in Online mode.
It does nothing when in line-interactive though, since that already has BYPASS mistakenly active in NUT, as stated above.

VincentKueszter, 2025-06-15:
> Can you point to a document that defines "ECO" as line-interactive, and says what line-interactive means? I think of it as a term from BEST Ferrups, and I don't understand what it means. I have perceived that the point of ECO is to be in bypass, with almost everything shut down, with the ability to turn on the inverter very quickly so that the load does not perceive a gap, if there is a problem.
> 
> At this point I don't even know what "OL" really means.

My current understanding of the terms as per [this website](https://www.fs.com/blog/comparison-of-ups-topologies-line-interactive-vs-online-vs-offline-3538.html)

1) Offline/VFD (Voltage and Frequency Dependent) or Standby UPS
Mains power is filtered and fed directly to load.
If mains fails, a relay connects load to the inverter output (4-8ms transfer time).
2) Line-Interactive
Inverter is always under power, on mains failure the inverter switches from reverse to regular operation and the battery is no longer charging, but feeds the load. Very low switching times.
3) Online/double conversion
Mains always charge the batteries, which power the load.

In regards to ECO:
If you click on "technical specifications" [here](https://www.riello-ups.com/products/1-ups/45-sentinel-dual-low-power), it says "Line-Interactive/Smart Active efficiency 98%".
Furthermore, the [manual](https://www.riello-ups.com/uploads/file/722/722/0MNSDH1K0RUENUC__MAN_SDH_1000-3000_EN_.pdf) defines the following modes:
> ON-LINE is the mode with the greatest load protection and the best quality of the output waveform
> ECO is the mode with which the UPS consumes the least amount of power, and is therefore the most efficient
> SMART ACTIVE: in this mode, the UPS decides whether to operate in ON-LINE or ECO mode according to a statistic about the quality of the mains power.
> STAND-BY OFF [Mode 1]: the UPS operates as an emergency power supply. If mains power is present, the load is not powered, however should the mains supply fail, the load is powered by the UPS.

This makes me think ECO is Line-Interactive and leads me to assume my UPS has the following modes:
- Online (regular double conversion)
- Line-Interactive (called Eco)
- Smart Active (Which is Line-Interactive until the mains power has issues, then it switches to online - and since [this table](https://www.riello-ups.com/questions/29-what-is-an-offline-ups) states the only issues line-interactive can't handle are harmonics and frequency variation, I assume it's about those issue types)
- a special mode that only powers the load if mains is gone.

Then there is the slightly confusing usage of bypass:
- Riello uses it to describe mains power, [e.g. on page 16 of the manual](https://www.riello-ups.com/uploads/file/722/722/0MNSDH1K0RUENUC__MAN_SDH_1000-3000_EN_.pdf)
- Then there is "manual bypass", which removes power from the battery circuits so you can swap them. Load is powered by mains and unprotected.

NUT currently outputs the following for my UPS:
ON-LINE: OL
ECO: OL BYPASS
SMART ACTIVE: OL BYPASS
STAND-BY OFF: OL BYPASS

The following would make sense to me:
ON-LINE: OL
ECO: ECO/LI/...
SMART ACTIVE: Either one of the above, depending on the current situation.
STAND-BY OFF: I don't know how to handle this one, but I also don't use it and I don't know if this falls under the use-case of an UPS.

BYPASS should only be active when I use the manual bypass, since most tools (like pfSense or NUT monitor) warn that the load is unprotected if BYPASS is sent.

gdt, 2025-06-15:
Vincent Küszter <notifications@github.com> writes:

> VincentKueszter left a comment (networkupstools/nut#2254)
>
> My current understanding of the terms as per [this website](https://www.fs.com/blog/comparison-of-ups-topologies-line-interactive-vs-online-vs-offline-3538.html)

That's one particular manufacturer.  Doesn't make them wrong about their
equipment, but doesn't make it industry-standard terms, like some IEEE
committee definition.

> 1) Offline/VFD (Voltage and Frequency Dependent) or Standby UPS
> Mains power is filtered and fed directly to load.
> If mains fails, a relay connects load to the inverter output (4-8ms transfer time).

In their diagram, this is a static property of a particular design.  I
have a UPS that I am 99% sure is like this.  When there is a power
glitch, it clicks, and makes more noise, and the battery voltage drops,
and then after 2s it clicks again and is quiet, and over a minute the
battery voltage recovers.

> 2) Line-Interactive
> Inverter is always under power, on mains failure the inverter switches from reverse to regular operation and the battery is no longer charging, but feeds the load. Very low switching times.

This is not a useful description; it leaves out all the hard/interesting
parts.  I sort it into marketing babble.  It seems like their specific
design, not a general approach, and I don't expect it to apply broadly
across brands.  As an example I'm pretty sure BEST used this to mean
something else in the 90s.   Needs link to journal article with
schematic and theory of operation of "reversible" inverter.

> 3) Online/double conversion
> Mains always charge the batteries, which power the load.

Makes sense but I do not believe anybody thinks nut's "OL" means that.
This is very inefficient (even if your unit is only using 8% doing it,
impressive) and makes sense only when you have super sensitive
equipment.  These days equipment is ~all power supplies to make some DC
and ~all of it can handle a few ms glitch without messing up the
internal DC, unless maybe it's some twitchy medical stuff or lab
instrumentation.  I don't believe computers need this.  My computers
(and switches) have experienced many glitches and transfer to inverter
with no issues.   Yesterday, 3 times (it's squirrels chewing transformer
wires season).

I expect online/double is pretty rare.

> In regards to ECO:
> If you click on "technical specifications" [here](https://www.riello-ups.com/products/1-ups/45-sentinel-dual-low-power), it says "Line-Interactive/Smart Active efficiency 98%". [...]

The idea that 'ON-LINE' is "best quality of output waveform" does not
make sense.  2% distortion is worse than what you get from the grid.
You are trading distortion and loss always for not glitching.  That's ok
if that's what you want of course.

> This makes me think ECO is Line-Interactive and leads me to assume my UPS has the following modes:

It doesn't make me think this.  It makes me think there is some other
mode and marketing people called it ECO because customers think things
called ECO are good.  What the manual actually says is:

  In order to optimise performance, in ECO mode, the load is normally
  powered by the bypass. If the mains goes out of the permitted
  tolerance range, the UPS switches to ON LINE operation. If the mains
  returns within the permitted tolerance range for at least five
  minutes, the UPS goes back to powering the load from the bypass

which is basically a standby system.  (That's sensible, but it's just
standby.)  It's interesting to have it stay for 5 minutes, but given
that it's capable of indefinite ON LINE, that makes sense.  (A regular
standby-only UPS is not going to be happy with the extra drain as
[... 104 lines cut ...]
status would be OB (only), where OB means "output power is present,
using battery in a way which is not sustainable if mains power were
present".

I don't think we should do that.  If BYPASS means manual bypass now, we
should leave that and clarify it.  And perhaps add DOUBLE and RAPID to
mean those styles, so existing data consumers don't have an ABI break.

For RAPID, there might also be a vendor:eco token for that vendor's take
on what eco means, when in RAPID/vendoreco mode.

Finally, I think we need to be careful about what's a status and what's
part of design.   For nut, we really care about:

  are we operating normally, in a way that can continue indefinitely

  are we operating using up battery, that's not going to last

  are we operating in manual bypass

and then secondarily we care a very little about

  is this online/double, standby(bypass to inverter), or something else?

  what is the amount of power consumed by the UPS

gdt, 2025-06-15:
> [nut is showing "BYPASS" in manual bypass, and not in other modes].

> > Makes sense but I do not believe anybody thinks nut's "OL" means that. This is very inefficient (even if your unit is only using 8% doing it, impressive) and makes sense only when you have super sensitive equipment. These days equipment is ~all power supplies to make some DC and ~all of it can handle a few ms glitch without messing up the internal DC, unless maybe it's some twitchy medical stuff or lab instrumentation. I don't believe computers need this. My computers (and switches) have experienced many glitches and transfer to inverter with no issues. Yesterday, 3 times (it's squirrels chewing transformer wires season).
> 
> I am surprised it isn't generally perceived as standing for a double conversion system, but the pair of "OL=on line" & "OB=on battery" makes sense.

Maybe it is.  But we  really need to read the docs, and see what they say, and then talk about if that's right or should be changed, and if various drivers respect it.   My own view of OL is colored by the bestfortress driver setting it.

> > The idea that 'ON-LINE' is "best quality of output waveform" does not make sense. 2% distortion is worse than what you get from the grid. You are trading distortion and loss always for not glitching. That's ok if that's what you want of course.
> 
> It's something I keep seeing in marketing pages by the way (e.g. [here](https://mitsubishicritical.com/technologies/double-conversion-vs-line-interactive/) or [here](https://www.vertiv.com/en-emea/about/news-and-insights/articles/educational-articles/what-are-the-different-types-of-ups-systems/)). I suppose they refer to the inverter output of standby UPS often being square wave when the load is powered by battery? If that's true I don't know, but I read it repeatedly when they want to advertise the advantages of double converter systems. Whether one needs that is a different question as well, of course.

We really need to be careful not to pay attention to inaccurate marketing text.  There's a lot of it out there.  Surely some is dishonest and some is a failure to understand.   Whether an inverter is square, modified sine, better sine, almost sine with 2%, or looking like the grid, is a separable issue, technically, from whether it is used in standby or double-conversion.    I do not believe that any double-conversion will have as clean an output *when the grid is not experiencing faults*.    I do believe that it will be less glitchy under faults.

As an example, I often notice my UPS transfer to battery and back 2s later.  Sometimes I see the lights blink at the same time.  Often when this happens there is an outage reported just afterwards.  I think that what's going on is that the fault current (from an incinerating squirrel, or a tree branch) causes a voltage drop, which causes the UPS to transfer to battery/inverter.  But either the squirrel is a self-clearing fault (being sort of a 1000A fuse), or some protection device on a lateral where it was opens, avoiding an outage on the rest of the distribution system.  But other than that, I think the input waveform is really clean 99.99% of the time.   So it's really a tradeoff of a less clean waveform -- that probably no actual load device cares about -- for reduced glitching under fault conditions, not what the marketing people say.

masterwishx, 2025-06-15:
> I have a Riello USV that has a similar issue, and I'd like to help.
> I'd like to help, how can I print a debug log you need?

Already added ECO mode for Eaton time ago you can check it in `mge_hid.c`

jimklimov, 2025-06-15:
Speaking of the docs, NUT definition of `OL` is:
```
$ grep OL docs/new-drivers.txt
UPS status flags like on line (OL) and on battery (OB) live in
        status_set(val) -- add a status word (OB, OL, etc)
        OL      -- On line (mains is present)
- the `OL` and `OB` flags are an indication of the input line status only.
```

gdt, 2025-06-16:
Thanks! The values are indeed written out pretty clearly.   I would say that we should probably head down the vendor:mode path for things like double conversion, same as eco, and also for line interactive, until there is adequate clarity  (but I'm not holding my breath!).