networkupstools/nut #2258: Tripp Lite SMX1200XLHG - protocol 3017

biohead, 2024-01-08:
I've just got hold of a Tripp Lite SMX1200XLHG, although disappointed to find it's not fully working with NUT... yet.
This is a 36v, 1000VA/750w UPS with 6 outlets, with expandable battery storage.

When using the tripplite_usb driver, it reports that it uses an unknown protocol:
```
root@proxmox2:~# upsdrvctl -ddddd start
Network UPS Tools - UPS driver controller 2.8.0
Network UPS Tools - Tripp Lite OMNIVS / SMARTPRO driver 0.33 (2.8.0)
Warning: This is an experimental driver.
Some features may not function correctly.

Detected a UPS: Tripp Lite /TRIPP LITE SMX1200XLHG 
Unit ID: 65535
Unknown protocol (3017)Attached to Tripp Lite TRIPP LITE SMX1200XLHG
```
As per the documentation, it's falling down with: ups.firmware.aux: protocol 3017

Here is the output when connecting using the tripplite_usb driver:
```
root@proxmox2:~# upsc tripp@localhost
Init SSL without certificate database
battery.voltage.nominal: 36
device.mfr: Tripp Lite
device.model: TRIPP LITE SMX1200XLHG
device.type: ups
driver.name: tripplite_usb
driver.parameter.bus: 001
driver.parameter.pollinterval: 15
driver.parameter.port: auto
driver.parameter.product: TRIPP LITE SMX1200XLHG
driver.parameter.productid: 0001
driver.parameter.synchronous: auto
driver.parameter.vendor: Tripp Lite
driver.parameter.vendorid: 09AE
driver.version: 2.8.0
driver.version.internal: 0.33
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
input.voltage.nominal: 230
ups.debug.0: 30 17 58 58 58 58 0d '0.XXXX.'
ups.debug.D: 37 42 38 38 0d 00 00 '7B88...'
ups.debug.L: 30 30 30 44 07 58 0d '000D.X.'
ups.debug.load_banks: 0
ups.debug.M: 37 39 37 44 0d 00 00 '797D...'
ups.debug.P: 30 31 30 30 30 58 0d '01000X.'
ups.debug.S: 31 34 30 04 64 30 0d '140.d0.'
ups.debug.T: 41 34 31 46 33 30 0d 'A41F30.'
ups.debug.V: 32 30 36 30 58 58 0d '2060XX.'
ups.delay.shutdown: 64
ups.firmware: F3009.C
ups.firmware.aux: protocol 3017
ups.id: 65535
ups.mfr: Tripp Lite
ups.model: TRIPP LITE SMX1200XLHG
ups.power.nominal: 1000
ups.productid: 0001
ups.status: 
ups.vendorid: 09ae 
```

I'm happy to try help assist working out this protocol and getting it into NUT, but not quite sure the best place to start this.
I'm guessing the debug arrays are the useful info and they just need interpreting properly first?

jimklimov, 2024-01-08:
CC @aquette @dzomaya

clepple, 2024-01-10:
> I'm guessing the debug arrays are the useful info and they just need interpreting properly first?

Yes. Here are examples for protocol 1001: https://github.com/networkupstools/nut/blob/v2.8.1/drivers/tripplite_usb.c#L33 and protocol 3003: https://github.com/networkupstools/nut/blob/v2.8.1/drivers/tripplite_usb.c#L67

From what I can tell, the main difference between the protocols is whether each value is sent in binary, BCD, hex, etc.: https://github.com/networkupstools/nut/blob/v2.8.1/drivers/tripplite_usb.c#L529 and what sort of scale factors need to be applied (usually based on the number of battery packs, or nominal AC voltage).

Hopefully this is just a matter of extending the driver with a few more case/switch statements. Unfortunately, I don't have a lot of time to maintain the driver these days (my 1001 unit needs a battery replacement, and the 3016 USB hardware was especially unreliable with newer USB controllers so I sold it to someone who didn't need monitoring), but I can probably answer a few questions here and there.

Along the lines of the 3016 USB hardware issue, you might be able to tell whether this will be a problem on the 3017 as well - just leave the USB cable plugged in for several days, and check the kernel log for disconnections. (Maybe run `lsusb -v -d 09ae:` occasionally to send some traffic to the UPS.) The general trend I saw was that later-generation USB 2.0+ host chipsets would stall when talking to the UPS (USB 1.1 low-speed, IIRC) and would eventually reconnect. Hopefully the 3017 doesn't have the same device-side USB chip. If it does, the driver modifications may not be worth the hassle.

biohead, 2024-01-12:
I've had it powered up and connected for about 24 hours now, and so far dmesg is looking ok.
I have configured NUT to talk to the UPS, and setup the upsstats.cgi to report on it - hopefuly that is keeping some traffic flowing. I'll keep it running like this for a bit longer to be sure.

This is the entire dmesg output since connecting the UPS yesterday (the device itself is sat doing nothing else, explaining the low message count):
```[  215.825662] usb 1-1.2: new low-speed USB device number 6 using xhci_hcd
[  215.933596] usb 1-1.2: New USB device found, idVendor=09ae, idProduct=0001, bcdDevice= 0.0a
[  215.933620] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  215.933628] usb 1-1.2: Product: TRIPP LI
[  215.933634] usb 1-1.2: Manufacturer: Tripp Lite 
[  215.940554] hid-generic 0003:09AE:0001.0004: hiddev0,hidraw0: USB HID v1.10 Device [Tripp Lite  TRIPP LI] on usb-0000:00:15.0-1.2/input0
[17145.851346] perf: interrupt took too long (2511 > 2500), lowering kernel.perf_event_max_sample_rate to 79500
[23295.285830] perf: interrupt took too long (3145 > 3138), lowering kernel.perf_event_max_sample_rate to 63500
[31446.963882] perf: interrupt took too long (3945 > 3931), lowering kernel.perf_event_max_sample_rate to 50500
[43787.983227] perf: interrupt took too long (4947 > 4931), lowering kernel.perf_event_max_sample_rate to 40250
[62903.362714] perf: interrupt took too long (6211 > 6183), lowering kernel.perf_event_max_sample_rate to 32000```

clepple, 2024-01-12:
> I have configured NUT to talk to the UPS, and setup the upsstats.cgi to report on it

but upsstats is just showing empty bar graphs? (just keeping the driver running should be sufficient - I think it sends each of the queries for the debug strings every two seconds)

That is a good sign, though - I saw issues with 3016 well before 24 hours.

biohead, 2024-01-12:
Yes, it's got nothing but the name reported, everything else is "Not Supported". I was hoping it would keep enough traffic for the purpose of testing.

biohead, 2024-01-24:
I've been running it like this, and not so much a sniff of any USB disconnections so I think it's good to continue.

What would be the best way to tackle this - just periodically poll and grab the debug strings? Or is there a logical method to breaking down these Tripp protocols?

clepple, 2024-01-25:
Yes, logging the `ups.debug.*` values is what I did. If you make a shell script with `upsc` in a loop, it helps to record the time as well, and also try to ballpark some values for things like ambient temperature and percent load. Mark down stats from the LCD (if applicable); the UPS usually returns the same data in short order.

Just paging through the code for other models, I am noticing one value that corresponds to 120V (first character of `ups.debug.V` is effectively an enum), and another that corresponds to 50 Hz (last `0` of `ups.debug.S`) - is that correct? Also, you mentioned six outlets - are any of them individually controllable? (last `0` of `ups.debug.V` implies no) The `4` in `ups.debug.S` usually implies that no self-test has been performed recently.

biohead, 2024-01-25:
Here is a first run at some outputs.
Ambient temp is around 20C
I'm UK - so nominal 240V/50Hz - NOT 120V.
This unit has no LCD, so can't check the load. It's going to be around the 50w mark though.
The unit is 1kVW/750w rated.
No switchable output.

A self test was carried out just after 09:45
I simulated a power loss just before 10:00
There should've been a slight increase in load around 10:50
Power restored 11:00
Slight decrease in power at 11:10

```
[2024-01-25 09:20:01] Command Output (filtered): ups.debug.0: 30 17 58 58 58 58 0d '0.XXXX.'
ups.debug.D: 37 44 38 38 0d 00 00 '7D88...'
ups.debug.L: 30 30 30 44 07 58 0d '000D.X.'
ups.debug.load_banks: 0
ups.debug.M: 36 43 38 31 0d 00 00 '6C81...'
ups.debug.P: 30 31 30 30 30 58 0d '01000X.'
ups.debug.S: 31 34 30 00 64 30 0d '140.d0.'
ups.debug.T: 39 43 31 46 34 30 0d '9C1F40.'
ups.debug.V: 32 30 36 30 58 58 0d '2060XX.'
[2024-01-25 09:25:01] Command Output (filtered): ups.debug.0: 30 17 58 58 58 58 0d '0.XXXX.'
ups.debug.D: 37 44 38 38 0d 00 00 '7D88...'
ups.debug.L: 30 30 30 44 07 58 0d '000D.X.'
ups.debug.load_banks: 0
ups.debug.M: 36 43 38 31 0d 00 00 '6C81...'
ups.debug.P: 30 31 30 30 30 58 0d '01000X.'
ups.debug.S: 31 34 30 00 64 30 0d '140.d0.'
ups.debug.T: 39 43 31 46 34 30 0d '9C1F40.'
ups.debug.V: 32 30 36 30 58 58 0d '2060XX.'
[2024-01-25 09:30:01] Command Output (filtered): ups.debug.0: 30 17 58 58 58 58 0d '0.XXXX.'
ups.debug.D: 37 44 38 38 0d 00 00 '7D88...'
ups.debug.L: 30 30 30 44 07 58 0d '000D.X.'
ups.debug.load_banks: 0
ups.debug.M: 36 43 38 31 0d 00 00 '6C81...'
ups.debug.P: 30 31 30 30 30 58 0d '01000X.'
ups.debug.S: 31 34 30 00 64 30 0d '140.d0.'
ups.debug.T: 39 43 31 46 32 30 0d '9C1F20.'
ups.debug.V: 32 30 36 30 58 58 0d '2060XX.'
[2024-01-25 09:35:01] Command Output (filtered): ups.debug.0: 30 17 58 58 58 58 0d '0.XXXX.'
ups.debug.D: 37 44 38 38 0d 00 00 '7D88...'
ups.debug.L: 30 30 30 44 07 58 0d '000D.X.'
ups.debug.load_banks: 0
ups.debug.M: 36 43 38 31 0d 00 00 '6C81...'
ups.debug.P: 30 31 30 30 30 58 0d '01000X.'
ups.debug.S: 31 34 30 00 64 30 0d '140.d0.'
ups.debug.T: 39 43 31 46 33 30 0d '9C1F30.'
ups.debug.V: 32 30 36 30 58 58 0d '2060XX.'
[2024-01-25 09:40:01] Command Output (filtered): ups.debug.0: 30 17 58 58 58 58 0d '0.XXXX.'
ups.debug.D: 37 44 38 38 0d 00 00 '7D88...'
ups.debug.L: 30 38 30 44 07 58 0d '080D.X.'
ups.debug.load_banks: 0
ups.debug.M: 36 43 38 31 0d 00 00 '6C81...'
ups.debug.P: 30 31 30 30 30 58 0d '01000X.'
ups.debug.S: 31 34 30 00 64 30 0d '140.d0.'
ups.debug.T: 39 43 31 46 34 30 0d '9C1F40.'
ups.debug.V: 32 30 36 30 58 58 0d '2060XX.'
[2024-01-25 09:45:01] Command Output (filtered): ups.debug.0: 30 17 58 58 58 58 0d '0.XXXX.'
ups.debug.D: 37 42 38 38 0d 00 00 '7B88...'
ups.debug.L: 30 38 30 44 07 58 0d '080D.X.'
ups.debug.load_banks: 0
ups.debug.M: 36 43 38 31 0d 00 00 '6C81...'
ups.debug.P: 30 31 30 30 30 58 0d '01000X.'
ups.debug.S: 31 34 30 00 64 30 0d '140.d0.'
ups.debug.T: 39 43 31 46 34 30 0d '9C1F40.'
ups.debug.V: 32 30 36 30 58 58 0d '2060XX.'
[2024-01-25 09:50:01] Command Output (filtered): ups.debug.0: 30 17 58 58 58 58 0d '0.XXXX.'
ups.debug.D: 37 42 38 44 0d 00 00 '7B8D...'
[... 226 lines cut ...]
ups.debug.load_banks: 0
ups.debug.M: 30 30 38 31 0d 00 00 '0081...'
ups.debug.P: 30 31 30 30 30 58 0d '01000X.'
ups.debug.S: 31 30 30 00 64 30 0d '100.d0.'
ups.debug.T: 41 35 31 46 34 30 0d 'A51F40.'
ups.debug.V: 32 30 36 30 58 58 0d '2060XX.'
[2024-01-25 12:00:01] Command Output (filtered): ups.debug.0: 30 17 58 58 58 58 0d '0.XXXX.'
ups.debug.D: 37 42 38 41 0d 00 00 '7B8A...'
ups.debug.L: 30 30 30 44 07 58 0d '000D.X.'
ups.debug.load_banks: 0
ups.debug.M: 30 30 38 31 0d 00 00 '0081...'
ups.debug.P: 30 31 30 30 30 58 0d '01000X.'
ups.debug.S: 31 30 30 00 64 30 0d '100.d0.'
ups.debug.T: 41 36 31 46 34 30 0d 'A61F40.'
ups.debug.V: 32 30 36 30 58 58 0d '2060XX.'
[2024-01-25 12:05:01] Command Output (filtered): ups.debug.0: 30 17 58 58 58 58 0d '0.XXXX.'
ups.debug.D: 37 44 38 36 0d 00 00 '7D86...'
ups.debug.L: 30 30 30 44 07 58 0d '000D.X.'
ups.debug.load_banks: 0
ups.debug.M: 30 30 38 31 0d 00 00 '0081...'
ups.debug.P: 30 31 30 30 30 58 0d '01000X.'
ups.debug.S: 31 30 30 00 50 30 0d '100.P0.'
ups.debug.T: 41 36 31 46 34 30 0d 'A61F40.'
ups.debug.V: 32 30 36 30 58 58 0d '2060XX.'
```

clepple, 2024-01-26:
Thanks, that's a good set of events for testing.

Okay, I guess that means that 3017 uses this weird voltage ordering: https://github.com/networkupstools/nut/blob/v2.8.1/drivers/tripplite_usb.c#L555

The `S` status digits are the same as basically all of the other models, which is good. (Note that after the self test, it went from `14...` to `10...`.) The "on battery" status shows up at 10pm (4th ASCII digit of `S`), then `M` switches from `6C81` (min: 207V) to `0081` (min: 0V), with a max voltage of 247V throughout.

It looks like the load (first part of `L`) is pretty low except when you increased it to ~ 50%. The two hex digits at the beginning of the `L` string can be concatenated, so `'32...` (or `33 32 ...` in the hex dump of the ASCII - confusing, I know) is 0x32 == 50.

clepple, 2024-01-26:
If you're up for testing the code, I pushed an update to the `2217-TL-proto-3017` branch.

I don't personally have experience with proxmox, but given that it seems to be based on Debian, the following may be useful: https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests

clepple, 2024-01-26:
I wonder if the temperature reading is intended to be a heatsink temperature rather than ambient. 0x9c - 0xa4 is roughly 35-38°C. At least it increases a bit when the inverter is on, so it isn't likely a NTC thermistor or anything wacky. Both of the commits on that branch put the 3017 protocol into the "smart" set, so they should both yield `ups.temperature` readings in `upsc`.

If you build the code from the branch, the driver version should be at least `0.36` - if not, you are still running the stock 2.8.0 driver.

biohead, 2024-01-26:
I think I've had a good go at compiling, but when I try to start the driver I'm getting this error in the debug output:

```   
   0.000000     [D5] send_to_all: SETINFO driver.state "init.starting"
Network UPS Tools - Tripp Lite OMNIVS / SMARTPRO driver 0.36 (2.8.1-405-g76d0602fb)
Warning: This is an experimental driver.
Some features may not function correctly.
   0.000284     [D5] send_to_all: SETINFO driver.version.usb "libusb-1.0.26 (API: 0x1000109)"
   0.000316     [D1] Using USB implementation: libusb-1.0.26 (API: 0x1000109)
   <snipped>
   0.600239     [D2] Checking device 5 of 8 (09AE/0001)
   0.605729     [D2] - VendorID: 09ae
   0.605787     [D2] - ProductID: 0001
   0.605811     [D2] - Manufacturer: Tripp Lite 
   0.605840     [D2] - Product: TRIPP LITE SMX1200XLHG 
   0.605937     [D2] - Serial Number: unknown
   0.605983     [D2] - Bus: 001
   0.606039     [D2] - Bus Port: 004
   0.606083     [D2] - Device: 005
   0.606111     [D2] - Device release number: 000a
   0.606140     [D2] Trying to match device
   0.606169     [D3] match_function_regex: matching a device...
   0.606211     [D2] Device matches
   0.606239     [D2] Reading configuration descriptor 1 of 1
   0.606296     [D3] libusb_kernel_driver_active() returned 0: Success
   0.606375     [D2] Claimed interface 0 successfully
   0.606421     [D3] nut_usb_set_altinterface: skipped libusb_set_interface_alt_setting(udev, 0, 0)
   0.607397     [D2] Retrieved HID descriptor (expected 9, got 9)
   0.607439     [D3] HID descriptor, method 1: (9 bytes) => 09 21 10 01 00 01 22 34 00
   0.607465     [D3] HID descriptor length (method 1) 52
   0.607496     [D4] i=0, extra[i]=09, extra[i+1]=21
   0.607530     [D3] HID descriptor, method 2: (9 bytes) => 09 21 10 01 00 01 22 34 00
   0.607562     [D3] HID descriptor length (method 2) 52
   0.607591     [D2] HID descriptor length 52
   0.609611     [D3] send_cmd(msg_len=2, type='U')
   0.609654     [D5] send_cmd: sending  3a 55 aa 0d 00 00 00 00 '.U......'
   0.717313     [D5] send_cmd: received 55 ff ff 0d 00 00 00 00 'U.......' (OK)
   0.717371     [D2] Caller doesn't like this device
```

clepple, 2024-01-27:
I think @sapireli may have broken the default case when adding the "matching by unit ID" code to `tripplite_usb`. Try adding `upsid = 65535` to `ups.conf`.

biohead, 2024-01-27:
Looking good so far. Values displayed so far look as expected.

```
root@proxmox2:/tmp/nut# upsc tripp@localhost
Init SSL without certificate database
battery.test.status: Battery OK
battery.voltage: 40.80
battery.voltage.nominal: 36
device.mfr: Tripp Lite
device.model: TRIPP LITE SMX1200XLHG
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.name: tripplite_usb
driver.parameter.pollinterval: 15
driver.parameter.port: auto
driver.parameter.productid: 0001
driver.parameter.synchronous: auto
driver.parameter.upsid: 65535
driver.parameter.vendorid: 09ae
driver.state: quiet
driver.version: 2.8.1-405-g76d0602fb
driver.version.internal: 0.36
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
input.frequency: 50.0
input.frequency.nominal: 50
input.voltage: 239
input.voltage.maximum: 247
input.voltage.minimum:   0
input.voltage.nominal: 230
ups.debug.D: 37 44 38 38 0d 00 00 '7D88...'
ups.debug.load_banks: 0
ups.debug.M: 30 30 38 31 0d 00 00 '0081...'
ups.debug.P: 30 31 30 30 30 58 0d '01000X.'
ups.debug.S: 31 30 30 00 64 30 0d '100.d0.'
ups.debug.T: 41 31 31 46 34 30 0d 'A11F40.'
ups.debug.V: 32 30 36 30 58 58 0d '2060XX.'
ups.delay.shutdown: 64
ups.firmware: F3009.C
ups.firmware.aux: protocol 3017
ups.id: 65535
ups.load: 0
ups.mfr: Tripp Lite
ups.model: TRIPP LITE SMX1200XLHG
ups.power.nominal: 1000
ups.productid: 0001
ups.status: OL
ups.temperature: 37.5
ups.vendorid: 09ae
```

biohead, 2024-01-31:
When putting everything together, and using the NUT cgistats as the visualising mechanism, the only things missing now is the battery % and output (VAC). 
Runtime is also missing, but I guess there is a dependency on %.

clepple, 2024-01-31:
There is a driver-side approximation `for battery.charge` that I just pushed. No guarantees as to the accuracy.

I don't know where the output voltage is - one of the other models stores it where this one reports load percentage. For the OMNIVS1000, I ramped the input voltage up and down with a variac, and watched what changed. Even if you get a variac, this one is going to complicate that because it has a voltage correction circuit.

None of the other models supported by this driver return runtime. That is usually estimated by the UPS firmware, because the UPS has better insight into the discharge statistics (voltage and current over time). Calculating runtime from battery percentage depends on the size of the battery, so you'd have an approximation on top of an approximation.

biohead, 2024-02-02:
Driver: 2.8.1-406-g49d4a5ed7 - looks as expected!
[image]

jimklimov, 2024-02-03:
Cross-referencing: fixes here are explored on branch https://github.com/networkupstools/nut/compare/2217-TL-proto-3017

jimklimov, 2025-09-17:
Hm, I think not, slipped through the cracks. Currently there's the branch above to build a custom NUT with proposed support.
I'll prepare a PR from that development brought up to spec with current NUT, thanks for the reminder.

jimklimov, 2025-09-22:
That seems like udev rules (or equivalent) did not get updated, so the NUT runtime user has no permissions to the device FS node as far as the OS is concerned.

For now, try adding `-x user=root` to the command line for testing - this would keep the `sudo`-started program running as `root` and  hop over this particular issue, if the rest works better we can see why that one went wrong.

fherreazcue, 2025-09-22:
thanks, I was not aware of that option... I tried with it and still it seems to return no regex match:
```
$ sudo ./drivers/nutdrv_qx -s tempups -d1 -DDDDDD -x port=auto -x user=root
   0.000000	[D5] send_to_all: SETINFO driver.state "init.starting"
Network UPS Tools 2.8.4.77.4-81+gec25f25a3 (development iteration after 2.8.4) - Generic Q* USB/Serial driver 0.45
USB communication driver (libusb 1.0) 0.50
   0.000114	[D1] upsdrv_makevartable...
   0.000141	[D5] send_to_all: SETINFO driver.version.usb "libusb-1.0.27 (API: 0x0100010A)"
   0.000148	[D1] Using USB implementation: libusb-1.0.27 (API: 0x0100010A)
   0.000197	[D3] main_arg: var='port' val='auto'
   0.000204	[D6] testinfo_reloadable: var=port, infoname=driver.parameter.port, newval=auto, reloadable=0, reload_flag=0
   0.000217	[D6] testinfo_reloadable: verdict for (re)loading var=port value: 1
   0.000225	[D5] send_to_all: SETINFO driver.parameter.port "auto"
   0.000231	[D3] main_arg: var='user' val='root'
   0.000237	[D6] testval_reloadable: var=user, oldval=nut, newval=root, reloadable=0, reload_flag=0
   0.000244	[D1] testval_reloadable: setting 'user' exists and differs: new value 'root' vs. 'nut'
   0.000248	[D6] testval_reloadable: verdict for (re)loading var=user value: 1
   0.000253	[D1] Overriding previously specified user 'nut' with 'root' specified for driver section
   0.000263	[D1] Network UPS Tools version 2.8.4.77.4-81+gec25f25a3 (development iteration after 2.8.4), 64-bit build for x86_64, built with gcc (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0 and configured with flags: --with-drvpath=/usr/local/ups/bin --sysconfdir=/etc/nut --with-group=nut --with-user=nut --enable-Wcolor --enable-configure-debug --enable-warnings --enable-Werror --enable-keep_nut_report_feature --with-all=auto --with-cgi=auto --with-serial=auto --with-dev=auto --with-nut_monitor=auto --with-pynut=auto --disable-force-nut-version-header --enable-check-NIT --enable-maintainer-mode --with-doc=skip --enable-docs-man-for-progs-built-only=no PKG_CONFIG_PATH=/usr/local/lib/x86_64-linux-gnu/pkgconfig:/usr/local/lib/pkgconfig:/usr/local/share/pkgconfig:/usr/lib/x86_64-linux-gnu/pkgconfig:/usr/lib/pkgconfig:/usr/share/pkgconfig --with-unmapped-data-points --with-debuginfo=auto
   0.000299	[D1] debug level is '6'
   0.000305	[D5] send_to_all: SETINFO driver.debug "6"
   0.000312	[D5] send_to_all: SETFLAGS driver.debug RW NUMBER
   0.001132	[D1] Succeeded to become_user(root): now UID=0 GID=0
   0.001159	[D1] Signalling UPS [tempups]: driver.exit (quietly, no fuss if no driver is running or responding)
   0.001175	Can't open /var/state/ups/nutdrv_qx-tempups: No such file or directory
   0.001183	[D1] Request for other driver to exit returned code -1
   0.001188	[D1] Socket dialog with the other driver instance (may be absent) failed: No such file or directory
   0.001196	[D5] send_to_all: SETINFO device.type "ups"
   0.001203	[D5] send_to_all: SETINFO driver.state "init.device"
   0.001210	[D1] upsdrv_initups...
   0.006776	[D2] Checking device 1 of 7 (1D6B/0003)
   0.027850	[D1] nut_libusb_open: invalid libusb bus number 0
   0.028106	[D2] - VendorID: 1d6b
   0.028116	[D2] - ProductID: 0003
   0.028122	[D2] - Manufacturer: Linux 6.14.0-29-generic xhci-hcd
   0.028128	[D2] - Product: xHCI Host Controller
   0.028133	[D2] - Serial Number: 0000:00:14.0
   0.028138	[D2] - Bus: 004
   0.028143	[D2] - Bus Port: unknown
   0.028148	[D2] - Device: 001
   0.028153	[D2] - Device release number: 0614
   0.028159	[D2] Trying to match device
   0.028165	[D3] match_function_regex: matching a device...
   0.028175	[D2] Device does not match - skipping

   0.028439	[D2] Checking device 2 of 7 (09AE/0001)
   0.032752	[D2] - VendorID: 09ae
   0.032778	[D2] - ProductID: 0001
   0.032782	[D2] - Manufacturer: Tripp Lite 
   0.032787	[D2] - Product: TRIPP LITE SMX1200XLHG 
   0.032791	[D2] - Serial Number: unknown
   0.032795	[D2] - Bus: 003
   0.032799	[D2] - Bus Port: 003
   0.032803	[D2] - Device: 003
   0.032807	[D2] - Device release number: 000a
   0.032811	[D2] Trying to match device
   0.032816	[D3] match_function_regex: matching a device...
   0.032822	[D2] Device does not match - skipping

   0.032847	[D2] Checking device 3 of 7 (8087/0033)
   0.126399	[D2] - VendorID: 8087
   0.126428	[D2] - ProductID: 0033
   0.126433	[D2] - Manufacturer: unknown
   0.126437	[D2] - Product: unknown
   0.126441	[D2] - Serial Number: unknown
   0.126444	[D2] - Bus: 003
   0.126448	[D2] - Bus Port: 010
   0.126452	[D2] - Device: 004
   0.126456	[D2] - Device release number: 0000
   0.126460	[D2] Trying to match device
[... 43 lines cut ...]
   0.265773	[D2] Checking device 7 of 7 (1D6B/0002)
   0.298109	[D1] nut_libusb_open: invalid libusb bus number 0
   0.298230	[D2] - VendorID: 1d6b
   0.298234	[D2] - ProductID: 0002
   0.298237	[D2] - Manufacturer: Linux 6.14.0-29-generic xhci-hcd
   0.298240	[D2] - Product: xHCI Host Controller
   0.298243	[D2] - Serial Number: 0000:00:0d.0
   0.298246	[D2] - Bus: 001
   0.298249	[D2] - Bus Port: unknown
   0.298254	[D2] - Device: 001
   0.298257	[D2] - Device release number: 0614
   0.298265	[D2] Trying to match device
   0.298282	[D3] match_function_regex: matching a device...
   0.298287	[D2] Device does not match - skipping
   0.299288	[D2] libusb1: No appropriate HID device found
   0.299312	No supported devices found. Please check your device availability with 'lsusb'
and make sure you have an up-to-date version of NUT. If this does not help,
try running the driver with at least 'subdriver', 'vendorid' and 'productid'
options specified. Please refer to the man page for details about these options
(man 8 nutdrv_qx).

   0.299331	[D5] send_to_all: SETINFO driver.state "cleanup.exit"
```

and i get similar results if i specify `-x vendorid=09ae -x productid=0001`.

jimklimov, 2025-09-22:
Oh, wait - when did we hop over to `nutdrv_qx`? The earlier discussion, and the fix, was for `tripplite_usb` :-D