networkupstools/nut #2265: NSS backend usage documentation and may code error

duncanwebb, 2024-01-16:
Version: Debian 2.8.0-7
Debian Release 12 bookworm

## Description
I'm failing to get upsd to run using with the [nss basckend](https://networkupstools.org/historic/v2.8.0/docs/user-manual.chunked/ar01s10.html#_nss_backend_usage)

The configuration is:
```config
LISTEN 0.0.0.0 3493
MAXAGE 25
DISABLE_WEAK_SSL true
CERTPATH /etc/nut/cert_db
CERTIDENT nut-server password
```
and
```
certutil -L -d /etc/nut/cert_db
Certificate Nickname                                         Trust Attributes
                                                             SSL,S/MIME,JAR/XPI
my-ca-cert                                                   CT,,
nut-server                                                   u,u,u
```
Keys:
```
certutil -K -d /etc/nut/cert_db/ -f ~/passwordfile
certutil: Checking token "NSS Certificate DB" in slot "NSS User Private Key and Certificate Services"
< 0> rsa      799fb1cb38f67481a0e6fffa252507ef59dce404   NSS Certificate DB:nut-server
```
## Problem
I've added printing the values just before the NSS_Init calls and this is what is shown:
```
upsd[1400123]: DJW: certname nut-server.
upsd[1400123]: DJW: certfile /etc/nut/cert_db.
upsd[1400123]: NSS_Init status -1.
upsd[1400123]: Can not initialize SSL context
upsd[1400123]: [D1] nss_error -8174 in upscli_init / NSS_[NoDB]_Init
```

The documentation says:
`CERTPATH /usr/local/ups/etc/cert_db`
Also tell which is the certificate to send to clients to authenticate itself and the password to decrypt private key associated to certificate:
`CERTIDENT 'certificate name' 'database password'`

If you add the CERTIDENT as described in the example, then it would read:
`CERTIDENT 'My nut server' password`
and in the code it would be passed as:
`DJW: certname 'My.`
Which means the quote is passed as the name and the name is truncated at the first space.

## Documentation
This could be improved if there is a working example and fix the space handling.

## Question
What am I doing incorrectly? I can't figure out what the problem is but suspect it might be something to do with signing the request and the trust attributes.

## More
`Sign a certificate request with the CA certificate (simulate a real CA signature): certutil -C -d CA_db -c "My Root CA" -a -i server.req -o server.crt -2 -6`
The -6 option is to add an extended key usage extension to a certificate. Which one to add?
Should this be:
`-6 serverAuth,critical`?

jimklimov, 2024-01-16:
I think NUT parser only uses the double quotes for token grouping, and your "screenshot" uses single quotes. So the line
````
CERTIDENT 'My nut server' password
````
actually carries four tokens (two of them involve a single quote and a word).

The dot in the end of DJW lines is added by the log printer, right?

duncanwebb, 2024-01-16:
The single quotes are in the "Notes on securing NUT" page, I'll try with double quotes.

Yes, the dot is from:
`upslogx(LOG_WARNING, "DJW: certname %s.", certname);`

duncanwebb, 2024-01-18:
> I think NUT parser only uses the double quotes for token grouping, and your "screenshot" uses single quotes.

Indeed, you are correct, double quotes group the tokens together. Thanks for that.

duncanwebb, 2024-01-20:
Thanks for fixing the documentation, this looks clearer.

The main problem is that NSS_Init returns with error -8174 and there is no error message.

Here is a patch with my messages and when running the server ./server/upsd -F -DDDD I see these messages:
```
upsd[1542706]: [D3] listen_add: added 0.0.0.0:3493
upsd[1542706]: [D1] debug level is '4'
upsd[1542706]: [D3] setuptcp: try to bind to 0.0.0.0 port 3493
upsd[1542706]: listening on 0.0.0.0 port 3493
upsd[1542706]: Connected to UPS [ups]: usbhid-ups-ups
upsd[1542706]: [D2] user_add_action: adding 'SET' for admin
upsd[1542706]: [D2] user_add_instcmd: adding 'ALL' for admin
upsd[1542706]: [D2] user_add_action: adding 'login' for admin
upsd[1542706]: [D2] user_add_action: adding 'master' for admin
upsd[1542706]: [D2] user_add_action: adding 'fsd' for admin
upsd[1542706]: [D2] user_add_action: adding 'login' for upsmon
upsd[1542706]: [D2] user_add_action: adding 'master' for upsmon
upsd[1542706]: [D2] user_add_action: adding 'fsd' for upsmon
upsd[1542706]: Running as foreground process, not saving a PID file
upsd[1542706]: [D1] DJW: PR_Initialized ok.
upsd[1542706]: [D1] DJW: certname nut-server.
upsd[1542706]: [D1] DJW: certfile /etc/nut/cert_db.
upsd[1542706]: Can not initialize SSL context
upsd[1542706]: [D1] DJW: error=-8174 in ssl_init / NSS_Init.
upsd[1542706]: [D1] DJW: os error=0.
upsd[1542706]: [D1] DJW: error text len=0.
```
The NSS_Init call fails.

[djw.patch](https://github.com/networkupstools/nut/files/13997890/djw.patch)

I've checked out the source for nss-3.87.1 and hacked nss-3.87.1/nss/cmd/tests/baddbdir.c and changed the NSS_InitReadWrite call to NSS_Init.

This code change results in "NSS_InitReadWrite succeeded unexpectedly" message. Which means that NSS_Init succeeded.
```
/* This Source Code Form is subject to the terms of the Mozilla Public
 * License, v. 2.0. If a copy of the MPL was not distributed with this
 * file, You can obtain one at http://mozilla.org/MPL/2.0/. */

#include <stdio.h>
#include <stdlib.h>

#include "nss.h"
#include "secerr.h"

/*
 * Regression test for bug 495097.
 *
 * NSS_InitReadWrite("sql:<dbdir>") should fail with SEC_ERROR_BAD_DATABASE
 * if the directory <dbdir> doesn't exist.
 */

int
main()
{
    SECStatus status;
    int error;

    /* status = NSS_InitReadWrite("sql:/etc/nut/cert_db"); // this succeeded */
    /* status = NSS_InitReadWrite("/etc/nut/cert_db"); // this succeeded */
    status = NSS_Init("/etc/nut/cert_db");
    if (status == SECSuccess) {
        fprintf(stderr, "NSS_InitReadWrite succeeded unexpectedly\n");
        exit(1);
    }
    error = PORT_GetError();
    if (error != SEC_ERROR_BAD_DATABASE) {
        fprintf(stderr, "NSS_InitReadWrite failed with the wrong error code: "
                        "%d\n",
                error);
        exit(1);
    }
    printf("PASS\n");
    return 0;
}
```

Does anybody have any idea why NSS_Init is failing? I've run out of ideas :(
Thanks

jimklimov, 2024-01-20:
Thanks for the exploration. So far - no idea really, but just looking at the method names, maybe `InitReadWrite` implies that the certdb files should be writeable (by the `nut` account)? Are they on your system?

duncanwebb, 2024-01-20:
You found the problem, and it was that the ownership was root and the permissions were `-rw-------` and these permissions are set by `certuil` so the database could not be opened. I used `cp -ax cert_db /etc/nut` so the ownership and permissions were retailed. Having already run `chown root:nut /etc/nut/*` and `chmod 640 /etc/nut/*` the files remained unchanged.

Now I see:
```
upsd[1544650]: listening on 0.0.0.0 port 3493
upsd[1544650]: Connected to UPS [ups]: usbhid-ups-ups
upsd[1544650]: Running as foreground process, not saving a PID file
upsd[1544650]: Intend to retrieve password for NSS User Private Key and Certificate Services / NSS Certificate DB: password configured
upsd[1544650]: SSL handshake done successfully with client 192.168.3.210
upsd[1544650]: SSL handshake done successfully with client 192.168.3.223
upsd[1544650]: User upsmon@192.168.3.210 logged into UPS [ups] (SSL)
upsd[1544650]: User upsmon@192.168.3.223 logged into UPS [ups] (SSL)
upsd[1544650]: SSL handshake done successfully with client 192.168.3.205
```

I see this in doc/rst/legacy/nss_sample_code/nss_sample_code_sample6/index.rst
```
        /* The NSS db must be initialized read-write since we'll be creating
         * keys in it. Once keys are generated, it can be opened without read-write
         * subsequently (NSS_Init).
         */
```

This issue is now solved.
Many thanks. :)

closed 2024-01-20 by duncanwebb

jimklimov, 2024-01-20:
> Having already run `chown root:nut /etc/nut/*` and `chmod 640 /etc/nut/*` the files remained unchanged.

Just in case, is your `/etc/nut/cert_db` a sub-directory or a prefix for filenames right in `/etc/nut`? If it is a directory, then `chmod 640 /etc/nut/*` would have been counter-productive (`750` or GNUish `-X` for dirs) ;)

jimklimov, 2024-01-20:
Double-checked, generally chown/chmod (and `0755`) are all covered in `docs/security.txt` - so nothing more to fix here :)

duncanwebb, 2024-01-20:
openssh fails when the permissions for "other" have read, so 750 and 640 are better. When the cert_db is copied to other machines, the permissions can change.

The code could check if the files in cert_db can be read ad the error code -8174 is not documented.

There are a few potential buffer overruns in the code caused by PR_GetErrorText(buffer) - better call PR_GetErrorTextLength() before PR_GetErrorText(buffer).

jimklimov, 2024-01-20:
Would you care to post your patch as a real PR to make it easier to follow up? :)

duncanwebb, 2024-01-20:
Will do, tomorrow or Monday.

PR #2268 (2024-01-19): Update upsd/upsmon conf docs regarding NSS and general asciidoc markup
Closes: #2265