networkupstools/nut #2267: Driver nutdrv_qx not working with Green Cell UPS04 but blazer_usb

JanGerritsen, 2024-01-17:
**I have a new Green Cell Uninterruptible Power Supply UPS 1500VA 900W with LCD Display | EU VERSION aka. UPS04** 

```sh
# nut-scanner -U
Scanning USB bus.
[nutdev1]
        driver = "nutdrv_qx"
        port = "auto"
        vendorid = "0001"
        productid = "0000"
        product = "MEC0003"
        vendor = "MEC"
        bus = "003"
```

**Syslog reports:**
```
nut-driver@nutdev1[1885]: Using protocol: Q1 0.07
nut-driver@nutdev1[1885]: Can't autodetect number of battery packs [-1/27.10]
nut-driver@nutdev1[1885]: Battery runtime will not be calculated (runtimecal not set)
nutdrv_qx[1885]: Can't autodetect number of battery packs [-1/27.10]
nutdrv_qx[1885]: Battery runtime will not be calculated (runtimecal not set)
nutdrv_qx[2402]: Startup successful
```

**Battery charge not detected:**
```sh
# upsc nutdev1@localhost
Init SSL without certificate database
battery.voltage: 27.10
device.type: ups
driver.name: nutdrv_qx
driver.parameter.bus: 003
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 1
driver.parameter.port: auto
driver.parameter.product: MEC0003
driver.parameter.productid: 0000
driver.parameter.synchronous: auto
driver.parameter.vendor: MEC
driver.parameter.vendorid: 0001
driver.version: 2.8.0
driver.version.data: Q1 0.07
driver.version.internal: 0.32
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
input.frequency: 50.0
input.voltage: 229.4
input.voltage.fault: 230.3
output.voltage: 230.3
ups.beeper.status: enabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.load: 8
ups.productid: 0000
ups.status: OL
ups.type: offline / line interactive
ups.vendorid: 0001
```

**When I switch to the `blazer_usb` driver it looks good and no problems are repoted in syslog:**
```sh
# upsc nutdev1@localhost
Init SSL without certificate database
battery.charge: 100
battery.voltage: 27.10
battery.voltage.high: 26.00
battery.voltage.low: 20.80
battery.voltage.nominal: 24.0
device.type: ups
driver.name: blazer_usb
driver.parameter.bus: 003
driver.parameter.pollinterval: 1
driver.parameter.port: auto
driver.parameter.product: MEC0003
driver.parameter.productid: 0000
driver.parameter.synchronous: auto
driver.parameter.vendor: MEC
driver.parameter.vendorid: 0001
driver.version: 2.8.0
driver.version.internal: 0.14
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
input.current.nominal: 6.0
input.frequency: 50.0
input.frequency.nominal: 50
input.voltage: 226.9
input.voltage.fault: 227.4
input.voltage.nominal: 230
output.voltage: 226.9
ups.beeper.status: enabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.load: 10
ups.productid: 0000
ups.status: OL
ups.type: offline / line interactive
ups.vendorid: 0001
```

I am reporting this, because the driver `blazer_usb` has been declared deprecated, but the successor `nutdrv_qx` is not working correctly for me.

```
nut-driver@nutdev1[9745]: Please note that this driver is deprecated and will not receive
nut-driver@nutdev1[9745]: new development. If it works for managing your devices - fine,
nut-driver@nutdev1[9745]: but if you are running it to try setting up a new device, please
nut-driver@nutdev1[9745]: consider the newer nutdrv_qx instead, which should handle all 'Qx'
nut-driver@nutdev1[9745]: protocol variants for NUT. (Please also report if your device works
nut-driver@nutdev1[9745]: with this driver, but nutdrv_qx would not actually support it with
nut-driver@nutdev1[9745]: any subdriver!)
```

jimklimov, 2024-01-20:
Thanks for the highlight. I think there were some fixes merged for these drivers after 2.8.0 release, so if you'd have a chance to build and test the current codebase (whether the issue remains or got solved), it would be helpful.

Note that you are not required to install the test build over your packaged system just for testing (you can do so if you like the results though).

See https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests for details

jimklimov, 2024-01-20:
Please see also https://github.com/networkupstools/nut/issues/2253#issuecomment-1902129651 for more suggestions and ideas...