networkupstools/nut #2271: Support for Liebert Corp PowerSure PST UPS

digitlman, 2024-01-19:
Need help getting driver support for this unit:

Bus 001 Device 096: ID 10af:0002 Liebert Corp. PowerSure PST UPS
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0         8
  idVendor           0x10af Liebert Corp.
  idProduct          0x0002 PowerSure PST UPS
  bcdDevice            0.02
  iManufacturer           3 Vertiv Co.
  iProduct                1 Liebert PST5
  iSerial                 5 2309310073A4A3C
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x003b
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0
      bInterfaceProtocol      0
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode           33 US
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      27
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               5
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0
      bInterfaceProtocol      0
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode           33 US
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     816
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval             255

Supposedly has kernel support:  https://linux-hardware.org/?id=usb:10af-0002

digitlman, 2024-01-19:
root@NUT01:/home/it#  sudo /lib/nut/usbhid-ups -a Liebert_LR -DDD
Network UPS Tools - Generic HID driver 0.47 (2.8.0)
USB communication driver (libusb 1.0) 0.43
   0.000000     [D1] debug level is '3'
   0.001752     [D2] Initializing an USB-connected UPS with library libusb-1.0.26 (API: 0x1000109) (NUT subdriver name='USB communication driver (libusb 1.0)' ver='0.43')
   0.001849     [D1] upsdrv_initups (non-SHUT)...
   0.037827     [D2] Checking device 1 of 5 (7392/7811)
   0.038055     [D1] Failed to open device (7392/7811), skipping: Access denied (insufficient permissions)
   0.038123     [D2] Checking device 2 of 5 (10AF/0002)
   0.038213     [D1] Failed to open device (10AF/0002), skipping: Access denied (insufficient permissions)
   0.038265     [D2] Checking device 3 of 5 (0424/EC00)
   0.038362     [D1] Failed to open device (0424/EC00), skipping: Access denied (insufficient permissions)
   0.038413     [D2] Checking device 4 of 5 (0424/9514)
   0.038498     [D1] Failed to open device (0424/9514), skipping: Access denied (insufficient permissions)
   0.038550     [D2] Checking device 5 of 5 (1D6B/0002)
   0.038634     [D1] Failed to open device (1D6B/0002), skipping: Access denied (insufficient permissions)
   0.038684     [D2] libusb1: No appropriate HID device found
   0.038723     libusb1: Could not open any HID devices: insufficient permissions on everything
   0.038793     No matching HID UPS found

digitlman, 2024-01-19:
root@NUT01:/# hostnamectl
 Static hostname: NUT01
       Icon name: computer
      Machine ID: 9191adec98c9415e8718d0ae390cd465
         Boot ID: 4ebcc863c80241e697596f316547cc7a
Operating System: Raspbian GNU/Linux 12 (bookworm)
          Kernel: Linux 6.1.72-v7+
    Architecture: arm

jimklimov, 2024-01-19:
This looks like it has a chance to be covered by `usbhid-ups` with a `belkin-hid` subdriver (per comments to some related devices, there may be scaling issues for some of the measurements that could need further adaptations of codebase to map them properly).

However, product ID 0x0002 is not currently known in the driver sources. This has two implications:

* driver would not try to match the IDs it does not know about -- I can only suggest trying either a custom build of the version of NUT you have packaged (update `drivers/belkin-hid.c` around line 70, in the `belkin_usb_device_table[]`, to add the ID) perhaps using its packaging recipes, or just a custom build of current NUT (notably `usbhid-ups` recently in #2058 had updates to try specified subdrivers for given vendorid+productid) - see https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests about building NUT and testing the driver from the build workspace; eventually installing if you like the results
* generated `udev.rules` or similar do not tell the OS to `chown` the device filesystem nodes to `nut` run-time user -- you can address this with `sudo /lib/nut/usbhid-ups -a Liebert_LR -DDD -x user=root` while testing (eventually a `user=root` line in its `ups.conf` section)

jimklimov, 2024-01-19:
I've proposed an exploratory fix in PR #2272 - you can try to build that codebase per wiki instructions linked earlier, using this for the initial git checkout:
````
:; git clone -b issue-2271 https://github.com/jimklimov/nut nut-issue-2271
:; cd nut-issue-2271
### ...go on with the rest of config/build/test(/install) routine
````

digitlman, 2024-01-19:
Jim,

Thanks for your prompt reply.  I will see if I can get some testing done this weekend.

jimklimov, 2024-01-22:
Cheers, do we have any lucky progress here? :)

digitlman, 2024-01-23:
I am trying to follow the directions for installing the test on the raspian
system.  Trying is the optimum word.

jimklimov, 2024-02-09:
Gentle bump? :) Note I've updated the previous post with a few more ideas...

jimklimov, 2024-02-10:
@electroflame : just in case, are your NUT containers the HomeAssistant plugin? If yes, see our Wiki for community hints about building one. Hope that helps.

Also, is a NUT container useful (e.g. to shut down its host and bring down the UPS in the end)?

jimklimov, 2024-02-12:
Following a discussion from another thread, just as a guess - check if `nutdrv_qx` or `blazer_usb` actually handle this device better than `usbhid-ups`? (e.g. if this device was released with VID:PID suggesting it is same as those which talk USB HID protocol, but in fact it talks one of Megatec Qx protocols and the HID support is a rudimentary stub)

jimklimov, 2024-02-12:
In that case, I'd suggest checking in the host system whether NUT drivers behave better, using a custom build or maybe by `chroot`ing into the mounted container filesystem, but avoiding the virtualization layers which might confuse the devfs access permissions. Thinking of the latter, is your container privileged, or does the host see it running as processes all owned by some one user account (which is not listed as the USB node owner via NUT `udev.rules` or similar snippet on the host)?

jimklimov, 2024-02-14:
Well, `nut-scanner` does not see the device because its VID:PID are not built in as known and supported (the crux of PR #2272 waiting for experimental confirmations like yours to get merged - whether just IDs suffice or some more work is needed for the device model).

Starting with NUT v2.8.1 (or master) you should be able to pass a `subdriver` option to `usbhid-ups` driver, check its run-time help for acceptable values (can specify substrings though, as a case-insensitive exact/regex match), e.g.:
````
:; ./drivers/usbhid-ups -h
...
Acceptable values for 'subdriver' via -x or ups.conf in this driver (exact names here,
case-insensitive sub-strings may be used, as well as regular expressions):
 "EXPLORE HID 0.2", "MGE HID 1.46", "APC HID 0.100", "Arduino HID 0.21",
 "Belkin/Liebert HID 0.18", "CyberPower HID 0.8", "Delta UPS HID 0.6",
 "Ever HID 0.10", "iDowell HID 0.2", "Legrand HID 0.2",
 "Phoenixtec/Liebert HID 0.41", "openUPS HID 0.5",
 "PowerCOM HID 0.7", "Powervar HID 0.20",
 "Salicru HID 0.4", "TrippLite HID 0.85"
````

Try Belkin/Liebert; also maybe Phoenixtec/Liebert or MGE as likely related variants.

jimklimov, 2024-02-17:
From indications in the issues so far, different Pi models may have had different quality USB chips. There were many reports about issues with Pi3 (maybe owing more to platform popularity than HW quality, though) which subsided around Pi4 and on the rise with Pi5. Take with a grain of salt - I have no 100% conclusions here.

Regarding a fresh install - which NUT version was there? Packaged (and 2.8.1?) or one you've built from Git master or the exploratory fix in PR #2272? (I guess building AND installing only the latter would get around the permission option problem, as this is the only NUT source which "knows" the new ID.) Did you try with the `subdriver` option?

jimklimov, 2024-02-17:
Well, name/model/etc. were the basics per USB HID, so the interesting part is if it pulls the actual UPS power device info :)

If you followed the Wiki article (and config prereqs that it references), you would have `ccache` installed and used by `ci_build.sh` script, which should speed up rebuilds of files where the sources did not change.

jimklimov, 2024-02-19:
For kicks, can you try adding a `pollonly` flag to `usbhid-ups` driver settings?

jimklimov, 2024-02-20:
Interesting idea... There are some devices and drivers using a non-zero interface, etc. IIRC Arduino-based experiments were much of the recent context on this, and the different numbers became driver config fields (with no exposure yet via config files, I think). Maybe an id-based tweak could help in upsdrv_init() or similar... Can you experiment on that?

jimklimov, 2024-03-01:
Hello, can you please check if https://github.com/networkupstools/nut/pull/2331 helps?

Follow https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests with this branch for the build:
````
git clone https://github.com/jimklimov/nut -b issue-2149
````

jimklimov, 2024-03-01:
I've got a confirmation there, so maybe would just release this change once CI is happy with syntax nuances :D and then it would be part of NUT `master` branch builds (per Wiki).

jimklimov, 2024-03-02:
Relevant PRs were merged to `master` by now, so "standard" text of https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests would apply for testing.

jimklimov, 2024-03-04:
Thanks! The `output.frequency: 6000.00` looks off by a couple of orders of magnitude. Are you in USA or another country with 60Hz-based utility power network? (Probably a data conversion model-dependent tweak in the subdriver mapping table can take care of that).

With `output.voltage: 0.0`, I am not sure this subdriver does report more than the other one :-}

jrjparks, 2024-03-05:
I was able to get the below with this change. I was very hacky with `liebert_line_voltage_fun` to get the right value for me, so will definitely need to be done better to merge in.

```diff
diff --git a/drivers/belkin-hid.c b/drivers/belkin-hid.c
index 686dc9261..21f3c9729 100644
--- a/drivers/belkin-hid.c
+++ b/drivers/belkin-hid.c
@@ -183,7 +183,11 @@ static const char *liebert_config_voltage_fun(double value)
 static const char *liebert_line_voltage_fun(double value)
 {
        if( value < 1 ) {
-               if( fabs(value - 1e-7) < 1e-9 ) {
+               if( fabs(value - 1e-3) < 1e-3 ) {
+                       liebert_line_voltage_mult = 1e5;
+                       upsdebugx(2, "Input/OutputVoltage = %g -> assuming correction factor = %g",
+                               value, liebert_line_voltage_mult);
+               } else if( fabs(value - 1e-7) < 1e-9 ) {
                        liebert_line_voltage_mult = 1e7;
                        upsdebugx(2, "Input/OutputVoltage = %g -> assuming correction factor = %g",
                                value, liebert_line_voltage_mult);
@@ -480,6 +484,17 @@ static hid_info_t belkin_hid2nut[] = {
   { "battery.voltage", 0, 0, "UPS.PowerSummary.Voltage", NULL, "%s", 0, liebert_line_voltage_info },
   { "battery.voltage.nominal", 0, 0, "UPS.PowerSummary.ConfigVoltage", NULL, "%s", HU_FLAG_STATIC, liebert_config_voltage_info },
   { "ups.load", 0, 0, "UPS.Output.PercentLoad", NULL, "%.0f", 0, NULL },
+  /* Liebert PSI5 */
+  { "input.voltage.nominal", 0, 0, "UPS.Flow.ConfigVoltage", NULL, "%.0f", 0, NULL },
+  { "input.frequency", 0, 0, "UPS.PowerConverter.Input.Frequency", NULL, "%s", 0, divide_by_100_conversion },
+  { "input.voltage", 0, 0, "UPS.PowerConverter.Input.Voltage", NULL, "%s", 0, liebert_line_voltage_info },
+  { "output.voltage.nominal", 0, 0, "UPS.Flow.ConfigVoltage", NULL, "%.0f", 0, NULL },
+  { "output.frequency", 0, 0, "UPS.PowerConverter.Output.Frequency", NULL, "%s", 0, divide_by_100_conversion },
+  { "output.voltage", 0, 0, "UPS.PowerConverter.Output.Voltage", NULL, "%s", 0, liebert_line_voltage_info },
+  { "ups.load", 0, 0, "UPS.OutletSystem.Outlet.PercentLoad", NULL, "%.0f", 0, NULL },
+  { "battery.voltage", 0, 0, "UPS.BatterySystem.Battery.Voltage", NULL, "%s", 0, liebert_line_voltage_info },
+  { "battery.voltage.nominal", 0, 0, "UPS.BatterySystem.Battery.ConfigVoltage", NULL, "%.0f", 0, NULL },
+  { "battery.capacity", 0, 0, "UPS.Flow.ConfigApparentPower", NULL, "%.0f", 0, NULL },
   /* status */
   { "BOOL", 0, 0, "UPS.PowerSummary.Discharging", NULL, NULL, HU_FLAG_QUICK_POLL, liebert_discharging_info }, /* might not need to be liebert_* version */
   { "BOOL", 0, 0, "UPS.PowerSummary.Charging", NULL, NULL, HU_FLAG_QUICK_POLL, liebert_charging_info },
diff --git a/drivers/usbhid-ups.c b/drivers/usbhid-ups.c
index 4775d38fe..6481092c6 100644
--- a/drivers/usbhid-ups.c
+++ b/drivers/usbhid-ups.c
@@ -553,6 +553,22 @@ info_lkp_t divide_by_10_conversion[] = {
        { 0, NULL, divide_by_10_conversion_fun, NULL }
 };
 
+/* returns statically allocated string - must not use it again before
+   done with result! */
+static const char *divide_by_100_conversion_fun(double value)
+{
+       static char buf[20];
+
+       snprintf(buf, sizeof(buf), "%0.1f", value * 0.01);
+
+       return buf;
+}
+
+/* FIXME? Do we need an inverse "nuf()" here? */
+info_lkp_t divide_by_100_conversion[] = {
+       { 0, NULL, divide_by_100_conversion_fun, NULL }
+};
+
 /* returns statically allocated string - must not use it again before
    done with result! */
 static const char *kelvin_celsius_conversion_fun(double value)
diff --git a/drivers/usbhid-ups.h b/drivers/usbhid-ups.h
index 2e8a3fced..8bc1f94f2 100644
--- a/drivers/usbhid-ups.h
+++ b/drivers/usbhid-ups.h
@@ -114,6 +114,7 @@ extern info_lkp_t date_conversion[];
 extern info_lkp_t hex_conversion[];
 extern info_lkp_t stringid_conversion[];
 extern info_lkp_t divide_by_10_conversion[];
+extern info_lkp_t divide_by_100_conversion[];
 extern info_lkp_t kelvin_celsius_conversion[];
 
 /* ---------------------------------------------------------------------- */
```

[image]

[image]

jimklimov, 2024-03-05:
Thanks for the suggested fixes. Actually does not look as "bad" as you made believe :)

One thing that looks like it needs more attention is the ordering of `if` clauses in `liebert_line_voltage_fun()` so the bigger delta does not rule out processing of the smaller ones.

Also, for whatever reason the original code went with two orders of magnitude in the difference, e.g. `if (fabs(value - 1e-7) < 1e-9) { liebert_line_voltage_mult = 1e7; ... }` and yours goes `if (fabs(value - 1e-3) < 1e-3) { liebert_line_voltage_mult = 1e5; ... }`

I think this needs a bit more investigation in `git blame` comments for the commit history of this line/function (scrolling on Github should be helpful, easier than in command line) or some spec. Even (or especially) if the different approaches for the earlier handled devices and your new one are the correct way to go, a comment in code would be helpful so nobody "fixes" it as a loose thread in the world later :)

jimklimov, 2024-03-19:
@jrjparks : would you please care to open a pull request with this change (and revise my earlier comment, if it does make sense or not)?

jrjparks, 2024-03-19:
Yeah I can make one this weekend

PR #2369 (2024-03-24): feat: Add support for Liebert PSI5
Adding support for the Liebert PSI5 model of UPS, as discussed in #2271 initially.

Maybe addresses similar complaints posted in #2024 

Closes: #2271
closed 2024-03-24 by jimklimov