networkupstools/nut #2278: V2.8.1 Nearly immediate shutdown on powerloss

GeneralGresi, 2024-01-25:
After the update from 2.8.0 to 2.8.1 the behaviour of nut, connected via snmp to an Eaton 5PX UPS, changed drastically.

On power loss, after about 30-60 seconds, all connected devices, master and slaves shut down immediately.
With 2.8.0, this only happened as soon as the UPS reached it's configured "left" battery time (300 seconds)

In the logfiles, there is an indication of a set FSB Flag:
Jan 25 18:33:03 upsmon 89442 Auto logout and shutdown proceeding
Jan 25 18:33:03 upsmon 89442 Executing automatic power-fail shutdown
Jan 25 18:33:03 upsd 22669 Client local-monitor@127.0.0.1 set FSD on UPS [Eaton_5PX2200]
Jan 25 18:33:03 upsmon 89442 UPS [Eaton_5PX2200] is reported as (administratively) OFF
Jan 25 18:33:03 upsmon 89442 UPS Eaton_5PX2200: administratively OFF or asleep
Jan 25 18:33:03 upsmon 89442 UPS Eaton_5PX2200 on battery

Comparing with a UPS on 2.8.0, I don't get the FSB message, also no indication of beeing "administratively OFF or asleep".
Jan 25 19:42:38 Mastermind upsmon[17548]: UPS Eaton_5PX2200@127.0.0.1 on battery
Jan 25 19:43:38 Mastermind upsmon[17548]: UPS Eaton_5PX2200@127.0.0.1 on line power

However, compared to the logdata of the UPS itself, there is no indication that this flag comes from the UPS.
2024/01/25 18:32:52 Normal AC NOK
2024/01/25 18:32:52 ABM state resting
2024/01/25 18:32:52 UPS on battery
2024/01/25 18:32:53 System shutdown in 2 h 17 mn 16 s
2024/01/25 18:32:53 Outlet group 1 shutdown in 2 h 49 mn 35 s
2024/01/25 18:32:53 Outlet group 2 shutdown in 2 h 49 mn 35 s
2024/01/25 18:32:53 ABM state Off
2024/01/25 18:32:53 Normal AC voltage out of tolerance

As logged there should be at least 2h of runtime. (which it normally has, just had an incident about 3 weeks ago which lasted more than two hours, and all boxes were still going strong). Back then, I did not have version 2.8.1 installed, I was still below. 

For testing, and for ruling out communication issues, I unplugged the UPS from mains about 5 times, every time, the box with 2.8.1 (and it's slaves) went black, every other box with 2.8.0 or 2.7.x are staying alive.
The UPS was still reachable an queryable during the time of my staged power loss tests.

Checking with "upsc", it seems like in a normal power loss situation, this type of UPS reports an "OFF" flag during power loss.
[image]

I stumbled accross issue #2104 which seems kind of related, however, there it only happened during self-tests or calibrations, not during a "normal" power loss situation.

jimklimov, 2024-01-26:
Thank you for the report, sounds serious.

Can you please experiment a bit with a neutered `SHUTDOWNCMD` (e.g. just logging to a file or `wall` but not actually shutting down), to check if the `OFF` flag dissipates after a few seconds or stays for a long time? With calibrations I did see it happen (in users' posted logs) for a few seconds, so it was possible to code some tolerance to ignore it if intermittent. Thinking of it (not at a computer now), I wonder if at least that use-case got already addressed on the master branch (eventually will be 2.8.2)?

desertwitch, 2024-01-26:
https://github.com/networkupstools/nut/blob/746cb86110982920b9db8f5066b766d560d19eef/clients/upsmon.c#L1101-L1108

I think the reason is the `||` condition here, either `linestate == 0` **or** `offstate == 1` will trigger a critical UPS when the UPS has a `ST_OFF` flag (which it automatically gets encountering `OFF`, only the `offstate` gets set later after `offdurationtime` is exceeded). While `offstate` is still 0 (due to `offdurationtime` not being exceeded yet) an `OB` state will set `linestate` to 0 immediately and this alone will satisfy this IF clause, because we'll have `ST_OFF`, `offdurationtime >= 0` and `linestate == 0`. It'll disregard the fact that `offstate` hasn't been set to 1 yet (and the `OFF` considered to be long enough to warrant a critical UPS)

We'll probably need `&&` to make sure this only happens when the UPS is **both** OB (has `linestate == 0`) **AND** is considered `OFF` for long enough (has `offstate == 1`):

```
 /* administratively OFF (long enough, see OFFDURATION) */ 
 if (flag_isset(ups->status, ST_OFF) && offdurationtime >= 0 
 && (ups->linestate == 0 && ups->offstate == 1)) { 
 	upslogx(LOG_WARNING, 
 		"UPS [%s] is reported as (administratively) OFF", 
 		ups->sys); 
 	return 1; 
 } 
```

desertwitch, 2024-01-26:
On another (sanity-checking) thought, perhaps the second piece of code I've suggested would be the better choice because it's likely possible that a UPS is `OL` (on mains power) while at the same time not powering the load itself (administratively `OFF`) so we shouldn't make a `OFF`-induced shutdown conditional on `OB` (not on mains power), just on exceeding the `offdurationtime` (`offstate == 1`). So this would likely be the better choice:

```
 /* administratively OFF (long enough, see OFFDURATION) */ 
 if (flag_isset(ups->status, ST_OFF) && offdurationtime >= 0 
 && ups->offstate == 1) { 
 	upslogx(LOG_WARNING, 
 		"UPS [%s] is reported as (administratively) OFF", 
 		ups->sys); 
 	return 1; 
 } 
```

GeneralGresi, 2024-01-26:
I did the checks with a neutralized SHUTDOWNCMD mentioned by @jimklimov
After 10 minutes being on battery, upsc still shows:
ups.status: FSD OB OFF OB

On a machine with 2.8.0 it shows just 
ups.status: OB OFF OB

So the OFF State isn't just short lived.
However as mentioned, on the Webinterface of the UPS there is no indication a cause for that OFF State.
Runtime still >2h, no errors whatsoever.

Therefore, If I understand correctly, even if the offdurationtime is set to a very big value, this would just shift the result of everything shutting down a bit in time.

I don't see why the UPS would be considered Administratively OFF, or it reports itself as OFF - at least that what nut says, as it's clearly running, and powering everything. Could this be an issue in the snmp driver?

desertwitch, 2024-01-26:
Some UPS report this state (wrongly so?), until this issue is fixed you can set `OFFDURATION -1` in `UPSMON.CONF`.
This will disable the OFF-timer entirely and not cause a shutdown because of `OFF` itself, which should be what you want.

jimklimov, 2024-01-26:
CC @aquette  for insights about Eaton SNMP nuances

jimklimov, 2024-01-26:
@GeneralGresi : would you have a chance to custom-build NUT per e.g. https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests to check if the updated upsmon binary from the PR branch would behave better for you?

GeneralGresi, 2024-01-26:
Built and installed the version from the PR, also removed the previously set (and working btw) OFFDURATION -1 setting in upsmon.conf.
Unfortunately the behaviour did not change, UPS still reports FSB OB OFF OB after unplugging from mains.
Shortly after, I get the messages for autologout and shutdown procedures, and the walled message from the neutralized SHUTDOWNCMD.

desertwitch, 2024-01-27:
> Built and installed the version from the PR, also removed the previously set (and working btw) OFFDURATION -1 setting in upsmon.conf.
>
> Unfortunately the behaviour did not change, UPS still reports [...]

Thanks for reporting back, in your case it's no surprise you're still getting a shutdown because your OFF status persists longer than the default OFFDURATION of 30 seconds. 

Can you confirm for us the shutdown doesn't happen instantly anymore, but very slightly delayed, and the logs now include a message about "assuming the line is not fed" as compared to before? At least this was the case in my testings of the pull request's code changes and my local replications of your issue.

GeneralGresi, 2024-01-27:
Definetly having a delay now.
Even a new Logmessage:
Jan 27 00:25:36 Mastermind nut-monitor[169159]: UPS Eaton_5PX2200@127.0.0.1 on battery
Jan 27 00:25:36 Mastermind nut-monitor[169159]: UPS Eaton_5PX2200@127.0.0.1: administratively OFF or asleep
[...]
Jan 27 00:26:11 Mastermind nut-monitor[169159]: ups_is_off: Eaton_5PX2200@127.0.0.1 is in state OFF for 35 sec, assuming the line is not fed (if it is calibrating etc., check the upsmon 'OFFDURATION' option)
[...]
Jan 27 00:26:11 Mastermind nut-monitor[169159]: UPS [Eaton_5PX2200@127.0.0.1] is reported as (administratively) OFF
Jan 27 00:26:11 Mastermind nut-monitor[169159]: Executing automatic power-fail shutdown
Jan 27 00:26:11 Mastermind nut-monitor[169159]: Auto logout and shutdown proceeding

jimklimov, 2024-01-27:
So I guess this boils down to a few follow-ups:
* want an option to NOT consider "administrative OFF" as a factor in decision-making, for broken devices/drivers which abuse this state report - hopefully `OFFDURATION = -1` does tick this box (maybe should be announced in the debug logs for shutdown decision e.g. if not configured)
* figure out of this particular NUT sub-driver is broken (looks at a wrong OID in the MIB, maybe prefers a less suitable one if it lists several as many SNMP mappings do; mis-interprets what it sees; etc.) and fix it, if possible.

closed 2024-01-29 by jimklimov