networkupstools/nut #2280: Devices with power monitoring support

Limeth, 2024-01-26:
Hello! I'd like to buy a UPS with power monitoring (in watts on the output sockets), but I can't figure out which ones support this feature. I was hoping it would be documented somewhere, but I'm not having much luck.
In particular, I'm currently looking at the Eaton 5E900UF, but wasn't able to deduce from its spec sheet much of anything about the supported features in the USB communications.
Any advice would be greatly appreciated, thanks!

trallnag, 2024-03-17:
I have the Ellipse PRO 650 and it works great.

```
battery.type: PbAc
device.mfr: EATON
device.model: Ellipse PRO 650 
device.type: ups
driver.name: usbhid-ups
driver.parameter.vendorid: 0463
driver.version: 2.8.0
driver.version.data: MGE HID 1.46
driver.version.internal: 0.47
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
ups.firmware: 01.16.0020
```

Limeth, 2024-07-09:
Thanks! I ended up getting the 5E900UF and it seems to provide the required information.

closed 2024-07-09 by Limeth