networkupstools/nut #2282: tripplite_usb: if `upsid` is not specified, any unit ID should match

clepple, 2024-01-27:
It appears that between 2.8.0 and 746cb86110982920b9db8f5066b766d560d19eef (including 2.8.1), `tripplite_usb` stopped matching the default unit ID of 65535/0xffff when `upsid` is not specified in the configuration file.

Although I don't have hardware available to confirm, I suspect it is commit 106e7c6fe7, so I am tagging @sapireli.

IMHO, if `upsid` is not passed, then the matching function should still be `NULL` here: https://github.com/networkupstools/nut/commit/106e7c6fe7e0c67a7964e254ae8983f4537d8103#diff-51229908f9631058c142955b4726606c68b280194b6aeb184fdc9a17766f924eR1652

I would also recommend adding some logging during the matching process - the current debug message does not indicate that the ID doesn't match: https://github.com/networkupstools/nut/issues/2258#issuecomment-1912097611

clepple, 2024-01-27:
I don't really like the idea of defaulting to unit ID 65535, but I could be convinced, especially if the code uses `DEFAULT_UPSID` in the matcher as well as the help string (and the man page is updated to reflect that): https://github.com/networkupstools/nut/commit/106e7c6fe7e0c67a7964e254ae8983f4537d8103#diff-51229908f9631058c142955b4726606c68b280194b6aeb184fdc9a17766f924eR337

As the man page indicates, it is possible to change the unit ID with `upsrw`, but that is only possible after the driver is up and running. This complicates changing the unit ID if an UPS is repurposed from elsewhere and the ID is no longer set to the default. Hence my suggestion to keep the original behavior of matching any UPS when `upsid` is not specified.

sapireli, 2024-01-27:
@clepple this is already the current behavior. If upsid is not in the config it will still match whichever unit. I.e. upsid  is only used if it is defined.

I tested it just now and it works. What’s your setup / issue you’re experiencing?

sapireli, 2024-01-27:
Just realized my test was invalid but I’ll retest tomorrow to confirm it indeed works as you defined. Otherwise it’s an easy fix.

sapireli, 2024-01-27:
Ok I can confirm the bug exists. If upsid is not defined nothing is matched. Yea this is annoying. 

what about just returning 1 if its not defined? wouldn’t that be same as the ‘null’ behavior?

clepple, 2024-01-27:
> what about just returning 1 if its not defined? wouldn’t that be same as the ‘null’ behavior?

I guess so, as long as it returns early. (I was originally thinking that matcher function gets called all the time, but it would only be after the matching struct is applied.)

sapireli, 2024-01-27:
Alright I’ll handle it next weekend

PR #2297 (2024-02-10): Fix: tripplite_usb: if upsid is not specified, make any unit ID match
small fix restoring matching when upsid is undefined in the ups.conf file closes #2282 as per the thread
cc @clepple
commit 46057fbe (2024-02-10): Fix for undefined upsid in config
small fix restoring matching when upsid is undefined in config closes #2282
closed 2024-02-10 by jimklimov