networkupstools/nut #2296: Issue -  stoping working with 2 similar Eaton

seblang, 2024-01-31:
I have added a second Etaon inverter on the same computer.
I have added 2 devices in config but the soft stop working with below logs :

s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-timezone: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
[10:43:03] INFO: Configuring timezone (Europe/Paris)...
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service base-addon-timezone successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/nut.sh
[10:43:04] INFO: Setting mode to netserver...
[10:43:04] INFO: Connected USB devices:
Bus 001 Device 053: ID 0463:ffff MGE UPS Systems UPS
Bus 001 Device 114: ID 0463:ffff MGE UPS Systems UPS
Bus 001 Device 006: ID 0658:0200 Sigma Designs, Inc. Aeotec Z-Stick Gen5 (ZW090) - UZB
Bus 001 Device 011: ID 1a86:7523 QinHeng Electronics CH340 serial converter
Bus 001 Device 002: ID 0438:7900 Advanced Micro Devices, Inc. Root Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 002: ID 8087:0a2b Intel Corp. Bluetooth wireless interface
Bus 002 Device 028: ID 10c4:ea60 Silicon Labs CP210x UART Bridge
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
[10:43:04] INFO: Generating /etc/nut/upsd.users...
[10:43:04] INFO: Configuring user: Slangolf
[10:43:06] INFO: Configuring Device named eatonpro...
[10:43:06] INFO: Configuring Device named eatoneco...
[10:43:06] INFO: Starting the UPS drivers...
Using subdriver: MGE HID 1.46
Network UPS Tools - Generic HID driver 0.47 (2.8.0)
USB communication driver (libusb 1.0) 0.43
Can't claim USB device [0463:ffff]@0/0: Entity not found
Network UPS Tools - Generic HID driver 0.47 (2.8.0)
USB communication driver (libusb 1.0) 0.43
Network UPS Tools - UPS driver controller 2.8.0
Driver failed to start (exit status=1)
cont-init: info: /etc/cont-init.d/nut.sh exited 1
cont-init: info: running /etc/cont-init.d/nutclient.sh
cont-init: info: /etc/cont-init.d/nutclient.sh exited 0
cont-init: warning: some scripts exited nonzero
s6-rc: warning: unable to start service legacy-cont-init: command exited 1
/run/s6/basedir/scripts/rc.init: warning: s6-rc failed to properly bring all the services up! Check your logs (in /run/uncaught-logs/current if you have in-container logging) for more information.
/run/s6/basedir/scripts/rc.init: fatal: stopping the container.
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service base-addon-timezone: stopping
s6-rc: info: service base-addon-log-level: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service base-addon-timezone successfully stopped
s6-rc: info: service base-addon-log-level successfully stopped
s6-rc: info: service base-addon-banner: stopping
s6-rc: info: service base-addon-banner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

My config is :

- name: eatonpro
  driver: usbhid-ups
  port: auto
  vendorid: "0465"
  productid: ffff
  desc: EATON PRO
  SERIAL: EATON_Ellipse_PRO_P357J06JCH
  config: []
- name: eatoneco
  driver: usbhid-ups
  port: auto
  vendorid: "0465"
  productid: ffff
  desc: EATON ECO
  SERIAL: EATON_Ellipse_ECO_000000000
  config: []

I have used the "serial" option because all other info are the same for both inverter. Enclosed the data for the 2 inverter.

How can I solve my issue?
[image]
[image]

jimklimov, 2024-01-31:
I think usbhid-ups deals with the short value (all-zeroes string on ECO's I saw in reports, but luckily unique-ish on your other unit). 

Also the `serial` keyword is lower-case, not sure OTOH if that matters.

Apparently you are on HA NUT plugin? Can you fiddle with NUT `debug_min` settings (see wiki or docs) via yaml to get more details about why the driver start fails?

jimklimov, 2024-02-09:
Hello again, while reformatting the original post, I've noticed you have `vendorid: "0465"` but for Eaton it should be `0463`. Kindly check if fixing the typo resolves the issue?