networkupstools/nut #2298: is it supported? Network UPS Tools upsstats 2.8.0 - energenie - [error: Driver not connected] on myups2@localhost

estatistics, 2024-02-03:
lsusb
`Bus 002 Device 003: ID 0001:0000 Fry's Electronics`

`sudo nut-scanner -U`

```
Scanning USB bus.
[nutdev1]
        driver = "nutdrv_qx"
        port = "auto"
        vendorid = "0001"
        productid = "0000"
        bus = "002"
```

`sudo /lib/nut/nutdrv_qx -a myups2`

Network UPS Tools - Generic Q* USB/Serial driver 0.32 (2.8.0)
USB communication driver (libusb 1.0) 0.43
libusb1: Could not open any HID devices: insufficient permissions on everything
No supported devices found. Please check your device availability with 'lsusb'
and make sure you have an up-to-date version of NUT. If this does not help,
try running the driver with at least 'subdriver', 'vendorid' and 'productid'
options specified. Please refer to the man page for details about these options
(man 8 nutdrv_qx).
sudo cat /etc/nut/ups.conf 

[myups]
    driver = usbhid-ups
    port = auto
    vendorid = "0001:0000"
    productid = "0000"
    bus = "002"

[myups2]
    driver = nutdrv_qx
    port = auto
    vendorid = "0001:0000"
    productid = "0000"
    bus = "002"

sudo cat /etc/nut/upsd.conf 
LISTEN 127.0.0.1 3493 

 sudo cat /etc/nut/upsd.users 

[admin1]
  password = s1
  admin1 m1

sudo cat /etc/nut/upsmon.conf 
MONITOR myups2@localhost 1 admin1 s1 m1

 sudo cat /etc/nut/upsset.conf 
I_HAVE_SECURED_MY_CGI_DIRECTORY

jimklimov, 2024-02-04:
As `nut-scanner` says, `vendorid` is just the 4 digits.

Regarding "no permissions", check if your udev/upower/... rules are set up to delegate the vid:pid to the `nut` run-time account (if that's what your packaging calls it). As a workaround for this bit, you can say `user=root` in the driver config section.

Finally, this vendor chose to use an unregistered vid:pid, so there are no surefire pre-sets to identify the device or the protocols it talks. You may have to scavenge the NUT DDL and/or github issue reports for fine-tuned settings (subdriver and others) if needed.

IRONHIDEUA, 2025-01-24:
did you manage to connect this ups to nut? if so, please describe the procedure for connecting and setting up

estatistics, 2025-01-25:
@IRONHIDEUA no. I run it but its 'widely' use native program UPSmart tool II. I succeeded by playing with drivers and ids to get some values but i was stuck. I thought not all necessary information was printed. So, i abandoned nut.

IRONHIDEUA, 2025-01-25:
At the moment, using chatgpt, I’m trying to connect it to unraid by forwarding usb to the win 10 virtual machine

jimklimov, 2025-01-25:
Thanks for the detail about UPSmart compatibility, that may help if someone gets to developing a driver/fix for this device. At least it confirms that some dialect of Megatec Qx protocol is likely spoken by the UPS.

While trying the NUT drivers, did you run them with bumped debug verbosity to see why it refused to work with a particular device?

I see I gave a few pointers above, but saw no replies... at that time:
* `vendorid="0001"` and `productid="0000"` as nut-scanner said; without that the driver probably did not match any device (and had no permissions for IDs unknown to packaged NUT in general)
* if IDs are right, and no permissions still, check if your `udev` rules or similar mechanism allows those IDs for `nut` user access (assuming that is used by the package)
* as noted, running the driver with debug can show a lot more hints, e.g. `sudo /lib/nut/nutdrv_qx -a myups2 -DDDDDD -d1` to try starting it, run one data-collection loop, dump the data and exit; you can also add `-u root` to this line to not drop privileges and see if permissions are the problem or not...

jimklimov, 2025-01-25:
Also note that NUT v2.8.0 is about 3 years old now; there was some development progress since then. If you have a chance to custom-build NUT (on some other computer if the NAS is a closed ecosystem, maybe on a VM with USB pass-through, as those were mentioned) from current sources, it can help you check if just the software upgrade brings some benefits. You can check from the build workspace, without changing the NUT installation in the system, see https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests