networkupstools/nut #2301: Hi, first thanks for your work. May it possible to add support for aeg ups protect d serie like model D3000 use usb connexion. using nut 2.8.0 with usbhid-ups driver, it got some basic information but not importante information like batterie state, online state etc  It will be a pleasure if i can help you but i'm not a dev!!!  siteweb@magnaval.org

MAGADE007, 2024-02-10:


jimklimov, 2024-02-10:
Hello and welcome!

Could you please check in the service or system logs whether the NUT driver (`usbhid-ups` in this case) has any complaints during initialization?

For example, you can stop the NUT services and run the driver from command line with higher debug verbosity in probing mode, and paste the result here, e.g. (assuming your `ups.conf` section for the device is named `aeg`):

````
:; /lib/nut/usbhid-ups -a aeg -DDDDDD -d 1
````

This should report many details about looking at different USB devices and trying the matching options, and finally make one walk to collect data and dump it similarly to `upsc` client, and exit the driver.

In particular, the "online state" is a rather fundamental piece of information. There were reports of it staying as "OB" regardless of device physical state, that were linked to USB connectivity/permissions problems (e.g. if the OS did not hand off the device completely to the NUT driver program -- more so if virtualization or some complex USB stacks with filter drivers are involved, like that on Windows).

There is also a possibility to generate a skeleton subdriver based on a special "explore"-mode device walk from `usbhid-ups`, see https://github.com/networkupstools/nut/blob/master/docs/hid-subdrivers.txt - it may be never too late to become a bit of a dev :)

For hints about installing pre-requisites to build custom NUT from sources (e.g. so you can test the custom driver right from that build workspace, without replacing packaged NUT until you are satisfied with the code customization), see https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests

MAGADE007, 2024-02-10:
Thanks to your answer, with usbhid-ups driver, the 

I'm using now nutq-qx driver cause it tell more information than usbhid-ups
usbhid-ups online state is OB
nutq-qx online state is OL and when I'm on battery with no AC input it is good value 

The test I had make
on AC input and after on 0V input:
the result:
root@DARYL:/etc/nut# upsc nutdev1
Init SSL without certificate database
battery.voltage: 2.30
device.type: ups
driver.name: nutdrv_qx
driver.parameter.bus: 001
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.product: USB UPS
driver.parameter.productid: 0003
driver.parameter.synchronous: auto
driver.parameter.vendor: PPC
driver.parameter.vendorid: 06DA
driver.version: 2.8.0
driver.version.data: Q1 0.07
driver.version.internal: 0.32
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
input.frequency: 49.9
input.voltage: 229.1
ups.beeper.status: disabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.load: 13
ups.productid: 0003
ups.status: OL
ups.temperature: 24.3
ups.type: online
ups.vendorid: 06da
                                                                               
Broadcast message from nut@DARYL (somewhere) (Fri Dec  8 00:39:00 2023):       
                                                                               
UPS nutdev1@localhost on battery                                               
                                                                               
                                                                               
Broadcast message from nut@DARYL (somewhere) (Fri Dec  8 00:39:00 2023):       
                                                                               
UPS nutdev1@localhost on battery                                               
                                                                               
upsc nutdev1
Init SSL without certificate database
battery.voltage: 2.00
device.type: ups
driver.name: nutdrv_qx
driver.parameter.bus: 001
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.product: USB UPS
driver.parameter.productid: 0003
driver.parameter.synchronous: auto
driver.parameter.vendor: PPC
driver.parameter.vendorid: 06DA
driver.version: 2.8.0
driver.version.data: Q1 0.07
driver.version.internal: 0.32
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
input.frequency: 0.0
input.voltage: 0.0
ups.beeper.status: disabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.load: 13
ups.productid: 0003
ups.status: OB
ups.temperature: 23.9
ups.type: online
ups.vendorid: 06da

In this test, it is missing battery the left and battery % rester time on battery

MAGADE007, 2024-02-10:
> :; /lib/nut/nutdrv-qx -a aeg -DDDDDD -d 1
root@DARYL:/home/magade# systemctl stop nut-client
root@DARYL:/home/magade# systemctl stop nut-server
root@DARYL:/home/magade# systemctl stop nut-monitor
root@DARYL:/home/magade# systemctl stop nut-driver
Failed to stop nut-driver.service: Unit nut-driver.service not loaded.
root@DARYL:/home/magade# :; /lib/nut/nutdrv_qx -a nutdev1 -DDDDDD -d 1
Network UPS Tools - Generic Q* USB/Serial driver 0.32 (2.8.0)
USB communication driver (libusb 1.0) 0.43
   0.000000	[D1] debug level is '6'
   0.001264	[D5] send_to_all: SETINFO device.type "ups"
   0.001359	[D1] upsdrv_initups...
   0.037643	[D2] Checking device 1 of 6 (174C/55AA)
   0.037887	[D1] Failed to open device (174C/55AA), skipping: Access denied (insufficient permissions)
   0.037960	[D2] Checking device 2 of 6 (1D6B/0003)
   0.038054	[D1] Failed to open device (1D6B/0003), skipping: Access denied (insufficient permissions)
   0.038125	[D2] Checking device 3 of 6 (0557/2304)
   0.038219	[D1] Failed to open device (0557/2304), skipping: Access denied (insufficient permissions)
   0.038283	[D2] Checking device 4 of 6 (06DA/0003)
   0.044482	[D2] - VendorID: 06da
   0.044558	[D2] - ProductID: 0003
   0.044619	[D2] - Manufacturer: PPC
   0.044678	[D2] - Product: USB UPS
   0.044736	[D2] - Serial Number: unknown
   0.044794	[D2] - Bus: 001
   0.044852	[D2] - Device: unknown
   0.044911	[D2] - Device release number: 0003
   0.044972	[D2] Trying to match device
   0.045032	[D3] match_function_regex: matching a device...
   0.045388	[D2] Device matches
   0.045457	[D2] Reading first configuration descriptor
   0.045541	[D3] libusb_kernel_driver_active() returned 0
   0.045614	[D2] failed to claim USB device: Resource busy
   0.046054	[D2] Kernel driver already detached
   0.046133	[D2] failed to claim USB device: Resource busy
   0.046202	[D2] Kernel driver already detached
   0.046267	[D2] failed to claim USB device: Resource busy
   0.046328	[D2] Kernel driver already detached
   0.046399	[D2] failed to claim USB device: Resource busy
   0.046465	[D2] Kernel driver already detached
   0.046532	Can't claim USB device [06da:0003]@0/0: Entity not found
root@DARYL:/home/magade# 

root@DARYL:/home/magade# lsusb
Bus 002 Device 002: ID 174c:55aa ASMedia Technology Inc. ASM1051E SATA 6Gb/s bridge, ASM1053E SATA 6Gb/s bridge, ASM1153 SATA 3Gb/s bridge, ASM1153E SATA 6Gb/s bridge
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 0557:2304 ATEN International Co., Ltd IO Module
Bus 001 Device 080: ID 06da:0003 Phoenixtec Power Co., Ltd 1300VA UPS
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
root@DARYL:/home/magade# 

my ups.conf:
nut.conf  ups.conf  upsd.conf  upsd.users  upsmon.conf	upssched.conf
root@DARYL:/etc/nut# cat ups.conf
# Network UPS Tools: example ups.conf
#
# --- SECURITY NOTE ---
#
# If you use snmp-ups and set a community string in here, you
# will have to secure this file to keep other users from obtaining
# that string.  It needs to be readable by upsdrvctl and any drivers,
# and by upsd.
#
# ---
#
# This is where you configure all the UPSes that this system will be
# monitoring directly.  These are usually attached to serial ports, but
# USB devices and SNMP devices are also supported.
#
[... 173 lines cut ...]
#   desc = "Web server"
#
# If your UPS driver requires additional settings, you can specify them
# here.  For example, if it supports a setting of "1234" for the
# variable "cable", it would look like this:
#
# [myups]
#   driver = mydriver
#   port = /dev/ttyS1
#   cable = 1234
#   desc = "Something descriptive"
#
# To find out if your driver supports any extra settings, start it with
# the -h option and/or read the driver's documentation.

[nutdev1]
	driver = "nutdrv_qx"
#        driver = "usbhid-ups"
	port = "auto"
	vendorid = "06DA"
	productid = "0003"
	product = "USB UPS"
	vendor = "PPC"
	bus = "001"
root@DARYL:/etc/nut#

jimklimov, 2024-02-10:
Nice, so at least this device talks a variant of Megatec Q<x> protocol, and not standard USB HID.

There were quite a few fixes to `nutdrv_qx` lately, so I encourage you to check that wiki article and try a custom build of current NUT master codebase. Chances are, more info would be seen then (maybe not, we'll see).

jimklimov, 2024-02-10:
For service-stopping, please check if you have a `nut-driver@nutdev1.service` instance which wraps the single driver?

MAGADE007, 2024-02-10:
> For service-stopping, please check if you have a `nut-driver@nutdev1.service` instance which wraps the single driver?
root@DARYL:/etc/nut# nut-driver@nutdev1.service
bash: nut-driver@nutdev1.service: command not found
root@DARYL:/etc/nut# systemctl list-units --type=service | grep nut
  nut-driver@nutdev1.service                                  loaded active running Network UPS Tools - device driver for NUT device 'nutdev1'
root@DARYL:/etc/nut#

MAGADE007, 2024-02-10:
WITH DRIVER USBHID-UPS activate in ups.conf

root@DARYL:/etc/nut# :; /lib/nut/usbhid-ups -a nutdev1 -DDDDDD -d 1
Network UPS Tools - Generic HID driver 0.47 (2.8.0)
USB communication driver (libusb 1.0) 0.43
   0.000000	[D1] debug level is '6'
   0.001209	[D5] send_to_all: SETINFO device.type "ups"
   0.001291	[D2] Initializing an USB-connected UPS with library libusb-1.0.26 (API: 0x1000109) (NUT subdriver name='USB communication driver (libusb 1.0)' ver='0.43')
   0.001355	[D1] upsdrv_initups (non-SHUT)...
   0.037151	[D2] Checking device 1 of 6 (174C/55AA)
   0.037363	[D1] Failed to open device (174C/55AA), skipping: Access denied (insufficient permissions)
   0.037432	[D2] Checking device 2 of 6 (1D6B/0003)
   0.037526	[D1] Failed to open device (1D6B/0003), skipping: Access denied (insufficient permissions)
   0.037628	[D2] Checking device 3 of 6 (0557/2304)
   0.037730	[D1] Failed to open device (0557/2304), skipping: Access denied (insufficient permissions)
   0.037794	[D2] Checking device 4 of 6 (06DA/0003)
   0.043704	[D2] - VendorID: 06da
   0.043779	[D2] - ProductID: 0003
   0.043839	[D2] - Manufacturer: PPC
   0.043898	[D2] - Product: USB UPS
   0.043956	[D2] - Serial Number: unknown
   0.044013	[D2] - Bus: 001
   0.044069	[D2] - Device: unknown
   0.044127	[D2] - Device release number: 0003
   0.044184	[D2] Trying to match device
   0.044243	[D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.044319	[D3] match_function_regex: matching a device...
   0.044661	[D2] Device matches
   0.044726	[D2] Reading first configuration descriptor
   0.044810	[D3] libusb_kernel_driver_active() returned 0
   0.044881	[D2] failed to claim USB device: Resource busy
   0.044948	[D2] Kernel driver already detached
   0.045014	[D2] failed to claim USB device: Resource busy
   0.045080	[D2] Kernel driver already detached
   0.045145	[D2] failed to claim USB device: Resource busy
   0.045211	[D2] Kernel driver already detached
   0.045276	[D2] failed to claim USB device: Resource busy
   0.045344	[D2] Kernel driver already detached
   0.045410	Can't claim USB device [06da:0003]@0/0: Entity not found
root@DARYL:/etc/nut# 

root@DARYL:/etc/nut# upsc nutdev1
Init SSL without certificate database
device.mfr: PPC
device.model: USB UPS
device.type: ups
driver.name: usbhid-ups
driver.parameter.bus: 001
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.product: USB UPS
driver.parameter.productid: 0003
driver.parameter.synchronous: auto
driver.parameter.vendor: PPC
driver.parameter.vendorid: 06DA
driver.version: 2.8.0
driver.version.data: Phoenixtec/Liebert HID 0.41
driver.version.internal: 0.47
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
ups.mfr: PPC
ups.model: USB UPS
ups.productid: 0003
ups.status: OB
ups.vendorid: 06da
root@DARYL:/etc/nut# 

see ups/status is wrong "OB" cause there are AC input on 230V

MAGADE007, 2024-02-10:
If it could be helpfulroot@DARYL:/etc/nut# nut-scanner
Scanning USB bus.
No start IP, skipping SNMP
Scanning XML/HTTP bus.
No start IP, skipping NUT bus (old connect method)
Scanning NUT bus (avahi method).
[nutdev1]
	driver = "nutdrv_qx"
	port = "auto"
	vendorid = "06DA"
	productid = "0003"
	product = "USB UPS"
	vendor = "PPC"
	bus = "001"
root@DARYL:/etc/nut#

jimklimov, 2024-02-10:
> see ups/status is wrong "OB" cause there are AC input on 230V

Yes, the device does not actually talk USB HID properly, but just seems to do so. Thus the driver remains at some default value for the status, which is probably a bug, come to think of it. Either this, or libusb provides some bogus reply (always same in such cases on different platforms) which the driver interprets honestly and not *our* bug then.

> failed to claim USB device: Resource busy

So, beside finding that `nut-driver@nutdev1.service` exists - did you stop it first (before running the driver programs from command-line)?

jimklimov, 2024-02-10:
Also, for large "text screenshots", please wrap them into lines of 3 or 4 backticks, so they would be formatted as code blocks and be better readable :)

MAGADE007, 2024-02-10:
> Yes, the device does not actually talk USB HID properly, but just seems to do so. Thus the driver remains at some default value for the status, which is probably a bug, come to think of it. Either [...]

I succed to stop only driver with the commande line root@DARYL:/etc/nut# systemctl stop nut-driver@nutdev1.service
This is what I got now:
root@DARYL:/etc/nut# :; /lib/nut/nutdrv_qx -a nutdev1 -DDDDDD -d 1
Network UPS Tools - Generic Q* USB/Serial driver 0.32 (2.8.0)
USB communication driver (libusb 1.0) 0.43
   0.000000	[D1] debug level is '6'
   0.001691	[D5] send_to_all: SETINFO device.type "ups"
   0.002274	[D1] upsdrv_initups...
   0.038947	[D2] Checking device 1 of 6 (174C/55AA)
   0.039672	[D1] Failed to open device (174C/55AA), skipping: Access denied (insufficient permissions)
   0.040234	[D2] Checking device 2 of 6 (1D6B/0003)
   0.040801	[D1] Failed to open device (1D6B/0003), skipping: Access denied (insufficient permissions)
   0.040872	[D2] Checking device 3 of 6 (0557/2304)
   0.041431	[D1] Failed to open device (0557/2304), skipping: Access denied (insufficient permissions)
   0.041953	[D2] Checking device 4 of 6 (06DA/0003)
   0.047907	[D2] - VendorID: 06da
   0.048188	[D2] - ProductID: 0003
   0.048253	[D2] - Manufacturer: PPC
   0.048313	[D2] - Product: USB UPS
   0.048367	[D2] - Serial Number: unknown
   0.048425	[D2] - Bus: 001
   0.048481	[D2] - Device: unknown
   0.048540	[D2] - Device release number: 0003
   0.048598	[D2] Trying to match device
   0.048658	[D3] match_function_regex: matching a device...
   0.049017	[D2] Device matches
   0.049082	[D2] Reading first configuration descriptor
   0.049164	[D3] libusb_kernel_driver_active() returned 0
   0.049281	[D2] Claimed interface 0 successfully
   0.049349	[D3] nut_usb_set_altinterface: skipped libusb_set_interface_alt_setting(udev, 0, 0)
   0.049428	[D5] send_to_all: SETINFO ups.vendorid "06da"
   0.049498	[D5] send_to_all: SETINFO ups.productid "0003"
   0.050703	[D3] send: QGS
   0.148177	[D5] read: (5 bytes) => 28 4e 41 4b 0d
   0.148253	[D3] read: (NAK
   0.148321	[D2] qx_process_answer: query rejected by the UPS (input.voltage)
   1.148942	[D3] send: Connection timed out (-7)
   1.149052	[D4] qx_process: failed to preprocess answer [input.voltage]
   1.151541	[D3] send: Pipe error (-9)
   1.922712	[D1] Device reset handled
   1.923335	[D4] qx_process: failed to preprocess answer [input.voltage]
   1.955031	[D2] Checking device 1 of 6 (174C/55AA)
   1.955155	[D1] Failed to open device (174C/55AA), skipping: Access denied (insufficient permissions)
   1.955207	[D2] Checking device 2 of 6 (1D6B/0003)
   1.955276	[D1] Failed to open device (1D6B/0003), skipping: Access denied (insufficient permissions)
   1.955327	[D2] Checking device 3 of 6 (0557/2304)
   1.955396	[D1] Failed to open device (0557/2304), skipping: Access denied (insufficient permissions)
   1.955445	[D2] Checking device 4 of 6 (06DA/0003)
   1.961286	[D2] - VendorID: 06da
   1.961343	[D2] - ProductID: 0003
   1.961387	[D2] - Manufacturer: PPC
   1.961430	[D2] - Product: USB UPS
   1.961474	[D2] - Serial Number: unknown
   1.961519	[D2] - Bus: 001
   1.961563	[D2] - Device: unknown
   1.961606	[D2] - Device release number: 0003
   1.961649	[D2] Trying to match device
   1.961693	[D3] match_function_exact: matching a device...
   1.961739	[D3] match_function_regex: matching a device...
   1.961814	[D2] Device matches
   1.961859	[D2] Reading first configuration descriptor
   1.961920	[D3] libusb_kernel_driver_active() returned 0
   1.962006	[D2] Claimed interface 0 successfully
   1.962058	[D3] nut_usb_set_altinterface: skipped libusb_set_interface_alt_setting(udev, 0, 0)
   1.964091	[D3] send: M
   2.053235	[D5] read: (5 bytes) => 28 4e 41 4b 0d
   2.053292	[D3] read: (NAK
   2.053349	[D2] voltronic_qs_protocol: invalid protocol [(NAK]
[... 136 lines cut ...]
  11.666396	[D1] Device reset handled
  11.666981	[D4] qx_process: failed to preprocess answer [input.voltage]
  11.698809	[D2] Checking device 1 of 6 (174C/55AA)
  11.698940	[D1] Failed to open device (174C/55AA), skipping: Access denied (insufficient permissions)
  11.698998	[D2] Checking device 2 of 6 (1D6B/0003)
  11.699069	[D1] Failed to open device (1D6B/0003), skipping: Access denied (insufficient permissions)
  11.699118	[D2] Checking device 3 of 6 (0557/2304)
  11.699196	[D1] Failed to open device (0557/2304), skipping: Access denied (insufficient permissions)
  11.699245	[D2] Checking device 4 of 6 (06DA/0003)
  11.705451	[D2] - VendorID: 06da
  11.705510	[D2] - ProductID: 0003
  11.705562	[D2] - Manufacturer: PPC
  11.705608	[D2] - Product: USB UPS
  11.705652	[D2] - Serial Number: unknown
  11.705696	[D2] - Bus: 001
  11.705739	[D2] - Device: unknown
  11.705782	[D2] - Device release number: 0003
  11.705826	[D2] Trying to match device
  11.705871	[D3] match_function_exact: matching a device...
  11.705918	[D3] match_function_regex: matching a device...
  11.705992	[D2] Device matches
  11.706037	[D2] Reading first configuration descriptor
  11.706100	[D3] libusb_kernel_driver_active() returned 0
  11.706193	[D2] Claimed interface 0 successfully
  11.706245	[D3] nut_usb_set_altinterface: skipped libusb_set_interface_alt_setting(udev, 0, 0)

MAGADE007, 2024-02-10:
> Also, for large "text screenshots", please wrap them into lines of 3 or 4 backticks (same count on start and finish), so they would be formatted as code blocks and be better readable :)

Sorry I don't understand what you mean. My English language is basic. ;)

jimklimov, 2024-02-10:
Thanks, this part looked promising:

````
   9.642409	[D3] send: Q1
   9.891440	[D5] read: (47 bytes) => 28 32 33 31 2e 31 20 32 33 30 2e 34 20 32 32 39 2e 39
   9.891566	[D5]  20 30 31 33 20 35 30 2e 30 20 32 2e 33 31 20 32 34 2e 33 20 30 30 30 30 30
   9.891634	[D5]  30 30 31 0d
   9.891696	[D3] read: (231.1 230.4 229.9 013 50.0 2.31 24.3 00000001
   9.891800	[D5] send_to_all: SETINFO input.voltage "231.1"
````

...however this did not:
````
  10.892427	[D3] send: Connection timed out (-7)
````

I am not quickly sure which `send` this is - from the driver to the device (may be a protocol issue, etc.), or from the driver to `upsd` data server. Is your `nut-server.service` still stopped? Can you start it (or run `upsd`) just in case?

There were some fixes for this driver merged in recent months, so if you manage to build NUT from github sources (per wiki article earlier) and test with that variant of the driver, it may be useful to check if that issue is already known and solved. There was something like it, but I can't find the specific PR that dealt with this (not sure if it was solved for a particular protocol/vendor subdriver or for the whole `nutdrv_qx` driver).

MAGADE007, 2024-02-10:
Just in case it is useful with usbhid-ups
root@DARYL:/etc/nut# systemctl stop nut-driver@nutdev1.service
root@DARYL:/etc/nut# :; /lib/nut/usbhid-ups -a nutdev1 -DDDDDD -d 1
Network UPS Tools - Generic HID driver 0.47 (2.8.0)
USB communication driver (libusb 1.0) 0.43
   0.000000	[D1] debug level is '6'
   0.001789	[D5] send_to_all: SETINFO device.type "ups"
   0.002561	[D2] Initializing an USB-connected UPS with library libusb-1.0.26 (API: 0x1000109) (NUT subdriver name='USB communication driver (libusb 1.0)' ver='0.43')
   0.003219	[D1] upsdrv_initups (non-SHUT)...
   0.033940	[D2] Checking device 1 of 6 (174C/55AA)
   0.034217	[D1] Failed to open device (174C/55AA), skipping: Access denied (insufficient permissions)
   0.034261	[D2] Checking device 2 of 6 (1D6B/0003)
   0.034323	[D1] Failed to open device (1D6B/0003), skipping: Access denied (insufficient permissions)
   0.034362	[D2] Checking device 3 of 6 (0557/2304)
   0.034417	[D1] Failed to open device (0557/2304), skipping: Access denied (insufficient permissions)
   0.034457	[D2] Checking device 4 of 6 (06DA/0003)
   0.040108	[D2] - VendorID: 06da
   0.040156	[D2] - ProductID: 0003
   0.040191	[D2] - Manufacturer: PPC
   0.040226	[D2] - Product: USB UPS
   0.040260	[D2] - Serial Number: unknown
   0.040294	[D2] - Bus: 001
   0.040327	[D2] - Device: unknown
   0.040361	[D2] - Device release number: 0003
   0.040395	[D2] Trying to match device
   0.040431	[D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.040478	[D3] match_function_regex: matching a device...
   0.040716	[D2] Device matches
   0.040756	[D2] Reading first configuration descriptor
   0.040807	[D3] libusb_kernel_driver_active() returned 0
   0.040883	[D2] Claimed interface 0 successfully
   0.040922	[D3] nut_usb_set_altinterface: skipped libusb_set_interface_alt_setting(udev, 0, 0)
   0.043384	[D3] HID descriptor, method 1: (9 bytes) => 09 21 11 01 00 01 22 1b 00
   0.043431	[D3] HID descriptor length (method 1) 27
   0.043467	[D4] i=0, extra[i]=09, extra[i+1]=21
   0.043508	[D3] HID descriptor, method 2: (9 bytes) => 09 21 11 01 00 01 22 1b 00
   0.043544	[D3] HID descriptor length (method 2) 27
   0.043579	[D2] HID descriptor length 27
   0.047363	[D2] Report Descriptor size = 27
   0.047414	[D3] Report Descriptor: (27 bytes) => 06 a0 ff 09 04 a1 01 09 06 15 00 26 ff 00
   0.047455	[D3]  75 08 95 08 81 82 09 07 95 40 91 82 c0
   0.047651	Using subdriver: Phoenixtec/Liebert HID 0.41
   0.047691	[D1] 2 HID objects found
   0.047728	[D4] Entering libusb_get_report
   0.050126	[D3] Report[get]: (2 bytes) => 00 00
   0.050170	[D5] PhyMax = 0, PhyMin = 0, LogMax = 255, LogMin = 0
   0.050207	[D5] Unit = 00000000, UnitExp = 0
   0.050241	[D5] Exponent = 0
   0.050279	[D5] hid_lookup_path: ffa00004 -> not found in lookup table
   0.050316	[D5] hid_lookup_path: ffa00006 -> not found in lookup table
   0.050356	[D1] Path: ffa00004.ffa00006, Type: Input, ReportID: 0x00, Offset: 0, Size: 8, Value: 0
   0.050393	[D3] Report[buf]: (2 bytes) => 00 00
   0.050429	[D5] PhyMax = 0, PhyMin = 0, LogMax = 255, LogMin = 0
   0.050464	[D5] Unit = 00000000, UnitExp = 0
   0.050499	[D5] Exponent = 0
   0.050534	[D5] hid_lookup_path: ffa00004 -> not found in lookup table
   0.050570	[D5] hid_lookup_path: ffa00007 -> not found in lookup table
   0.050607	[D1] Path: ffa00004.ffa00007, Type: Output, ReportID: 0x00, Offset: 0, Size: 8, Value: 0
   0.050652	[D5] send_to_all: SETINFO ups.mfr "PPC"
   0.050693	[D5] send_to_all: SETINFO ups.model "USB UPS"
   0.050734	[D5] send_to_all: SETINFO ups.vendorid "06da"
   0.050773	[D5] send_to_all: SETINFO ups.productid "0003"
   0.050810	[D2] Report descriptor retrieved (Reportlen = 27)
   0.050844	[D2] Found HID device
   0.050883	[D1] Detected a UPS: PPC/USB UPS
   0.050927	[D5] hid_lookup_usage: UPS -> 00840004
   0.050965	[D5] hid_lookup_usage: PowerSummary -> 00840024
   0.051001	[D5] hid_lookup_usage: Voltage -> 00840030
   0.051036	[D4] string_to_path: depth = 3
   0.051073	[D5] hid_lookup_usage: UPS -> 00840004
[... 137 lines cut ...]
device.type: ups
driver.name: usbhid-ups
driver.parameter.bus: 001
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.product: USB UPS
driver.parameter.productid: 0003
driver.parameter.synchronous: auto
driver.parameter.vendor: PPC
driver.parameter.vendorid: 06DA
driver.version: 2.8.0
driver.version.data: Phoenixtec/Liebert HID 0.41
driver.version.internal: 0.47
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
ups.mfr: PPC
ups.model: USB UPS
ups.productid: 0003
ups.status: OB
ups.vendorid: 06da
   2.308921	[D1] upsdrv_cleanup...
                                                                               
Broadcast message from nut@DARYL (somewhere) (Fri Dec  8 06:02:36 2023):       
                                                                               
Executing automatic power-fail shutdown

MAGADE007, 2024-02-10:
> Thanks, this part looked promising:
>
> ```
>    9.642409	[D3] send: Q1
>    9.891440	[D5] read: (47 bytes) => 28 32 33 31 2e 31 20 32 33 30 2e 34 20 32 32 39 2e 39
>    9.891566	[D5]  20 30 31 33 20 35 30 [...]

Okay, i will try to get this more update version of nut and test same this this version to send you the results

MAGADE007, 2024-02-11:
blazer_usb seem to get a little bit more inforoot@DARYL:/home/magade# upsc nutdev1
Init SSL without certificate database
battery.voltage: 2.20
battery.voltage.high: -1.08
battery.voltage.low: -0.87
device.type: ups
driver.name: blazer_usb
driver.parameter.bus: 001
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: auto
driver.version: 2.8.0
driver.version.internal: 0.14
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
input.frequency: 50.0
input.voltage: 226.3
input.voltage.fault: 230.4
output.voltage: 230.1
ups.beeper.status: enabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.load: 14
ups.productid: 0003
ups.status: OL
ups.temperature: 23.1
ups.type: online
ups.vendorid: 06da
root@DARYL:/home/magade# 

root@DARYL:/home/magade# :; /lib/nut/blazer_usb -a nutdev1 -DDDDDD -d 1
Network UPS Tools - Megatec/Q1 protocol USB driver 0.14 (2.8.0)
   0.000000	[D1] debug level is '6'
   0.000953	[D5] send_to_all: SETINFO device.type "ups"
   0.023375	[D2] Checking device 1 of 6 (174C/55AA)
   0.023505	[D1] Failed to open device (174C/55AA), skipping: Access denied (insufficient permissions)
   0.023537	[D2] Checking device 2 of 6 (1D6B/0003)
   0.023586	[D1] Failed to open device (1D6B/0003), skipping: Access denied (insufficient permissions)
   0.023614	[D2] Checking device 3 of 6 (0557/2304)
   0.023657	[D1] Failed to open device (0557/2304), skipping: Access denied (insufficient permissions)
   0.023685	[D2] Checking device 4 of 6 (06DA/0003)
   0.029764	[D2] - VendorID: 06da
   0.029801	[D2] - ProductID: 0003
   0.029827	[D2] - Manufacturer: PPC
   0.029854	[D2] - Product: USB UPS
   0.029874	[D2] - Serial Number: unknown
   0.029894	[D2] - Bus: 001
   0.029914	[D2] - Device: unknown
   0.029934	[D2] - Device release number: 0003
   0.029953	[D2] Trying to match device
   0.029975	[D3] match_function_regex: matching a device...
   0.030178	[D2] Device matches
   0.030204	[D2] Reading first configuration descriptor
   0.030246	[D3] libusb_kernel_driver_active() returned 1 (driver active)
   0.030271	[D2] successfully set kernel driver auto-detach flag
   0.033271	[D2] Claimed interface 0 successfully
   0.033680	[D3] nut_usb_set_altinterface: skipped libusb_set_interface_alt_setting(udev, 0, 0)
   0.034083	[D5] send_to_all: SETINFO ups.vendorid "06da"
   0.034422	[D5] send_to_all: SETINFO ups.productid "0003"
   0.034765	[D5] send_to_all: SETINFO driver.version "2.8.0"
   0.035095	[D5] send_to_all: SETINFO driver.version.internal "0.14"
   0.035419	[D5] send_to_all: SETINFO driver.name "blazer_usb"
   0.035716	Please note that this driver is deprecated and will not receive
new development. If it works for managing your devices - fine,
but if you are running it to try setting up a new device, please
consider the newer nutdrv_qx instead, which should handle all 'Qx'
protocol variants for NUT. (Please also report if your device works
with this driver, but nutdrv_qx would not actually support it with
any subdriver!)

   0.037627	[D2] Trying megatec protocol...
[... 140 lines cut ...]
battery.voltage.high: -1.08
battery.voltage.low: -0.87
device.type: ups
driver.name: blazer_usb
driver.parameter.bus: 001
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: auto
driver.version: 2.8.0
driver.version.internal: 0.14
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
input.frequency: 50.0
input.voltage: 227.8
input.voltage.fault: 230.4
output.voltage: 230.1
ups.beeper.status: enabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.load: 14
ups.productid: 0003
ups.status: OL
ups.temperature: 23.2
ups.type: online
ups.vendorid: 06da
root@DARYL:/home/magade#

jimklimov, 2024-02-12:
Good to hear! The drivers *are* a bit different, with `nutdrv_qx` aiming to replace several earlier drivers for Megatec Qx protocol family and so including their code (hence the deprecation warning), but perhaps something got implemented differently in the years after the initial split. The negative battery high/low limit voltages look weird though. IIRC there was some bit-shift maths problem solved after 2.8.0, maybe that was it.

MAGADE007, 2024-02-12:
With some add in ups.cong, I got this, not perfect but at least lot of overwrites info
root@DARYL:/home/magade# cd /etc/nut
root@DARYL:/etc/nut# upsc nutdev1
Init SSL without certificate database
battery.charge: 100
battery.packs: 6
battery.runtime: 1810
battery.voltage: 2.31
battery.voltage.high: 2.31
battery.voltage.low: 2.21
battery.voltage.nominal: 72.0
device.type: ups
driver.name: blazer_usb
driver.parameter.chargetime: 3600
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.product: USB UPS
driver.parameter.productid: 0003
driver.parameter.runtimecal: 181,100,540,50,1320,25,4200,7
driver.parameter.synchronous: auto
driver.parameter.vendor: PPC
driver.parameter.vendorid: 06DA
driver.version: 2.8.0
driver.version.internal: 0.14
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
input.frequency: 50.0
input.voltage: 223.3
input.voltage.fault: 230.4
output.voltage: 230.1
ups.beeper.status: disabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.load: 13
ups.productid: 0003
ups.status: OL
ups.temperature: 23.5
ups.type: online
ups.vendorid: 06da
root@DARYL:/etc/nut# 

ups.conf
[nutdev1]
#	driver = "nutdrv_qx"
#       driver = "usbhid-ups"
        driver = "blazer_usb"
        port = "auto"
        vendorid = "06DA"
        productid = "0003"
        product = "USB UPS"
        vendor = "PPC"
        bus = "001

#THIS IS FOR CORRECT VOLTAGE BATTERY INFO
	default.battery.voltage.nominal = 72.0
	override.battery.voltage.nominal = 72.0

#this for correct Voltage value wrong 24V 
        override.battery.packs = 6

#THIS IS FOR GET % BATTERY CHARGE STATE
	default.battery.voltage.high = 2.31
	default.battery.voltage.low = 2.21

#THIS IS FOR CORRECT DATA VOLTAGE
#default.battery.voltage.nominal =
#override.battery.voltage.nominal =
#this for correct Voltage value wrong 24V 
#        override.battery.packs = 12

#THIS IS FOR KNOW TIME REMINING FONCTION ON LOAD 
	runtimecal = 181,100,540,50,1320,25,4200,7

#THIS IS TO KNOW HOW LONG BEFORE CHARGE BATTERY TO 100%
	chargetime = 3600
root@DARYL:/etc/nut#

MAGADE007, 2024-02-12:
I dont succed to update nut to 2.8.1-3, how can I do that?

jimklimov, 2024-02-14:
> I don't succeed to update nut to 2.8.1-3, how can I do that?

Depends on the OS (seems to be Debian/Ubuntu like, by the numbering scheme?) In some you can add "experimental" repositories or download package files from newer distribution versions.

https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests and installing the result (if it behaves better in tests from the build workspace) may be a more simple option.