networkupstools/nut #2304: @aquette : any idea if this is intentional, or if the `"NULL"` entry can be removed?

jimklimov, 2024-02-10:
https://github.com/networkupstools/nut/blob/dcbb2ba041fd1feaff924f3edd388d2f75c350ae/drivers/raritan-px2-mib.c#L55-L59

              @aquette : any idea if this is intentional, or if the `"NULL"` (string) entry can be removed in favor of a `NULL` (pointer) sentinel? Comes all the way from original commit https://github.com/networkupstools/nut/commit/cc63de084424b64ed394081604bc4ce0ed83a945

_Originally posted by @jimklimov in https://github.com/networkupstools/nut/pull/2285#discussion_r1484311381_

aquette, 2024-02-10:
Probably a typo, should be unquoted NULL, check others and lookup value() to be sure

jimklimov, 2024-02-10:
BTW, looking at OIDs now - got to https://cdn.raritan.com/download/px3/version-3.6.31/MIB_Usage_3.6.31_47637.pdf and it seems the last `.1` before `%i` is a "PDU ID", so the number could be not a "1" in case of daisy-chain... 

https://github.com/networkupstools/nut/blob/dcbb2ba041fd1feaff924f3edd388d2f75c350ae/drivers/raritan-px2-mib.c#L133

jimklimov, 2024-02-10:
And https://www.circitor.fr/Mibs/Html/P/PDU2-MIB.php#SensorStateEnumeration offers many more values that can be seen in this enumeration.

closed 2024-02-11 by jimklimov