networkupstools/nut #2307: Fry's Electronic China UPS

denveronly, 2024-02-10:
Howdy!

Ive been struggling a full day to hook  a Chineese UPS to a NUT server.
Tried it on homeassistant and Unraid instances, just to make sure there is no USB driver issues.

So i have a brand "2E" RE650VA UPS, which goes to a server 2u Rack. It is pretty neat, but lacks of NUT support.
Unraid Monitor gave me a setup like this to try.

`[nutdev1]
driver = "nutdrv_qx"
port = "auto"
vendorid = "0001"
productid = "0000"
bus = "001"
device = "006"
busport = "004"
###NOTMATCHED-YET###bcdDevice = "0100"`

Ubuntu VM with a nut-scanner proposed nutdrv_actl_usb

All are failing to start, but the original program that came on a CD UPSmon works.

**lsusb -v** give such an answer on connected device

`Bus 001 Device 006: ID 0001:0000 Fry's Electronics 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0001 Fry's Electronics
  idProduct          0x0000 
  bcdDevice            1.00
  iManufacturer           0 
  iProduct                1 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0029
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     624
          Report Descriptor: (length is 624)
            Item(Global): Usage Page, data= [ 0x86 ] 134
[... 458 lines cut ...]
            Item(Main  ): End Collection, data=none
            Item(Main  ): End Collection, data=none
            Item(Main  ): End Collection, data=none
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              10
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              10
Device Status:     0x0000
  (Bus Powered)`

jimklimov, 2024-02-10:
Hello, can you try stopping the NUT programs and starting the drivers from command line in probe mode, e.g. `nutdrv_qx -a myups -DDDDDD -d 1` so they would report in more detail what they tried and how that failed?

jimklimov, 2024-02-10:
In the screenshot, "Mega" protocol is mentioned. I assume it means a variant of Megatec Q<x> protocol family, so generally `nutdrv_qx` should be a good match.

There were some fixes merged recently for this driver, to help it handle devices this played loose with the spec (e.g. ending some messages with zero-bytes instead of zero characters in a sort of flag bit map entry). I am not sure if that was done for a specific subdriver (IIRC Ablerex) or for the whole ecosystem, however.

If you have a chance to build NUT from git sources - it may be possible to see if the fix takes effect and helps with your device.

More details at https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests - note you do not have to replace a packaged NUT just to test the driver (can do from build workspace).

denveronly, 2024-02-10:
> Hello, can you try stopping the NUT programs and starting the drivers from command line in probe mode, e.g. `nutdrv_qx -a myups -DDDDDD -d 1` so they would report in more detail what they tried and how that failed?

im not sure how to start it in probe mode,
the only idea i tried is 
[image]

denveronly, 2024-02-10:
> In the screenshot, "Mega" protocol is mentioned. I assume it means a variant of Megatec Q protocol family, so generally `nutdrv_qx` should be a good match.
> 
> There were some fixes merged recently for this driver, to help it handle devices this played loose with the spec (e.g. ending some messages with zero-bytes instead of zero characters in a sort of flag bit map entry). I am not sure if that was done for a specific subdriver (IIRC Ablerex) or for the whole ecosystem, however.
> 
> If you have a chance to build NUT from git sources - it may be possible to see if the fix takes effect and helps with your device.
> 
> More details at https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests - note you do not have to replace a packaged NUT just to test the driver (can do from build workspace).

Maybe it is easier just to replace the driver file from /usr/lib/nut , the file nutdrv_qx from other version?

jimklimov, 2024-02-10:
In Unix/Linux, you have to provide a path to programs (if not in a `PATH` listed location), e.g. `./nutdrv_qx -a myups -DDDDDD -d 1` (assuming `myups` is the device configuration section name in your `ups.conf`).

To "`rm` and `cp`" you have to build that newer driver program first :) And if after testing from the build workspace it behaves better, the `make install` would do a better job to copy the new files into your system (the wiki article ends with this optional replacement step).

denveronly, 2024-02-10:
> In Unix/Linux, you have to provide a path to programs (if not in a `PATH` listed location), e.g. `./nutdrv_qx -a myups -DDDDDD -d 1` (assuming `myups` is the device configuration section name in [...]

oh i get, still learning. sorry.
I tried the probe mode, but the info is not so informative

[image]

jimklimov, 2024-02-10:
Hm, I was under impression you had NUT 2.8.x, must have mistaken with another issue of the day. Probably have to build a new one or find packaged.

CristiXP, 2024-03-16:
Any fix for this? I have the same problem in Homeassistant.

marios8543, 2024-11-22:
Any news about this ? Also struggling with a similar UPS

jimklimov, 2024-11-23:
"Similar" being "2E" branded, or with ID 0000:0001 indicating that the vendor did not even bother to get an ID?

marios8543, 2024-11-23:
It is ID 0000:0001 with a "Fry's Electronics" vendor. It also works with the UPSmart with the Mega (USB) option as per the original post.

emukhin, 2024-12-11:
I also happen to have a piece of Fry's Electronics UPS. It's zeros everywhere.

device.type "ups"
driver.version "2.7.4"
driver.version.internal "0.28"
driver.name "nutdrv_qx"

```
0.319356     Checking device (0001/0000) (001/004)
   0.324043     - VendorID: 0001
   0.324063     - ProductID: 0000
   0.324072     - Manufacturer: unknown
   0.324080     - Product: MEC0003
   0.324091     - Serial Number: unknown
   0.324101     - Bus: 001
   0.324113     - Device release number: 0100
   0.324128     Trying to match device
   0.324191     Device matches
   0.324222     nut_usb_set_altinterface: skipped usb_set_altinterface(udev, 0)
   0.324238     send_to_all: SETINFO ups.vendorid "0001"
   0.324250     send_to_all: SETINFO ups.productid "0000"
   0.324266     send: QGS
   0.324277     read: QGS
   0.324288     qx_process_answer: short reply (input.voltage)
   0.324301     send: QGS
   0.324313     read: QGS
   0.324323     qx_process_answer: short reply (input.voltage)
   0.324335     send: QGS
   0.324345     read: QGS
   0.324355     qx_process_answer: short reply (input.voltage)
   0.324368     send: M
   0.324380     read: M
   0.324394     voltronic_qs_protocol: invalid protocol [M]
   0.324407     ups_infoval_set: failed to preprocess value [ups.firmware.aux: M]
   0.324421     send: M
   0.324432     read: M
   0.324444     voltronic_qs_protocol: invalid protocol [M]
   0.324454     ups_infoval_set: failed to preprocess value [ups.firmware.aux: M]
   0.324465     send: M
   0.324475     read: M
   0.324485     voltronic_qs_protocol: invalid protocol [M]
   0.324495     ups_infoval_set: failed to preprocess value [ups.firmware.aux: M]
   0.324506     send: M
   0.324517     read: M
   0.324534     voltronic_qs_hex_protocol: invalid protocol [M]
   0.324544     ups_infoval_set: failed to preprocess value [ups.firmware.aux: M]
   0.324556     send: M
   0.324566     read: M
   0.324575     voltronic_qs_hex_protocol: invalid protocol [M]
   0.324585     ups_infoval_set: failed to preprocess value [ups.firmware.aux: M]
   0.324597     send: M
   0.324607     read: M
   0.324619     voltronic_qs_hex_protocol: invalid protocol [M]
   0.324632     ups_infoval_set: failed to preprocess value [ups.firmware.aux: M]
   0.324646     send: QS
   0.324655     read: QS
   0.324668     qx_process_answer: short reply (input.voltage)
   0.324679     send: QS
   0.324690     read: QS
   0.324702     qx_process_answer: short reply (input.voltage)
   0.324713     send: QS
   0.324723     read: QS
   0.324734     qx_process_answer: short reply (input.voltage)
   0.324748     send: D
   0.324761     read: D
   0.324776     qx_process_answer: short reply (input.voltage)
   0.324787     send: D
   0.324797     read: D
   0.324808     qx_process_answer: short reply (input.voltage)
   0.324817     send: D
[... 153 lines cut ...]
   0.436176     Using protocol: Q1 0.07
   0.436195     send_to_all: SETINFO device.type "ups"
   0.436207     send_to_all: SETINFO driver.version "2.7.4"
   0.436219     send_to_all: SETINFO driver.version.internal "0.28"
   0.436235     send_to_all: SETINFO driver.name "nutdrv_qx"
   0.436245     upsdrv_initinfo...
   0.436257     send_to_all: SETINFO driver.version.data "Q1 0.07"
   0.436287     send: Q1
   0.443130     received 47 (40)
   0.443170     read: (47 bytes) => 28 30 30 30 2e 30 20 30 30 30 2e 30 20 30 30 30 2e 30
   0.443184      20 30 30 30 20 30 30 2e 30 20 30 2e 30 30 20 30 30 2e 30 20 30 30 30 30 30
   0.443194      30 30 30 0d
   0.443206     read: (000.0 000.0 000.0 000 00.0 0.00 00.0 00000000
   0.443229     send_to_all: SETINFO input.voltage.fault "0.0"
   0.443245     send_to_all: SETINFO output.voltage "0.0"
   0.443260     send_to_all: SETINFO ups.load "0"
   0.443274     send_to_all: SETINFO input.frequency "0.0"
   0.443289     send_to_all: SETINFO battery.voltage "0.00"
   0.443303     send_to_all: SETINFO ups.temperature "0.0"
   0.443318     update_status: OL
   0.443330     update_status: !LB
   0.443348     send_to_all: SETINFO ups.type "online"
   0.443361     update_status: !CAL
   0.443378     update_status: !FSD
```