networkupstools/nut #2308: Problem with Eaton PW9130 - ConnectedUPS Web/SNMP Card V4.38

WBHWebDev, 2024-02-12:
Hi everybody, 

I've been struggling for quite some time to get the NUT package up and running with an Eaton PW9130, which is connected to a ConnectedUPS Web/SNMP Card V4.38. Manual invocation works smoothly, and all UPS attributes are displayed without any issues. However, when monitoring, I consistently encounter the following warning: 

_"Warning: excessive poll failures, limiting error reporting (OID = .1.3.6.1.4.1.534.6.8.1.1.2.1.9.1)"._

It appears that the system is unable to retrieve the "SensorSerialNumber". How can I prevent this message from being generated? Is it possible to exclude this attribute from the query altogether? Or did a make any misconfiguration?

Other than this issue, everything seems to be functioning as expected. However, having the warning message in the logs every few seconds is less than ideal. Any guidance or suggestions would be greatly appreciated.

Thank you in advance!

jimklimov, 2024-02-12:
Thanks for the report. Just in case, which version of NUT is involved?

I think "limited error reporting" should have taken hold so you would not see it "every few seconds" (might be printed just every few polling cycles though, a new one running some 5 seconds apart by default) - arguably a bug if this does not happen. Maybe it was addressed in recent years (after NUT v2.7.4 which is still deployed on many systems - hence asking).

WBHWebDev, 2024-02-12:
Thank you for the prompt response.

I'm using version 2.8.0-7 on Debian 12 Bookworm.

With a polling interval of 10 seconds, I receive the error message almost every 20 seconds. Adjusting the polling frequency results in corresponding changes to the error message frequency. Is there a way to increase the "limited error reporting"? Or can it be even deactiveded for this warning?

[image]

Thanks in advance.

jimklimov, 2024-02-12:
Looking at code, there's a counter `numerr` which is increased for every error hit, and the warning is spewed only when first `SU_ERR_LIMIT` (10) or yet another `SU_ERR_RATE` (100) issues were seen. I suppose it fails to read hundreds of your OIDs but only exposes one here (or it is the hundredth one read in your MIB, if `numerr` is reset for each polling loop?)

Check https://github.com/networkupstools/nut/wiki/Changing-NUT-daemon-debug-verbosity about starting the driver with more debug verbosity; at level 4 or more this routine would report each hit to make some sense of what happens.

https://github.com/networkupstools/nut/blob/bf105fe/drivers/snmp-ups.c#L1148-L1187

WBHWebDev, 2024-02-13:
Thank you for your guidance. I changed the loglevel and here is the log. 

I think I found the relevant part of the log by myself - see at the end. The full log file is attached...
It seems that ten values can not be found. At the tenth failure the warning is logged. So I unfortunately get a Logentry every pool.

Do you have any suggestion what I could do now to avoid many warnings in the log?
Or can I provide you more needed informations?

[log.txt](https://github.com/networkupstools/nut/files/14259818/log_short.txt)

_Feb 13 09:59:55 debian snmp-ups[1011]: [D1] ambient template definition found (ambient.%i.present)...
Feb 13 09:59:55 debian snmp-ups[1011]: [D1] guestimate_template_count(.1.3.6.1.4.1.534.6.8.1.1.4.1.1.%i)
Feb 13 09:59:55 debian snmp-ups[1011]: [D3] nut_snmp_get(.1.3.6.1.4.1.534.6.8.1.1.4.1.1.0)
Feb 13 09:59:55 debian snmp-ups[1011]: [D3] nut_snmp_walk(.1.3.6.1.4.1.534.6.8.1.1.4.1.1.0)
Feb 13 09:59:55 debian snmp-ups[1011]: [D4] nut_snmp_walk: max. iteration = 1
Feb 13 09:59:55 debian snmp-ups[1011]: [D2] [nutdev1] nut_snmp_walk: .1.3.6.1.4.1.534.6.8.1.1.4.1.1.0: (noSuchName) There is no such variable name in this MIB.
Feb 13 09:59:55 debian snmp-ups[1011]: [D3] nut_snmp_get(.1.3.6.1.4.1.534.6.8.1.1.4.1.1.1)
Feb 13 09:59:55 debian snmp-ups[1011]: [D3] nut_snmp_walk(.1.3.6.1.4.1.534.6.8.1.1.4.1.1.1)
Feb 13 09:59:55 debian snmp-ups[1011]: [D4] nut_snmp_walk: max. iteration = 1
Feb 13 09:59:55 debian snmp-ups[1011]: [D2] [nutdev1] nut_snmp_walk: .1.3.6.1.4.1.534.6.8.1.1.4.1.1.1: (noSuchName) There is no such variable name in this MIB.
Feb 13 09:59:55 debian snmp-ups[1011]: [D3] guestimate_template_count: 0
Feb 13 09:59:55 debian snmp-ups[1011]: [D1] 0 instances found...
Feb 13 09:59:55 debian snmp-ups[1011]: [D1] No ambient present, discarding template definition...
Feb 13 09:59:55 debian snmp-ups[1011]: [D1] snmp_ups_walk: processing unitary device (1)
Feb 13 09:59:55 debian snmp-ups[1011]: [D1] ambient template definition found (ambient.%i.name)...
Feb 13 09:59:55 debian snmp-ups[1011]: [D1] guestimate_template_count(.1.3.6.1.4.1.534.6.8.1.1.3.1.1.%i)
Feb 13 09:59:55 debian snmp-ups[1011]: [D3] nut_snmp_get(.1.3.6.1.4.1.534.6.8.1.1.3.1.1.0)
Feb 13 09:59:55 debian snmp-ups[1011]: [D3] nut_snmp_walk(.1.3.6.1.4.1.534.6.8.1.1.3.1.1.0)
Feb 13 09:59:55 debian snmp-ups[1011]: [D4] nut_snmp_walk: max. iteration = 1
Feb 13 09:59:55 debian snmp-ups[1011]: [D2] [nutdev1] nut_snmp_walk: .1.3.6.1.4.1.534.6.8.1.1.3.1.1.0: (noSuchName) There is no such variable name in this MIB.
Feb 13 09:59:55 debian snmp-ups[1011]: [D3] nut_snmp_get(.1.3.6.1.4.1.534.6.8.1.1.3.1.1.1)
Feb 13 09:59:55 debian snmp-ups[1011]: [D3] nut_snmp_walk(.1.3.6.1.4.1.534.6.8.1.1.3.1.1.1)
Feb 13 09:59:55 debian snmp-ups[1011]: [D4] nut_snmp_walk: max. iteration = 1
Feb 13 09:59:55 debian snmp-ups[1011]: [D2] [nutdev1] nut_snmp_walk: .1.3.6.1.4.1.534.6.8.1.1.3.1.1.1: (noSuchName) There is no such variable name in this MIB.
Feb 13 09:59:55 debian snmp-ups[1011]: [D3] guestimate_template_count: 0
Feb 13 09:59:55 debian snmp-ups[1011]: [D1] 0 instances found...
Feb 13 09:59:55 debian snmp-ups[1011]: [D1] No ambient present, discarding template definition...
Feb 13 09:59:55 debian snmp-ups[1011]: [D1] snmp_ups_walk: processing unitary device (1)
Feb 13 09:59:55 debian snmp-ups[1011]: [D1] ambient template definition found (ambient.%i.mfr)...
Feb 13 09:59:55 debian snmp-ups[1011]: [D1] guestimate_template_count(.1.3.6.1.4.1.534.6.8.1.1.2.1.6.%i)
Feb 13 09:59:55 debian snmp-ups[1011]: [D3] nut_snmp_get(.1.3.6.1.4.1.534.6.8.1.1.2.1.6.0)
Feb 13 09:59:55 debian snmp-ups[1011]: [D3] nut_snmp_walk(.1.3.6.1.4.1.534.6.8.1.1.2.1.6.0)
Feb 13 09:59:55 debian snmp-ups[1011]: [D4] nut_snmp_walk: max. iteration = 1
Feb 13 09:59:55 debian snmp-ups[1011]: [D2] [nutdev1] nut_snmp_walk: .1.3.6.1.4.1.534.6.8.1.1.2.1.6.0: (noSuchName) There is no such variable name in this MIB.
Feb 13 09:59:55 debian snmp-ups[1011]: [D3] nut_snmp_get(.1.3.6.1.4.1.534.6.8.1.1.2.1.6.1)
Feb 13 09:59:55 debian snmp-ups[1011]: [D3] nut_snmp_walk(.1.3.6.1.4.1.534.6.8.1.1.2.1.6.1)
Feb 13 09:59:55 debian snmp-ups[1011]: [D4] nut_snmp_walk: max. iteration = 1
Feb 13 09:59:55 debian snmp-ups[1011]: [D2] [nutdev1] nut_snmp_walk: .1.3.6.1.4.1.534.6.8.1.1.2.1.6.1: (noSuchName) There is no such variable name in this MIB.
Feb 13 09:59:55 debian snmp-ups[1011]: [D3] guestimate_template_count: 0
Feb 13 09:59:55 debian snmp-ups[1011]: [D1] 0 instances found...
Feb 13 09:59:55 debian snmp-ups[1011]: [D1] No ambient present, discarding template definition...
Feb 13 09:59:55 debian snmp-ups[1011]: [D1] snmp_ups_walk: processing unitary device (1)
Feb 13 09:59:55 debian snmp-ups[1011]: [D1] ambient template definition found (ambient.%i.model)...
Feb 13 09:59:55 debian snmp-ups[1011]: [D1] guestimate_template_count(.1.3.6.1.4.1.534.6.8.1.1.2.1.7.%i)
Feb 13 09:59:55 debian snmp-ups[1011]: [D3] nut_snmp_get(.1.3.6.1.4.1.534.6.8.1.1.2.1.7.0)
Feb 13 09:59:55 debian snmp-ups[1011]: [D3] nut_snmp_walk(.1.3.6.1.4.1.534.6.8.1.1.2.1.7.0)
Feb 13 09:59:55 debian snmp-ups[1011]: [D4] nut_snmp_walk: max. iteration = 1
Feb 13 09:59:55 debian snmp-ups[1011]: [D2] [nutdev1] nut_snmp_walk: .1.3.6.1.4.1.534.6.8.1.1.2.1.7.0: (noSuchName) There is no such variable name in this MIB.
Feb 13 09:59:55 debian snmp-ups[1011]: [D3] nut_snmp_get(.1.3.6.1.4.1.534.6.8.1.1.2.1.7.1)
Feb 13 09:59:55 debian snmp-ups[1011]: [D3] nut_snmp_walk(.1.3.6.1.4.1.534.6.8.1.1.2.1.7.1)
[... cut ...]
Feb 13 09:59:55 debian snmp-ups[1011]: [D3] nut_snmp_walk(.1.3.6.1.4.1.534.6.8.1.1.2.1.9.1)
Feb 13 09:59:55 debian snmp-ups[1011]: [D4] nut_snmp_walk: max. iteration = 1
Feb 13 09:59:55 debian snmp-ups[1011]: [nutdev1] Warning: excessive poll failures, limiting error reporting (OID = .1.3.6.1.4.1.534.6.8.1.1.2.1.9.1)
Feb 13 09:59:55 debian snmp-ups[1011]: [D3] guestimate_template_count: 0
Feb 13 09:59:55 debian snmp-ups[1011]: [D1] 0 instances found...
Feb 13 09:59:55 debian snmp-ups[1011]: [D1] No ambient present, discarding template definition..._

jimklimov, 2024-02-14:
Looking at these entries, it seems like they are about supporting an array (with `%i` wildcard iterator here and there) of ambient sensors and your UPS probably has none attached.

I'd expect these to be tried, not found, and thus discarded from the checklist for active re-queries when the `snmp-ups` driver does an initial walk, which is more or less the reason of it doing one (also to find "static" fields like model name to remember and never re-query them). The log actually says `No ambient present, discarding template definition...` but given this happens in `upsdrv_updateinfo()` cycles and not in `upsdrv_initinfo()` alone, probably something is broken about the "discarding" part.

CC @aquette : any insights on this one, please?

@WBHWebDev  : for a quick alleviation of your syslog, one idea that comes to mind is to bump that `SU_ERR_LIMIT` value from `10` to a somewhat higher number in `drivers/snmp-ups.h`, if building and running a customized NUT codebase is an option for you...

WBHWebDev, 2024-02-16:
@jimklimov: Thank you for your suggestion. I would prefer not to use an adapted version, mainly due to potential updates, etc. That would be more of a last resort.

In the meantime, I reached out to EATON for a solution, but unfortunately, they cannot offer any support for PROXMOX. Quite disappointing!

Perhaps @aquette has another idea, or maybe the issue will be patched in the next version? In that case, I could manage with an adapted version temporarily. What are your thoughts on this?

arnaudquette-eaton, 2024-02-16:
sorry for the eternity taken to answer. offloading from @aquette to @arnaudquette-eaton :D

I thought I shared already long ago with @jimklimov, but this is a false positive log, due to indeed non-existent OIDs.
Can you please confirm with an snmpget ... .1.3.6.1.4.1.534.6.8.1.1.2.1.9.1?

I once fixed this years ago, but not pushed and lost the tip (or well hidden).

@jimklimov can you please log a real-fix task for the future, and possibly just adapt the logging for now?

note that this driver is still a mess (the initial PoC I made in 2002, that chaotically grown :D) and still subject to a full rewrite.
Definitely not for a retired developer ;)

WBHWebDev, 2024-02-16:
Hi arnaudquette,

this is the output I received when performing an SNMPget on my test enviroment:

[image]

I hope this is what you were expecting. Otherwise, please feel free to ask for executing other commands. 
Additionally, if you need any other logs or information, I will do my best to assist you.

Thank you.

jimklimov, 2024-02-16:
@aquette : just in case, I've double-checked that the possibly made solution you mention is not in the FTY branch :( But your and upstream repos have a lot of branches slightly ahead of respective masters to trawl and look for gems...

At least, there are no significant changes around the lines which report this alert:

* https://github.com/networkupstools/nut/compare/master...jimklimov:nut:FTY-remerge-20240123?expand=1#diff-179abee8003a5af3a7bb9f01fe6a3a9d3748c4818cc51038a1df8da398d61c69R1271