networkupstools/nut #2309: nutdrv_qx: memory leak

Yellow-Horror, 2024-02-13:
In nutdrv_qx version 2023.07.11 on XigmaNAS (FreeBSD based) there is a memory leak, somehow related to the loss of communication with the UPS.

Symptoms: When communication with the UPS is lost, the memory consumption of the nutdrv_qx process increases by 4 kb every few seconds. Consumed memory will never be released by the process, even if the connection is later restored.

jimklimov, 2024-02-13:
Thanks for the report! Are you in position to check if this also happens with current NUT codebase (a build from github master) to make sure this was not solved since?

A blind guess would be that in case of connection loss the loop allocates some string to print error messages and never frees it, or something of the sort...

Yellow-Horror, 2024-02-13:
Unfortunately, I cannot check the relevance of the problem with the latest code, since I am using the “embedded” installation of XigmaNAS, which does not provide the possibility of changing system components. I can promise to check for the problem in the next XigmaNAS release, but the project is developing slowly and does not have a regular release schedule.

jimklimov, 2024-02-29:
Finally got to glance at the code, and it seems there are two situations that can lead to reconnections, with some code paths (and/or platform depenent builds) maybe mis-balancing `libusb_open()` attempts with `libusb_close()` clean-up. This seems to be a pattern in several USB-capable drivers after a recent-ish refactoring pass.

Maybe the library re-allocates something for the `udev` handle upon each `open()`?.. Didn't quickly find examples or docs stating if it should be `free()`'d by caller, either.

jimklimov, 2024-02-29:
Ok, tried and failed to reproduce with `usbhid-ups` on Linux, ad least, also after commenting away `ifdef`s that avoided `libusb_close()` on some OSes...

jimklimov, 2024-03-01:
I don't have a true QX UPS on hand, so hacked `nutdrv_qx.c` to accept whatever device I had to readily plug around. 
* https://github.com/jimklimov/nut/tree/issue-2309-hacks
* `make -s && valgrind --leak-check=full  --show-leak-kinds=all ./drivers/nutdrv_qx -s test -x port=auto -x vendorid=0463 -x productid=ffff -x subdriver=null -DD`

Reconnection attempts after the first disconnection did increase the memory footprint by a few KB (12 to ~100 between runs) according to `top` and `getrusage()`, but the number usually did not move (rarely decreased) after plugging the device back into same or different port.

Tracing with `valgrind` reports a number of allocations remaining not freed, mostly related to dynamic loading and some to libusb. It may well be that the library misbehaves differently on Linux and FreeBSD, but there's little NUT can do about it.

There's also a suspect possible re-allocation of USB matcher structures (exact/regex); can at least tackle those...

````
^C  52.698408   [D1] set_exit_flag: raising exit flag due to signal 2
  52.699204     Signal 2: exiting
  52.700571     [D1] upsdrv_cleanup...
==3825797==
==3825797== HEAP SUMMARY:
==3825797==     in use at exit: 8,272 bytes in 55 blocks
==3825797==   total heap usage: 12,848 allocs, 12,793 frees, 6,327,228 bytes allocated
==3825797==
==3825797== 4 bytes in 1 blocks are still reachable in loss record 1 of 24
==3825797==    at 0x483877F: malloc (vg_replace_malloc.c:307)
==3825797==    by 0x4B1BB7A: strdup (strdup.c:42)
==3825797==    by 0x4EC42F6: udev_monitor_filter_add_match_subsystem_devtype (in /usr/lib/x86_64-linux-gnu/libudev.so.1.7.0)
==3825797==    by 0x4A6481B: ??? (in /usr/lib/x86_64-linux-gnu/libusb-1.0.so.0.3.0)
==3825797==    by 0x4A622BC: ??? (in /usr/lib/x86_64-linux-gnu/libusb-1.0.so.0.3.0)
==3825797==    by 0x4A59C93: libusb_init (in /usr/lib/x86_64-linux-gnu/libusb-1.0.so.0.3.0)
==3825797==    by 0x123B2F: nut_libusb_open (libusb1.c:205)
==3825797==    by 0x11C121: qx_command (nutdrv_qx.c:3529)
==3825797==    by 0x120E6D: qx_process (nutdrv_qx.c:4367)
==3825797==    by 0x122041: qx_ups_walk (nutdrv_qx.c:3932)
==3825797==    by 0x122C3C: upsdrv_updateinfo (nutdrv_qx.c:3078)
==3825797==    by 0x11A605: main (main.c:2491)
==3825797==
==3825797== 11 bytes in 1 blocks are still reachable in loss record 2 of 24
==3825797==    at 0x483877F: malloc (vg_replace_malloc.c:307)
==3825797==    by 0x4B1BB7A: strdup (strdup.c:42)
==3825797==    by 0x4EC430F: udev_monitor_filter_add_match_subsystem_devtype (in /usr/lib/x86_64-linux-gnu/libudev.so.1.7.0)
==3825797==    by 0x4A6481B: ??? (in /usr/lib/x86_64-linux-gnu/libusb-1.0.so.0.3.0)
==3825797==    by 0x4A622BC: ??? (in /usr/lib/x86_64-linux-gnu/libusb-1.0.so.0.3.0)
==3825797==    by 0x4A59C93: libusb_init (in /usr/lib/x86_64-linux-gnu/libusb-1.0.so.0.3.0)
==3825797==    by 0x123B2F: nut_libusb_open (libusb1.c:205)
==3825797==    by 0x11C121: qx_command (nutdrv_qx.c:3529)
==3825797==    by 0x120E6D: qx_process (nutdrv_qx.c:4367)
==3825797==    by 0x122041: qx_ups_walk (nutdrv_qx.c:3932)
==3825797==    by 0x122C3C: upsdrv_updateinfo (nutdrv_qx.c:3078)
==3825797==    by 0x11A605: main (main.c:2491)
==3825797==
==3825797== 15 bytes in 1 blocks are possibly lost in loss record 3 of 24
==3825797==    at 0x483877F: malloc (vg_replace_malloc.c:307)
==3825797==    by 0x4B1BB7A: strdup (strdup.c:42)
==3825797==    by 0x13CA4B: xstrdup (common.c:1884)
==3825797==    by 0x119C1B: splitxarg (main.c:1406)
==3825797==    by 0x119C1B: main (main.c:1894)
==3825797==
==3825797== 16 bytes in 1 blocks are still reachable in loss record 4 of 24
==3825797==    at 0x483877F: malloc (vg_replace_malloc.c:307)
==3825797==    by 0x4EBF4EB: udev_new (in /usr/lib/x86_64-linux-gnu/libudev.so.1.7.0)
==3825797==    by 0x4A647D6: ??? (in /usr/lib/x86_64-linux-gnu/libusb-1.0.so.0.3.0)
==3825797==    by 0x4A622BC: ??? (in /usr/lib/x86_64-linux-gnu/libusb-1.0.so.0.3.0)
==3825797==    by 0x4A59C93: libusb_init (in /usr/lib/x86_64-linux-gnu/libusb-1.0.so.0.3.0)
==3825797==    by 0x123B2F: nut_libusb_open (libusb1.c:205)
==3825797==    by 0x11C121: qx_command (nutdrv_qx.c:3529)
==3825797==    by 0x120E6D: qx_process (nutdrv_qx.c:4367)
==3825797==    by 0x122041: qx_ups_walk (nutdrv_qx.c:3932)
==3825797==    by 0x122C3C: upsdrv_updateinfo (nutdrv_qx.c:3078)
==3825797==    by 0x11A605: main (main.c:2491)
==3825797==
==3825797== 24 bytes in 1 blocks are still reachable in loss record 5 of 24
==3825797==    at 0x483877F: malloc (vg_replace_malloc.c:307)
==3825797==    by 0x4EC3C2F: udev_monitor_new_from_netlink (in /usr/lib/x86_64-linux-gnu/libudev.so.1.7.0)
[... 234 lines cut ...]
==3825797==    by 0x483ADE7: realloc (vg_replace_malloc.c:834)
==3825797==    by 0x4A617ED: ??? (in /usr/lib/x86_64-linux-gnu/libusb-1.0.so.0.3.0)
==3825797==    by 0x4A63D36: ??? (in /usr/lib/x86_64-linux-gnu/libusb-1.0.so.0.3.0)
==3825797==    by 0x4A64CAC: ??? (in /usr/lib/x86_64-linux-gnu/libusb-1.0.so.0.3.0)
==3825797==    by 0x4A62196: ??? (in /usr/lib/x86_64-linux-gnu/libusb-1.0.so.0.3.0)
==3825797==    by 0x4A59C93: libusb_init (in /usr/lib/x86_64-linux-gnu/libusb-1.0.so.0.3.0)
==3825797==    by 0x123B2F: nut_libusb_open (libusb1.c:205)
==3825797==    by 0x11C121: qx_command (nutdrv_qx.c:3529)
==3825797==    by 0x120E6D: qx_process (nutdrv_qx.c:4367)
==3825797==    by 0x122041: qx_ups_walk (nutdrv_qx.c:3932)
==3825797==    by 0x122C3C: upsdrv_updateinfo (nutdrv_qx.c:3078)
==3825797==
==3825797== LEAK SUMMARY:
[... cut ...]
==3825797==    still reachable: 4,292 bytes in 15 blocks
==3825797==         suppressed: 48 bytes in 1 blocks
==3825797==
==3825797== For lists of detected and suppressed errors, rerun with: -s
==3825797== ERROR SUMMARY: 9 errors from 9 contexts (suppressed: 0 from 0)
==3825797== could not unlink /tmp/vgdb-pipe-from-vgdb-to-3825797-by-root-on-???
==3825797== could not unlink /tmp/vgdb-pipe-to-vgdb-from-3825797-by-root-on-???
==3825797== could not unlink /tmp/vgdb-pipe-shared-mem-vgdb-3825797-by-root-on-???
````

jimklimov, 2024-03-01:
Tried tracing with just that "hack" commit, and physical reconnections (not simulated by libusb pipe error), and without the potential fix for matcher re-allocation - can not say matchers leak. The set of errors this time is different (maybe Ctrl+C landed in the middle of reading) but all paths lead into libusb:
````
^C  65.443552   [D1] set_exit_flag: raising exit flag due to signal 2
  65.444538     Signal 2: exiting
  65.446274     [D1] upsdrv_cleanup...
==1395203==
==1395203== HEAP SUMMARY:
==1395203==     in use at exit: 8,836 bytes in 61 blocks
==1395203==   total heap usage: 12,699 allocs, 12,638 frees, 6,112,973 bytes allocated
==1395203==
==1395203== 4 bytes in 1 blocks are still reachable in loss record 1 of 30
==1395203==    at 0x483877F: malloc (vg_replace_malloc.c:307)
==1395203==    by 0x4B1BB7A: strdup (strdup.c:42)
==1395203==    by 0x4EC42F6: udev_monitor_filter_add_match_subsystem_devtype (in /usr/lib/x86_64-linux-gnu/libudev.so.1.7.0)
==1395203==    by 0x4A6481B: ??? (in /usr/lib/x86_64-linux-gnu/libusb-1.0.so.0.3.0)
==1395203==    by 0x4A622BC: ??? (in /usr/lib/x86_64-linux-gnu/libusb-1.0.so.0.3.0)
==1395203==    by 0x4A59C93: libusb_init (in /usr/lib/x86_64-linux-gnu/libusb-1.0.so.0.3.0)
==1395203==    by 0x123B2F: nut_libusb_open (libusb1.c:205)
==1395203==    by 0x11C121: qx_command (nutdrv_qx.c:3529)
==1395203==    by 0x120E6D: qx_process (nutdrv_qx.c:4367)
==1395203==    by 0x122041: qx_ups_walk (nutdrv_qx.c:3932)
==1395203==    by 0x122C3C: upsdrv_updateinfo (nutdrv_qx.c:3078)
==1395203==    by 0x11A605: main (main.c:2491)
==1395203==
==1395203== 4 bytes in 1 blocks are possibly lost in loss record 2 of 30
==1395203==    at 0x483877F: malloc (vg_replace_malloc.c:307)
==1395203==    by 0x4B1BB7A: strdup (strdup.c:42)
==1395203==    by 0x4A61714: ??? (in /usr/lib/x86_64-linux-gnu/libusb-1.0.so.0.3.0)
==1395203==    by 0x4A63D36: ??? (in /usr/lib/x86_64-linux-gnu/libusb-1.0.so.0.3.0)
==1395203==    by 0x4A6409F: ??? (in /usr/lib/x86_64-linux-gnu/libusb-1.0.so.0.3.0)
==1395203==    by 0x4A645B9: ??? (in /usr/lib/x86_64-linux-gnu/libusb-1.0.so.0.3.0)
==1395203==    by 0x4A64716: ??? (in /usr/lib/x86_64-linux-gnu/libusb-1.0.so.0.3.0)
==1395203==    by 0x4A77EA6: start_thread (pthread_create.c:477)
==1395203==    by 0x4B8DA2E: clone (clone.S:95)
==1395203==
==1395203== 11 bytes in 1 blocks are still reachable in loss record 3 of 30
==1395203==    at 0x483877F: malloc (vg_replace_malloc.c:307)
==1395203==    by 0x4B1BB7A: strdup (strdup.c:42)
==1395203==    by 0x4EC430F: udev_monitor_filter_add_match_subsystem_devtype (in /usr/lib/x86_64-linux-gnu/libudev.so.1.7.0)
==1395203==    by 0x4A6481B: ??? (in /usr/lib/x86_64-linux-gnu/libusb-1.0.so.0.3.0)
==1395203==    by 0x4A622BC: ??? (in /usr/lib/x86_64-linux-gnu/libusb-1.0.so.0.3.0)
==1395203==    by 0x4A59C93: libusb_init (in /usr/lib/x86_64-linux-gnu/libusb-1.0.so.0.3.0)
==1395203==    by 0x123B2F: nut_libusb_open (libusb1.c:205)
==1395203==    by 0x11C121: qx_command (nutdrv_qx.c:3529)
==1395203==    by 0x120E6D: qx_process (nutdrv_qx.c:4367)
==1395203==    by 0x122041: qx_ups_walk (nutdrv_qx.c:3932)
==1395203==    by 0x122C3C: upsdrv_updateinfo (nutdrv_qx.c:3078)
==1395203==    by 0x11A605: main (main.c:2491)
==1395203==
==1395203== 15 bytes in 1 blocks are possibly lost in loss record 4 of 30
==1395203==    at 0x483877F: malloc (vg_replace_malloc.c:307)
==1395203==    by 0x4B1BB7A: strdup (strdup.c:42)
==1395203==    by 0x13CA0B: xstrdup (common.c:1884)
==1395203==    by 0x119C1B: splitxarg (main.c:1406)
==1395203==    by 0x119C1B: main (main.c:1894)
==1395203==
==1395203== 16 bytes in 1 blocks are still reachable in loss record 5 of 30
==1395203==    at 0x483877F: malloc (vg_replace_malloc.c:307)
==1395203==    by 0x4EBF4EB: udev_new (in /usr/lib/x86_64-linux-gnu/libudev.so.1.7.0)
==1395203==    by 0x4A647D6: ??? (in /usr/lib/x86_64-linux-gnu/libusb-1.0.so.0.3.0)
==1395203==    by 0x4A622BC: ??? (in /usr/lib/x86_64-linux-gnu/libusb-1.0.so.0.3.0)
==1395203==    by 0x4A59C93: libusb_init (in /usr/lib/x86_64-linux-gnu/libusb-1.0.so.0.3.0)
==1395203==    by 0x123B2F: nut_libusb_open (libusb1.c:205)
==1395203==    by 0x11C121: qx_command (nutdrv_qx.c:3529)
==1395203==    by 0x120E6D: qx_process (nutdrv_qx.c:4367)
==1395203==    by 0x122041: qx_ups_walk (nutdrv_qx.c:3932)
==1395203==    by 0x122C3C: upsdrv_updateinfo (nutdrv_qx.c:3078)
==1395203==    by 0x11A605: main (main.c:2491)
==1395203==
==1395203== 24 bytes in 1 blocks are still reachable in loss record 6 of 30
[... 293 lines cut ...]
==1395203==    by 0x483ADE7: realloc (vg_replace_malloc.c:834)
==1395203==    by 0x4A617ED: ??? (in /usr/lib/x86_64-linux-gnu/libusb-1.0.so.0.3.0)
==1395203==    by 0x4A63D36: ??? (in /usr/lib/x86_64-linux-gnu/libusb-1.0.so.0.3.0)
==1395203==    by 0x4A64CAC: ??? (in /usr/lib/x86_64-linux-gnu/libusb-1.0.so.0.3.0)
==1395203==    by 0x4A62196: ??? (in /usr/lib/x86_64-linux-gnu/libusb-1.0.so.0.3.0)
==1395203==    by 0x4A59C93: libusb_init (in /usr/lib/x86_64-linux-gnu/libusb-1.0.so.0.3.0)
==1395203==    by 0x123B2F: nut_libusb_open (libusb1.c:205)
==1395203==    by 0x11C121: qx_command (nutdrv_qx.c:3529)
==1395203==    by 0x120E6D: qx_process (nutdrv_qx.c:4367)
==1395203==    by 0x122041: qx_ups_walk (nutdrv_qx.c:3932)
==1395203==    by 0x122C3C: upsdrv_updateinfo (nutdrv_qx.c:3078)
==1395203==
==1395203== LEAK SUMMARY:
==1395203==    definitely lost: 39 bytes in 3 blocks
==1395203==    indirectly lost: 0 bytes in 0 blocks
[... cut ...]
==1395203==    still reachable: 4,316 bytes in 16 blocks
==1395203==         suppressed: 48 bytes in 1 blocks
==1395203==
==1395203== For lists of detected and suppressed errors, rerun with: -s
==1395203== ERROR SUMMARY: 14 errors from 14 contexts (suppressed: 0 from 0)
==1395203== could not unlink /tmp/vgdb-pipe-from-vgdb-to-1395203-by-root-on-???
==1395203== could not unlink /tmp/vgdb-pipe-to-vgdb-from-1395203-by-root-on-???
==1395203== could not unlink /tmp/vgdb-pipe-shared-mem-vgdb-1395203-by-root-on-???
````

jimklimov, 2024-03-01:
I think at the moment this is as far as I can take this quest, without a device and the OS on metal.

If you are able to build NUT on your system (perhaps using a VM with an equivalent FreeBSD release and passing through the USB from UPS), following the `config-prereqs.txt` and building with `ci_build.sh` for quickly wrapped defaults, you can run the driver (as root to avoid USB devfs permissions hassle) under `valgrind` as I outlined above, and reproduce the actual problem with not-hacked driver, maybe those traces would help find a bug in NUT if its code does leak somewhere.

Maybe the actual leak is still in libusb, but some change to call structure from NUT would help (e.g. diligently close/open upon reconnection attempts, take care to `libusb_init()` only once, etc.)?..

biergaizi, 2024-03-03:
> If you are able to build NUT on your system (perhaps using a VM with an equivalent FreeBSD release and passing through the USB from UPS)

Second this. If one wants to see the problem getting fixed, contributing to the test effort by running the git upstream code is often necessary to accelerate progress (which otherwise may take *years* because of the slow release cycle). If it's inconvenient to test it on your main server, a workaround is to test it on a virtual machine in combination with USB passthrough, or test it on a spare single-board computer like a Raspberry Pi (the last time I've checked, FreeBSD works perfectly fine). These methods allow a quick test without disturbing the server.

jimklimov, 2024-07-22:
@Yellow-Horror : gentle bump? were you able to test the updated `master` codebase? Note that the PR above should have been released as part of official NUT v2.8.2 too.

jimklimov, 2024-12-03:
@Yellow-Horror : gentle bump, or I'll close the issue as presumed fixed :D

jimklimov, 2025-03-19:
Closing for now, feel free to re-open or reference in new issue if the problem is still around after NUT v2.8.2 release.

closed 2025-03-19 by jimklimov