networkupstools/nut #2319: UPS Model "Powerwalker Basic VI 800 STL" not recognized - Driver startup issue

tamenmd, 2024-02-20:
Hello everyone,

I hope this is the right place to seek assistance. I am currently trying to install NUT on my server to work with my UPS, the "Powerwalker Basic VI 800 STL." Unfortunately, it seems that my UPS is not supported out of the box. I have attempted to test various drivers during the configuration, as suggested in the documentation, but without success.

When attempting to start the UPS driver with the command `upsdrvctl start`, I encounter the following error message:

```
Network UPS Tools - UPS driver controller 2.8.0
Network UPS Tools - Generic HID driver 0.47 (2.8.0)
USB communication driver (libusb 1.0) 0.43
libusb1: Could not open any HID devices: insufficient permissions on everything
No matching HID UPS found
Driver failed to start (exit status=1)
```
I attempted to explore alternative drivers, but the problem persists.

I would greatly appreciate any guidance or assistance in resolving this issue. Could it be that my UPS model is not supported, or is there a specific driver that I should try?

System Information:
UPS Model: Powerwalker Basic VI 800 STL
NUT Version: 2.8.0
OS: Proxmox 8.1.4

Thank you in advance for your help!

jimklimov, 2024-02-20:
Did you run `upsdrvctl` as `root` or `nut`? A simple other user may have no rights to the USB devfs node.

Is NUT installed as a package? If yes - the nut-driver-enumerator (see wiki) may have already created and started a service unit based on the ups.conf section (would conflict with manual experiments).

jimklimov, 2024-03-14:
Updated the https://github.com/networkupstools/nut/wiki/nut%E2%80%90driver%E2%80%90enumerator-(NDE) article implied earlier with regard to specifically conflicts vs. `upsdrvctl`