networkupstools/nut #2320: UPS Load and Battery Charge always zero using nutdrv_qx

LucasTor, 2024-02-27:
I just recently got my UPS back from maintenance (it's a TS Shara 1500VA pure sinewave rackmount model), and now it does not report the battery percentage nor the load anymore (it used to do it normally), in the meantime, while it was under maintenance, I updated all the docker containers I have running to the latest version with docker compose pull (I am using instantlinux/nut-upsd), do you guys have any clue if this could be an issue with an updated driver not being able to read or a firmware update from when the UPS went for the maintenance in the factory?
Any help is appreciated!

Here is an upsc log

```
battery.charge: 0
battery.voltage: 13.6
battery.voltage.high: 26.00
battery.voltage.low: 20.80
battery.voltage.nominal: 24.0
device.mfr: TS SHARA 230727
device.model: Sen 2U  15
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.name: nutdrv_qx
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: /dev/serial/by-id/usb-STMicroelectronics_STM32_Virtual_ComPort_00000000001A-if00
driver.parameter.synchronous: auto
driver.state: quiet
driver.version: 2.8.1
driver.version.data: Megatec 0.07
driver.version.internal: 0.36
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
input.current.nominal: 100.0
input.frequency: 60.0
input.frequency.nominal: 60
input.voltage: 218.0
input.voltage.fault: 218.0
output.voltage: 112.0
ups.beeper.status: enabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.firmware: V01030503A
ups.load: 0
ups.status: OL
ups.temperature: 27.0
ups.type: offline / line interactive
```

LucasTor, 2024-02-27:
Thanks @jimklimov, that was it, just changed the docker image to [2.8.0-r4](https://hub.docker.com/layers/instantlinux/nut-upsd/2.8.0-r4/images/sha256-84ccea9e4b6f6000ee772395b2fbb9a657dfd84c17361bad4ab073b48d025d50?context=explore), and now I have data for the load and battery percentage back

I am using USB for it, just as before

Here is another upsc log, in case that helps with debugging

```
battery.charge: 100
battery.voltage: 13.40
battery.voltage.high: 26.00
battery.voltage.low: 20.80
battery.voltage.nominal: 24.0
device.mfr: TS SHARA 230727
device.model: Sen 2U  15
device.type: ups
driver.name: nutdrv_qx
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: /dev/serial/by-id/usb-STMicroelectronics_STM32_Virtual_ComPort_00000000001A-if00
driver.parameter.synchronous: auto
driver.version: 2.8.0
driver.version.data: Megatec 0.06
driver.version.internal: 0.32
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
input.current.nominal: 100.0
input.frequency: 60.0
input.frequency.nominal: 60
input.voltage: 223.0
input.voltage.fault: 223.0
output.voltage: 115.0
ups.beeper.status: enabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.firmware: V01030503A
ups.load: 13
ups.status: OL
ups.temperature: 27.0
ups.type: offline / line interactive
```

jimklimov, 2024-02-27:
Well, the UPS/firmware is not broken - that's good news :)

Figuring out why we lost the info in 2.8.1 is gonna be a head-scratcher...

Are you by chance in position to build NUT from source and test the current master code base (in case some of the changes after 2.8.1 fixed this bit)? https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests outlines how to build and test drivers right from the build workspace, without necessarily changing your OS installed variant of NUT.

Or at least to run that 2.8.1 container with a higher debug verbosity, check https://github.com/networkupstools/nut/wiki/Changing-NUT-daemon-debug-verbosity for suggestions in this regard...

LucasTor, 2024-02-28:
Hey! Sorry for the delayed response, had a busy day today, I will read through the instructions on how to build from source, and let you know if I have any luck getting the missing data from the UPS
By the way, can I build it on an M1 Mac? As I plan on doing on my work PC instead of my homelab server

jimklimov, 2024-02-28:
Macs should work too, as far as compilation goes at least - each NUT iteration of the past couple of years is tested on CircleCI Mac environments. There were some recently posted issues (search by label) regarding some XML manifests for further OS integration. Don't have one, so that's as much as I can elaborate.

Guiorgy, 2025-12-24:
On an Ubuntu 24.04 LTS running nut 2.8.1 I get `battery.charge: 100` and `ups.load: 0`, so at least one is working. I tried manually downloading the deb packages (nut, nut-client and nut-server) meant for the upcoming 26.04, which is 2.8.4 and installing it, but I quickly run into a dependency hell. I'll try manually compiling when I have time later.