networkupstools/nut #2338: apc_modbus: follow-up wanted regarding Main Outlet Group operations

jimklimov, 2024-03-02:
> Cheers, how is it going here? Do you plan to post some updates shortly, or is (from your viewpoint) the change OK to merge as is and follow up on specific problems and improvements later?

Sorry, it has been difficult for me to find time to work on this lately. I just was able to test the outlet commands and it seems that the outlet specific commands for outlet group 1, which is the only one I have, all work for me. What does not work are the outlet commands that specify the MOG (main outlet group). I'm not sure if that group is actually supposed to be switchable. I'll check the documentation some more.

_Originally posted by @EchterAgo in https://github.com/networkupstools/nut/issues/2184#issuecomment-1889691433_
            

NOTE: Possibly some more questions raised in that issue are to be followed up too.

piotrekszala, 2024-05-09:
Hi, I have APC SMT-750IC connected using driver apc_modbus (v2.8.2) over TCP.
Reading data from UPS also works as expected but commands to shutdown or switch load/output  do not work at all.
However commands to trigger: test.battery.start or test.panel.start work just fine.
This issue actually prevents nut from performing a UPS shutdown when FSD occurs.

EchterAgo, 2026-04-02:
> > @cardil wrote:
> >
> > @EchterAgo I deployed this on my SMT1500RMI2U (FW 15.1, serial). The quirk detection doesn't trigger, and `load.on` still fails.
> > 
> > Driver startup at debug level 3 shows no quirk-related messages at all: no "Detected word swap quirk", no "Failed to detect word swap quirk", nothing. Here's the version string reported:
> > 
> > ```
> > nut-driver@apc[176583]: Network UPS Tools 2.8.4.1689.2-1691+g95dbdb614 (development iteration after 2.8.4) - NUT APC Modbus driver without USB support (libmodbus link type: dynamic) 0.19
> > ```
> > 
> > I re-ran my [pymodbus probe](https://gist.github.com/cardil/ab1e8e243210e36e8066239cf7097211) to double-check; same results as before: BE `[0x0000, 0x0007]` gives exception 4, LE `[0x0007, 0x0000]` succeeds.
> > 
> > The PR probe sends LE order `[0x0007, 0x0000]`, which **succeeds** on my unit (firmware accepts it silently), so the `if (... < 0)` block is never entered and `needs_word_swap_quirk` stays 0.
> > 
> > I think the detection logic is inverted. On your unit, the probe fails with EMBXILVAL because the firmware validates reg 1538 properly, meaning BE order already works for you. On mine, the probe succeeds because the firmware doesn't validate the combo, but that's exactly the unit that needs the swap.
> > 
> > The fix should be: set `needs_word_swap_quirk = 1` when the probe succeeds, not when it fails.
> > 
> > Also the `EMBXSFAIL` comment mentioning SMT1500RMI2U comes from my [BE-order test](https://github.com/networkupstools/nut/issues/2338#issuecomment-2761193946), not from the LE probe. The LE probe returns SUCCESS on my unit.

> @EchterAgo wrote:
> 
> ```
>  resrvd  flags
> [0x0007, 0x0000]
> ```
> 
> is what is supposed to succeed because according to spec the flags are all in the reserved area and there is no command, so this succeeds on my unit.
> 
> `_apc_modbus_from_uint64` produces `[0x0000, 0x0007]` from the same flags, which fails with `EMBXILVAL` on my unit because the combination of flags is invalid.
> 
> It seems that your unit behaves the same way mine does. Are you sure swapping the words actually makes the commands work on your unit or does it just accept them because all the bits are then in the reserved area?
> 
> For example `shutdown.return` will fail with `EMBXILVAL` on my unit because ~it does not have a main outlet group~ the switched outlet groups have to be turned off first. I'm using `outlet.3.load.off` and `outlet.3.load.on` to test because I don't have anything connected to that outlet group. I'm actually not sure what to do there anyway, my device has 3 switched outlet groups, what to do in that case? I don't think we can even detect what outlet groups there are, and if I shut them down sequentially, what happens if the first outlet group is the one that powers the machine running the apc_modbus driver?

> > @cardil wrote:
> >
> > @EchterAgo Fair question about whether LE word order actually executes commands or just gets silently accepted. I haven't proven that yet - my pymodbus test sent LE `load.on` while the load was already on, so the UPS accepted it without error but I can't confirm the command actually did anything. I need to move my production workloads off this UPS first so I can safely toggle outlets. Once that's done I'll connect a laptop via serial, test both word orders with actual state changes (outlet OFF -> send `load.on` -> verify it turns ON), and report back here.
> > 
> > On your question about what to switch off and what happens if the driver's outlet loses power - that's the normal FSD scenario. By the time `upsdrvctl shutdown` runs, the OS has already halted everything. The driver sends `shutdown.return` to tell the UPS "turn off all outlets now (using off-delay), turn back on when mains returns." The machine running the driver losing power is the desired behavior - it preserves remaining charge on the UPS, and when mains comes back the UPS cleanly starts up all servers. The off-delay gives the OS time to finish halting before the UPS cuts power.
> > 
> > Looking at the code, the apc_modbus command map already has [`shutdown.return` registered](https://github.com/networkupstools/nut/blob/95dbdb6141/drivers/apc_modbus.c#L1373) with `CMD_OUTPUT_SHUTDOWN | TARGET_MAIN_OUTLET_GROUP | MOD_USE_OFF_DELAY`, and it goes through `_apc_modbus_instcmd()` which does the proper 2-register FC 0x10 write plus your word swap. Both [`usbhid-ups`](https://github.com/networkupstools/nut/blob/95dbdb6141/drivers/usbhid-ups.c#L1263) and [`snmp-ups`](https://github.com/networkupstools/nut/blob/95dbdb6141/drivers/snmp-ups.c#L531) use `shutdown.return` as the first command in their default sdcommands list: `"shutdown.return,shutdown.reboot,load.off.delay,shutdown.stayoff"` - the framework tries them in order until one succeeds. The apc_modbus driver should probably do the same instead of its current [`upsdrv_shutdown()`](https://github.com/networkupstools/nut/blob/95dbdb6141/drivers/apc_modbus.c#L1611) which does its own FC 0x06 single-register write bypassing both the command map and the word swap logic.
> > 
> > For your unit with only SOGs and no MOG, the spec section 3.3 you quoted says that issuing the Shutdown command in `SimpleSignalCommand_BF` will "issue an on command to every SOG (and MOG) to turn off using its specified OffDelay program" and "when AC is present the outlets that were on prior to issuing this command will be turned on automatically." That's exactly what `shutdown.return`  [... cut ...]
> I think my unit acutally does have an MOG but you need to have switched off all SOGs before operating it. I have a good idea now how to correctly map the commands, but I need to figure out whether we can just specify all SOGs in `shutdown.*` and `load.*` or whether we need a probe mechanism and only specify the SOGs that actually exist on the unit. I just got out my old RS232 / USB SMT1500I that has only 1 SOG, hopefully I can do some testing with it today. I think this specific discussion should continue in #2338 because it has nothing to do with the word swap quirk.

I think this discussion belongs in here rather than #3381

jimklimov, 2026-04-05:
Does PR #3395 fully address this issue (so can they be closed in tandem when the PR is merged), or is there some more work needed/planned for the issue's goals?