networkupstools/nut #2347: APC Back-UPS BX1600MI spurious LOWBATT/REPLACEBATT events

grifferz, 2024-03-02:
I have an APC Back-UPS BX1600MI connected by its supplied USB cable. With the version of nut in Debian 10 (2.7.4-8) this was unusable as the usbhid-ups driver kept disconnecting every few seconds.

I upgraded to the git HEAD of nut and communication is now stable, but spurious events come in every so often (once or twice an hour at the moment). When I call a notify script that calls `upsc` at the time of those spurious events, I can see that the `ups.status` does bear that out, but other values do not. Example:

```
battery.charge: 100
battery.charge.low: 10
battery.mfr.date: 2001/01/01
battery.runtime: 659
battery.runtime.low: 120
battery.type: PbAc
battery.voltage: 27.3
battery.voltage.nominal: 24.0
device.mfr: American Power Conversion
device.model: Back-UPS BX1600MI
device.serial: 9B2339A15993
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.name: usbhid-ups
driver.parameter.poll freq: 30
driver.parameter.poll interval: 2
driver.parameter.port : auto
driver.parameter.synchronous: auto
driver.state: quiet
driver.version: 2.8.1-884-gf2fc47067
driver.version.data: APC HID 0.100
driver.version.internal: 0.52
driver.version.usb: libusb-1.0.22 (API: 0x1000106)
input.sensitivity: medium
input.transfer.high: 295
input.transfer.low: 145
input.voltage: 234.0
input.voltage.nominal: 230
ups.beeper.status: disabled
ups.delay.shutdown: 20
ups.load: 30
ups.mfr: American Power Conversion
ups.mfr.date: 2023/10/05
ups.model: Back-UPS BX1600MI
ups.productid: 0002
ups.realpower.nominal: 900
ups.serial: 9B2339A15993
ups.status: OL LB RB
ups.test.result: Done and passed
ups.timer.reboot: 0
ups.timer.shutdown: -1
ups.vendorid: 051d
```

As you can see, the status does include `LB` and `RB`, but the charge is still 100 and the runtime is as expected. The bad status lasts less than 2 seconds before returning to simply `OL`.

I went to the effort of installing a Windows VM, passing USB through to it and trying APC's own Powerchute Serial Shutdown software in there. This does not report any spurious events. Both Powerchute and nut do report true events that I induce. A self-test of the device passed. I am unable to demonstrate any behaviour that APC consider incorrect.

I returned the UPS to the vendor as faulty and they sent a replacement. The replacement behaves the same.

I installed `apcupsd` just to see how it behaved. It maintained a connection but its rate of spurious events was even worse: every couple of minutes. Another `apcupsd` user reports the same symptoms as me:

https://sourceforge.net/p/apcupsd/mailman/message/58740970/

Interestingly, they had a BX1600MI working with `apcupsd`, replaced it with another BX1600MI and now they see what I see, implying that newer models of Back-UPS have something different about them even though they are the same model number.

Is there any way to work around this sort of thing? It's almost like I need a way to not believe such statuses unless they persist for at least 5 seconds, or something.

ShiroDN, 2024-03-04:
I have a new APC Back-UPS BX1200MI (basically the same model, just with a smaller battery) connected to TrueNAS that is using NUT 2.8.0, and there is some weird behavior. Sometimes, 1-2 times an hour, it triggers LOWBATT/REPLACEBATT events. In a debug log, I can see "[D2] parse_status: [OL DISCHRG CHRG LB RB]".

When it works and the battery is charged to 100%, it seems to be switching between "OL" and "OL CHRG" every few seconds. I am not sure if that is a problem too, or just normal behavior, though.

Here is a debug log from the driver at the time of an event:

```log
22097.933876    [D1] upsdrv_updateinfo...
22098.033355    [D2] file_report_buffer: expected 5 bytes, but got 512 instead
22098.033486    [D3] Report[err]: (512 bytes) => 16 63 82 42 fd 7f 00 00 01 00 00 00 00 00 00 00
22098.033549    [D3]  00 00 00 00 00 00 00 00 18 69 e6 c0 77 55 00 00 d5 ad e7 c0 77 55 00 00 80
22098.033611    [D3]  b0 82 42 fd 7f 00 00 16 00 00 00 00 00 00 00 35 61 e6 c0 77 55 00 00 00 00
22098.033671    [D3]  00 00 00 00 00 00 00 00 00 00 00 00 00 00 60 b2 82 42 fd 7f 00 00 90 b1 82
22098.033731    [D3]  42 fd 7f 00 00 2d 82 e5 65 00 00 00 00 b2 6c 00 00 00 00 00 00 02 00 00 00
22098.033792    [D3]  00 00 00 00 00 ba 75 9f c5 fe 19 5f ff ff ff ff 77 55 00 00 e0 b0 82 42 fd
22098.033856    [D3]  7f 00 00 00 00 00 00 00 00 00 00 e0 d2 1c 0f 3c 7f 00 00 00 00 00 00 00 00
22098.033917    [D3]  00 00 73 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ba 75 9f c5 fe 19
22098.033978    [D3]  5f 01 00 00 00 00 00 00 00 00 ba 75 9f c5 fe 19 5f 7b e7 e6 c0 77 55 00 00
22098.034038    [D3]  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
22098.034099    [D3]  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
22098.034159    [D3]  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
22098.034219    [D3]  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
22098.034280    [D3]  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
22098.034340    [D3]  00 00 00 00 00 00 00 00 00 00 00 50 4f 4e 47 0a 00 00 00 00 ba 75 9f c5 fe
22098.034399    [D3]  19 5f 6d 6f 64 65 6c 73 20 28 d1 e4 e6 c0 77 55 00 00 da de 5d 00 00 00 00
22098.034459    [D3]  00 17 00 00 00 00 00 00 00 e0 af 82 42 fd 7f 00 00 00 00 00 00 00 00 00 00
22098.034519    [D3]  01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 48 4a e8 c0 77 55 00 00 01
22098.034578    [D3]  00 00 00 00 00 00 00 1d 05 00 00 00 00 00 00 99 b0 e5 c0 77 55 00 00 50 4f
22098.034638    [D3]  4e 47 0a 00 00 00 97 94 e6 c0 77 55 00 00 6d 6f 64 65 00 00 00 00 70 b2 82
22098.034696    [D3]  42 fd 7f 00 00 98 e4 e6 c0 77 55 00 00 01 00 00 00 00 00 00 00
22098.034751    [D1] Got 9 HID objects...
22098.034807    [D3] Report[buf]: (5 bytes) => 16 63 82 42 fd
22098.034862    [D5] PhyMax = 0, PhyMin = 0, LogMax = 1, LogMin = 0
22098.034915    [D5] Unit = 00000000, UnitExp = 0
22098.034968    [D5] Exponent = 0
22098.035022    [D5] hid_lookup_path: 00840004 -> UPS
22098.035076    [D5] hid_lookup_path: 00840024 -> PowerSummary
22098.035130    [D5] hid_lookup_path: 00840002 -> PresentStatus
22098.035185    [D5] hid_lookup_path: 00850044 -> Charging
22098.035243    [D2] Path: UPS.PowerSummary.PresentStatus.Charging, Type: Input, ReportID: 0x16, Offset: 0, Size: 1, Value: 1
22098.035299    [D5] hu_find_infoval: found chrg (value: 1)
22098.035352    [D5] process_boolean_info: chrg
22098.035408    [D3] Report[buf]: (5 bytes) => 16 63 82 42 fd
22098.035462    [D5] PhyMax = 0, PhyMin = 0, LogMax = 1, LogMin = 0
22098.035515    [D5] Unit = 00000000, UnitExp = 0
22098.035568    [D5] Exponent = 0
22098.035621    [D5] hid_lookup_path: 00840004 -> UPS
22098.035674    [D5] hid_lookup_path: 00840024 -> PowerSummary
22098.035727    [D5] hid_lookup_path: 00840002 -> PresentStatus
22098.035781    [D5] hid_lookup_path: 00850045 -> Discharging
22098.035836    [D2] Path: UPS.PowerSummary.PresentStatus.Discharging, Type: Input, ReportID: 0x16, Offset: 1, Size: 1, Value: 1
22098.035891    [D5] hu_find_infoval: found dischrg (value: 1)
22098.035946    [D5] process_boolean_info: dischrg
22098.036001    [D3] Report[buf]: (5 bytes) => 16 63 82 42 fd
22098.036054    [D5] PhyMax = 0, PhyMin = 0, LogMax = 1, LogMin = 0
22098.036107    [D5] Unit = 00000000, UnitExp = 0
22098.036159    [D5] Exponent = 0
22098.036212    [D5] hid_lookup_path: 00840004 -> UPS
22098.036265    [D5] hid_lookup_path: 00840024 -> PowerSummary
22098.036318    [D5] hid_lookup_path: 00840002 -> PresentStatus
22098.036372    [D5] hid_lookup_path: 008500d0 -> ACPresent
22098.036427    [D2] Path: UPS.PowerSummary.PresentStatus.ACPresent, Type: Input, ReportID: 0x16, Offset: 2, Size: 1, Value: 0
22098.036482    [D5] hu_find_infoval: found !online (value: 0)
22098.036534    [D5] process_boolean_info: !online
22098.036589    [D3] Report[buf]: (5 bytes) => 16 63 82 42 fd
22098.036642    [D5] PhyMax = 0, PhyMin = 0, LogMax = 1, LogMin = 0
22098.036695    [D5] Unit = 00000000, UnitExp = 0
22098.036747    [D5] Exponent = 0
22098.036800    [D5] hid_lookup_path: 00840004 -> UPS
22098.036853    [D5] hid_lookup_path: 00840024 -> PowerSummary
[... 88 lines cut ...]
22098.040892    [D5] Unit = 00000000, UnitExp = 0
22098.040904    [D5] Exponent = 0
22098.040916    [D2] Path: UPS.PowerSummary.PresentStatus.Charging, Type: Feature, ReportID: 0x16, Offset: 0, Size: 1, Value: 1
22098.040927    [D5] hu_find_infoval: found chrg (value: 1)
[... cut ...]
22098.042616    [D5] hu_find_infoval: found !lowbatt (value: 0)
22098.042628    [D5] process_boolean_info: !lowbatt
22098.042641    ups_status_set: seems that UPS [apc] is in OL+DISCHRG state now. Is it calibrating or do you perhaps want to set 'onlinedischarge' option? Some UPS models (e.g. CyberPower UT series) emit OL+DISCHRG when offline.
22098.042656    [D5] send_to_all: SETINFO ups.status "OL DISCHRG CHRG LB RB"
22098.042679    [D6] send_to_all: write 43 bytes to socket 11 succeeded (ret=43): SETINFO ups.status "OL DISCHRG CHRG LB RB"
```

jimklimov, 2024-03-14:
CC @desertwitch : I think some of your earlier investigations were about similar behavior; any ideas here?

Notably, in NUT master (after release 2.8.1) we had PR #2216 to address such notifications with new configuration options. From research and educated guesswork there, this may be just part of battery charge management which happens under the hood on all UPS devices that became visibly exposed on some firmwares.

desertwitch, 2024-03-15:
Hi Jim, seems curious - I'll take a look, thanks for the CC - will report back.

desertwitch, 2024-03-17:
It seems to be driver related, something that the UPS is sending byte-wise that's being parsed wrongly into `RB` (or `LB`)

```
22098.038416    [D3] Report[buf]: (5 bytes) => 16 63 82 42 fd
22098.038469    [D5] PhyMax = 0, PhyMin = 0, LogMax = 1, LogMin = 0
22098.038522    [D5] Unit = 00000000, UnitExp = 0
22098.038573    [D5] Exponent = 0
22098.038625    [D5] hid_lookup_path: 00840004 -> UPS
22098.038679    [D5] hid_lookup_path: 00840024 -> PowerSummary
22098.038732    [D5] hid_lookup_path: 00840002 -> PresentStatus
22098.038785    [D5] hid_lookup_path: 0085004b -> NeedReplacement
22098.038844    [D2] Path: UPS.PowerSummary.PresentStatus.NeedReplacement, Type: Input, ReportID: 0x16, Offset: 6, Size: 1, Value: 1
22098.038855    [D5] hu_find_infoval: found replacebatt (value: 1)
22098.038867    [D5] process_boolean_info: replacebatt
```
 I couldn't find an issue in the upsmon handling or the later processes. I've since had another report from a user with an APC BX1200MI-GR also experiencing intermediate `LB` events, so it seems to be that series. I'm in touch with the other user if it works on an older version of NUT so hopefully we can narrow it down by comparison.

desertwitch, 2024-03-18:
Got another report of an affected Back-UPS BX750MI with spurious RB events.
I'm hoping to get my hands on at least a BX750MI this week so I can mess around with the driver a bit.

PilaScat, 2024-03-18:
Same thing for me, I've the same problem with BX1200MI-GR

desertwitch, 2024-03-26:
Unfortunately my contact didn't come through with the UPS so we're back to square one.

bitmario, 2024-03-26:
I am seeing the same behavior on a newly purchased BX1600MI-GR, manufactured October 2023, on Debian 12's nut 2.8.0-7.

@desertwitch let me know if I can help by collecting logs or trying any patches

marcgarciamarti, 2024-04-02:
Hello there! 
I'm using the containerized version of nut available in home assistant through the official add-on https://github.com/hassio-addons/addon-nut/tree/main

I can confirm this behavior is also happening with that version.

[image]

The pattern seems to be random. It sometimes lasts a few seconds or a few minutes.

desertwitch, 2024-04-03:
> I am seeing the same behavior on a newly purchased BX1600MI-GR, manufactured October 2023, on Debian 12's nut 2.8.0-7.
> 
> @desertwitch let me know if I can help by collecting logs or trying any patches

Any chance you could try if the problem persists putting this flag in your UPS configuration in `UPS.CONF`:
```
maxreport
```
So an example configuration would look like this:
```
[ups]
        driver = usbhid-ups
        port = auto
        maxreport
```

bitmario, 2024-04-03:
@desertwitch I just tried `maxreport` but it doesn't seem to help - I got two instances of "battery is low"+"battery needs to be replaced" within 3 hours.

desertwitch, 2024-04-03:
> @desertwitch I just tried `maxreport` but it doesn't seem to help - I got two instances of "battery is low"+"battery needs to be replaced" within 3 hours.

Thanks for the attempt, it was a long shot. Is there any logic to the statuses, as in do you notice a rapid succession or cycling through of certain statuses (LB, RB, LB, OL or OL, LB, RB, OL) in the logs? **In general, do you have any logs you can provide us from NUT itself with timestamps so we can try to investigate this some more?** Particularly interesting would be if the LB (low battery state) is happening at the same time the UPS is also OL or if the UPS goes OB before/after.

bitmario, 2024-04-03:
The general pattern is that it will usually be in a normal status like `OL CHRG` and then every hour or two I'll see this (on Debian 12, nut 2.8.0-7):

```
Apr 03 14:04:43 nut-ups-test usbhid-ups[556]: ups_status_set: seems that UPS [apc1] is in OL+DISCHRG state now. Is it calibrating or do you perhaps want to set 'onlinedischarge' option? Some UPS models (e.g. CyberPower UT series) emit OL+DISCHRG when offline.
```

```
Apr 03 14:04:44 nut-ups-test nut-monitor[560]: UPS apc1@localhost battery is low
Apr 03 14:04:44 nut-ups-test nut-monitor[560]: UPS apc1@localhost battery needs to be replaced
```

Afterwards it goes back to a normal status.

I've been running a patched `apcupsd` so don't have any long-running logs at the moment, but I'm happy to run NUT as long as needed to help figure this out. Should I enable any debug logging options? Do you want me to try a fresh `master` build or should I keep using 2.8.0?

KillianMelsen, 2024-04-03:
> > @desertwitch I just tried `maxreport` but it doesn't seem to help - I got two instances of "battery is low"+"battery needs to be replaced" within 3 hours.
> 
> Thanks for the attempt, it was a long shot. Is there any logic to the statuses, as in do you notice a rapid succession or cycling through of certain statuses (LB, RB, LB, OL or OL, LB, RB, OL) in the logs? **In general, do you have any logs you can provide us from NUT itself with timestamps so we can try to investigate this some more?** Particularly interesting would be if the LB (low battery state) is happening at the same time the UPS is also OL or if the UPS goes OB before/after.
> 
> > The only thing that comes to mind is to add a yet another throttle for such reports - so to not expose the status if RB appears and dissipates quickly (I'd be wary of tweaking LB like that... maybe optionally-throttle tied to known OL/OB/BYPASS status?)
> 
> I'm curious though if this is something else entirely, which the driver misinterprets, or the UPS is actually sending these bogus statuses. I'm thinking it probably doesn't happen on their own APC software (if there is one for that series) or there would be more complaints on the APC forums, but considering that both APCUPSD and NUT are affected in a similar fashion it does seem to be some change in the firmware when compared to other functioning APC devices or even older models of the BX series which seem to work (firmware bug in newer models?). I agree with a LB throttle on its own being less than ideal, unless tied to another status or succession of states (if we can figure out any logic behind what's happening here).

I just got home and installed NUT through the unraid apps. I'd be happy to provide logs but not sure how.

desertwitch, 2024-04-03:
> The general pattern is that it will usually be in a normal status like `OL CHRG` and then every hour or two I'll see this (on Debian 12, nut 2.8.0-7):
>
> ```
> Apr 03 14:04:43 nut-ups-test [...]

This is very valuable, so we can also see here that the UPS is doing some kind of - perhaps - calibration before these LB and RB states. Maybe it is cycling through those states as part of some calibration mechanism. Can you confirm you're never having a "is on battery" status and you're also not getting any shutdowns from these intermittent low battery occurrences?

The other user's log above has shown a status of "OL DISCHRG CHRG LB RB" which seems to be also what's happening on your side (CHRG does not get reported in the logs as an event). @jimklimov perhaps if users use the `onlinedischarge_calibration` to turn `OL DISCHRG` into `CAL` an additional toggle to basically ignore every other status while `CAL` is present could be a solution here (effectively suppressing `CHRG LB RB` in such cases and only seeing `CAL`)? We've already seen other UPS cycling through statuses while doing calibration, perhaps such a toggle could also be beneficial to other users?

grifferz, 2024-04-03:
> I upgraded to the git HEAD of nut and communication is now stable, but spurious events come in every so often (once or twice an hour at the moment). When I call a notify script that calls `upsc` at [...]

I don't know whether it is of any use, but I did not change anything (kept running git HEAD of nut) and the above strange behaviour stopped after a few days. It has been behaving normally for weeks now.

Thanks,
Andy

desertwitch, 2024-04-03:
> I just got home and installed NUT through the unraid apps. I'd be happy to provide logs but not sure how.

Just set NUT up normally through the GUI as you want to have it and watch the SYSLOG (Tools->System Log) for any such strange events being reported while NUT is started, they should pop up in the log by themselves if they occur on your system.

> I don't know whether it is of any use, but I did not change anything (kept running git HEAD of nut) and the above strange behaviour stopped after a few days. It has been behaving normally for weeks [...]

That is extremely curious, makes me wonder even more what the UPS is doing there - thanks though!

bitmario, 2024-04-03:
> Maybe it is cycling through those states as part of some calibration mechanism. Can you confirm you're never having a "is on battery" status and you're also not getting any shutdowns from these intermittent low battery occurrences?

Correct - I only get "is low" and "needs to be replaced".

As a sanity check, I just tried pulling the plug for a few seconds and this is what I see in that case:

```
Apr 03 15:56:06 nut-ups-test nut-monitor[579]: UPS apc1@localhost on battery
Apr 03 15:56:06 nut-ups-test nut-monitor[643]: Network UPS Tools upsmon 2.8.0
Apr 03 15:56:26 nut-ups-test nut-monitor[579]: UPS apc1@localhost on line power
Apr 03 15:56:26 nut-ups-test nut-monitor[646]: Network UPS Tools upsmon 2.8.0
```

```
Apr 03 15:56:23 nut-ups-test usbhid-ups[574]: ups_status_set: seems that UPS [apc1] is in OL+DISCHRG state now. Is it calibrating or do you perhaps want to set 'onlinedischarge' option? Some UPS models (e.g. CyberPower UT series) emit OL+DISCHRG when offline.
```

Thanks for taking the time to look into this :)

KillianMelsen, 2024-04-03:
> Just set NUT up normally through the GUI as you want to have it and watch the SYSLOG (Tools->System Log) for any such strange events being reported while NUT is started, they should pop up in the log by themselves if they occur on your system.

Ah got it. They system log does show some low battery indications as seen below. I also pulled the plug and that all seems to be working fine. UPS switches to battery, I get an unraid notification, and it goes back online when plugged back in.

```
Apr  3 17:06:09 Tower upsmon[13245]: Startup successful
Apr  3 17:06:09 Tower upsmon[13245]: Warning: running as one big root process by request (upsmon -p)
Apr  3 17:06:09 Tower upsd[13239]: User monuser@127.0.0.1 logged into UPS [ups]
Apr  3 17:07:14 Tower ool www[12635]: /usr/local/emhttp/plugins/nut-dw/scripts/write_config
Apr  3 17:07:14 Tower rc.nut: Writing NUT configuration...
Apr  3 17:07:15 Tower rc.nut: Updating permissions for NUT...
Apr  3 17:07:15 Tower rc.nut: Checking if the NUT Runtime Statistics Module should be enabled...
Apr  3 17:07:15 Tower rc.nut: Disabling the NUT Runtime Statistics Module...
...
Apr  3 17:14:09 Tower upsmon[13245]: UPS ups@127.0.0.1 battery is low
...
Apr  3 17:48:09 Tower upsmon[13245]: UPS ups@127.0.0.1 battery is low
...
Apr  3 18:09:34 Tower upsmon[13245]: UPS ups@127.0.0.1 on battery
Apr  3 18:09:34 Tower nut-notify: [ups] UPS is on battery. The system will shutdown when the UPS battery charge reaches 25%.
Apr  3 18:13:09 Tower upsmon[13245]: UPS ups@127.0.0.1 on line power
Apr  3 18:13:09 Tower nut-notify: [ups] UPS is online, any planned shutdowns were cancelled.
```

desertwitch, 2024-04-03:
Thanks for the logs everyone - just to double-check back here, **has anyone gotten a shutdown from this problem?**
Just to make sure that this is for the most part a log and notification problem and not an unwarranted shutdown one.

KillianMelsen, 2024-04-03:
> Thanks for the logs everyone - just to double-check back here, **has anyone gotten a shutdown from this problem?** Just to make sure that this is for the most part a log and notification problem and not an unwarranted shutdown one.

My unraid was running fine until I hooked it up to the UPS yesterday evening. This morning it was off, and it started a parity check after turning it on even though nothing was scheduled.

I can't find anything that logged the reason for the shutdown though. That was also using the built-in UPS utility, so I'll keep running NUT and see if I get any other unexpected shutdowns.

KillianMelsen, 2024-04-03:
So I just checked the system log again and found a whole bunch of errors. Don't know if it's related. I left all configuration options on the default values except for the setting the shutdown rules to 25% battery left. I'm also running powertop --auto-tune in case that's relevant.

```
Apr  3 18:27:50 Tower usbhid-ups[13220]: nut_libusb_get_report: Input/Output Error
Apr  3 18:27:52 Tower usbhid-ups[13220]: #012Reconnecting. If you saw "nut_libusb_get_interrupt: Input/Output Error" or similar message in the log above, try setting "pollonly" flag in "ups.conf" options section for this driver!
Apr  3 18:41:45 Tower usbhid-ups[13220]: nut_libusb_get_report: Input/Output Error
Apr  3 18:41:47 Tower usbhid-ups[13220]: #012Reconnecting. If you saw "nut_libusb_get_interrupt: Input/Output Error" or similar message in the log above, try setting "pollonly" flag in "ups.conf" options section for this driver!
Apr  3 19:06:52 Tower usbhid-ups[13220]: nut_libusb_get_report: Input/Output Error
Apr  3 19:06:54 Tower usbhid-ups[13220]: #012Reconnecting. If you saw "nut_libusb_get_interrupt: Input/Output Error" or similar message in the log above, try setting "pollonly" flag in "ups.conf" options section for this driver!
Apr  3 19:29:52 Tower usbhid-ups[13220]: nut_libusb_get_report: Input/Output Error
Apr  3 19:29:54 Tower usbhid-ups[13220]: #012Reconnecting. If you saw "nut_libusb_get_interrupt: Input/Output Error" or similar message in the log above, try setting "pollonly" flag in "ups.conf" options section for this driver!
Apr  3 19:37:19 Tower upsmon[13245]: UPS ups@127.0.0.1 battery is low
Apr  3 19:38:58 Tower usbhid-ups[13220]: nut_libusb_get_report: Input/Output Error
Apr  3 19:39:00 Tower usbhid-ups[13220]: #012Reconnecting. If you saw "nut_libusb_get_interrupt: Input/Output Error" or similar message in the log above, try setting "pollonly" flag in "ups.conf" options section for this driver!
```

desertwitch, 2024-04-03:
> So I just checked the system log again and found a whole bunch of errors. Don't know if it's related. I left all configuration options on the default values except for the setting the shutdown rules [...]

These messages are most certainly unrelated to the issue and caused by `powertop --auto-tune` which is known to modify the USB power management settings and cause these input/output errors due to hibernating of USB-connected UPS devices at times when they should communicate with NUT. On Unraid in the NUT Settings there's a "USB Power Management Override" which you can test activating, it should stop these messages from appearing and prevent the UPS device from wrongful hibernations even in conjunction with `powertop --auto-tune`. This is also a very possible reason Unraid's in-built UPS Utility (which is `apcupsd`) caused the unwanted shutdown for you because it might not handle a non-communicating UPS well, particularly when in combination with simultaneous LOWBATT (low battery) states. Bottom line, best get rid of `powertop --auto-tune` if you can and set up your power saving settings in a way that do not interfere with the USB devices (or your UPS device at least). 

Please continue to watch out for unwanted shutdowns now that you're using NUT and report back here if you should experience one which you suspect to be caused by this issue, for (unrelated) other issues surrounding the operation on Unraid you can also chime in at the respective support topic on the Unraid forums (so that we don't hijack this issue too much 😉). 

Thanks for everyone's time and efforts helping investigate this issue so far. 💯

bitmario, 2024-04-04:
Following my previous comments (https://github.com/networkupstools/nut/issues/2347#issuecomment-2034878731 and https://github.com/networkupstools/nut/issues/2347#issuecomment-2035001225) I let this run for 24 hours and compiled my configs/logs below.

In short, I had 20 `OL+DISCHRG` warnings in the logs, of which:
- 1 immediately preceded simultaneous `battery is low` + `battery needs to be replaced`
- 9 immediately preceded `battery is low`

No power failures, shutdowns or anything noteworthy happened during this time.

<details>

<summary><b>Environment and config</b></summary>

```shell
# lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 12 (bookworm)
Release:	12
Codename:	bookworm

# uname -a
Linux nut-ups-test 6.1.0-18-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.76-1 (2024-02-01) x86_64 GNU/Linux

# grep -v "^#" /etc/nut/nut.conf | grep -v "^$"
MODE=standalone

# grep -v "^#" /etc/nut/ups.conf | grep -v "^$"
maxretry = 3
[apc1]
	driver = usbhid-ups
	port = auto
	maxreport

# grep -v "^#" /etc/nut/upsd.conf | grep -v "^$"
LISTEN 127.0.0.1 3493

# grep -v "^#" /etc/nut/upsd.users | grep -v "^$"
[admin]
	password = test1234
	actions = SET
	instcmds = ALL
[upsmon]
        password = test1234
        upsmon primary

# grep -v "^#" /etc/nut/upsmon.conf | grep -v "^$"
MONITOR apc1@localhost 1 upsmon test1234 primary
MINSUPPLIES 1
SHUTDOWNCMD "/sbin/shutdown -h +0"
POLLFREQ 5
POLLFREQALERT 5
HOSTSYNC 15
DEADTIME 15
POWERDOWNFLAG /etc/killpower
RBWARNTIME 43200
NOCOMMWARNTIME 300
FINALDELAY 5

# grep -v "^#" /etc/nut/upssched.conf | grep -v "^$"
CMDSCRIPT /bin/upssched-cmd

# upsc apc1@localhost
Init SSL without certificate database
battery.charge: 97
battery.charge.low: 10
battery.mfr.date: 2024/03/19
battery.runtime: 950
battery.runtime.low: 120
battery.type: PbAc
battery.voltage: 28.5
[... 84 lines cut ...]
Apr 04 11:45:56 nut-ups-test usbhid-ups[557]: ups_status_set: seems that UPS [apc1] is in OL+DISCHRG state now. Is it calibrating or do you perhaps want to set 'onlinedischarge' option? Some UPS models (e.g. CyberPower UT series) emit OL+DISCHRG when offline.
Apr 04 13:36:27 nut-ups-test usbhid-ups[557]: ups_status_set: seems that UPS [apc1] is in OL+DISCHRG state now. Is it calibrating or do you perhaps want to set 'onlinedischarge' option? Some UPS models (e.g. CyberPower UT series) emit OL+DISCHRG when offline.
Apr 04 15:27:06 nut-ups-test nut-monitor[560]: UPS apc1@localhost battery is low
Apr 04 15:27:06 nut-ups-test nut-monitor[1907]: Network UPS Tools upsmon 2.8.0
Apr 04 15:51:38 nut-ups-test usbhid-ups[557]: ups_status_set: seems that UPS [apc1] is in OL+DISCHRG state now. Is it calibrating or do you perhaps want to set 'onlinedischarge' option? Some UPS models (e.g. CyberPower UT series) emit OL+DISCHRG when offline.
```

## Running a manual self-test by holding the button on the UPS
```
Apr 04 16:10:57 nut-ups-test usbhid-ups[557]: ups_status_set: seems that UPS [apc1] is in OL+DISCHRG state now. Is it calibrating or do you perhaps want to set 'onlinedischarge' option? Some UPS models (e.g. CyberPower UT series) emit OL+DISCHRG when offline.
Apr 04 16:10:59 nut-ups-test usbhid-ups[557]: ups_status_set: seems that UPS [apc1] is in OL+DISCHRG state now. Is it calibrating or do you perhaps want to set 'onlinedischarge' option? Some UPS models (e.g. CyberPower UT series) emit OL+DISCHRG when offline.
Apr 04 16:11:01 nut-ups-test usbhid-ups[557]: ups_status_set: seems that UPS [apc1] is in OL+DISCHRG state now. Is it calibrating or do you perhaps want to set 'onlinedischarge' option? Some UPS models (e.g. CyberPower UT series) emit OL+DISCHRG when offline.
Apr 04 16:11:03 nut-ups-test usbhid-ups[557]: ups_status_set: seems that UPS [apc1] is in OL+DISCHRG state now. Is it calibrating or do you perhaps want to set 'onlinedischarge' option? Some UPS models (e.g. CyberPower UT series) emit OL+DISCHRG when offline.
```

## Pulling the plug to simulate a power failure
```
Apr 04 16:12:21 nut-ups-test nut-monitor[560]: UPS apc1@localhost on battery
Apr 04 16:12:21 nut-ups-test nut-monitor[1918]: Network UPS Tools upsmon 2.8.0
Apr 04 16:12:35 nut-ups-test usbhid-ups[557]: ups_status_set: seems that UPS [apc1] is in OL+DISCHRG state now. Is it calibrating or do you perhaps want to set 'onlinedischarge' option? Some UPS models (e.g. CyberPower UT series) emit OL+DISCHRG when offline.
Apr 04 16:12:36 nut-ups-test nut-monitor[560]: UPS apc1@localhost on line power
Apr 04 16:12:36 nut-ups-test nut-monitor[1921]: Network UPS Tools upsmon 2.8.0
```

</details>

mauro-dasilva, 2024-04-11:
I have a new APC BX950MI and I am seeing the same as @ShiroDN noted, mainly around the switching between "OL" and "OL CHRG" every few seconds. 

The strange thing is that I have the exact same model from about 2 years ago that doesn't output that switching behaviour, so it is probably something on the newer versions if the models. 

I'd be happy to provide any information that you may require to help diagnose the issue, just let me know.

tsopokis, 2024-04-21:
I have the same issue with BX2200MI, essentially the same ups with more capacity:

`Apr 21 10:52:41 localhost nut-driver@BX2200MI[1386987]:    0.287346        [D3] Report[buf]: (2 bytes) => 18 01
Apr 21 10:52:41 localhost nut-driver@BX2200MI[1386987]:    0.287365        [D5] PhyMax = 0, PhyMin = 0, LogMax = 3, LogMin = 1
Apr 21 10:52:41 localhost nut-driver@BX2200MI[1386987]:    0.287382        [D5] Unit = 00000000, UnitExp = 0
Apr 21 10:52:41 localhost nut-driver@BX2200MI[1386987]:    0.287395        [D5] Exponent = 0
Apr 21 10:52:41 localhost nut-driver@BX2200MI[1386987]:    0.287410        [D2] Path: UPS.PowerSummary.AudibleAlarmControl, Type: Feature, ReportID: 0x18, Offset: 0, Size: 8, Value: 1
Apr 21 10:52:41 localhost nut-driver@BX2200MI[1386987]:    0.287422        [D3] Adding command 'beeper.enable' using Path 'UPS.PowerSummary.AudibleAlarmControl'
Apr 21 10:52:41 localhost nut-driver@BX2200MI[1386987]:    0.287437        [D5] send_to_all: ADDCMD beeper.enable
Apr 21 10:52:41 localhost nut-driver@BX2200MI[1386987]:    0.287452        [D5] hid_lookup_usage: UPS -> 00840004
Apr 21 10:52:41 localhost nut-driver@BX2200MI[1386987]:    0.287467        [D5] hid_lookup_usage: PowerSummary -> 00840024
Apr 21 10:52:41 localhost nut-driver@BX2200MI[1386987]:    0.287489        [D5] hid_lookup_usage: AudibleAlarmControl -> 0084005a
Apr 21 10:52:41 localhost nut-driver@BX2200MI[1386987]:    0.287507        [D4] string_to_path: depth = 3
Apr 21 10:52:41 localhost nut-driver@BX2200MI[1386987]:    0.287523        [D3] Report[buf]: (2 bytes) => 18 01
Apr 21 10:52:41 localhost nut-driver@BX2200MI[1386987]:    0.287537        [D5] PhyMax = 0, PhyMin = 0, LogMax = 3, LogMin = 1
Apr 21 10:52:41 localhost nut-driver@BX2200MI[1386987]:    0.287550        [D5] Unit = 00000000, UnitExp = 0
Apr 21 10:52:41 localhost nut-driver@BX2200MI[1386987]:    0.287562        [D5] Exponent = 0
Apr 21 10:52:41 localhost nut-driver@BX2200MI[1386987]:    0.287576        [D2] Path: UPS.PowerSummary.AudibleAlarmControl, Type: Feature, ReportID: 0x18, Offset: 0, Size: 8, Value: 1
Apr 21 10:52:41 localhost nut-driver@BX2200MI[1386987]:    0.287588        [D3] Adding command 'beeper.disable' using Path 'UPS.PowerSummary.AudibleAlarmControl'
Apr 21 10:52:41 localhost nut-driver@BX2200MI[1386987]:    0.287607        [D5] send_to_all: ADDCMD beeper.disable
Apr 21 10:52:41 localhost nut-driver@BX2200MI[1386987]:    0.287626        [D5] hid_lookup_usage: UPS -> 00840004
Apr 21 10:52:41 localhost nut-driver@BX2200MI[1386987]:    0.287642        [D5] hid_lookup_usage: PowerSummary -> 00840024
Apr 21 10:52:41 localhost nut-driver@BX2200MI[1386987]:    0.287658        [D5] hid_lookup_usage: AudibleAlarmControl -> 0084005a
Apr 21 10:52:41 localhost nut-driver@BX2200MI[1386987]:    0.287670        [D4] string_to_path: depth = 3
Apr 21 10:52:41 localhost nut-driver@BX2200MI[1386987]:    0.287687        [D3] Report[buf]: (2 bytes) => 18 01
Apr 21 10:52:41 localhost nut-driver@BX2200MI[1386987]:    0.287701        [D5] PhyMax = 0, PhyMin = 0, LogMax = 3, LogMin = 1
Apr 21 10:52:41 localhost nut-driver@BX2200MI[1386987]:    0.287713        [D5] Unit = 00000000, UnitExp = 0
Apr 21 10:52:41 localhost nut-driver@BX2200MI[1386987]:    0.287730        [D5] Exponent = 0
Apr 21 10:52:41 localhost nut-driver@BX2200MI[1386987]:    0.287748        [D2] Path: UPS.PowerSummary.AudibleAlarmControl, Type: Feature, ReportID: 0x18, Offset: 0, Size: 8, Value: 1
Apr 21 10:52:41 localhost nut-driver@BX2200MI[1386987]:    0.287761        [D3] Adding command 'beeper.mute' using Path 'UPS.PowerSummary.AudibleAlarmControl'
Apr 21 10:52:41 localhost nut-driver@BX2200MI[1386987]:    0.287775        [D5] send_to_all: ADDCMD beeper.mute
Apr 21 10:52:41 localhost nut-driver@BX2200MI[1386987]:    0.287798        [D5] send_to_all: ADDCMD load.off
Apr 21 10:52:41 localhost nut-driver@BX2200MI[1386987]:    0.287816        [D2] find_nut_info: unknown info type: load.on.delay
Apr 21 10:52:41 localhost nut-driver@BX2200MI[1386987]:    0.287838        [D2] find_nut_info: unknown info type: load.on.delay
Apr 21 10:52:41 localhost nut-driver@BX2200MI[1386987]:    0.287859        [D5] send_to_all: SETINFO driver.version "2.8.0"
Apr 21 10:52:41 localhost nut-driver@BX2200MI[1386987]:    0.287880        [D5] send_to_all: SETINFO driver.version.internal "0.47"
Apr 21 10:52:41 localhost nut-driver@BX2200MI[1386987]:    0.287900        [D5] send_to_all: SETINFO driver.name "usbhid-ups"
Apr 21 10:52:41 localhost nut-driver@BX2200MI[1386987]:    0.287914        [D1] upsdrv_initinfo...
Apr 21 10:52:41 localhost nut-driver@BX2200MI[1386987]:    0.287931        [D5] send_to_all: SETINFO driver.version.data "APC HID 0.98"
Apr 21 10:52:41 localhost nut-driver@BX2200MI[1386987]:    0.287948        [D5] send_to_all: SETINFO driver.parameter.pollfreq "30"
Apr 21 10:52:41 localhost nut-driver@BX2200MI[1386987]:    0.287962        [D1] upsdrv_updateinfo...
Apr 21 10:52:41 localhost nut-driver@BX2200MI[1386987]:    0.306759        [D2] file_report_buffer: expected 5 bytes, but got 512 instead
Apr 21 10:52:41 localhost nut-driver@BX2200MI[1386987]:    0.306804        [D3] Report[err]: (512 bytes) => 16 0d 0d 01 fd 7f 00 00 d0 f2 92 aa 96 55 00 00
Apr 21 10:52:41 localhost nut-driver@BX2200MI[1386987]:    0.306829        [D3]  d0 bd 36 01 fd 7f 00 00 50 bf 36 01 fd 7f 00 00 1f c5 67 a9 96 55 00 00 fb
[... cut ...]
Apr 21 10:52:53 localhost usbhid-ups[1387007]: [D5] PhyMax = 0, PhyMin = 0, LogMax = 1, LogMin = 0
Apr 21 10:52:53 localhost usbhid-ups[1387007]: [D5] Unit = 00001001, UnitExp = 0
Apr 21 10:52:53 localhost usbhid-ups[1387007]: [D5] Exponent = 0
Apr 21 10:52:53 localhost usbhid-ups[1387007]: [D2] Path: UPS.PowerSummary.BelowRemainingCapacityLimit, Type: Feature, ReportID: 0x14, Offset: 0, Size: 8, Value: 0
Apr 21 10:52:53 localhost usbhid-ups[1387007]: [D5] hu_find_infoval: found !lowbatt (value: 0)
Apr 21 10:52:53 localhost usbhid-ups[1387007]: [D5] process_boolean_info: !lowbatt
`

tsopokis, 2024-04-21:
and the upc output:

upsc BX2200MI@localhost
Init SSL without certificate database
battery.charge: 99
battery.charge.low: 10
battery.mfr.date: 2024/04/04
battery.runtime: 3777
battery.runtime.low: 120
battery.type: PbAc
battery.voltage: 27.2
battery.voltage.nominal: 24.0
device.mfr: American Power Conversion
device.model: Back-UPS BX2200MI
device.serial: XXXXXXXXXXXX
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: auto
driver.parameter.vendorid: 051d
driver.version: 2.8.0
driver.version.data: APC HID 0.98
driver.version.internal: 0.47
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
input.sensitivity: low
input.transfer.high: 295
input.transfer.low: 145
input.transfer.reason: input voltage out of range
input.voltage: 224.0
input.voltage.nominal: 230
ups.beeper.status: disabled
ups.delay.shutdown: 20
ups.load: 0
ups.mfr: American Power Conversion
ups.mfr.date: 2023/11/09
ups.model: Back-UPS BX2200MI
ups.productid: 0002
ups.realpower.nominal: 1200
ups.serial: XXXXXXXXXXXX
ups.status: OL CHRG
ups.test.result: Aborted
ups.timer.reboot: 0
ups.timer.shutdown: -1
ups.vendorid: 051d

tsopokis, 2024-04-21:
I think version 2.8.2 should fix this one.

statte, 2024-05-24:
> I think version 2.8.2 should fix this one.

Anyone that can confirm this is a solution ?

PilaScat, 2024-05-24:
> > I think version 2.8.2 should fix this one.
> 
> Anyone that can confirm this is a solution ?

its not, not working for months now
I have a completely useless UPS, and its new

desertwitch, 2024-05-24:
> **TL;DR** I got rid of the incorrect status values by adding `pollonly` for my UPS in `ups.conf`. I am on nut v2.8.2 and the UPS is a brand new APC BX1200MI.
> 
> Once my UPS is fully charged and on-line, the nominal data bytes for the report with ID 0x16 is `16 0c 00 00 00`. The few bits that are set in these databytes correspond correctly to _UPS.PowerSummary.PresentStatus.BatteryPresent_ and _UPS.PowerSummary.PresentStatus.ACPresent_ (OL). As as side note, my UPS is actually only returning three bytes for this report (i.e., `16 0c 00`) and the remaining two (zero)bytes are being padded by nut on the fly. These two padding bytes are, however, defined as constants in the UPS' report descriptor, so they are nevertheless ignored.
> 
> Incorrect status values for my UPS only show up when report ID 0x16 is being received in `file_report_buffer()` in `drivers/libhid.c`. These HID events correspond to situations, when the reports come in from the UPS without the reports being requested. Running in debug mode, I see that these non-requested reports come in sizes of 512 bytes while nut usually is expecting much less. For example, receiving report 0x16 will show up in the debug log as `[D2] file_report_buffer: expected 5 bytes, but got 512 instead`. Most of the bytes included in these 512 bytes are leftovers from old buffers and in my case the first five bytes for 0x16 usually is `16 0d 0d c6 db`. In addition to _BatteryPresent_ and _ACPresent_ these would incorrectly indicate _UPS.PowerSummary.PresentStatus.Charging_ (CHRG) and _UPS.PowerSummary.PresentStatus.VoltageNotRegulated_. Occasionally I also get other incorrect datavalues corresponding for example to _UPS.PowerSummary.PresentStatus.NeedReplacement_ (RB).
> 
> By setting `pollonly` in my UPS config, I can make the parameter `use_interrupt_pipe` become `FALSE`. This in turn will lead to all those strange non-requested reports being ignored by nut (in `upsdrv_updateinfo()` in `drivers/usbhid-ups.c`).

Thank you very much for this detailed analysis with your (previously) affected UPS.
This is sounding to me like some firmware problem sending such bogus intermittent 0x16 reports with false statuses.

> **TL;DR I got rid of the incorrect status values by adding pollonly for my UPS in ups.conf. I am on nut v2.8.2 and the UPS is a brand new APC BX1200MI.**

**Can anyone else with such an UPS please test the suggested solution and see if that also resolves the situation for them?**

statte, 2024-05-24:
@s7uben Thank you for spending the time to write this out.

I've just made the changes suggested , I will report back tomorrow.

PilaScat, 2024-05-24:
[image]

Tried but not persistent

PilaScat, 2024-05-26:
Update: not fixed
[image]
and stopped polling from UPS
[image]

statte, 2024-05-26:
For me adding the _pollonly_ seems to have solved the issues.

For the other people still having issues 
I also set the battery manufacturing date and did a quick battery test aswel.

Sanderluc5, 2024-06-05:
Running latest NUT with my MX2200-MI

I have enabled the setting pollonly In the settings. and also forced the USB to be permanent powered on.

Still getting this errors:

Already tried a different USB-cable. But i can't seem to fix the problem.

`
Jun  5 12:05:44 srv01 upsd[3109]: UPS [ups] data is no longer stale
Jun  5 12:05:46 srv01 upsmon[3112]: Communications with UPS ups@127.0.0.1 established
Jun  5 12:05:46 srv01 upsmon[3112]: UPS ups@127.0.0.1 battery is low
Jun  5 12:21:34 srv01 usbhid-ups[30958]: nut_libusb_get_report: Input/Output Error
Jun  5 12:37:53 srv01 usbhid-ups[30958]: nut_libusb_get_report: Input/Output Error
Jun  5 12:38:11 srv01 usbhid-ups[30958]: nut_libusb_get_report: Input/Output Error
Jun  5 12:54:12 srv01 usbhid-ups[30958]: nut_libusb_get_report: Input/Output Error`

statte, 2024-06-07:
I'm still on 2.8.0 (no update due to using truenas scale).
After adding pollonly the main issue went away. Getting flooded by alerts about battery status. Eg. Replace battery or battery low. Once "pollonly" was added 👌. 

I did still notice in the kernel log that every 3 minutes (exactly 3 min)  the ups's usb connection was lost and restored. I've changed usb cable , usb port but this didn't go away untill I disconnected the ups from the server, stopped the ups service. Plugged in the ups again and started the service afterwards.  Now I haven't had the message for 6h+ .

Hope this helps someone.

Sanderluc5, 2024-06-11:
> I'm still on 2.8.0 (no update due to using truenas scale). After adding pollonly the main issue went away. Getting flooded by alerts about battery status. Eg. Replace battery or battery low. Once [...]

I was playing with my server aand changed some settings in the bios and reconnected everything and the problem was gone. I don't know what the cause was.

ivanjx, 2024-06-11:
might be a bit unrelated since i use apcupsd directly but i tried compiling apcupsd with `./configure --enable-usb --with-generic-usb` and it seems the battery disconnected/reattached events are gone now. the only downside is that the syslog for power failure and power restored are gone too now but the apcaccess is still able to detect the on battery status when i unplugged the ups from the power socket.

gabrieleancora, 2024-06-19:
> Got another report of an affected Back-UPS BX750MI with spurious RB events. I'm hoping to get my hands on at least a BX750MI this week so I can mess around with the driver a bit.

Hello, I do have the same exact model and I'm having the same problem. Let me know if you need any log/report and I'll provide them!

JoshNansoz1, 2024-07-06:
I’m getting a bunch of the same logs here with a BX950MI.
 

Is there anything we can do to help fix this? 
 

Provide some logs or perform any tests for you?

rioachim, 2024-07-18:
Same issues here with BX1600MI exchanged under warranty. The old one was fine, the new one is unusable with _apcupsd_ on freshtomato router firmware. The log is flooded by "battery disconnected/UPS battery must be replaced/Battery reattached" followed after some time by "usbhid control queue full". I have tried the patch in the following thread, but it behaves the same: https://sourceforge.net/p/apcupsd/mailman/apcupsd-users/thread/9c51020c-9d8b-4eda-b4d4-3ff8dc08749c%40okazoo.eu/#msg58741334

I am aware that apcupsd's \drivers\usb\usb.c is totally different that nut's drivers/usbhid-ups.c but in the end it would be a similar approach on the misbehaving UPS status info.

jimklimov, 2024-07-18:
Thanks for the different reports and ideas and the recent link to `apcupsd` seeing the same problem - so at least it is more likely an APC's HW/FW regression in roughly 2024's makes of their devices, than NUT's, right?

Not really sure which logs to ask for (thanks for offering), but thought of one point that can impact development of a fix: when the status flips about "battery disconnected -> must replace -> was reattached", how long does such cycle take? Is it always a matter of a few seconds with a (relatively) long quiet phase of normal work in between?

It may make sense to add an option to suppress such messages if the issue gets resolved within a specified timeframe... (and/or following the apcupsd patch - immediately report repeated offenses that are not resolved at the moment).

Would anyone here be brave enough to try coding such a change and making a PR?

JoshNansoz1, 2024-07-19:
@jimklimov 

Here are the event logs from my unraid server.

`Jul 18 00:03:39 Garage-Server apcupsd[3492]: Power is back. UPS running on mains.
Jul 18 00:03:39 Garage-Server apcupsd[3492]: Battery disconnected.
Jul 18 00:03:40 Garage-Server apcupsd[3492]: Battery reattached.
Jul 18 00:03:56 Garage-Server apcupsd[3492]: Power failure.
Jul 18 00:03:57 Garage-Server apcupsd[3492]: Power is back. UPS running on mains.
Jul 18 00:03:57 Garage-Server apcupsd[3492]: Battery disconnected.
Jul 18 00:03:58 Garage-Server apcupsd[3492]: Battery reattached.
Jul 18 00:05:34 Garage-Server apcupsd[3492]: Power failure.
Jul 18 00:05:34 Garage-Server apcupsd[3492]: Power is back. UPS running on mains.
Jul 18 00:05:34 Garage-Server apcupsd[3492]: Battery disconnected.
Jul 18 00:05:35 Garage-Server apcupsd[3492]: Battery reattached.
Jul 18 00:05:55 Garage-Server apcupsd[3492]: Power failure.
Jul 18 00:05:56 Garage-Server apcupsd[3492]: Power is back. UPS running on mains.
Jul 18 00:05:56 Garage-Server apcupsd[3492]: Battery disconnected.
Jul 18 00:05:57 Garage-Server apcupsd[3492]: Battery reattached.
Jul 18 00:07:39 Garage-Server apcupsd[3492]: Power failure.
Jul 18 00:07:39 Garage-Server apcupsd[3492]: Power is back. UPS running on mains.
Jul 18 00:07:39 Garage-Server apcupsd[3492]: Battery disconnected.
Jul 18 00:07:40 Garage-Server apcupsd[3492]: Battery reattached.
Jul 18 00:09:49 Garage-Server apcupsd[3492]: Power failure.
Jul 18 00:09:49 Garage-Server apcupsd[3492]: Power is back. UPS running on mains.
Jul 18 00:09:49 Garage-Server apcupsd[3492]: Battery disconnected.
Jul 18 00:09:50 Garage-Server apcupsd[3492]: Battery reattached.
Jul 18 00:10:22 Garage-Server apcupsd[3492]: Power failure.
Jul 18 00:10:23 Garage-Server apcupsd[3492]: Power is back. UPS running on mains.
Jul 18 00:10:23 Garage-Server apcupsd[3492]: Battery disconnected.
Jul 18 00:10:24 Garage-Server apcupsd[3492]: Battery reattached.`

jimklimov, 2024-07-19:
Thanks. So it is fairly frequent, but each situation takes a couple of seconds to resolve. No particular rhythm or rhyme to frequency of the situations, though.

desertwitch, 2024-07-19:
Sorry I haven't been able to get much further on this issue either, as there's no apparent logic to these conditions.
I'm just hoping that this will be fixed in future devices from the affected series and manufacturer.

grifferz, 2024-07-19:
> I'm just hoping that this will be fixed in future devices from the affected series and manufacturer.

As the original reporter of this issue, I did go through this with APC support and they are not prepared to accept any software reports unless they are from their own Powerchute software on Windows. As these spurious events do not happen there, they closed my issue as not reproducible.

So in short, APC do not accept there is any kind of bug to fix I'm afraid.

Strangely enough as mentioned in earlier comment:

https://github.com/networkupstools/nut/issues/2347#issuecomment-2034944128

the behaviour seems to have stopped for me by itself. It's been months and it hasn't happened. I have no idea why as all I did was change to what was at the time the HEAD of main.

owenperkins111, 2024-07-21:
> @jimklimov
>
> Here are the event logs from my unraid server.
>
> Jul 18 00:03:39 Garage-Server apcupsd[3492]: Power is back. UPS running on mains. Jul 18 00:03:39 Garage-Server apcupsd[3492]: Battery [...]

I am getting the exact same on my proxmox server with apcupsd and a APC Back UPS BX950

ivanjx, 2024-07-24:
anyone else also getting lots of self test switch messages from apcupsd? i have BX750MI

jimklimov, 2024-07-29:
Hello all, thank you for all the reports and logs. I've posted a prospective PR to address the situation by delaying LB/RB status propagation on impacted devices (user-configurable) and so hiding it from `upsmon` etc.  clients which emit the warnings. Testing would be welcome, see #2565 for details.

desertwitch, 2024-08-07:
For anyone here on Unraid, I have rolled out a testing package with the latest plugin update today. Please read the instructions shown on the "NUT Settings" page on how to switch to the testing package and if possible do report back if the problems are resolved for you (or not) - thanks a lot for your support. 😉

YahangWu, 2024-08-11:
I’m on 6.12.11 and using nut-2.8.2-x86_64-2master.ssl11 at the moment. My device is Back-UPS BX1200MI. The change seems to reduced the replacement battery warning frequency, precious it was every 9 hours. I’m running the new version for about two days and I’m getting one warning every 19 hours or so.

jimklimov, 2024-08-11:
As suggested in the PR discussion, try setting
````
lbrb_log_delay_without_calibrating
onlinedischarge_calibration
````
flags in your `ups.conf` section for the device, so the `OL+DISCHRG` combo would also be treated as a calibration and might tune into the change made here better.

closed 2024-08-12 by jimklimov

joshenders, 2024-10-12:
If it wasn't obvious by that last comment and you've arrived here from a search engine looking for a quick fix:

> set these flags in your `ups.conf` section for the device:

```
[ups]
...
lbrb_log_delay_without_calibrating = 1
onlinedischarge_calibration = 1
```

>  .... so the `OL+DISCHRG` combo would also be treated as a calibration and might tune into the change made here better.

Korvenwin, 2024-12-25:
Is this problem finally solved? I've just bought an BX1200MI and I've connected to an UnRAID server and I'm receiving the RB notifications twice a day.

jimklimov, 2024-12-25:
To the best of my knowledge - yes. But it is not part of a NUT release yet (2.8,3 expected soonish) so probably not packaged in official repos. I think @rysz made some testing builds back then.

desertwitch, 2024-12-25:
> To the best of my knowledge - yes. But it is not part of a NUT release yet (2.8,3 expected soonish) so probably not packaged in official repos. I think @desertwitch made some testing builds back then.

Yes, we've had good results with the recent master builds and the new configuration toggles introduced. For Unraid, this seemed to help the majority of users with this problem: https://forums.unraid.net/topic/60217-plugin-nut-v2-network-ups-tools/page/70/#findComment-1494093

```
pollonly
lbrb_log_delay_sec = 3
lbrb_log_delay_without_calibrating
```

invario, 2025-05-06:
Hi, just wanted to add to the that this occurs in APC model BVK750M2 also.

miguelbrandao, 2025-08-13:
For TrueNAS users affected by this issue and stuck on version 2.8.0, a [feature request](https://forums.truenas.com/t/update-nut-to-the-latest-version/50033) exists on the TrueNAS forums to update NUT to version 2.8.3 or later.

hydrolytic-melon, 2025-08-27:
I tried with 2.8.4 compiling from the sources on Proxmox 9.0.6 and unfortunately the problem persists; my `ups.conf` is:

```
user = nut
pollinterval = 5
maxretry = 3

offdelay = 120
ondelay = 240

[apc]
  driver = "usbhid-ups"
  port = "auto"
  vendorid = "051D"
  productid = "0002"
  product = "Back-UPS BX750MI"
  serial = "<XXX>"
  vendor = "American Power Conversion"
  bus = "001"
  device = "002"
  busport = "001"
```
Does anyone know on any workaround for this issue?

[2 more replies]