networkupstools/nut #2348: Build a driver from source for an existing installation

lewisxy, 2024-03-03:
I have an APC SmartUPS SMT3000RM2UC, which supports both usb (`usbhid-ups`) and modbus (`apc_modbus`) driver. My servers runs ubuntu, so I installed `nut` from the system's package manager. However, it does not contains `apc_modbus` driver. Since drivers are separate binaries located in `/lib/nut`, is it possible for me to only build the driver I needed and copy that binary to the existing installation and use it, without installing everything from source? Thanks

biergaizi, 2024-03-03:
Yes, it's possible. As a driver developer, I've used this method extensively for testing purpose.

In order to do that, the driver must be built using the same search paths as the system-wide NUT installation when running `./configure`, otherwise they would be unable to find each other's files. For example, on Debian, it would be:

```
./configure --with-modbus \
    --with-user=nut --with-group=dialout \
    --datadir=/usr/share/nut \
    --sysconfdir=/etc/ups \
    --with-statepath=/var/run/nut \
    --with-pidpath=/var/run/nut \
    --with-altpidpath=/var/run/nut \
    --with-drvpath=/usr/sbin \
    --with-systemdsystemunitdir=/usr/lib/systemd/system \
    --with-pkgconfig-dir=/usr/lib/pkgconfig \
    --with-udev-dir=/usr/lib/udev \
    --libdir=/usr/lib/
```

All of these parameters must be set identical to the system-wide NUT package. Thus, it's often necessary to examine the source code of your distro package's build files to find the correct values. Don't use random values, it won't work and may mess your system up.

But if you're able to do that, all you need is to:

```
# compile include/ separately
cd include
make
cd ..

# compile common/ separately
cd common/
make
cd ..

# compire driver-name separately
cd drivers
make driver-name
```

Then, simply copy the driver to your system's path.

jimklimov, 2024-03-03:
For `apc_modbus` in particular, if you plan to use USB connection, you would need a custom-built libmodbus and then configure NUT against that (see wiki).

For Serial and TCP links the stock version should do.

lewisxy, 2024-03-04:
Thanks for the help. I managed to build it. Here is some note.
My server is running Ubuntu 20.04 LTS. So I created an VM with the same OS to build things.

My configuration command is
```bash
./configure --prefix= \
                             --with-user=nut --with-group=nut \
                             --with-modbus \
                             --sysconfdir=/etc/nut \
                             --includedir=/usr/include \
                             --with-dev \
                             --enable-static \
                             --with-statepath=/run/nut \
                             --with-altpidpath=/run/nut \
                             --with-drvpath=/lib/nut \
                             --with-pidpath=/run/nut \
                             --datadir=/usr/share/nut \
                             --with-pkgconfig-dir=/usr/lib/pkgconfig
```
This is partially adapted from @biergaizi 's and partially from `rules` defined in the [package source][nut-ubuntu]. Note that `libmodbus-dev` need to be installed for configuration to succeed. After configuration, I just followed the build instructions from @biergaizi , and copy the binary to my server. I also installed `libmodbus5` on the server.

The modbus driver provides way more information and controls than the `usbhid-ups` I was using before. For example, I can now read the `ups.load` and shutdown/power cycle each outlet groups independently, which are previously unavailable.

[nut-ubuntu]: http://archive.ubuntu.com/ubuntu/pool/main/n/nut/nut_2.7.4-11ubuntu4.debian.tar.xz

closed 2024-03-04 by lewisxy

jimklimov, 2024-06-04:
See also https://github.com/networkupstools/nut/wiki/APC-UPS-with-Modbus-protocol - especially regarding the hint that custom libmodbus can be built and linked into the NUT driver statically, so eliminating a possible conflict with the OS-provided library.

jhavens12, 2024-10-04:
@biergaizi Sorry I don't understand, where are the include, common and drivers folders?

jimklimov, 2024-10-04:
@jhavens12  : Those would be part of NUT sources. Now that this is all part of NUT master branch, I suggest you follow the wiki article above about building them together. If you want it aligned with a package, you may want to mix it with "in-place" build configuration detailed in another wiki article (linked from wiki footer).