networkupstools/nut #2353: Problem with my UPS on NUT (HomeAssistant)

studioIV, 2024-03-06:
I am new to Homelab and just got my new ups [Powerwalker VI 1100 CW](https://powerwalker.com/product/10121130/) but I can't find a driver that works with NUT(Running in HomeAssistant). To be honest I found this driver ( [nutdrv_qx](https://networkupstools.org/docs/man/nutdrv_qx.html) ). But most of the time the ups disconnect randomly and show up as anavailable. I use this computer as a proxmox server: [M83 Desktop (ThinkCentre) SFF - Type 10AH](https://pcsupport.lenovo.com/gr/el/products/desktops-and-all-in-ones/thinkcentre-m-series-desktops/thinkcentre-m83/10ah/10aha0sx00/s4f06151/downloads/driver-list).

andyevans-eng, 2024-07-05:
I have setup my Powerwalker UPS using the NUT integration, it kept losing connection with the stale data error after a few hours, I have configured the maxage and polling interval etc and is would run for about 20 hours before getting the same error. After reading through many forums I changed the polling interval to pollonly = "enabled"
It worked perfectly for about 36 hours, but am now back to the stale data problem. I really need to get this working properly as I have safety automations that depend on the UPS data, which obviously don't work with stale data. HA running on PI5 8Gb Ram, everything up to date. Any suggestions please? Yes I can create an automation to restart when UPS becomes unavailable, but that defeats the object, plus I don't want my HA out of action for a minute or 2 almost every day. Thanks in advance, Andy

jimklimov, 2024-07-05:
One more idea that comes to mind is adding cron scripts (generally, maybe some HA actions would be more idiomatic for you) to restart NUT and/or just try to soft-reset the USB port.

Check examples of such ideas at:
* https://github.com/networkupstools/nut/tree/master/scripts/usb_resetter
* https://github.com/networkupstools/nut/blob/master/scripts/Solaris/reset-ups-usb-solaris.sh.sample