networkupstools/nut #2357: nJoy Keen 1500 support

andrey-ivanov, 2024-03-10:
Manufacturer: nJoy
Device: Keen 600 USB
Website: [https://www.njoy.ro/UPS/keen-600 No separate site for USB version](https://www.njoy.global/product/keen-1500)

I can confirm Keen 1500 is recognized by nutdrv_qx. I used [Keen 600 ticket](https://github.com/networkupstools/nut/issues/867) as reference.

/var/log/syslog
```
upsdrvctl[522234]: Network UPS Tools - UPS driver controller 2.7.4
upsd[486317]: Can't connect to UPS [njoykeen] (nutdrv_qx-njoykeen): No such file or directory
upsdrvctl[522236]: Using protocol: Q1 0.07
upsdrvctl[522236]: Can't autodetect number of battery packs [-1/13.20]
upsdrvctl[522236]: Battery runtime will not be calculated (runtimecal not set)
upsdrvctl[522235]: Network UPS Tools - UPS driver controller 2.7.4
upsd[486317]: Connected to UPS [njoykeen]: nutdrv_qx-njoykeen
```

upsc
```
battery.voltage: 13.60
device.type: ups
driver.name: nutdrv_qx
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 10
driver.parameter.port: /dev/ttyUSB0
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.data: Q1 0.07
driver.version.internal: 0.28
input.frequency: 49.8
input.voltage: 217.0
input.voltage.fault: 0.0
output.voltage: 217.0
ups.beeper.status: enabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.load: 12
ups.status: OL
ups.temperature: 25.0
ups.type: offline / line interactive
```

Driver has an exactly same limitations as Keen 600:
- No **battery.voltage.high** and**battery.voltage.low** reported
- **ups.temperature** is always reported as 25.0

I observed USB reconnection after every on-line to on-battery switching (and vise versa). It leads to "connection loss" event and require nut-driver server restarting, Not sure if this is UPS related or specific to my hardware.

jimklimov, 2024-03-10:
Thanks for the report! Would you have a chance to check if modern NUT 2.8.1 release would see more data (and/or did not regress and see at least this much)?

andrey-ivanov, 2024-03-10:
I'll back with a report about 2.8.1 in the next few months. Not ready for Ubuntu 24.04 challenge yet. :)

andrey-ivanov, 2024-04-17:
Percentage values are not reported for Keen 1500. However, battery voltage is available. It can be used as threshold for automatic shutdown.

andrey-ivanov, 2024-09-28:
> Would you have a chance to check if modern NUT 2.8.1 release would see more data (and/or did not regress and see at least this much)?

upsc
```
battery.voltage: 13.6
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.name: nutdrv_qx
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 10
driver.parameter.port: /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
driver.parameter.synchronous: auto
driver.state: quiet
driver.version: 2.8.1
driver.version.data: Q1 0.08
driver.version.internal: 0.36
driver.version.usb: libusb-1.0.27 (API: 0x100010a)
input.frequency: 49.8
input.voltage: 213.0
input.voltage.fault: 0.0
output.voltage: 213.0
ups.beeper.status: enabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.load: 0
ups.status: OL
ups.temperature: 25.0
ups.type: offline / line interactive
```

Looks like there is no regression in 2,8,1

No new data, but there are new driver entries:
driver.debug: 0
driver.flag.allow_killpower: 0
driver.state: quiet
driver.version.usb: libusb-1.0.27 (API: 0x100010a)

and driver.parameter.synchronous changed from no to auto

racinmat, 2026-09-13:
Hi, I tried the same UPS (nJoy Keen 1500, official cable, CH340 1a86:7523) on an Odroid-C4 running Armbian with NUT 2.8.1-5, but unfortunately it didn't work for me. I'll try a warranty claim, maybe just this unit of mine is broken.

For anyone who lands here with the same symptom, here's what I found in case it's useful. `nutdrv_qx` gives me "Device not supported!" whether I let it autodetect or force protocol=q1. Rather than assume I'd messed up the config, I dropped below NUT entirely and poked the serial port directly with a small Python script (termios/fcntl, no pyserial) to figure out if this was a config problem or the UPS just wasn't talking.

Turns out 2400 baud, 8N1 is definitely right - UPS answers instantly, every time, and it doesn't care what I do with DTR/RTS (tried all four combinations). Drop to any other baud rate and it's dead silent.

Here's the annoying part: no matter what I send, I get back the same four bytes - 23 2d 31 0d, which is `#-1\r`. Q1, QS, QGS, QPI, QMD, QRI, QMOD, I, F, doesn't matter. Even a bare `\r` with nothing in front of it gets the same reply. Even garbage like `ZZZZ\r`. Drop the `\r` terminator and there's nothing at all, so whatever's answering only cares that a line ended in `\r` — not what was in it.

I also wondered if this thing wants a checksum before it'll actually parse anything, so I tried CRC16-XMODEM, CRC16-CCITT, and a dumb 1-byte XOR on Q1/QPI/QPIGS. Same `#-1\r` every time.

A few other things I ruled out: no unsolicited data, if I just sit and listen for a few seconds, and a full power cycle of the UPS (unplugged, off, 60s, back on) changed nothing - same behavior before and after, so it's not some MCU that needed a cold boot. The UPS's own front panel is happy the whole time - green LED, no beeping - so it seems like it's specifically the USB port.

This also doesn't line up with the Megatec spec, which says an unrecognized command should get echoed back, not swapped for a canned reply. So whatever's on the other end doesn't seem to be reading the command at all - my best guess is the USB-to-Megatec translation board inside this unit is either faulty or running something that isn't any Q*/Megatec dialect, and `#-1` is just its generic "something's wrong" response.

Since the original report in this issue got a clean Q1 0.07 autodetect on what's presumably the same model, this looks like a hardware or firmware revision difference rather than a config issue on my end.

Npeca75, 2026-09-23:
> I get back the same four bytes - 23 2d 31 0d,

me too :(
reproduced all the simptoms
same UPS, same behaviour
`#-1` is only answer i get whatever test data push to him