networkupstools/nut #2358: PowerWalker VI 3000 RLP (Cyber Power System, Inc. PR1500LCDRT2U UPS) - Unable to get HID descriptor (Overflow)

mablebee, 2024-03-11:
Built nut-2.8.1 on raspberry pi with PR1500LCDRT2U UPS attached via USB:

`Bus 001 Device 005: ID 0764:0601 Cyber Power System, Inc. PR1500LCDRT2U UPS
`
and configured the UPS in the ups.conf File:
`[powerwalker]
        driver = usbhid-ups
        vendorid = 0764
        productid = 0601
        pollinterval = 15
        port = auto
`
When running the `/usr/local/ups/bin/usbhid-ups -a powerwalker -DD` command I get following output:

`Network UPS Tools - Generic HID driver 0.52 (2.8.1)
USB communication driver (libusb 1.0) 0.46
   0.000000     [D1] Network UPS Tools version 2.8.1 (release/snapshot of 2.8.1) built with gcc (Debian 8.3.0-6) 8.3.0 and configured with flags: --with-user=nut --with-group=nut --with-usb
   0.000218     [D1] debug level is '2'
   0.002223     [D1] Succeeded to become_user(nut): now UID=117 GID=127
   0.002489     [D1] upsdrv_initups (non-SHUT)...
   0.002640     [D2] Initializing an USB-connected UPS with library libusb-1.0.22 (API: 0x1000106) (NUT subdriver name='USB communication driver (libusb 1.0)' ver='0.46')
   0.017363     [D2] Checking device 1 of 4 (1D6B/0003)
   0.017651     [D1] Failed to open device (1D6B/0003), skipping: Access denied (insufficient permissions)
   0.017803     [D2] Checking device 2 of 4 (0764/0601)
   1.018513     [D1] nut_libusb_open get iManufacturer failed, retrying...
   2.019171     [D1] nut_libusb_open get iManufacturer failed, retrying...
   3.019700     [D1] nut_libusb_open get iManufacturer failed, retrying...
   4.020214     [D1] nut_libusb_open get iProduct failed, retrying...
   5.020756     [D1] nut_libusb_open get iProduct failed, retrying...
   6.021286     [D1] nut_libusb_open get iProduct failed, retrying...
   7.021813     [D1] nut_libusb_open get iSerialNumber failed, retrying...
   8.022366     [D1] nut_libusb_open get iSerialNumber failed, retrying...
   9.022818     [D1] nut_libusb_open get iSerialNumber failed, retrying...
   9.022877     [D2] - VendorID: 0764
   9.022893     [D2] - ProductID: 0601
   9.022908     [D2] - Manufacturer: unknown
   9.022923     [D2] - Product: unknown
   9.022938     [D2] - Serial Number: unknown
   9.022952     [D2] - Bus: 001
   9.022967     [D2] - Bus Port: 002
   9.022982     [D2] - Device: 005
   9.022997     [D2] - Device release number: 0200
   9.023012     [D2] Trying to match device
   9.023030     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   9.023121     [D2] Device matches
   9.023141     [D2] Reading first configuration descriptor
   9.023216     [D2] Claimed interface 0 successfully
   9.025366     [D2] Unable to get HID descriptor (Overflow)
   9.025422     [D2] HID descriptor length 739
  14.025790     [D2] Unable to get Report descriptor: Resource temporarily unavailable
  14.025974     [D2] Checking device 3 of 4 (2109/3431)
  14.026035     [D1] Failed to open device (2109/3431), skipping: Access denied (insufficient permissions)
  14.026057     [D2] Checking device 4 of 4 (1D6B/0002)
  14.026089     [D1] Failed to open device (1D6B/0002), skipping: Access denied (insufficient permissions)
  14.026111     [D2] libusb1: No appropriate HID device found
  14.026130     libusb1: Could not open any HID devices: insufficient permissions on everything
  14.026149     No matching HID UPS found
  14.026177     upsnotify: failed to notify about state 4: no notification tech defined, will not spam more about it
`

jimklimov, 2024-03-11:
Hello, a couple of quick ideas first:

* make sure that udev/upower rules got applied (e.g. services restarted, including `systemctl restart udev` or a reboot), especially if you had `make install` deliver the NUT rules files and none were present before (as from packaging, earlier release, etc.) - this is often behind permissions issues
* check also https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests for recently tested hints about building and installing in a systemd-based environment (closer to the end)
* for CPS-derived devices there's also https://github.com/networkupstools/nut/wiki/CyberPower-Systems-(CPS)-know-how - notably, their controllers tend to go to sleep at inconvenient times (more frequent polling may help)
* there were also quite a few stability reports with Raspberries involved, still not sure if that is a problem with the platform or testament to its popularity :) It seems RPi3 and RPi5 are involved more frequently than RPi4, for what it's worth...

Also, can you show `lsusb -vvv -d 0764:0601` - specifically I wonder what it says where it reports how many "interfaces" it has, with their transfer sizes. Recently more and more devices are on the market whose USB HID interface is not number zero, NUT is catching up with those (`master` branch recently grew driver options for forcing a custom number, helpful at least for experiments)...

mablebee, 2024-03-11:
Thanks for your first quick ideas. It seams that if I do a `usbreset 0764:0601`  before starting the services or manually running `usbhid-ups ` then the connection is established. 

I'm currently building the current master and will retry. Meanwhile: here is the output from `lsusb -vvv -d 0764:0601`

```
Bus 001 Device 003: ID 0764:0601 Cyber Power System, Inc. PR1500LCDRT2U UPS
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x0764 Cyber Power System, Inc.
  idProduct          0x0601 PR1500LCDRT2U UPS
  bcdDevice            2.00
  iManufacturer           3 1
  iProduct                1 3000
  iSerial                 2 000000000000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0029
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0
      bInterfaceProtocol      0
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     739
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval              20
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval              20
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered
```

jimklimov, 2024-03-11:
Thanks! So it seems there is one descriptor, so at least this part should be pretty normal:
````
          bDescriptorType        34 Report
          wDescriptorLength     739
````

mablebee, 2024-03-13:
OK, so I did a complete fresh reinstall, changed the config files and it's working fine now (and persistently stable through reboots). 

I have absolutely no idea what the issue was before - probably a reboot after the make install would have been a good idea.

A Raspberry is now master and there are 2 WinNUT clients (Windows 2022 VMs), 3 Proxmox Servers and two QNAP Systems as slaves configured and working.

Thanks for your help!

closed 2024-03-13 by mablebee