networkupstools/nut #2365: Riello IDG1200 Partial Data

Simonv92, 2024-03-19:
Hi, 
I'm trying to configure a Riello IDG1200 UPS with NUT.
I've setup it up with these parameters:

driver = "riello_usb" 
port = "auto"

But I have only data about battery % (wrong data), Volta IN, Voltage OUT ans Status.
Am I doing something wrong? On other guides I've seen that they have much more data from the UPS..
Thank you very much!
Simone

jimklimov, 2024-03-19:
Which NUT version? Many distros still ship 2.7.4 (~8 years old) while current is 2.8.1.

Simonv92, 2024-03-19:
If I run upsc riello I see:
driver version 2.7.4

jimklimov, 2024-03-19:
Officially - get a newer distro release (some have "experimental" repos with newer packages also).

Recommended - check wiki article here about "in-place upgrade" builds, to roll your own NUT from current sources. You can test the built driver from the build workspace, to only install the new programs if they are proven better for your device.

Simonv92, 2024-03-19:
Sorry but I'm a newbie.. how do I get a new distro? running sudo apt install nut-client nut-server always show me the 2.7.4-13 release..

jimklimov, 2024-03-19:
Newer release of the distro of apparently the Debian or Ubuntu (based?) system you seem to have.

Following that wiki article could be in fact easier a d leas disruptive - install some packages for prerequisite tools and dependencies and copy-paste commands to build NUT.

Simonv92, 2024-03-19:
I'm using Raspberry Pi OS Lite on a Raspberry Pi Zero 2W.
I'm trying to install the latest release I found here: https://github.com/networkupstools/nut/releases
Trying to copy and then install the  nut-2.8.1.tar.gz file doesn't work.. How can I do that?

jimklimov, 2024-03-20:
Now I'm at a real computer so can post some links: :)

* https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests
* notably https://networkupstools.org/docs/user-manual.chunked/_build_prerequisites_to_make_nut_from_scratch_on_various_operating_systems.html about prerequisites

The "tar.gz" archive on site is a snapshot of NUT sources for the release (some errata applies, fixed since in Git and should be part of upcoming next release snapshot) which should be built similar to the procedure above, just instead of checking out a Git branch you would unpack that archive. At the moment, due to that errata, better follow the Wiki for a current Git-source build.