networkupstools/nut #2370: Maths puzzle: revise Liebert/Belkin USB HID parsing for line voltage orders of magnitude and don't loose track of zeroes

jimklimov, 2024-03-24:
Following up from #2271 / #2363 discussions and changes (CC @jrjparks @clepple @aquette @digitlman @electroflame)

The differing maths approaches in the older and newer methods still nag me in the back of my mind, and I also wonder if these can be collapsed into `d_equal()` or related macros we have defined in NUT for comparing values that are almost equal, within an `epsilon` error apart...

Spilling it out here to think out loud :)

Looking at https://github.com/networkupstools/nut/blob/31e9979/drivers/belkin-hid.c#L168-L222 I see the two methods in question:

````
OLD> liebert_line_voltage_fun(double value) {
	if( value < 1 ) {
		if( fabs(value - 1e-7) < 1e-9 ) {
			liebert_line_voltage_mult = 1e7;
...
````

````
NEW> liebert_psi5_line_voltage_fun(double value) {
	if( value < 1 ) {
		if( fabs(value - 1e-3) < 1e-3 ) {
			liebert_line_voltage_mult = 1e5;
...
````

I believe the goal here is to find such a `liebert_line_voltage_mult` that we would multiply the miniscule `value` reading (coming from hardware) by, and get something in the hundreds range for the line voltage (typically 120/127/220/230/240V - so somewhat over an exact hundred).

I can also accept that the different HW/FW models (PSI5 vs. others) can serve their data differently and not "just" shifted by some orders of magnitude compared to other models, although posting some examples from low-level readings of different models could be useful to make sense of different use-cases (adding debug prints into the driver builds?)

Namely, what concerns me here is how:
* in the old case we compare `if abs(value - (10^-N)) < (10^-N)/100) then mult=(10^N)` (where `N=7`)
* and for the new code `if abs(value - (10^-N)) < (10^-N)) then mult=100*(10^N)` (where `N=3`)
* ... note the hundreds moving around. Is that correct?

Also, what do these `abs(diff)` achieve?

Assuming larger scaling numbers just for easier visible maths, with `N=0` (`10^0 == 1`) and reasonable values of 120V or 220V (seen *after* multiplication by `100*N` when the test succeeds, so `double value=1.2` and `value=2.2`), we would see either:
* OLD LOGIC: want 120V/220V *after* multiplication by `mult = 10^N = 1` when the test succeeds, so `double value=120.0` and `value=220.0`
  * `if (abs(120-1) = 119 < 0.01) then mult=1` -- does not quite hold true
  * `if (abs(220-1) = 221 < 0.01) then mult=1` -- does not quite hold true
* NEW LOGIC: want 120V/220V *after* multiplication by `mult = 100*(10^N) = 100` when the test succeeds, so `double value=1.2` and `value=2.2`
  * `if (abs(1.20-1) = 0.2 < 1) then mult=100` -- seems true
  * `if (abs(2.20-1) = 1.2 < 1) then mult=100` -- does not quite hold true

Quite may be that I got the napkin math (without even a napkin) during commute wrong, and maybe that's why this puzzles me =D

jimklimov, 2024-03-24:
A few raw readings:

* PSI5 (new) per https://github.com/networkupstools/nut/issues/2271#issuecomment-1976540478
  * raw value is circa `0.001211` to become approx `121V` (so `mult=1e5` is right)
  * Belkin/Liebert HID mapping as discussed here:
````
   0.162809	[D1] Path: UPS.PowerConverter.Input.Voltage, Type: Feature, ReportID: 0x18, Offset: 0, Size: 16, Value: 0.001212
   0.168331	[D1] Path: UPS.PowerConverter.Output.Voltage, Type: Feature, ReportID: 0x1b, Offset: 0, Size: 16, Value: 0.001214
````

  * Same device, another HID mapping (Phoenixtec/Liebert):
````
   0.147460	[D1] Path: UPS.PowerConverter.Input.Voltage, Type: Feature, ReportID: 0x18, Offset: 0, Size: 16, Value: 0.001213
   0.152961	[D1] Path: UPS.PowerConverter.Output.Voltage, Type: Feature, ReportID: 0x1b, Offset: 0, Size: 16, Value: 0.001211
````

clepple, 2024-03-24:
I can't speak for the new code, but off the top of my head, the old numbers were just a heuristic that happened to catch the 1e-7 factor. I don't know if this is the same issue as the CPS HID report descriptor parsing issue, where NUT and Windows differ about how the HID parser state gets manipulated when collections are pushed and popped on the stack. If that is the case, then the 1e7 multiplier cancels out something that should have never been added. Maybe there is a 1e-5/1e5 factor hanging around for the other case. But as in #169 I'm not sure of another good way to handle incorrect HID descriptors without introducing bugs if/when the vendor fixes the descriptor.

Now that NUT has a bit of a testing framework, we should really factor out some of this math into unit tests. People shouldn't have to wait until they hook up hardware to see whether this has been fixed (and it will help developers not break existing workarounds.)

jimklimov, 2024-03-25:
A bit of info from the mailing list, with NUT v2.7.4 and a "LiebertPSA" device:
````
   0.278176 Path: UPS.Input.Voltage, Type: Feature, ReportID: 0x20, Offset: 0, Size: 16, Value: 2.201e-05
   0.278274 Input/OutputVoltage = 2.201e-05 -> assuming correction factor = 1e+07
   0.278384 send_to_all: SETINFO input.voltage "220.1"
   0.278517 hid_lookup_usage: UPS -> 00840004
   0.278621 hid_lookup_usage: Output -> 0084001c
   0.278769 hid_lookup_usage: Voltage -> 00840030
   0.278913 string_to_path: depth = 3
   0.279025 Report[buf]: (3 bytes) => 2b 9e 08
   0.279133 PhyMax = 0, PhyMin = 0, LogMax = 65535, LogMin = 0
   0.279240 Unit = 00f0d121, UnitExp = -1
   0.279304 Exponent = -8
   0.279412 Path: UPS.Output.Voltage, Type: Feature, ReportID: 0x2b, Offset: 0, Size: 16, Value: 2.206e-05
   0.279523 Input/OutputVoltage = 2.206e-05 -> assuming correction factor = 1e+07
   0.279655 send_to_all: SETINFO output.voltage "220.6"
   0.279793 hid_lookup_usage: UPS -> 00840004
   0.279893 hid_lookup_usage: PowerSummary -> 00840024
   0.279999 hid_lookup_usage: Voltage -> 00840030
   0.280100 string_to_path: depth = 3
   0.280209 Report[buf]: (3 bytes) => 1d 8b 00
   0.280313 PhyMax = 0, PhyMin = 0, LogMax = 65535, LogMin = 0
   0.280426 Unit = 00f0d121, UnitExp = -1
   0.280488 Exponent = -8
   0.280596 Path: UPS.PowerSummary.Voltage, Type: Feature, ReportID: 0x1d, Offset: 0, Size: 16, Value: 1.39e-06
   0.280707 Input/OutputVoltage = 1.39e-06 -> assuming correction factor = 1e+07
   0.280822 send_to_all: SETINFO battery.voltage "13.9"
   0.280929 hid_lookup_usage: UPS -> 00840004
   0.281029 hid_lookup_usage: PowerSummary -> 00840024
   0.281136 hid_lookup_usage: ConfigVoltage -> 00840040
   0.281249 string_to_path: depth = 3
   0.281361 Report[buf]: (3 bytes) => 1f 0c 00
   0.281440 PhyMax = 0, PhyMin = 0, LogMax = 65535, LogMin = 0
   0.281539 Unit = 00f0d121, UnitExp = -1
````

jimklimov, 2024-03-25:
I think I've found the "regression" in Belkin/Liebert USB HID support which worked for NUT v2.7.4 and reports zero voltages in 2.8.0 -- it was actually caused by a bug fix: practically the only functional difference per `git diff v2.7.4..v2.8.0 drivers/belkin-hid.c` is a change from `abs()` to `fabs()` in order-of-magnitude comparisons like `if( abs(value - 1e-7) < 1e-9 ) {...}` which feel odd on their own, as discussed above.

The old code worked because `abs()` is for `int` numbers, and the miniscule argument became an always-zero after casting (which is less than `10^-9`), and for existing devices it just happened to be the correct codepath to pick an exponent for USB readings. 

The new code made the comparisons mathematically correct, to work on `double` precision numbers using `fabs()` (to not confuse with `fabsf()` for `float` types, go figure). And boom! Something no longer clicks for `1.39e-06 => 13.9V` and `2.206e-05 => 220V` (with correction factor `1e7`) =D

Obligatory inspiring motivator : https://xkcd.com/1172/

Thanks to Mr. Fischer from the mailing list discussion for the right logs (and version-dissection findings) at the right time :)

jimklimov, 2024-04-06:
A TDD solution was merged and released as part of NUT v2.8.2 - we'll see how well it fares...

closed 2024-04-06 by jimklimov

digitlman, 2024-04-08:
Any idea how long before package managers offer this version in Debian?  I
tried to build from source and was not able to....

jimklimov, 2024-04-08:
CC @bigon regarding packaging  :)

As for the build - what failed for you?Do you have prerequisites installed? (See wiki for links into nut docs).

digitlman, 2024-04-08:
***@***.***:/home/it# systemctl status nut-driver-enumerator.service
× nut-driver-enumerator.service - Network UPS Tools - enumeration of
configure-file devices into systemd unit instances
     Loaded: loaded (/lib/systemd/system/nut-driver-enumerator.service;
enabled; preset: enabled)
     Active: failed (Result: exit-code) since Mon 2024-04-08 14:41:51 EDT;
2min 55s ago
TriggeredBy: ● nut-driver-enumerator.path
    Process: 367 ExecStartPre=/usr/bin/systemd-tmpfiles --create
/usr/lib/tmpfiles.d/nut-common-tmpfiles.conf (code=exited, sta>
    Process: 372 ExecStart=/usr/local/ups/libexec/nut-driver-enumerator.sh
(code=exited, status=2)
   Main PID: 372 (code=exited, status=2)
        CPU: 86ms

Apr 08 14:41:50 NUT systemd[1]: Starting nut-driver-enumerator.service -
Network UPS Tools - enumeration of configure-file devi>
Apr 08 14:41:51 NUT nut-driver-enumerator[372]: FATAL: The
'/usr/local/ups/etc/ups.conf' file does not exist or is not readable
Apr 08 14:41:51 NUT systemd[1]: nut-driver-enumerator.service: Main process
exited, code=exited, status=2/INVALIDARGUMENT
Apr 08 14:41:51 NUT systemd[1]: nut-driver-enumerator.service: Failed with
result 'exit-code'.
Apr 08 14:41:51 NUT systemd[1]: Failed to start
nut-driver-enumerator.service - Network UPS Tools - enumeration of
configure-fi>
***@***.***:/home/it#

jimklimov, 2024-04-08:
>The
'/usr/local/ups/etc/ups.conf' file does not exist or is not readable

Did you actually `configute` your build?