networkupstools/nut #2372: Update Eaton MIBs 9PX - M2 network card

arghhh40k, 2024-03-26:
Eaton has been adding data, alarms and status from their newer UPS. They have updated the MIB files to reflect this. Can the new MIB file be included in the driver?

I have a Eaton 9PX6KSP model using SNMP v3 with the M2 network card.

[EATON-EMP-MIB.txt](https://github.com/networkupstools/nut/files/14759121/EATON-EMP-MIB.txt)

arghhh40k, 2024-03-26:
I have also noticed incorrect ups.power and only showing 1 phase of the split phase and not the total output of the UPS.

28% of 5520 is 1545w not 627w 

[image]

jimklimov, 2024-03-26:
CC @aquette

desertwitch, 2024-09-24:
**For what it's worth, here's a rudimentary driver stub with dumped data from a simulated SNMP device on the new MIB:**
https://gist.github.com/desertwitch/b9208033c3ab8817f2836d0674c12a35

**This should help get the ball rolling with some additional (new) data points and mappings, most likely to enhance:**
https://github.com/networkupstools/nut/blob/master/drivers/eaton-ups-pxg-mib.c
which is based on what seems to be a previous revision of `Xups.mib`, so probably best merged into.