networkupstools/nut #2377: blazer_usb failing after update all

ripf77, 2024-03-28:
My nut was scanning the ups (Fyd MEC0003) with blazer_usb properly. After running an update on raspbian level nut is exiting after start with 

===================================================
systemctl status nut-driver delivers

Mar 28 16:03:43 susi upsdrvctl[2819]: Using 'guestimation' (low: 10.400000, high: 13.000000)!
Mar 28 16:03:43 susi upsdrvctl[2819]: Battery runtime will not be calculated (runtimecal not set)
Mar 28 16:03:43 susi upsdrvctl[2818]: Network UPS Tools - UPS driver controller 2.7.4
Mar 28 16:03:43 susi systemd[1]: Started Network UPS Tools - power device driver controller.
Mar 28 16:03:43 susi blazer_usb[2821]: Startup successful
Mar 28 16:03:44 susi systemd[1]: Stopping Network UPS Tools - power device driver controller...
Mar 28 16:03:44 susi upsdrvctl[2822]: Network UPS Tools - UPS driver controller 2.7.4
Mar 28 16:03:45 susi blazer_usb[2821]: Signal 15: exiting
Mar 28 16:03:45 susi systemd[1]: nut-driver.service: Succeeded.
Mar 28 16:03:45 susi systemd[1]: Stopped Network UPS Tools - power device driver controller.

===================================================
ups.conf 
[susi]
        driver = "blazer_usb"
        port = "auto"
        vendorid = "0001"
        productid = "0000"
        bus = "001"

=====================================================
pi@susi:/etc/nut $ lsusb -v

Bus 001 Device 005: ID 0001:0000 Fry's Electronics MEC0003
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x0001 Fry's Electronics
  idProduct          0x0000 
  bcdDevice            1.00
  iManufacturer           1 MEC
  iProduct                2 MEC0003
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0022
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     624
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0

===================================================
pi@susi:/etc/nut $ sudo upsdrvctl -DDD start
Network UPS Tools - UPS driver controller 2.7.4
   0.000000	
If you're not a NUT core developer, chances are that you're told to enable debugging
to see why a driver isn't working for you. We're sorry for the confusion, but this is
the 'upsdrvctl' wrapper, not the driver you're interested in.

Below you'll find one or more lines starting with 'exec:' followed by an absolute
path to the driver binary and some command line option. This is what the driver
starts and you need to copy and paste that line and append the debug flags to that
line (less the 'exec:' prefix).

   0.001026	Starting UPS: susi
   0.001094	3 remaining attempts
   0.001133	exec:  /lib/nut/blazer_usb -a susi
Network UPS Tools - Megatec/Q1 protocol USB driver 0.12 (2.7.4)
Supported UPS detected with megatec protocol
Vendor information unavailable
No values provided for battery high/low voltages in ups.conf

Using 'guestimation' (low: 10.400000, high: 13.000000)!
Battery runtime will not be calculated (runtimecal not set)

===================================================
pi@susi:/etc/nut $ sudo /lib/nut/blazer_usb -a susi -DDD
Network UPS Tools - Megatec/Q1 protocol USB driver 0.12 (2.7.4)
   0.000000	debug level is '3'
   0.002148	Checking device (0001/0000) (001/005)
   0.004788	- VendorID: 0001
   0.005086	- ProductID: 0000
   0.005329	- Manufacturer: MEC
   0.005570	- Product: MEC0003
   0.005810	- Serial Number: unknown
   0.006050	- Bus: 001
   0.006289	- Device release number: 0100
   0.006526	Trying to match device
   0.006860	Device matches
   0.007156	nut_usb_set_altinterface: skipped usb_set_altinterface(udev, 0)
   0.007443	Trying megatec protocol...
   0.007705	send: Q1
   0.231596	received 47 (40)
   0.232189	read: (236.2 236.7 236.7 012 50.0 13.9 --.- 00001001
   0.232694	blazer_status: non numerical value [--.-]
   0.233152	Status read in 1 tries
   0.233507	Supported UPS detected with megatec protocol
   0.233868	send: F
   0.348098	received 22 (35)
   0.348683	read: #230.0 002 12.00 50.0
   0.349110	Ratings read in 1 tries
   0.349476	send: I
   0.393323	received 5 (35)
   0.393875	read: #  .
   0.394239	blazer_vendor: short reply
   0.394601	Vendor information read 1 failed
   0.394960	send: I
   0.439937	received 5 (35)
   0.440478	read: #  .
   0.440843	blazer_vendor: short reply
   0.441206	Vendor information read 2 failed
   0.441565	send: I
   0.485314	received 5 (35)
   0.485865	read: #  .

Any help appreciated

ripf77, 2024-03-28:
I am running a RASPBERRY 3B with
raspbian bullseye

jimklimov, 2024-03-28:
I did not quite understand the wording - was it working before the update? with same configuration? What was the NUT (and OS) version before the update? 2.7.4 is almost 8 years old...

Generally speaking, tons of unrelated devices use unregistered vendor/product IDs like 0000, 0001, ffff which indicate nothing about their capabilities. Yours at least seems to talk some dialect of Megatec Qx protocol family, so you should try different `protocol` and `subdriver` values.

It may help to get a newer NUT package or build your own using current sources.

ripf77, 2024-03-28:
jimklimov, thank you for the response. Sorry that I was not clear enough.
The make of the UPS is DIGITUS. Before performing the upgrade all worked fine with the same configuration. I have a second look-alike installation with the same UPS and another raspbian w/o upgrade and its still working.
upsc output here is
pi@babsi:~ $ upsc ups
Init SSL without certificate database
battery.charge: 100
battery.voltage: 13.70
battery.voltage.high: 13.00
battery.voltage.low: 10.40
battery.voltage.nominal: 12.0
device.type: ups
driver.name: blazer_usb
driver.parameter.bus: 001
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: 0000
driver.parameter.synchronous: no
driver.parameter.vendorid: 0001
driver.version: 2.7.4
driver.version.internal: 0.12
input.current.nominal: 2.0
input.frequency: 50.0
input.frequency.nominal: 50
input.voltage: 230.5
input.voltage.fault: 230.5
input.voltage.nominal: 230
output.voltage: 230.5
ups.beeper.status: enabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.load: 4
ups.productid: 0000
ups.status: OL
ups.type: offline / line interactive
ups.vendorid: 0001

But since I learned that the nut package that is delivered from the raspbian repositories is that old I may go for buildiing a new package. (not my favourite option)

ripf77, 2024-03-28:
I made my way through the build (in multiple attempts) and managed to get it going.
I have now installed 2.8.1 on a raspbian bullseye and can see my ups again. 

THANK YOU !

closed 2024-03-28 by ripf77