networkupstools/nut #2378: Support of linux hwmon (ina219)

jviki, 2024-03-28:
Would it be possible to integrate support for standard linux hwmon that can monitor power of the system? I am running a small system (baseboard: [CM4-POE-UPS-BASE](http://www.waveshare.com/wiki/CM4-POE-UPS-BASE)) with batteries that contains the [ina219](https://docs.kernel.org/hwmon/ina2xx.html) I2C device. I hoped that I can find some generic hwmon UPS-like support here but I didn't.

I've got a short bash script that can be used as a basis for a nut driver:

```
#! /bin/bash

find_ina219()
{
	for d in $(ls /sys/class/hwmon/); do
		path="/sys/class/hwmon/${d}/name"
		name=$(cat ${path})

		if [ "${name}" == ina219 ]; then
			echo "/sys/class/hwmon/${d}"
			return 0
		fi
	done

	return 1
}

ina219=$(find_ina219)

if [ "$?" != 0 ]; then
	echo "could not find ina219 device" >&2
	exit 1
fi

voltage=`cat ${ina219}/in1_input`
current=`cat ${ina219}/curr1_input`

echo "Status: "$(expr \( ${voltage} - 3000 \) / 12)" % (${voltage} mV, ${current} mA)"

if [ "${voltage}" -lt 3150 ] && [ "${current}" -gt 50 ]; then 
	echo "Low power"
	sync
fi
```

Quick test:
```
# status: OL, HB
$ ./check-ups.sh 
Status: 95 % (4140 mV, 0 mA)

# status: OB, DISCHRG (unplugged AC)
$ ./check-ups.sh 
Status: 90 % (4084 mV, 87 mA)
$ ./check-ups.sh 
Status: 89 % (4072 mV, 114 mA)

# status: OL, CHRG (plugged AC back)
$ ./check-ups.sh 
Status: 102 % (4224 mV, -150 mA)
```

> if the current is positive, it means the output current of the battery (which can be considered as the Raspberry Pi load current), and if the current is negative, it means the battery charging current.
> -- [Battery Level Detection](https://www.waveshare.com/wiki/CM4-POE-UPS-BASE#INA219_Battery_Level_Detection)

Where should I start when trying to integrate this?

jimklimov, 2024-03-29:
There are a few NUT drivers using the Linux i2c bus, a relatively recent one is pijuice. I suppose taking a look at those (and their integration to build system in makefiles) would be a helpful start.

jimklimov, 2024-08-04:
I wonder if the new driver (or its future iterations) could also talk to devices like "Waveshare UPS HAT (E) for Raspberry Pi 5/4/3B+ that takes four 21700 Lithium batteries, supports USB PD 3.0"
* https://www.cnx-software.com/2024/08/03/waveshare-ups-hat-e-for-raspberry-pi-5-4-3b-takes-four-21700-lithium-batteries-supports-usb-pd-3-0/ 
* https://www.waveshare.com/wiki/UPS_HAT_(E)

jviki, 2024-08-04:
It should, or at least I cannot see any reason why not... It's mostly about having the DTS right and loading the kernel driver.

I've got https://www.waveshare.com/wiki/UPS_HAT on the table and want to play with that as soon as I have some time.

jimklimov, 2024-08-05:
Sounds great! If there are any more PRs to add support, would be great to get them :) And then perhaps let WaveShare know (or even ask for support via more devices to test against)