networkupstools/nut #2381: Invalid signature on nut-2.8.2

atsampson, 2024-04-03:
The NUT 2.8.2 tarball is [signed](https://www.networkupstools.org/source/2.8/nut-2.8.2.tar.gz.sig) with key C8F2DB717B416C4DDAB2ED9BFFA7A68295C8BA96. This key is not in the maintainers' keyring that's linked from [the verification instructions](https://networkupstools.org/docs/user-manual.chunked/NUT_Security.html#verifySourceSig), so it cannot be verified.

jimklimov, 2024-04-03:
Thanks for the heads-up, I'll recheck which key carrier got used this time.

jimklimov, 2024-04-03:
It seems the key chains were updated on nut-source-archive but did not get published to "production" repo which GH renders websites from.

To confirm: did your gpg import only see two keys from Arnaud Quette?

jimklimov, 2024-04-03:
Cross-checking vs. older releases:

````
$ gpg --verify nut-2.8.1.tar.gz.sig
gpg: assuming signed data in 'nut-2.8.1.tar.gz'
gpg: Signature made 2023-10-31 23:30:24 +0100 CET
gpg:                using EDDSA key BFA06D7C653B64C11DFDAF0442061031267D11B1
gpg: Good signature from "Jim Klimov (Doing FOSS since last millennium) <jimklimov@gmail.com>" [unknown]
gpg:                 aka "Jim Klimov (Jenkins CI plugin contributor and maintainer) <jimklimov+jenkinsci@gmail.com>" [unknown]
gpg:                 aka "Jim Klimov (Network UPS Tools maintainer) <jimklimov+nut@gmail.com>" [unknown]
gpg:                 aka "Jim Klimov (Illumos-based OS collaborator) <jimklimov+illumos@gmail.com>" [unknown]
gpg: WARNING: This key is not certified with a trusted signature!
gpg:          There is no indication that the signature belongs to the owner.
Primary key fingerprint: B834 59F7 76B9 0224 988F  36C0 DE01 84DA 7043 DCF7
     Subkey fingerprint: BFA0 6D7C 653B 64C1 1DFD  AF04 4206 1031 267D 11B1
````

and

````
gpg --verify nut-2.8.0.tar.gz.sig
gpg: assuming signed data in 'nut-2.8.0.tar.gz'
gpg: Signature made 2023-06-01 00:10:16 +0200 CEST
gpg:                using EDDSA key BFA06D7C653B64C11DFDAF0442061031267D11B1
gpg: Good signature from "Jim Klimov (Doing FOSS since last millennium) <jimklimov@gmail.com>" [unknown]
gpg:                 aka "Jim Klimov (Jenkins CI plugin contributor and maintainer) <jimklimov+jenkinsci@gmail.com>" [unknown]
gpg:                 aka "Jim Klimov (Network UPS Tools maintainer) <jimklimov+nut@gmail.com>" [unknown]
gpg:                 aka "Jim Klimov (Illumos-based OS collaborator) <jimklimov+illumos@gmail.com>" [unknown]
gpg: WARNING: This key is not certified with a trusted signature!
gpg:          There is no indication that the signature belongs to the owner.
Primary key fingerprint: B834 59F7 76B9 0224 988F  36C0 DE01 84DA 7043 DCF7
     Subkey fingerprint: BFA0 6D7C 653B 64C1 1DFD  AF04 4206 1031 267D 11B1
````

but indeed, still

````
$ gpg --verify nut-2.8.2.tar.gz.sig
gpg: assuming signed data in 'nut-2.8.2.tar.gz'
gpg: Signature made 2024-04-02 03:25:26 +0200 CEST
gpg:                using RSA key C8F2DB717B416C4DDAB2ED9BFFA7A68295C8BA96
gpg: BAD signature from "Jim Klimov (Doing FOSS since last millennium) <jimklimov@gmail.com>" [unknown]
````

so probably got the wrong machine signing it.

jimklimov, 2024-04-04:
Oddly, it seems the key factually used is a sub-key listed in the published chain. Maybe that way of import falters somehow.

It also seems I can't pick a way to sign with the main key itself - as the previous releases seem to have been done, although the sub keys were made early on (before the other 2.8.x sigs) :\

Gotta scratch my head some more about this a bit later...

jimklimov, 2024-04-04:
````
gpg --list-packets < nut-key.gpg
# off=0 ctb=99 tag=6 hlen=3 plen=418
:public key packet:
        version 4, algo 17, created 1026746276, expires 0
        pkey[0]: [1024 bits]
        pkey[1]: [160 bits]
        pkey[2]: [1022 bits]
        pkey[3]: [1024 bits]
        keyid: DB6414CA204DDF1B
# off=421 ctb=b4 tag=13 hlen=2 plen=37
:user ID packet: "Arnaud Quette <aquette.dev@gmail.com>"
# off=460 ctb=88 tag=2 hlen=2 plen=94
:signature packet: algo 17, keyid DB6414CA204DDF1B
        version 4, created 1127129449, md5len 0, sigclass 0x13
        digest algo 2, begin of digest e0 2c
        hashed subpkt 2 len 4 (sig created 2005-09-19)
        hashed subpkt 27 len 1 (key flags: 03)
        hashed subpkt 11 len 5 (pref-sym-algos: 9 8 7 3 2)
        hashed subpkt 21 len 2 (pref-hash-algos: 2 3)
        hashed subpkt 22 len 2 (pref-zip-algos: 2 1)
        hashed subpkt 30 len 1 (features: 01)
        hashed subpkt 23 len 1 (keyserver preferences: 80)
        subpkt 16 len 8 (issuer key ID DB6414CA204DDF1B)
        data: [155 bits]
        data: [160 bits]
# off=556 ctb=89 tag=2 hlen=3 plen=540
:signature packet: algo 1, keyid 3A936196C095D941
        version 4, created 1260809411, md5len 0, sigclass 0x10
        digest algo 10, begin of digest 15 94
        hashed subpkt 2 len 4 (sig created 2009-12-14)
        subpkt 16 len 8 (issuer key ID 3A936196C095D941)
        data: [4096 bits]
# off=1099 ctb=b4 tag=13 hlen=2 plen=34
:user ID packet: "Arnaud Quette <aquette@debian.org>"
# off=1135 ctb=88 tag=2 hlen=2 plen=97
:signature packet: algo 17, keyid DB6414CA204DDF1B
        version 4, created 1178388895, md5len 0, sigclass 0x13
        digest algo 2, begin of digest 9f 5f
        hashed subpkt 27 len 1 (key flags: 03)
        hashed subpkt 11 len 5 (pref-sym-algos: 9 8 7 3 2)
        hashed subpkt 21 len 2 (pref-hash-algos: 2 3)
        hashed subpkt 22 len 2 (pref-zip-algos: 2 1)
        hashed subpkt 30 len 1 (features: 01)
        hashed subpkt 23 len 1 (keyserver preferences: 80)
        hashed subpkt 2 len 4 (sig created 2007-05-05)
        hashed subpkt 25 len 1 (primary user ID)
        subpkt 16 len 8 (issuer key ID DB6414CA204DDF1B)
        data: [159 bits]
        data: [157 bits]
# off=1234 ctb=88 tag=2 hlen=2 plen=94
:signature packet: algo 17, keyid DB6414CA204DDF1B
        version 4, created 1065988349, md5len 0, sigclass 0x13
        digest algo 2, begin of digest 69 d1
        hashed subpkt 2 len 4 (sig created 2003-10-12)
        hashed subpkt 27 len 1 (key flags: 03)
        hashed subpkt 11 len 5 (pref-sym-algos: 9 8 7 3 2)
        hashed subpkt 21 len 2 (pref-hash-algos: 2 3)
        hashed subpkt 22 len 2 (pref-zip-algos: 2 1)
        hashed subpkt 30 len 1 (features: 01)
        hashed subpkt 23 len 1 (keyserver preferences: 80)
        subpkt 16 len 8 (issuer key ID DB6414CA204DDF1B)
        data: [156 bits]
        data: [160 bits]
# off=1330 ctb=89 tag=2 hlen=3 plen=540
:signature packet: algo 1, keyid 3A936196C095D941
        version 4, created 1260809411, md5len 0, sigclass 0x10
        digest algo 10, begin of digest c9 05
        hashed subpkt 2 len 4 (sig created 2009-12-14)
        subpkt 16 len 8 (issuer key ID 3A936196C095D941)
[... 373 lines cut ...]
# off=18320 ctb=89 tag=2 hlen=3 plen=1132
:signature packet: algo 1, keyid DE0184DA7043DCF7
        version 4, created 1685536245, md5len 0, sigclass 0x18
        digest algo 10, begin of digest cc 99
        hashed subpkt 33 len 21 (issuer fpr v4 B83459F776B90224988F36C0DE0184DA7043DCF7)
        hashed subpkt 2 len 4 (sig created 2023-05-31)
        hashed subpkt 27 len 1 (key flags: 02)
        subpkt 16 len 8 (issuer key ID DE0184DA7043DCF7)
        subpkt 32 len 563 (signature: v4, class 0x19, algo 1, digest algo 10)
        data: [4093 bits]
jim@pve:~/nut-website/source$ cd /tmp
jim@pve:/tmp$ gpg -s -u 0xDE0184DA7043DCF7 --default-key 0xDE0184DA7043DCF7 -bv nut-2.8.2.tar.gz
gpg: using subkey FFA7A68295C8BA96 instead of primary key DE0184DA7043DCF7
File 'nut-2.8.2.tar.gz.sig' exists. Overwrite? (y/N) y
gpg: writing to 'nut-2.8.2.tar.gz.sig'
gpg: RSA/SHA512 signature from: "FFA7A68295C8BA96 Jim Klimov (Doing FOSS since last millennium) <jimklimov@gmail.com>"
jim@pve:/tmp$ gpg --verify *sig
gpg: assuming signed data in 'nut-2.8.2.tar.gz'
gpg: Signature made Thu Apr  4 02:48:29 2024 CEST
gpg:                using RSA key C8F2DB717B416C4DDAB2ED9BFFA7A68295C8BA96
gpg: Good signature from "Jim Klimov (Doing FOSS since last millennium) <jimklimov@gmail.com>" [ultimate]
gpg:                 aka "Jim Klimov (Jenkins CI plugin contributor and maintainer) <jimklimov+jenkinsci@gmail.com>" [ultimate]
gpg:                 aka "Jim Klimov (Network UPS Tools maintainer) <jimklimov+nut@gmail.com>" [ultimate]
gpg:                 aka "Jim Klimov (Illumos-based OS collaborator) <jimklimov+illumos@gmail.com>" [ultimate]
````

jimklimov, 2024-04-04:
Note the short keyids are tails fo the long, so `FFA7A68295C8BA96` is `C8F2DB717B416C4DDAB2ED9BFFA7A68295C8BA96`.

arjendekorte, 2024-04-04:
Where can I find the new signature file?

arjendekorte, 2024-04-04:
In that case, I'm probably looking in the wrong location. In [https://github.com/networkupstools/nut/releases](url) I still see the (wrong) signature from three days ago. Is it located elsewhere?

jimklimov, 2024-04-04:
Ah, you mean the very new one, per "screenshots" above? It was not published yet, as it used the same subkey. Maybe I should focus on documenting some way to get subkeys trusted - they are mentioned in the (published) nut-key gpg chain anyhow, rather than finding a way to re-sign with master key.

On the road this week though, so it can take some time to materialize.

atsampson, 2024-04-04:
Looking more carefully at the output above, I think I misdiagnosed the problem originally - sorry! Signing with the subkey should work - 2.8.0 and 2.8.1 used a subkey and those verify fine:

```
$ gpg -vv --always-trust --verify nut-2.8.1.tar.gz.sig 
# off=0 ctb=88 tag=2 hlen=2 plen=117
:signature packet: algo 22, keyid 42061031267D11B1
	version 4, created 1698791424, md5len 0, sigclass 0x00
	digest algo 8, begin of digest 46 d9
	hashed subpkt 33 len 21 (issuer fpr v4 BFA06D7C653B64C11DFDAF0442061031267D11B1)
	hashed subpkt 2 len 4 (sig created 2023-10-31)
	subpkt 16 len 8 (issuer key ID 42061031267D11B1)
	data: [254 bits]
	data: [255 bits]
gpg: assuming signed data in 'nut-2.8.1.tar.gz'
gpg: Signature made Tue 31 Oct 2023 22:30:24 GMT
gpg:                using EDDSA key BFA06D7C653B64C11DFDAF0442061031267D11B1
gpg: using subkey 42061031267D11B1 instead of primary key DE0184DA7043DCF7
gpg: using subkey 42061031267D11B1 instead of primary key DE0184DA7043DCF7
gpg: Good signature from "Jim Klimov (Doing FOSS since last millennium) <jimklimov@gmail.com>" [unknown]
gpg:                 aka "Jim Klimov (Jenkins CI plugin contributor and maintainer) <jimklimov+jenkinsci@gmail.com>" [unknown]
gpg:                 aka "Jim Klimov (Network UPS Tools maintainer) <jimklimov+nut@gmail.com>" [unknown]
gpg:                 aka "Jim Klimov (Illumos-based OS collaborator) <jimklimov+illumos@gmail.com>" [unknown]
gpg: using subkey 42061031267D11B1 instead of primary key DE0184DA7043DCF7
gpg: WARNING: Using untrusted key!
gpg: binary signature, digest algorithm SHA256, key algorithm ed25519
```

But there's something about the 2.8.2 signature that GPG doesn't like (I tried Debian's GPG 2.2.27 and GPG 2.4.5 built from source):

```
$ gpg -vv --always-trust --verify nut-2.8.2.tar.gz.sig 
# off=0 ctb=89 tag=2 hlen=3 plen=563
:signature packet: algo 1, keyid FFA7A68295C8BA96
	version 4, created 1712021126, md5len 0, sigclass 0x00
	digest algo 10, begin of digest cd 97
	hashed subpkt 33 len 21 (issuer fpr v4 C8F2DB717B416C4DDAB2ED9BFFA7A68295C8BA96)
	hashed subpkt 2 len 4 (sig created 2024-04-02)
	subpkt 16 len 8 (issuer key ID FFA7A68295C8BA96)
	data: [4095 bits]
gpg: assuming signed data in 'nut-2.8.2.tar.gz'
gpg: Signature made Tue 02 Apr 2024 02:25:26 BST
gpg:                using RSA key C8F2DB717B416C4DDAB2ED9BFFA7A68295C8BA96
gpg: using subkey FFA7A68295C8BA96 instead of primary key DE0184DA7043DCF7
gpg: using subkey FFA7A68295C8BA96 instead of primary key DE0184DA7043DCF7
gpg: BAD signature from "Jim Klimov (Doing FOSS since last millennium) <jimklimov@gmail.com>" [unknown]
gpg: binary signature, digest algorithm SHA512, key algorithm rsa4096
```

Note the "BAD signature" there, which you also got when you tried to verify it. When you regenerated the sig file with the same subkey and algorithms, you got "Good signature" instead...

jimklimov, 2024-04-06:
Ha, thanks for the re-check. My rough criterion was that the older signatures were 120 or so bytes long, and 2.8.2's keeps being 560 or so - thus the rough feeling was that it is still not what we wanted.

It may be that the bad/good detection differs on my computers/containers that only did the `gpg --fetch-keys ...` into a clean slate as per NUT docs, vs. those that actually have the keys locally and/or imported them back from the internet key shares I uploaded to a year ago or so.

jimklimov, 2024-04-06:
Updated in nut-source-archive, rendered website, and github release artifacts.

jimklimov, 2024-04-06:
````
~/tmp $ gpg --verify nut-2.8.2.tar.gz.sig
gpg: assuming signed data in 'nut-2.8.2.tar.gz'
gpg: Signature made 2024-04-01 22:20:21 +0200 CEST
gpg:                using EDDSA key BFA06D7C653B64C11DFDAF0442061031267D11B1
gpg: Good signature from "Jim Klimov (Doing FOSS since last millennium) <jimklimov@gmail.com>" [unknown]
gpg:                 aka "Jim Klimov (Jenkins CI plugin contributor and maintainer) <jimklimov+jenkinsci@gmail.com>" [unknown]
gpg:                 aka "Jim Klimov (Network UPS Tools maintainer) <jimklimov+nut@gmail.com>" [unknown]
gpg:                 aka "Jim Klimov (Illumos-based OS collaborator) <jimklimov+illumos@gmail.com>" [unknown]
gpg: WARNING: This key is not certified with a trusted signature!
gpg:          There is no indication that the signature belongs to the owner.
Primary key fingerprint: B834 59F7 76B9 0224 988F  36C0 DE01 84DA 7043 DCF7
     Subkey fingerprint: BFA0 6D7C 653B 64C1 1DFD  AF04 4206 1031 267D 11B1
````

jimklimov, 2024-04-06:
Not quite sure what differed on this workstation - it also has a subkey of its own and used it:
````
LD_PRELOAD=/usr/lib/x86_64-linux-gnu/faketime/libfaketime.so.1 FAKETIME="-5d" gpg -svb nut-2.8.2.tar.gz
gpg: using pgp trust model
gpg: using subkey 42061031267D11B1 instead of primary key DE0184DA7043DCF7
gpg: writing to 'nut-2.8.2.tar.gz.sig'
gpg: EDDSA/SHA256 signature from: "42061031267D11B1 Jim Klimov (Doing FOSS since last millennium) <jimklimov@gmail.com>"
````

The ring is generally similar to all other workstations:
````
gpg -K
/home/jim/.gnupg/pubring.kbx
----------------------------
sec#  rsa4096 2023-05-31 [SC]
      B83459F776B90224988F36C0DE0184DA7043DCF7
uid           [ unknown] Jim Klimov (Doing FOSS since last millennium) <jimklimov@gmail.com>
uid           [ unknown] Jim Klimov (Jenkins CI plugin contributor and maintainer) <jimklimov+jenkinsci@gmail.com>
uid           [ unknown] Jim Klimov (Network UPS Tools maintainer) <jimklimov+nut@gmail.com>
uid           [ unknown] Jim Klimov (Illumos-based OS collaborator) <jimklimov+illumos@gmail.com>
ssb#  rsa4096 2023-05-31 [S]
ssb#  rsa4096 2023-05-31 [S]
ssb#  rsa4096 2023-05-31 [E]
ssb#  ed25519 2023-05-31 [S]
ssb   ed25519 2023-05-31 [S]
````

But for whatever reason, the file generated here was 119 bytes like for earlier releases.

FWIW, gpg seems the same on both boxes:
````
:; gpg --version
gpg (GnuPG) 2.2.27
libgcrypt 1.9.4
````

jimklimov, 2024-04-06:
@arjendekorte @atsampson : Can you please re-check if https://networkupstools.org/source/2.8/nut-2.8.2.tar.gz.sig now fits https://networkupstools.org/source/2.8/nut-2.8.2.tar.gz for you per https://networkupstools.org/docs/user-manual.chunked/NUT_Security.html#verifySourceSig instructions, so this issue can be closed?

PS: Hope to learn more about this before future release cycles :)

jimklimov, 2024-04-06:
@arjendekorte : while our paths have crossed, would you mind adding your older e-mail to github known aliases for your account, so that early NUT commits are properly attributed, please? :)

As a random example, see e.g. https://github.com/networkupstools/nut/commit/390da12bdc0fbd08836d8e41ead482d08fb2bd7c

And how come you don't have a nut repo fork? :)

arjendekorte, 2024-04-06:
> @arjendekorte @atsampson : Can you please re-check if https://networkupstools.org/source/2.8/nut-2.8.2.tar.gz.sig now fits https://networkupstools.org/source/2.8/nut-2.8.2.tar.gz for you per [...]

New signature validates fine, so problem is resolved

atsampson, 2024-04-06:
Yep, that validates fine for me now with GPG 2.2.17 and 2.4.5. The new signature uses SHA256/ed25519 like the 2.8.0 and 2.8.1 signatures, rather than SHA512/rsa4096 like the broken one. Thanks!

closed 2024-04-06 by jimklimov

jimklimov, 2024-04-06:
Huh, so one trick to force use of desired key is to suffix it with `!`
And to look up the IDs (not shown by default in current versions), colons are the way: `gpg --list-secret-keys --with-colons | grep ssb`

* How gpg ignores the requested ID:
```
:; gpg -sv -u '0x5484E1C503BCDC6E' -b nut-2.8.2.tar.gz
gpg: using subkey FFA7A68295C8BA96 instead of primary key DE0184DA7043DCF7
...
```

* How it finally does what its yelled to do:
```
:; gpg -sv -u '0x5484E1C503BCDC6E!' -b nut-2.8.2.tar.gz
gpg: using subkey 5484E1C503BCDC6E instead of primary key DE0184DA7043DCF7
...
```