networkupstools/nut #2390: Driver riello_ser have the same issue #1685 of riello_usb about battery capacity and runtime

ramponis, 2024-04-10:
This issue
https://github.com/networkupstools/nut/issues/1685

was already solved with localcalculation feature

Is it possible to integrate the localcalculation functionality also in serial driver?

Thank you

ramponis, 2024-04-11:
I don't think to be able to modify the source by myself.
If you can do it, I will try to build the source and test it.
Then I will give you a feedback.

Thank you

jimklimov, 2024-04-11:
Give this a shot: #2391

When following the https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests instructions, be sure to check out the PR source branch:
````
:; git clone https://github.com/jimklimov/nut -b issue-2390
:; cd nut
...
````

ramponis, 2024-04-11:
It works!!!

Thank you

jimklimov, 2024-04-11:
Super, thanks!

PR #2391 (2024-04-11): `riello_ser`: port `localcalculation` feature from `riello_usb`
Import from sibling (Riello/USB) driver the ability to guesstimate battery charge and runtime from voltage. Same caveats apply, including expected obsoletion in favor of `runtimecal` at some later point in time.

Closes: #2390 

Follows up from #1692
closed 2024-04-11 by jimklimov