networkupstools/nut #2392: Driver `main.c`: use Unix socket protocol to kill off "Duplicate other instance"

jimklimov, 2024-04-12:
Currently we use PID files where available, or skip that if not (and possibly fail to start the new driver program as a result, if the device is held by the older copy). Follows up from investigation for #2384

Enhance the `upsdrvquery*()` based protocol with an `INSTCMD driver.exit` for similar effect as a SIGTERM from the sibling program, where we do not have a caller's explicit request for the old driver instance to try reloading and keep running (and/or exit if reload failed so a systemd unit would restart it) - so by default the caller wants its shiny new instance of the driver program to remain the one running after the dust settles.

OS based privileges for signalling or accessing the Unix socket (or Windows pipe) are deemed equivalently safe for these two approaches.

CC @clepple @aquette : WDYT?

clepple, 2024-04-13:
The driver socket protocol is not something I ever did much with, so I am probably not the best one to review this.

jimklimov, 2024-04-15:
Well, at least regarding general architectural decision: did I miss any potential flaws exposing such a command via local Unix socket protocol as compared to signal-sending? :)

PR #2393 (2024-04-16): Implement an `INSTCMD driver.exit` as alternative to PID files and signals
When a NUT driver starts, by default it checks the PID file (if present) to kill off its earlier running instance, if any, to have the device connection available and no logical conflicts at run-time.

Problems can arise when the PID file is not available (e.g. not saved due to service unit set-up). Technically we have a fallback capability of checking for the old driver instance via Unix socket made for driver-server communications, and since #1903 and numerous later issues and PRs we can have driver instances (and `up [... cut]
closed 2024-04-16 by jimklimov