networkupstools/nut #2395: UPS Visench C1K not support

Unreal-System, 2024-04-13:
I have a Visench C1K UPS that is unsupported, and its control chip comes from Richcomm but is not compatible with the existing Richcomm protocol.

The official software was last released in 2006 (PowerManager II).

Because its USB port is actually a CH340 USB-to-serial port chip from WCH, the communication protocol provided by both its COM port and USB port is the same.

I have reverse-engineered its communication protocol using tools.

I don’t know how to write C code, so I hope someone can help write a driver to adapt it (this should be able to adapt all UPSs that use the PowerManager II management software from Richcomm).”

[image]

Unreal-System, 2024-04-14:
root@user-virtual-machine:/lib/nut# ./nutdrv_qx -a c1k -DDDDD
Network UPS Tools - Generic Q* USB/Serial driver 0.28 (2.7.4)
USB communication driver 0.33
   0.000000	debug level is '5'
   0.001336	upsdrv_initups...
   0.034524	Checking device (1D6B/0002) (002/001)
   0.034552	- VendorID: 1d6b
   0.034558	- ProductID: 0002
   0.034562	- Manufacturer: unknown
   0.034571	- Product: unknown
   0.034575	- Serial Number: unknown
   0.034579	- Bus: 002
   0.034588	- Device release number: 0605
   0.034593	Trying to match device
   0.034598	Device does not match - skipping
   0.034607	Checking device (1A86/7523) (001/004)
   0.034621	- VendorID: 1a86
   0.034627	- ProductID: 7523
   0.034630	- Manufacturer: unknown
   0.034634	- Product: unknown
   0.034642	- Serial Number: unknown
   0.034647	- Bus: 001
   0.034652	- Device release number: 0264
   0.034657	Trying to match device
   0.034664	Device does not match - skipping
   0.034671	Checking device (0E0F/0002) (001/003)
   0.034685	- VendorID: 0e0f
   0.034693	- ProductID: 0002
   0.034697	- Manufacturer: unknown
   0.034701	- Product: unknown
   0.034708	- Serial Number: unknown
   0.034716	- Bus: 001
   0.034723	- Device release number: 0100
   0.034729	Trying to match device
   0.034734	Device does not match - skipping
   0.034740	Checking device (0E0F/0003) (001/002)
   0.034757	- VendorID: 0e0f
   0.034765	- ProductID: 0003
   0.034769	- Manufacturer: unknown
   0.034772	- Product: unknown
   0.034777	- Serial Number: unknown
   0.034784	- Bus: 001
   0.034789	- Device release number: 0103
   0.034797	Trying to match device
   0.034802	Device does not match - skipping
   0.034807	Checking device (1D6B/0001) (001/001)
   0.034822	- VendorID: 1d6b
   0.034826	- ProductID: 0001
   0.034830	- Manufacturer: unknown
   0.034834	- Product: unknown
   0.034841	- Serial Number: unknown
   0.034850	- Bus: 001
   0.034856	- Device release number: 0605
   0.034863	Trying to match device
   0.034873	Device does not match - skipping
   0.034880	No appropriate HID device found
   0.034886	No supported devices found. Please check your device availability with 'lsusb'
and make sure you have an up-to-date version of NUT. If this does not help,
try running the driver with at least 'subdriver', 'vendorid' and 'productid'
options specified. Please refer to the man page for details about these options
(man 8 nutdrv_qx).

root@user-virtual-machine:/lib/nut# lsusb
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 004: ID 1a86:7523 QinHeng Electronics CH340 serial converter
Bus 001 Device 003: ID 0e0f:0002 VMware, Inc. Virtual USB Hub
Bus 001 Device 002: ID 0e0f:0003 VMware, Inc. Virtual Mouse
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

jimklimov, 2024-04-14:
On one hand, this is a rather ancient version of NUT. On another, gotta check if 1a86:7523 is known to NUT. You can try passing `-x vendorid=1a86 -x productid=7523` to the driver so it knows to try grabbing this device even if the ID is not compiled in as "known".

Unreal-System, 2024-04-14:
root@user-virtual-machine:/lib/nut# ./nutdrv_qx -a c1k -DDDDDD -x vendorid=1a86 -x productid=7523
Network UPS Tools - Generic Q* USB/Serial driver 0.28 (2.7.4)
USB communication driver 0.33
   0.000000	send_to_all: SETINFO driver.parameter.vendorid "1a86"
   0.000039	send_to_all: SETINFO driver.parameter.productid "7523"
   0.000054	debug level is '6'
   0.001053	upsdrv_initups...
   0.046753	Checking device (1D6B/0002) (002/001)
   0.062561	- VendorID: 1d6b
   0.062588	- ProductID: 0002
   0.062596	- Manufacturer: unknown
   0.062603	- Product: unknown
   0.062627	- Serial Number: unknown
   0.062636	- Bus: 002
   0.062648	- Device release number: 0605
   0.062655	Trying to match device
   0.062693	Device does not match - skipping
   0.078325	Checking device (1A86/7523) (001/004)
   0.078361	- VendorID: 1a86
   0.078370	- ProductID: 7523
   0.078377	- Manufacturer: unknown
   0.078385	- Product: unknown
   0.078390	- Serial Number: unknown
   0.078395	- Bus: 001
   0.078407	- Device release number: 0264
   0.078419	Trying to match device
   0.078446	Device does not match - skipping
   0.078454	Checking device (0E0F/0002) (001/003)
   0.078471	- VendorID: 0e0f
   0.078478	- ProductID: 0002
   0.078488	- Manufacturer: unknown
   0.078493	- Product: unknown
   0.078497	- Serial Number: unknown
   0.078502	- Bus: 001
   0.078512	- Device release number: 0100
   0.078517	Trying to match device
   0.078522	Device does not match - skipping
   0.078530	Checking device (0E0F/0003) (001/002)
   0.078546	- VendorID: 0e0f
   0.078554	- ProductID: 0003
   0.078558	- Manufacturer: unknown
   0.078562	- Product: unknown
   0.078568	- Serial Number: unknown
   0.078575	- Bus: 001
   0.078579	- Device release number: 0103
   0.078584	Trying to match device
   0.078589	Device does not match - skipping
   0.078598	Checking device (1D6B/0001) (001/001)
   0.078614	- VendorID: 1d6b
   0.078619	- ProductID: 0001
   0.078623	- Manufacturer: unknown
   0.078628	- Product: unknown
   0.078638	- Serial Number: unknown
   0.078642	- Bus: 001
   0.078646	- Device release number: 0605
   0.078651	Trying to match device
   0.078661	Device does not match - skipping
   0.078671	No appropriate HID device found
   0.078676	No supported devices found. Please check your device availability with 'lsusb'
and make sure you have an up-to-date version of NUT. If this does not help,
try running the driver with at least 'subdriver', 'vendorid' and 'productid'
options specified. Please refer to the man page for details about these options
(man 8 nutdrv_qx).

jimklimov, 2024-04-15:
Hm, at least I see it is trying, and the ID's are correct:
````
0.000000 send_to_all: SETINFO driver.parameter.vendorid "1a86"
0.000039 send_to_all: SETINFO driver.parameter.productid "7523"
...
0.078325 Checking device (1A86/7523) (001/004)
0.078361 - VendorID: 1a86
0.078370 - ProductID: 7523
0.078377 - Manufacturer: unknown
0.078385 - Product: unknown
0.078390 - Serial Number: unknown
0.078395 - Bus: 001
0.078407 - Device release number: 0264
0.078419 Trying to match device
0.078446 Device does not match - skipping
````

If nothing else, the newer NUT versions should log their device matching attempts in more detail, so we could see why it decides this one can not be handled - e.g. "tried such and such protocol and subdriver, got some response that I could not parse", and maybe go from there if it is a Qx variant.

Looking at IDs, the only near match in current NUT sources is `1a86:5523` seen on devices supported by `huawei-ups2000` driver (added in NUT v2.8.0 and improved a few times later) - so maybe that is something to try as well.

Please check https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests about installing dependencies and custom-building NUT and running the new driver program from such build workspace. Either it would help directly (if new drivers work for the device), or would be a stepping stone to try tweaking the sources and iterating until you make a working driver :)

Unreal-System, 2024-04-15:
```
root@user-virtual-machine:/lib/nut# ls -l /dev/ttyCH341USB0 
crwxrwxrwx 1 nut dialout 169, 0  4月 15 19:00 /dev/ttyCH341USB0
root@user-virtual-machine:/lib/nut# ./nutdrv_qx -DDDDDD -a c1k -x vendorid=1a86 -x productid=7523
   0.000000	[D5] send_to_all: SETINFO driver.state "init.starting"
Network UPS Tools - Generic Q* USB/Serial driver 0.36 (2.8.2)
USB communication driver (libusb 0.1) 0.47
   0.000073	[D1] upsdrv_makevartable...
   0.000092	[D5] send_to_all: SETINFO driver.version.usb "libusb-0.1 (or compat)"
   0.000101	[D1] Using USB implementation: libusb-0.1 (or compat)
   0.000196	[D5] do_upsconf_args: confupsname=c1k, var=driver, val=nutdrv_qx
   0.000206	[D5] do_upsconf_args: call main_arg()
   0.000210	[D3] main_arg: var='driver' val='nutdrv_qx'
   0.000223	[D5] do_upsconf_args: not a main_arg()
   0.000228	[D5] do_upsconf_args: this is a 'driver' setting, may we proceed?
   0.000236	[D6] testval_reloadable: var=driver, oldval=nutdrv_qx, newval=nutdrv_qx, reloadable=0, reload_flag=0
   0.000240	[D6] testval_reloadable: verdict for (re)loading var=driver value: -1
   0.000245	[D5] do_upsconf_args: 'driver' setting already applied with this value
   0.000251	[D5] do_upsconf_args: confupsname=c1k, var=port, val=auto
   0.000254	[D5] do_upsconf_args: call main_arg()
   0.000258	[D3] main_arg: var='port' val='auto'
   0.000262	[D6] testinfo_reloadable: var=port, infoname=driver.parameter.port, newval=auto, reloadable=0, reload_flag=0
   0.000268	[D6] testinfo_reloadable: verdict for (re)loading var=port value: 1
   0.000273	[D5] send_to_all: SETINFO driver.parameter.port "auto"
   0.000288	[D3] main_arg: var='vendorid' val='1a86'
   0.000297	[D5] send_to_all: SETINFO driver.parameter.vendorid "1a86"
   0.000301	[D3] main_arg: var='productid' val='7523'
   0.000305	[D5] send_to_all: SETINFO driver.parameter.productid "7523"
   0.000309	[D1] Network UPS Tools version 2.8.2 (release/snapshot of 2.8.2) built with gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0 and configured with flags: --with-modbus --with-user=nut --with-group=nut --datadir=/usr/share/nut --sysconfdir=/etc/nut --includedir=/usr/include --with-statepath=/run/nut --with-pidpath=/run/nut --with-altpidpath=/run/nut --with-drvpath=/lib/nut --with-udev-dir=/lib/udev --with-systemdsystemunitdir=/lib/systemd/system --with-systemdshutdowndir=/lib/systemd/system-shutdown
   0.000317	[D1] debug level is '6'
   0.000328	[D5] send_to_all: SETINFO driver.debug "6"
   0.000338	[D5] send_to_all: SETFLAGS driver.debug RW NUMBER
   0.001156	[D1] Succeeded to become_user(nut): now UID=130 GID=138
   0.001179	[D5] send_to_all: SETINFO device.type "ups"
   0.001188	[D5] send_to_all: SETINFO driver.state "init.device"
   0.001192	[D1] upsdrv_initups...
   0.046341	[D3] usb_busses=0x64da9e3901c0
   0.046373	[D2] Checking device (1D6B/0002) (002/001)
   0.062345	[D1] libusb_open get iManufacturer failed, retrying...
   0.062365	[D1] libusb_open get iManufacturer failed, retrying...
   0.062370	[D1] libusb_open get iManufacturer failed, retrying...
   0.062378	[D1] libusb_open get iProduct failed, retrying...
   0.062386	[D1] libusb_open get iProduct failed, retrying...
   0.062392	[D1] libusb_open get iProduct failed, retrying...
   0.062396	[D1] libusb_open get iSerialNumber failed, retrying...
   0.062401	[D1] libusb_open get iSerialNumber failed, retrying...
   0.062405	[D1] libusb_open get iSerialNumber failed, retrying...
   0.062411	[D2] - VendorID: 1d6b
   0.062415	[D2] - ProductID: 0002
   0.062420	[D2] - Manufacturer: unknown
   0.062424	[D2] - Product: unknown
   0.062427	[D2] - Serial Number: unknown
   0.062434	[D2] - Bus: 002
   0.062443	[D2] - Device: 001
   0.062448	[D2] - Device release number: 0605
   0.062451	[D2] Trying to match device
   0.062456	[D3] match_function_regex: matching a device...
   0.062482	[D2] match_function_regex: failed match of VendorID: 1d6b
   0.062488	[D2] Device does not match - skipping
   0.062498	[D2] Checking device (1A86/7523) (001/004)
   0.062526	[D1] libusb_open get iProduct failed, retrying...
   0.062534	[D1] libusb_open get iProduct failed, retrying...
   0.062543	[D1] libusb_open get iProduct failed, retrying...
   0.062548	[D2] - VendorID: 1a86
   0.062552	[D2] - ProductID: 7523
   0.062555	[D2] - Manufacturer: unknown
   0.062564	[D2] - Product: unknown
   0.062573	[D2] - Serial Number: unknown
   0.062578	[D2] - Bus: 001
   0.062581	[D2] - Device: 004
[... 95 lines cut ...]
   0.000398	[D1] debug level is '6'
   0.000405	[D5] send_to_all: SETINFO driver.debug "6"
   0.000419	[D5] send_to_all: SETFLAGS driver.debug RW NUMBER
   0.001670	[D1] Succeeded to become_user(nut): now UID=130 GID=138
   0.001694	[D5] send_to_all: SETINFO device.type "ups"
   0.001704	[D5] send_to_all: SETINFO driver.state "init.device"
   0.001712	[D2] upsdrv_initups
   0.731487	[D5] send_to_all: SETINFO driver.state "init.quiet"
   0.731513	[D5] send_to_all: SETINFO driver.version "2.8.2"
   0.731517	[D5] send_to_all: SETINFO driver.version.internal "0.06"
   0.731522	[D5] send_to_all: SETINFO driver.name "huawei-ups2000"
   0.731525	[D5] send_to_all: SETINFO driver.state "init.info"
   0.731528	[D2] upsdrv_initinfo
   0.731532	[D2] ser_send_buf
   2.132752	unable to read from serial port /dev/ttyCH341USB0, retry...
   2.132775	[D2] ser_send_buf
   3.534019	unable to read from serial port /dev/ttyCH341USB0, retry...
   3.534044	[D2] ser_send_buf
   4.935284	unable to read from serial port /dev/ttyCH341USB0, retry...
   4.935304	unable to read from serial port /dev/ttyCH341USB0!
   4.935315	[D5] send_to_all: SETINFO driver.state "cleanup.upsdrv"
   4.938056	[D5] send_to_all: SETINFO driver.state "cleanup.exit"
   4.938073	upsnotify: failed to notify about state 4: no notification tech defined, will not spam more about it

```

jimklimov, 2024-04-15:
Just in case, does `/dev/ttyCH341USB0` exist in your filesystem? I'm not sure how its OS driver would call the devfs node...

Unreal-System, 2024-04-15:
The ttyCH341USB0 exists, and its driver uses the one officially compiled from
https://www.wch-ic.com/downloads/CH341SER_LINUX_ZIP.html

jimklimov, 2024-04-15:
Hm, given how `libusb_open get iProduct failed, retrying...` and it still fails to match the devices (`Device does not match - skipping` without much detail), and how the IDs are not known to NUT so not added to its udev/upower/... rules, I wonder if the FS permissions to the devfs node are the additional problem.

Can you please retry with `-x user=root` added to the command line (or `ups.conf` section, without the `-x` as just `user=root`) to work around that bit - so the driver would not drop privileges before work?

Unreal-System, 2024-04-15:
```
root@user-virtual-machine:/lib/nut# /lib/nut/huawei-ups2000 -DDDDDD -a c1k -x user=root
   0.000000	[D5] send_to_all: SETINFO driver.state "init.starting"
Network UPS Tools - NUT Huawei UPS2000 (1kVA-3kVA) RS-232 Modbus driver 0.06 (2.8.2)
Warning: This is an experimental driver.
Some features may not function correctly.

   0.000260	[D5] do_upsconf_args: confupsname=c1k, var=driver, val=huawei-ups2000
   0.000266	[D5] do_upsconf_args: call main_arg()
   0.000277	[D3] main_arg: var='driver' val='huawei-ups2000'
   0.000284	[D5] do_upsconf_args: not a main_arg()
   0.000290	[D5] do_upsconf_args: this is a 'driver' setting, may we proceed?
   0.000296	[D6] testval_reloadable: var=driver, oldval=huawei-ups2000, newval=huawei-ups2000, reloadable=0, reload_flag=0
   0.000302	[D6] testval_reloadable: verdict for (re)loading var=driver value: -1
   0.000307	[D5] do_upsconf_args: 'driver' setting already applied with this value
   0.000313	[D5] do_upsconf_args: confupsname=c1k, var=port, val=/dev/ttyCH341USB0
   0.000317	[D5] do_upsconf_args: call main_arg()
   0.000320	[D3] main_arg: var='port' val='/dev/ttyCH341USB0'
   0.000326	[D6] testinfo_reloadable: var=port, infoname=driver.parameter.port, newval=/dev/ttyCH341USB0, reloadable=0, reload_flag=0
   0.000330	[D6] testinfo_reloadable: verdict for (re)loading var=port value: 1
   0.000335	[D5] send_to_all: SETINFO driver.parameter.port "/dev/ttyCH341USB0"
   0.000346	[D3] main_arg: var='user' val='root'
   0.000351	[D6] testval_reloadable: var=user, oldval=nut, newval=root, reloadable=0, reload_flag=0
   0.000358	[D1] testval_reloadable: setting 'user' exists and differs: new value 'root' vs. 'nut'
   0.000364	[D6] testval_reloadable: verdict for (re)loading var=user value: 1
   0.000369	[D1] Overriding previously specified user 'nut' with 'root' specified for driver section
   0.000374	[D1] Network UPS Tools version 2.8.2 (release/snapshot of 2.8.2) built with gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0 and configured with flags: --with-modbus --with-user=nut --with-group=nut --datadir=/usr/share/nut --sysconfdir=/etc/nut --includedir=/usr/include --with-statepath=/run/nut --with-pidpath=/run/nut --with-altpidpath=/run/nut --with-drvpath=/lib/nut --with-udev-dir=/lib/udev --with-systemdsystemunitdir=/lib/systemd/system --with-systemdshutdowndir=/lib/systemd/system-shutdown
   0.000382	[D1] debug level is '6'
   0.000388	[D5] send_to_all: SETINFO driver.debug "6"
   0.000396	[D5] send_to_all: SETFLAGS driver.debug RW NUMBER
   0.000814	[D1] Succeeded to become_user(root): now UID=0 GID=0
   0.000837	[D5] send_to_all: SETINFO device.type "ups"
   0.000846	[D5] send_to_all: SETINFO driver.state "init.device"
   0.000855	[D2] upsdrv_initups
   0.032293	[D5] send_to_all: SETINFO driver.state "init.quiet"
   0.032319	[D5] send_to_all: SETINFO driver.version "2.8.2"
   0.032324	[D5] send_to_all: SETINFO driver.version.internal "0.06"
   0.032336	[D5] send_to_all: SETINFO driver.name "huawei-ups2000"
   0.032343	[D5] send_to_all: SETINFO driver.state "init.info"
   0.032348	[D2] upsdrv_initinfo
   0.032353	[D2] ser_send_buf
   1.432695	unable to read from serial port /dev/ttyCH341USB0, retry...
   1.432718	[D2] ser_send_buf
   2.833968	unable to read from serial port /dev/ttyCH341USB0, retry...
   2.834014	[D2] ser_send_buf
   4.235256	unable to read from serial port /dev/ttyCH341USB0, retry...
   4.235277	unable to read from serial port /dev/ttyCH341USB0!
   4.235287	[D5] send_to_all: SETINFO driver.state "cleanup.upsdrv"
   4.238043	[D5] send_to_all: SETINFO driver.state "cleanup.exit"
   4.238063	upsnotify: failed to notify about state 4: no notification tech defined, will not spam more about it
```

jimklimov, 2024-04-15:
Maybe @biergaizi can clarify why `unable to read from serial port` appears, but it may well be that the UPS is not one handled by the `huawei-2000` driver after all; maybe does not really serve a serial-port like interface either.

Can you please also try with `nutdrv_qx` again and `-x user=root`, for completeness?

Unreal-System, 2024-04-15:
```
root@user-virtual-machine:/lib/nut# ./nutdrv_qx -DDDDDD -a c1k -x user=root
   0.000000	[D5] send_to_all: SETINFO driver.state "init.starting"
Network UPS Tools - Generic Q* USB/Serial driver 0.36 (2.8.2)
USB communication driver (libusb 0.1) 0.47
   0.000080	[D1] upsdrv_makevartable...
   0.000101	[D5] send_to_all: SETINFO driver.version.usb "libusb-0.1 (or compat)"
   0.000109	[D1] Using USB implementation: libusb-0.1 (or compat)
   0.000201	[D5] do_upsconf_args: confupsname=c1k, var=driver, val=nutdrv_qx
   0.000211	[D5] do_upsconf_args: call main_arg()
   0.000214	[D3] main_arg: var='driver' val='nutdrv_qx'
   0.000231	[D5] do_upsconf_args: not a main_arg()
   0.000236	[D5] do_upsconf_args: this is a 'driver' setting, may we proceed?
   0.000241	[D6] testval_reloadable: var=driver, oldval=nutdrv_qx, newval=nutdrv_qx, reloadable=0, reload_flag=0
   0.000245	[D6] testval_reloadable: verdict for (re)loading var=driver value: -1
   0.000249	[D5] do_upsconf_args: 'driver' setting already applied with this value
   0.000254	[D5] do_upsconf_args: confupsname=c1k, var=port, val=/dev/ttyCH341USB0
   0.000259	[D5] do_upsconf_args: call main_arg()
   0.000263	[D3] main_arg: var='port' val='/dev/ttyCH341USB0'
   0.000267	[D6] testinfo_reloadable: var=port, infoname=driver.parameter.port, newval=/dev/ttyCH341USB0, reloadable=0, reload_flag=0
   0.000271	[D6] testinfo_reloadable: verdict for (re)loading var=port value: 1
   0.000281	[D5] send_to_all: SETINFO driver.parameter.port "/dev/ttyCH341USB0"
   0.000292	[D3] main_arg: var='user' val='root'
   0.000296	[D6] testval_reloadable: var=user, oldval=nut, newval=root, reloadable=0, reload_flag=0
   0.000300	[D1] testval_reloadable: setting 'user' exists and differs: new value 'root' vs. 'nut'
   0.000304	[D6] testval_reloadable: verdict for (re)loading var=user value: 1
   0.000308	[D1] Overriding previously specified user 'nut' with 'root' specified for driver section
   0.000313	[D1] Network UPS Tools version 2.8.2 (release/snapshot of 2.8.2) built with gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0 and configured with flags: --with-modbus --with-user=nut --with-group=nut --datadir=/usr/share/nut --sysconfdir=/etc/nut --includedir=/usr/include --with-statepath=/run/nut --with-pidpath=/run/nut --with-altpidpath=/run/nut --with-drvpath=/lib/nut --with-udev-dir=/lib/udev --with-systemdsystemunitdir=/lib/systemd/system --with-systemdshutdowndir=/lib/systemd/system-shutdown
   0.000322	[D1] debug level is '6'
   0.000326	[D5] send_to_all: SETINFO driver.debug "6"
   0.000335	[D5] send_to_all: SETFLAGS driver.debug RW NUMBER
   0.000865	[D1] Succeeded to become_user(root): now UID=0 GID=0
   0.000892	[D5] send_to_all: SETINFO device.type "ups"
   0.000902	[D5] send_to_all: SETINFO driver.state "init.device"
   0.000906	[D1] upsdrv_initups...
   0.176231	[D3] send: 'QGS'
   0.256190	[D5] read: (4 bytes) => 4e 41 4b 0d
   0.256208	[D3] read: 'NAK'
   0.256213	[D2] qx_process_answer: short reply (input.voltage)
   0.256301	[D3] send: 'QGS'
   0.326173	[D5] read: (4 bytes) => 4e 41 4b 0d
   0.326195	[D3] read: 'NAK'
   0.326201	[D2] qx_process_answer: short reply (input.voltage)
   0.326292	[D3] send: 'QGS'
   0.396218	[D5] read: (4 bytes) => 4e 41 4b 0d
   0.396241	[D3] read: 'NAK'
   0.396247	[D2] qx_process_answer: short reply (input.voltage)
   0.396339	[D3] send: 'M'
   0.466246	[D5] read: (4 bytes) => 4e 41 4b 0d
   0.466266	[D3] read: 'NAK'
   0.466273	[D2] voltronic_qs_protocol: invalid protocol [NAK]
   0.466283	[D4] ups_infoval_set: failed to preprocess value [ups.firmware.aux: NAK]
   0.466368	[D3] send: 'M'
   0.536225	[D5] read: (4 bytes) => 4e 41 4b 0d
   0.536243	[D3] read: 'NAK'
   0.536249	[D2] voltronic_qs_protocol: invalid protocol [NAK]
   0.536258	[D4] ups_infoval_set: failed to preprocess value [ups.firmware.aux: NAK]
   0.536343	[D3] send: 'M'
   0.606277	[D5] read: (4 bytes) => 4e 41 4b 0d
   0.606295	[D3] read: 'NAK'
   0.606301	[D2] voltronic_qs_protocol: invalid protocol [NAK]
   0.606310	[D4] ups_infoval_set: failed to preprocess value [ups.firmware.aux: NAK]
   0.606394	[D3] send: 'M'
   0.676278	[D5] read: (4 bytes) => 4e 41 4b 0d
   0.676296	[D3] read: 'NAK'
   0.676302	[D2] voltronic_qs_hex_protocol: invalid protocol [NAK]
   0.676312	[D4] ups_infoval_set: failed to preprocess value [ups.firmware.aux: NAK]
   0.676396	[D3] send: 'M'
   0.746295	[D5] read: (4 bytes) => 4e 41 4b 0d
   0.746314	[D3] read: 'NAK'
[... 389 lines cut ...]
  28.746605	[D5] update_status: !FSD
  28.746616	[D5] send_to_all: SETINFO driver.state "quiet"
  30.469513	[D5] send_to_all: SETINFO driver.state "updateinfo"
  30.469536	[D1] upsdrv_updateinfo...
  30.469543	[D1] Quick update...
  30.469633	[D3] send: 'Q1'
  30.746647	[D5] read: (47 bytes) => 28 32 32 39 2e 31 20 30 30 30 2e 30 20 32 32 39 2e 32
  30.746678	[D5]  20 30 32 37 20 35 30 2e 30 20 32 2e 32 35 20 34 39 2e 30 20 30 30 31 30 30
  30.746682	[D5]  30 30 30 0d
  30.746686	[D3] read: '(229.1 000.0 229.2 027 50.0 2.25 49.0 00100000'
  30.746696	[D5] update_status: OL
  30.746702	[D5] update_status: !LB
  30.746710	[D5] update_status: BYPASS
  30.746716	[D5] update_status: !CAL
  30.746719	[D5] update_status: !FSD
  30.746728	[D5] send_to_all: SETINFO driver.state "quiet"
^C  31.503653	[D1] set_exit_flag: raising exit flag due to signal 2
  31.503675	Signal 2: exiting
  31.503688	[D5] send_to_all: SETINFO driver.state "cleanup.upsdrv"
  31.503694	[D1] upsdrv_cleanup...
  31.508085	[D5] send_to_all: SETINFO driver.state "cleanup.exit"

```

It seems to be looping, so I cancelled it.

Unreal-System, 2024-04-15:
```
root@user-virtual-machine:/lib/nut# upsdrvctl -u root start
Network UPS Tools - UPS driver controller 2.7.4
Network UPS Tools - Generic Q* USB/Serial driver 0.28 (2.7.4)
USB communication driver 0.33
Using protocol: Megatec 0.06
No values for battery high/low voltages
Using 'guesstimation' (low: 20.800000, high: 26.000000)!
Battery runtime will not be calculated (runtimecal not set)
```
It looks like it needs to be tinkered with version information and battery voltage calculations.
Also, the battery voltage seems to be charged to 28V, because it automatically charges so high when it first boots up, and it was also 27V± when I tested it later.🤔

jimklimov, 2024-04-15:
Great, so the main supportability problem is just about adding the IDs to the driver source!

New driver versions may be more capable and industrious about battery voltage range => charge/runtime "guesstimations"; note you can use driver options like `default.battery.voltage.high = 28.0` to set something that suits your hardware.

In physical practice this likely means your device has two PbAc battery packs with nominal 12V each (so hovers around 13.5-14V when charged) and reports them as one unit.

jimklimov, 2024-04-15:
Can you please add a data dump of `upsc c1k` for the two NUT versions? 

Note that in case of the new custom build, you can instead run the driver program with added `-d1` argument to dump data once and end the loop - so you don't have to install it into the OS if for some reason you are reluctant to.

Unreal-System, 2024-04-16:
```
root@user-virtual-machine:/lib/nut# upsc c1k
Init SSL without certificate database
battery.charge: 100
battery.voltage: 2.23
battery.voltage.high: 26.00
battery.voltage.low: 20.80
battery.voltage.nominal: 24.0
device.mfr: 222222222222222
device.model: 2222222222
device.type: ups
driver.name: nutdrv_qx
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: /dev/ttyCH341USB0
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.data: Megatec 0.06
driver.version.internal: 0.28
input.current.nominal: 4.0
input.frequency: 50.0
input.frequency.nominal: 50
input.voltage: 229.4
input.voltage.fault: 0.0
input.voltage.nominal: 220
output.voltage: 229.4
ups.beeper.status: disabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.firmware: R1.00.47
ups.load: 27
ups.status: OL BYPASS
ups.temperature: 48.0
ups.type: online
root@user-virtual-machine:/lib/nut# upsc c1k -d1
Init SSL without certificate database
Error: old-style variable names are not supported
root@user-virtual-machine:/lib/nut# ./nutdrv_qx -d1 
Network UPS Tools - Generic Q* USB/Serial driver 0.28 (2.7.4)
USB communication driver 0.33
./nutdrv_qx: invalid option -- 'd'
Error: unknown option -?. Try -h for help.
```

jimklimov, 2024-04-16:
Thanks! As noted, `-d` is for your "new custom build" (since 2.8.0).

jimklimov, 2024-04-16:
Are `mfr` and `model` populated as literally strings of `2`s, or did you obfuscate them?

jimklimov, 2024-04-16:
Did you try any other `subdriver` options, in case some others discover more info than `megatec` does? e.g. depending on the build/version, the possible options are like this (as of current master branch):
````
:; ./drivers/nutdrv_qx -h
...
Acceptable values for 'subdriver' via -x or ups.conf in this driver: cypress, phoenixtec, phoenix, ippon, krauler, fabula, hunnox, fuji, sgs, snr, ablerex, armac

Acceptable values for 'protocol' via -x or ups.conf in this driver: voltronic, voltronic-qs, voltronic-qs-hex, mustek, megatec/old, bestups, mecer, megatec, zinto, masterguard, hunnox, ablerex, q1
````

jimklimov, 2024-04-16:
Looking at sources, it has tried 2/3 of the list:
```
/* Reference list of available non-USB subdrivers */
static subdriver_t      *subdriver_list[] = {
        &voltronic_subdriver,
        &voltronic_qs_subdriver,
        &voltronic_qs_hex_subdriver,
        &mustek_subdriver,
        &megatec_old_subdriver,
        &bestups_subdriver,
        &mecer_subdriver,
        &megatec_subdriver,
        &zinto_subdriver,
        &masterguard_subdriver,
        &hunnox_subdriver,
        &ablerex_subdriver,
        /* Fallback Q1 subdriver */
        &q1_subdriver,
        NULL
};
```

* (note historic misnomer - structure for "subdrivers" is exposed as `protocol` setting)

Unreal-System, 2024-04-16:
```
root@user-virtual-machine:/home/user/nut-2.8.2/drivers# ./nutdrv_qx -a c1k -x user=root -x protocol=ablerex
Network UPS Tools - Generic Q* USB/Serial driver 0.36 (2.8.2)
USB communication driver (libusb 0.1) 0.47
Using protocol: Ablerex 0.01
Autodetected 12 as number of battery packs [24/2.23]
Battery runtime will not be calculated (runtimecal not set)

root@user-virtual-machine:/home/user/nut-2.8.2/drivers# upsc c1k
Init SSL without certificate database
battery.charge: 100
battery.voltage: 2.23
battery.voltage.high: 27.30
battery.voltage.low: 19.20
battery.voltage.nominal: 24.0
device.mfr: 222222222222222
device.model: 2222222222
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.name: nutdrv_qx
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: /dev/ttyCH341USB0
driver.parameter.protocol: ablerex
driver.parameter.synchronous: auto
driver.state: quiet
driver.version: 2.8.2
driver.version.data: Ablerex 0.01
driver.version.internal: 0.36
driver.version.usb: libusb-0.1 (or compat)
input.frequency: 50.0
input.voltage: 228.2
input.voltage.fault: 0.0
output.current.nominal: 4.0
output.frequency.nominal: 50.0
output.voltage: 228.2
output.voltage.nominal: 220.0
ups.beeper.status: disabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.firmware: R1.00.47
ups.load: 26
ups.status: BYPASS OL
ups.temperature: 48.0
ups.type: online

root@user-virtual-machine:/home/user/nut-2.8.2/drivers# ./nutdrv_qx -a c1k -x user=root -x protocol=q1
Network UPS Tools - Generic Q* USB/Serial driver 0.36 (2.8.2)
USB communication driver (libusb 0.1) 0.47
Using protocol: Q1 0.08
Can't autodetect number of battery packs [-1/2.18]
Battery runtime will not be calculated (runtimecal not set)
root@user-virtual-machine:/home/user/nut-2.8.2/drivers# upsc c1k
Init SSL without certificate database
battery.voltage: 2.18
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.name: nutdrv_qx
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: /dev/ttyCH341USB0
driver.parameter.protocol: q1
driver.parameter.synchronous: auto
driver.state: quiet
driver.version: 2.8.2
driver.version.data: Q1 0.08
driver.version.internal: 0.36
driver.version.usb: libusb-0.1 (or compat)
input.frequency: 49.9
input.voltage: 228.5
input.voltage.fault: 0.0
output.voltage: 228.4
ups.beeper.status: disabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.load: 25
ups.status: OL BYPASS
ups.temperature: 48.0
ups.type: online
```

jimklimov, 2024-04-16:
@Unreal-System : can you please also post the ultimate working `ups.conf`? My understanding is that in the end you did not use `libusb` based communications (no `vendorid`/`productid`) and just connected `nutdrv_qx` to the serial port `/dev/ttyCH341USB0` - right?

So in the end the problem was about setting the correct permissions to the devfs node for the serial port device, and not about permissions for USB devices as such? (worked around here by running as `root`, but probably some targeted `chown nut:nut /dev/ttyCH341USB0` can suffice if your devfs implementation remembers permissions across reboots).

@aquette @clepple : I'm now wondering what the "fix" for such issue can be in terms of NUT code base. Just documenting for humans (`data/driver.list.in`) that this device model works with NUT but needs such setup? With such a "strange" serial port device name, is it likely to be auto-discovered by `nut-scanner` etc.?

Unreal-System, 2024-04-16:
```
[c1k]
    driver = nutdrv_qx
#    protocol = abler
#    port = auto
    port = /dev/ttyCH341USB0
#    default.battery.voltage.high = 28.0
#    vendorid = 1a86
#    productid = 7523
#    desc = "Debug Visench C1K UPS" 
#    maxretry = 3

```

jimklimov, 2024-04-16:
@Unreal-System : thanks again.

So one more thing to try is if `chown nut:nut /dev/ttyCH341USB0` on your system would:
* help start the driver without the `user` option and still let it see the UPS
* survive reboots, re-plug of the UPS, etc. (permissions would remain, driver would still work)

Alternately, adding the `user=root` line in `ups.conf` is a less preferable (security) but functional solution too.

Unreal-System, 2024-04-16:
```
root@user-virtual-machine:/home/user/nut-2.8.2/drivers# ls -l /dev/ttyCH341USB0 
crw-rw---- 1 nut nut 169, 0  4月 16 17:05 /dev/ttyCH341USB0
root@user-virtual-machine:/home/user/nut-2.8.2/drivers# systemctl start nut-server
root@user-virtual-machine:/home/user/nut-2.8.2/drivers# systemctl status nut-server
● nut-server.service - Network UPS Tools - power devices information server
     Loaded: loaded (/lib/systemd/system/nut-server.service; enabled; vendor preset: enabled)
     Active: active (running) since Tue 2024-04-16 17:22:17 CST; 3s ago
    Process: 3211 ExecStart=/sbin/upsd (code=exited, status=0/SUCCESS)
   Main PID: 3213 (upsd)
      Tasks: 1 (limit: 7009)
     Memory: 632.0K
        CPU: 9ms
     CGroup: /system.slice/nut-server.service
             └─3213 /lib/nut/upsd

4月 16 17:22:17 user-virtual-machine systemd[1]: Starting Network UPS Tools - power devices information server...
4月 16 17:22:17 user-virtual-machine upsd[3211]: fopen /run/nut/upsd.pid: No such file or directory
4月 16 17:22:17 user-virtual-machine upsd[3211]: listening on 127.0.0.1 port 3493
4月 16 17:22:17 user-virtual-machine upsd[3211]: listening on ::1 port 3493
4月 16 17:22:17 user-virtual-machine upsd[3211]: listening on 127.0.0.1 port 3493
4月 16 17:22:17 user-virtual-machine upsd[3211]: listening on ::1 port 3493
4月 16 17:22:17 user-virtual-machine upsd[3211]: Can't connect to UPS [c1k] (nutdrv_qx-c1k): Connection refused
4月 16 17:22:17 user-virtual-machine upsd[3211]: Can't connect to UPS [c1k] (nutdrv_qx-c1k): Connection refused
4月 16 17:22:17 user-virtual-machine upsd[3213]: Startup successful
4月 16 17:22:17 user-virtual-machine systemd[1]: Started Network UPS Tools - power devices information server.
root@user-virtual-machine:/home/user/nut-2.8.2/drivers# upsc c1k
Init SSL without certificate database
Error: Driver not connected
root@user-virtual-machine:/home/user/nut-2.8.2/drivers# ./nutdrv_qx -a c1k -DDDDDD
   0.000000	[D5] send_to_all: SETINFO driver.state "init.starting"
Network UPS Tools - Generic Q* USB/Serial driver 0.36 (2.8.2)
USB communication driver (libusb 0.1) 0.47
   0.000085	[D1] upsdrv_makevartable...
   0.000102	[D5] send_to_all: SETINFO driver.version.usb "libusb-0.1 (or compat)"
   0.000109	[D1] Using USB implementation: libusb-0.1 (or compat)
   0.000143	[D5] do_upsconf_args: confupsname=c1k, var=driver, val=nutdrv_qx
   0.000146	[D5] do_upsconf_args: call main_arg()
   0.000149	[D3] main_arg: var='driver' val='nutdrv_qx'
   0.000155	[D5] do_upsconf_args: not a main_arg()
   0.000158	[D5] do_upsconf_args: this is a 'driver' setting, may we proceed?
   0.000161	[D6] testval_reloadable: var=driver, oldval=nutdrv_qx, newval=nutdrv_qx, reloadable=0, reload_flag=0
   0.000166	[D6] testval_reloadable: verdict for (re)loading var=driver value: -1
   0.000169	[D5] do_upsconf_args: 'driver' setting already applied with this value
   0.000172	[D5] do_upsconf_args: confupsname=c1k, var=port, val=auto
   0.000175	[D5] do_upsconf_args: call main_arg()
   0.000178	[D3] main_arg: var='port' val='auto'
   0.000181	[D6] testinfo_reloadable: var=port, infoname=driver.parameter.port, newval=auto, reloadable=0, reload_flag=0
   0.000183	[D6] testinfo_reloadable: verdict for (re)loading var=port value: 1
   0.000187	[D5] send_to_all: SETINFO driver.parameter.port "auto"
   0.000197	[D1] Network UPS Tools version 2.8.2 (release/snapshot of 2.8.2) built with gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0 and configured with flags: --with-modbus --with-user=nut --with-group=nut --datadir=/usr/share/nut --sysconfdir=/etc/nut --includedir=/usr/include --with-statepath=/run/nut --with-pidpath=/run/nut --with-altpidpath=/run/nut --with-drvpath=/lib/nut --with-udev-dir=/lib/udev --with-systemdsystemunitdir=/lib/systemd/system --with-systemdshutdowndir=/lib/systemd/system-shutdown
   0.000204	[D1] debug level is '6'
   0.000210	[D5] send_to_all: SETINFO driver.debug "6"
   0.000217	[D5] send_to_all: SETFLAGS driver.debug RW NUMBER
   0.000872	[D1] Succeeded to become_user(nut): now UID=130 GID=138
   0.000887	[D5] send_to_all: SETINFO device.type "ups"
   0.000891	[D5] send_to_all: SETINFO driver.state "init.device"
   0.000894	[D1] upsdrv_initups...
   0.033859	[D3] usb_busses=0x60d7fe6156a0
   0.033890	[D2] Checking device (1D6B/0002) (002/001)
   0.033915	[D1] libusb_open get iManufacturer failed, retrying...
   0.033928	[D1] libusb_open get iManufacturer failed, retrying...
   0.033933	[D1] libusb_open get iManufacturer failed, retrying...
   0.033949	[D1] libusb_open get iProduct failed, retrying...
   0.033959	[D1] libusb_open get iProduct failed, retrying...
   0.033964	[D1] libusb_open get iProduct failed, retrying...
   0.033971	[D1] libusb_open get iSerialNumber failed, retrying...
   0.033976	[D1] libusb_open get iSerialNumber failed, retrying...
   0.033981	[D1] libusb_open get iSerialNumber failed, retrying...
   0.033989	[D2] - VendorID: 1d6b
[... 72 lines cut ...]
   0.034551	[D2] - ProductID: 0001
   0.034557	[D2] - Manufacturer: unknown
   0.034561	[D2] - Product: unknown
   0.034567	[D2] - Serial Number: unknown
   0.034571	[D2] - Bus: 001
   0.034575	[D2] - Device: 001
   0.034581	[D2] - Device release number: 0605
   0.034587	[D2] Trying to match device
   0.034590	[D3] match_function_regex: matching a device...
   0.034595	[D2] Device does not match - skipping
   0.034602	[D2] libusb0: No appropriate HID device found
[... cut ...]
try running the driver with at least 'subdriver', 'vendorid' and 'productid'
options specified. Please refer to the man page for details about these options
(man 8 nutdrv_qx).

   0.034623	[D5] send_to_all: SETINFO driver.state "cleanup.exit"
   0.034634	upsnotify: failed to notify about state 4: no notification tech defined, will not spam more about it
root@user-virtual-machine:/home/user/nut-2.8.2/drivers# cat /etc/nut/ups.conf
[c1k]
driver = nutdrv_qx
port = auto
#port = /dev/ttyCH341USB0
```

jimklimov, 2024-04-16:
No no, `port=auto` is for USB auto-detection (libusb does not look at devfs nodes to initialize a connection, but at USB IDs and ultimately bus/device number etc.)

So keep the `port = /dev/ttyCH341USB0` active in the configuration, and try starting this without `user=root` among options (now that `/dev/ttyCH341USB0` is owned by `nut:nut`).

Unreal-System, 2024-04-16:
```
root@user-virtual-machine:/home/user/nut-2.8.2/drivers# cat /etc/nut/ups.conf
[c1k]
driver = nutdrv_qx
port = auto
protocol = Megatec
#port = /dev/ttyCH341USB0
root@user-virtual-machine:/home/user/nut-2.8.2/drivers# ./nutdrv_qx -a c1k -DDDDDD
   0.000000	[D5] send_to_all: SETINFO driver.state "init.starting"
Network UPS Tools - Generic Q* USB/Serial driver 0.36 (2.8.2)
USB communication driver (libusb 0.1) 0.47
   0.000104	[D1] upsdrv_makevartable...
   0.000141	[D5] send_to_all: SETINFO driver.version.usb "libusb-0.1 (or compat)"
   0.000156	[D1] Using USB implementation: libusb-0.1 (or compat)
   0.000234	[D5] do_upsconf_args: confupsname=c1k, var=driver, val=nutdrv_qx
   0.000247	[D5] do_upsconf_args: call main_arg()
   0.000267	[D3] main_arg: var='driver' val='nutdrv_qx'
   0.000278	[D5] do_upsconf_args: not a main_arg()
   0.000285	[D5] do_upsconf_args: this is a 'driver' setting, may we proceed?
   0.000293	[D6] testval_reloadable: var=driver, oldval=nutdrv_qx, newval=nutdrv_qx, reloadable=0, reload_flag=0
   0.000302	[D6] testval_reloadable: verdict for (re)loading var=driver value: -1
   0.000308	[D5] do_upsconf_args: 'driver' setting already applied with this value
   0.000318	[D5] do_upsconf_args: confupsname=c1k, var=port, val=auto
   0.000328	[D5] do_upsconf_args: call main_arg()
   0.000336	[D3] main_arg: var='port' val='auto'
   0.000346	[D6] testinfo_reloadable: var=port, infoname=driver.parameter.port, newval=auto, reloadable=0, reload_flag=0
   0.000354	[D6] testinfo_reloadable: verdict for (re)loading var=port value: 1
   0.000364	[D5] send_to_all: SETINFO driver.parameter.port "auto"
   0.000373	[D5] do_upsconf_args: confupsname=c1k, var=protocol, val=Megatec
   0.000381	[D5] do_upsconf_args: call main_arg()
   0.000386	[D3] main_arg: var='protocol' val='Megatec'
   0.000401	[D5] do_upsconf_args: not a main_arg()
   0.000409	[D5] do_upsconf_args: process as value
   0.000415	[D6] testvar_reloadable: searching for var=protocol, vartype=2, reload_flag=0
   0.000424	[D6] testvar_reloadable: found var=protocol, val='(null)' => 'Megatec', vartype=2 => 2, found=0, reloadable=0, reload_flag=0
   0.000431	[D6] testvar_reloadable: verdict for (re)loading var=protocol value: 1
   0.000439	[D5] send_to_all: SETINFO driver.parameter.protocol "Megatec"
   0.000459	[D1] Network UPS Tools version 2.8.2 (release/snapshot of 2.8.2) built with gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0 and configured with flags: --with-modbus --with-user=nut --with-group=nut --datadir=/usr/share/nut --sysconfdir=/etc/nut --includedir=/usr/include --with-statepath=/run/nut --with-pidpath=/run/nut --with-altpidpath=/run/nut --with-drvpath=/lib/nut --with-udev-dir=/lib/udev --with-systemdsystemunitdir=/lib/systemd/system --with-systemdshutdowndir=/lib/systemd/system-shutdown
   0.000473	[D1] debug level is '6'
   0.000481	[D5] send_to_all: SETINFO driver.debug "6"
   0.000495	[D5] send_to_all: SETFLAGS driver.debug RW NUMBER
   0.001317	[D1] Succeeded to become_user(nut): now UID=130 GID=138
   0.001335	[D5] send_to_all: SETINFO device.type "ups"
   0.001339	[D5] send_to_all: SETINFO driver.state "init.device"
   0.001342	[D1] upsdrv_initups...
   0.047285	[D3] usb_busses=0x596bb0cd1790
   0.047310	[D2] Checking device (1D6B/0002) (002/001)
   0.063137	[D1] libusb_open get iManufacturer failed, retrying...
   0.063154	[D1] libusb_open get iManufacturer failed, retrying...
   0.063159	[D1] libusb_open get iManufacturer failed, retrying...
   0.063168	[D1] libusb_open get iProduct failed, retrying...
   0.063173	[D1] libusb_open get iProduct failed, retrying...
   0.063178	[D1] libusb_open get iProduct failed, retrying...
   0.063183	[D1] libusb_open get iSerialNumber failed, retrying...
   0.063187	[D1] libusb_open get iSerialNumber failed, retrying...
   0.063193	[D1] libusb_open get iSerialNumber failed, retrying...
   0.063200	[D2] - VendorID: 1d6b
   0.063205	[D2] - ProductID: 0002
   0.063213	[D2] - Manufacturer: unknown
   0.063218	[D2] - Product: unknown
   0.063221	[D2] - Serial Number: unknown
   0.063224	[D2] - Bus: 002
   0.063228	[D2] - Device: 001
   0.063235	[D2] - Device release number: 0605
   0.063240	[D2] Trying to match device
   0.063244	[D3] match_function_regex: matching a device...
   0.063251	[D2] Device does not match - skipping
   0.063263	[D2] Checking device (1A86/7523) (001/004)
   0.063293	[D1] libusb_open get iProduct failed, retrying...
   0.063299	[D1] libusb_open get iProduct failed, retrying...
[... 54 lines cut ...]
   0.063638	[D1] libusb_open get iProduct failed, retrying...
   0.063648	[D1] libusb_open get iSerialNumber failed, retrying...
   0.063652	[D1] libusb_open get iSerialNumber failed, retrying...
   0.063657	[D1] libusb_open get iSerialNumber failed, retrying...
   0.063661	[D2] - VendorID: 1d6b
   0.063665	[D2] - ProductID: 0001
   0.063671	[D2] - Manufacturer: unknown
   0.063675	[D2] - Product: unknown
   0.063681	[D2] - Serial Number: unknown
   0.063685	[D2] - Bus: 001
   0.063691	[D2] - Device: 001
   0.063697	[D2] - Device release number: 0605
   0.063703	[D2] Trying to match device
   0.063708	[D3] match_function_regex: matching a device...
   0.063714	[D2] Device does not match - skipping
   0.063724	[D2] libusb0: No appropriate HID device found
   0.063731	No supported devices found. Please check your device availability with 'lsusb'
and make sure you have an up-to-date version of NUT. If this does not help,
try running the driver with at least 'subdriver', 'vendorid' and 'productid'
options specified. Please refer to the man page for details about these options
(man 8 nutdrv_qx).

   0.063746	[D5] send_to_all: SETINFO driver.state "cleanup.exit"
   0.063752	upsnotify: failed to notify about state 4: no notification tech defined, will not spam more about it
```

Unreal-System, 2024-04-16:
Nut works after setting chown nut:nut /dev/ttyCH341USB0, but the system can’t remember this change after rebooting.

Unreal-System, 2024-04-16:
Additionally, I am passing ttyCH341USB0 through virtualization. 
Hot-swapping it will cause the virtualization to not be able to pass it through to this virtual machine again. 
It also does not support modification while booting.

jimklimov, 2024-04-16:
Well, for "productization" of the monitoring on your system, probably keeping `user=root` in `ups.conf` is the way to go then.

If your system uses systemd, you can add a drop-in file for `nut-driver.service` (older NUT) or `nut-driver@c1k.service` (2.8.x with https://github.com/networkupstools/nut/wiki/nut%E2%80%90driver%E2%80%90enumerator-(NDE) helper active) with a payload to `ExecStartPre=/bin/chown nut:nut /dev/ttyCH341USB0` and keep the driver running unprivileged.

Something like this:
````
# /etc/systemd/system/nut-driver@c1k.service.d/perms.conf
[Service]
ExecStartPre=/bin/chown nut:nut /dev/ttyCH341USB0
````

...modulo the service and program path names in your system.

PR #2404 (2024-04-16): [HCL] `data/driver.list.in`: Note known support for Visench C1K UPS
Closes: #2395

Given that this uses a serial port connection, not sure there is much more to "fix" on NUT code base side.
closed 2024-04-16 by jimklimov