networkupstools/nut #2397: Support for CyberPower Smart App online double conversion UPS

DaveTSG, 2024-04-14:
Hello,

I am looking at buying my first UPS and want to use NUT with it. 
I am considering [this CyberPower Smart App UPS](https://www.scorptec.com.au/product/ups/ups/81568-ols1000ert2ua), which I note isn't in the compatible hardware list.

I also note that there aren't any similar CyberPower model numbers/names in the list, either, and so I am concerned that using the drivers for a different CyberPower UPS won't work with this particular model.

I realise nothing can be guaranteed, but am I right to be concerned or is it more likely that CyberPower uses the same protocols in all their devices (and so the one I want to buy _should_ work just fine)?

Of course, I won't blame anyone if it doesn't work if I do buy it; I can just use the CyberPower software directly if needs be... but I'm just wondering what the likelihood of it working is.

Thanks in advance for any advice.
David

jimklimov, 2024-04-14:
I'm worried about "Smart App" in the name. Some vendors have UPSes with a network port which only talks (initiates connections) to their cloud servers and is useless for local monitoring. Double-check if something of the sort is the case here or not.

DaveTSG, 2024-04-14:
Ok, thank you for your advice. That's much appreciated.