networkupstools/nut #2398: New device FSP FP650

axet, 2024-04-14:
Hello!

I just got installed FP650 device on my desktop. And NUT well recognized it with nut-scanner. Device is working, reporting current load, battery voltage, allow to switch beeper off and on.

[____nutdrv_qx__2.8.0__01.zip](https://github.com/networkupstools/nut/files/14969745/____nutdrv_qx__2.8.0__01.zip)

The only problem I found so far: upower does not recognize the device, therefore system / GNOME does not know if desktop powered by battery or AC.

* https://www.fsp-ps.de/en/product/ups/1505956307-926.html

jimklimov, 2024-04-14:
Well, if NUT grabs the USB link diligently, GNOME would not know (not from the device) anyway.

IIRC there were side projects about NUT serving such info on dbus to represebt the device. Maybe @aquette and SmartNUT would revive something like that.

axet, 2024-04-17:
Duplicate of #2069 ?