networkupstools/nut #2424: RFE: Enhance `upsd` to `LISTEN host:port` with one colon-separated parameter token

jimklimov, 2024-04-28:
> tcp6 0 0 :::3493 :::* LISTEN 1447/upsd
> I remember being surprised by the space in the LISTEN directives ...

Thinking of it... might make sense to parse a single token on a `LISTEN <token1>:<token2>` (with at least one colon character), as an `addr`-colon-`port` so spelling of `LISTEN :::3493` would get split into `::` and `3493`, and some `LISTEN 127.0.0.1:3493` would become `127.0.0.1` and `3493`.

That said, in ~25 years of NUT's existence I think this is the first time such issue was raised (to the best of my personal knowledge) - so I guess it is a rare surprise after all the RTFM :)

_Originally posted by @jimklimov in https://github.com/networkupstools/nut/issues/2417#issuecomment-2081668113_

jimklimov, 2024-04-29:
For kicks, tried if it would just work - no, it does not; the whole single/first token is seen as an address, and the lack of second token causes an attempt to use the default port number:
````
:; rm -rf /tmp/nnnn ; mkdir -p /tmp/nnnn && touch /tmp/nnnn/{ups.conf,upsd.conf,upsd.users}
:; echo 'LISTEN :::3493' > /tmp/nnnn/upsd.conf
:; ALLOW_NO_DEVICE=yes NUT_STATEPATH=/tmp/nnnn NUT_CONFPATH=/tmp/nnnn ./server/upsd -DDDDDD ; echo $?
Network UPS Tools upsd 2.8.2-137-g3c19d79de
   0.000001     fopen /tmp/nnnn/upsd.pid: No such file or directory
   0.000043     Could not find PID file '/tmp/nnnn/upsd.pid' to see if previous upsd instance is already running!
   0.000208     /tmp/nnnn/upsd.conf is world readable
   0.000333     [D3] listen_add: added :::3493:3493
   0.000373     [D1] debug level is '6'
   0.000413     [D3] setuptcp: try to bind to :::3493 port 3493
   0.000530     getaddrinfo: Name or service not known
````

user01010111, 2026-09-20:
Hi @jimklimov, I’ve looked through this and would be happy to help with the implementation and tests, including the corresponding `nutconf` and Augeas changes.

The main detail to settle is IPv6 compatibility. Bare values such as `::3493` and `2001:db8::1:3493` are valid complete IPv6 addresses, so splitting every single-token argument at its final colon could change the meaning of existing configurations.

Would your preferred behaviour be to preserve bare IPv6 addresses, accept `host:port` and `[IPv6]:port`, and additionally support the `:::3493` spelling from your example? Or did you have a broader rule in mind for unbracketed IPv6 with a port?

If you’d like me to take this on, clarification of that behaviour would give me a clear basis for preparing a patch and regression tests across all three parsers. I’d keep service-name port support separate, as you suggested.

jimklimov, 2026-09-23:
Hi, I think the bracketed IPv6 addresses are a fair expectation here when coupled with ports in the same string (like they are in `nutauth.conf` sections etc.)

A three-colon spelling looks like a reasonable shortcut for `::` and `:port` (ending the string), since the official IPv6 stringification does not deal with such triplets.

closed 2026-10-10 by jimklimov