networkupstools/nut #2431: Libraries not found (Ubuntu 24.04 Server arm)

sccfit, 2024-05-03:
Hi all... just tried to install nut on fresh install Ubuntu 24 LTS on raspberry pi, installed fine but running sudo nut-scanner -C results in following:
```
Cannot load USB library (libusb-1.0.so) : file not found. USB search disabled.
Cannot load SNMP library (libnetsnmp.so) : file not found. SNMP search disabled.
Cannot load XML library (libneon.so) : file not found. XML search disabled.
Cannot load AVAHI library (libavahi-client.so) : file not found. AVAHI search disabled.
Cannot load IPMI library (libfreeipmi.so) : file not found. IPMI search disabled.
Cannot load NUT library (libupsclient.so) : file not found. NUT search disabled.
```
I switched to Ubuntu because the package in it is closer to current than RPI OS, and I need the tools to connect to identical UPS on one device. I've confirmed libusb-1.0 is installed, but for some reason nut can't find it, do I need to create extra symlinks or something like that?

jimklimov, 2024-05-03:
Please retry with higher debug verbosity, e.g.: 
```
:; sudo nut-scanner -DDDDDD -C > /tmp/nutscan.log 2>&1
```

There would be a huge wall of text about directories it looks into and file names it sees and tries to match there.

I wonder if your system delivers e.g. `libusb-1.0.so.1.27.0` and no symlink like `libusb-1.0.so` which could be in another package.

jimklimov, 2024-05-03:
Also, just in case: how did you install NUT (packages? custom build?) and which version?

danmrossi, 2024-05-07:
Must be the OS that NUT was compiled on, I’ve just compiled and installed on Ubuntu 24.04 no issues.

sccfit, 2024-05-17:
Sorry, have been on holidays, looping back to this now. I'm not great with Linux/packages/github, I installed using apt which I believe should be version 2.8.1-3.1ubuntu2.

Have run with higher debug verbosity, the following seem to be the most pertinent lines:

```
   0.001188     [D1]    Built-in:       /usr/lib/gcc/aarch64-unknown-linux-gnu
   0.001239     [D2] Looking for lib libusb-1.0.so in directory #0 : /usr/lib/aarch64-linux-gnu

   -- snip --
   
   0.012290     [D5] Comparing lib libusb-1.0.so with dirpath entry libusb-1.0.so.0
   0.012298     [D5] Comparing lib libusb-1.0.so with dirpath entry libusb-1.0.so.0.4.0

   -- snip --
   
   0.014574     [D1] Looking for lib libusb-1.0.so, found NULL
   
   -- snip --
   0.026829     [D1] nutscan_init: get_libname() did not resolve libname for LibUSB, trying to load it with libtool default resolver
   Cannot load USB library (libusb-1.0.so) : file not found. USB search disabled.
```

Am I right in thinking that creating a symlink 'libusb-1.0.so' that points to 'libusb-1.0.so.0' in the same directory should resolve the issue in this case?

jimklimov, 2024-05-17:
Yes, I wonder why one was not created in the first place (maybe devel packages deliver such a link, or some distro conventions had changed?) :\

At least, not directly a NUT problem, or not sure quickly how to address such distro specifics in common codebase, short of adding a `nut-scanner` config file/options about such paths.

jimklimov, 2024-05-17:
Hm, yeah, the name without trailing numbers seems to come from development packages:
````
:; grep libusb-1.0.so /var/lib/dpkg/info/*.list
/var/lib/dpkg/info/libusb-1.0-0-dev:amd64.list:/usr/lib/x86_64-linux-gnu/libusb-1.0.so
/var/lib/dpkg/info/libusb-1.0-0:amd64.list:/usr/lib/x86_64-linux-gnu/libusb-1.0.so.0.3.0
/var/lib/dpkg/info/libusb-1.0-0:amd64.list:/usr/lib/x86_64-linux-gnu/libusb-1.0.so.0
````

Wondering how nobody stumbled on that before. Did everyone have a larger system footprint installed?..

At least, this does seem like something NUT codebase can address - beside the idea above for direct run-time modifiable configuration of file names/paths to use, it may be possible to add the name(s) of the library seen at configuration/build time (e.g. `libusb-1.0.so.0` or resolved through all the symlinks to "real file" at `libusb-1.0.so.0.3.0`) to the built-in search list.

I would be wary of just allowing random suffixes to be accepted in the search method; to that end, `get_libname_in_dir()` explicitly has checks to `avoid "*.dll.a", ".so.1.2.3" etc.` and only pick the specified file name exactly.

jimklimov, 2024-05-17:
A practical search on a test system does use `realpath()` (where available) and practically use the ultimate file name when it looks at a symlink (which does exist):
````
   0.002097     [D5] Comparing lib libusb-1.0.so with dirpath entry libusb-1.0.so
   0.002108     [D2] Candidate path for lib libusb-1.0.so is /usr/lib/x86_64-linux-gnu/libusb-1.0.so (realpath /usr/lib/x86_64-linux-gnu/libusb-1.0.so.0.3.0)
   0.002119     [D1] Looking for lib libusb-1.0.so, found /usr/lib/x86_64-linux-gnu/libusb-1.0.so.0.3.0
   0.002121     [D1] nutscan_init: get_libname() resolved '/usr/lib/x86_64-linux-gnu/libusb-1.0.so.0.3.0' for LibUSB, loading it
   0.002324     [D1] nutscan_init: succeeded to load the library for LibUSB
````

Looking at code, I see that for each library of interest, `nut-scanner` tries one or more patterns via `get_libname()` and if any of those fails to find an exact filename match in searched directories, it falls back to the likes of `nutscan_load_usb_library()` (half a dozen of those for different libs) which would try to `lt_dlopen()` the specified pattern(s) in hopes that the OS resolver would take care of locating the library in some location unexpected by NUT (and maybe under a slightly different name - up to the OS not NUT at this point). This is the part seen in your log too, as:
````
   0.026829     [D1] nutscan_init: get_libname() did not resolve libname for LibUSB, trying to load it with libtool default resolver
````

jimklimov, 2024-05-17:
Running some experiments with another lib that is easier to fiddle with (NUT only built in this system, not installed into common paths, so its own `libupsclient` is not resolved via OS paths):

````
:; ./tools/nut-scanner/nut-scanner -O
Cannot load NUT library (libupsclient.so) : file not found. NUT search disabled.
````

With the local build in the library path, it is seen:
````
:; LD_LIBRARY_PATH="`pwd`/clients/.libs:$LD_LIBRARY_PATH" ./tools/nut-scanner/nut-scanner -O -DDDDDD 2>&1  | grep -B5 -A5 'libupsclient\.so\.'

...
   0.014299     [D5] Comparing lib libupsclient.so with dirpath entry libupsclient.so
   0.014308     [D2] Candidate path for lib libupsclient.so is /home/jim/nut/clients/.libs/libupsclient.so (realpath /home/jim/nut/clients/.libs/libupsclient.so.6.0.1)
   0.014312     [D2] Looking for lib libupsclient.so, found in LD_LIBRARY_PATH
   0.014313     [D1] Looking for lib libupsclient.so, found /home/jim/nut/clients/.libs/libupsclient.so.6.0.1
   0.014313     [D1] nutscan_init: get_libname() resolved '/home/jim/nut/clients/.libs/libupsclient.so.6.0.1' for NUT Client library, loading it
   0.014361     [D1] nutscan_init: succeeded to load the library for NUT Client library
...
````

Removing the `.so` symlink out of the way:
````
:; ls -la clients/.libs/libupsclient.so*
lrwxrwxrwx 1 jim jim     21 May 15 17:35 clients/.libs/libupsclient.so -> libupsclient.so.6.0.1
lrwxrwxrwx 1 jim jim     21 May 15 17:35 clients/.libs/libupsclient.so.6 -> libupsclient.so.6.0.1
-rwxr-xr-x 1 jim jim 223376 May 15 17:35 clients/.libs/libupsclient.so.6.0.1

:; mv clients/.libs/libupsclient.so{,.x}

:; LD_LIBRARY_PATH="`pwd`/clients/.libs:$LD_LIBRARY_PATH" ./tools/nut-scanner/nut-scanner -O -DDDDDD 2>&1  | grep -B5 -A5 'libupsclient\.so\.'

...
   0.013544     [D2] Looking for lib libupsclient.so in directory #1 : /home/jim/nut/clients/.libs
   0.013569     [D5] Comparing lib libupsclient.so with dirpath entry libnutclient.so.2.0.2
   0.013586     [D5] Comparing lib libupsclient.so with dirpath entry libupsclient.a
   0.013588     [D5] Comparing lib libupsclient.so with dirpath entry upslog
   0.013589     [D5] Comparing lib libupsclient.so with dirpath entry libnutclientstub.la
   0.013590     [D5] Comparing lib libupsclient.so with dirpath entry libupsclient.so.6.0.1
   0.013591     [D5] Comparing lib libupsclient.so with dirpath entry libupsclient.exp
...
   0.013598     [D5] Comparing lib libupsclient.so with dirpath entry libupsclient.so.x
   0.013599     [D5] Comparing lib libupsclient.so with dirpath entry libnutclient.a
   0.013600     [D5] Comparing lib libupsclient.so with dirpath entry libnutclientstub.a
   0.013600     [D5] Comparing lib libupsclient.so with dirpath entry upsset.cgi
   0.013601     [D5] Comparing lib libupsclient.so with dirpath entry libupsclient.so.6
...
   0.013613     [D1] Got no strong candidate path for lib libupsclient.so in /home/jim/nut/clients/.libs, but saw seemingly related names (are you missing a symbolic link, perhaps?) e.g.: libupsclient.so.6.0.1
   0.013619     [D2] Looking for lib libupsclient.so in directory #2 : /usr/lib/x86_64-linux-gnu
...
````

So it even recognizes that file, just not as a "strong" candidate.

jimklimov, 2024-06-15:
As a side note, this may also be a concern for DMF effort (currently awaiting its destiny in another branch):
````
...
Error loading Neon library required for DMF: libneon.so not found by dynamic loader; please verify it is in your /usr/lib or some otherwise searched dynamic-library path, under this exact name (maybe you just need a symlink?)
Error loading Neon library required for DMF: libneon-gnutls.so not found by dynamic loader; please verify it is in your /usr/lib or some otherwise searched dynamic-library path, under this exact name (maybe you just need a symlink?)
=== DMF-Test: Error parsing data: -1
````

alexojegu, 2024-06-18:
Same issue here, with Alpine Linux: https://gitlab.alpinelinux.org/alpine/aports/-/issues/16216

The [libusb](https://pkgs.alpinelinux.org/contents?branch=v3.20&name=libusb&arch=armv7&repo=main) package only provides:
```
/usr/lib/libusb-1.0.so.0
/usr/lib/libusb-1.0.so.0.4.0
```

The [libusb-dev](https://pkgs.alpinelinux.org/contents?branch=v3.20&name=libusb-dev&arch=armv7&repo=main) package provides:
```
/usr/include/libusb-1.0/libusb.h
/usr/lib/libusb-1.0.a
/usr/lib/libusb-1.0.so
/usr/lib/pkgconfig/libusb-1.0.pc
```

Maybe a regression of this old issue https://github.com/networkupstools/nut/issues/233

jimklimov, 2024-06-18:
Thanks for confirming this is more wide-spread ;( and also on the radar for so long...

sertonix, 2024-06-22:
This patch seems to work:
```patch
--- a/common/common.c
+++ b/common/common.c
@@ -3282,6 +3282,8 @@ static char * get_libname_in_dir(const char* base_libname, size_t base_libname_l
 
 	if (libname_alias) {
 		if (!libname_path) {
+			if (libname_alias[base_libname_length] == '.')
+				return libname_alias;
 			upsdebugx(1, "Got no strong candidate path for lib %s in %s"
 				", but saw seemingly related names (are you missing"
 				" a symbolic link, perhaps?) e.g.: %s",
```

I am not certain if it is robust though.

jimklimov, 2024-06-22:
My concern would be about picking a somewhat random library this way. Say `libusb.so.(0|1)` on some system meant the two different ABIs of separate generations of the library. Catastrophic!

sertonix, 2024-06-22:
Well, that is kind of already the case since there is no guarantee where the `libusb.so` symlink leads to. Ideally the sonames of the libraries nut was compiled against would be baked in the binary. No idea how to do that though.

jimklimov, 2024-07-03:
Seems to work, even on a Mac:
````
:; LD_LIBRARY_PATH="`pwd`/clients/.libs" ./tools/nut-scanner/nut-scanner -D
...
   0.000915     [D1] Looking for lib libusb-1.0.0.dylib, found /usr/local/Cellar/libusb/1.0.27/lib/libusb-1.0.0.dylib
   0.000933     [D1] nutscan_init: get_libname() resolved '/usr/local/Cellar/libusb/1.0.27/lib/libusb-1.0.0.dylib' for LibUSB, loading it
   0.006119     [D1] nutscan_init: succeeded to load the library for LibUSB
   0.006129     [D1] nutscan_init: skipped loading the library for LibSNMP: was absent during NUT build
   0.006401     [D1] Looking for lib libneon.27.dylib, found /usr/local/Cellar/neon/0.33.0/lib/libneon.27.dylib
   0.006409     [D1] nutscan_init: get_libname() resolved '/usr/local/Cellar/neon/0.33.0/lib/libneon.27.dylib' for LibNeon, loading it
   0.007187     [D1] nutscan_init: succeeded to load the library for LibNeon
   0.007196     [D1] nutscan_init: skipped loading the library for LibAvahi: was absent during NUT build
   0.007451     [D1] Looking for lib libfreeipmi.17.dylib, found /usr/local/Cellar/freeipmi/1.6.14/lib/libfreeipmi.17.dylib
   0.007459     [D1] nutscan_init: get_libname() resolved '/usr/local/Cellar/freeipmi/1.6.14/lib/libfreeipmi.17.dylib' for LibFreeIPMI, loading it
   0.010409     [D1] nutscan_init: succeeded to load the library for LibFreeIPMI
   0.010499     [D1] Looking for lib libupsclient.dylib, found /Users/abuild/nut/clients/.libs/libupsclient.6.dylib
   0.010506     [D1] nutscan_init: get_libname() resolved '/Users/abuild/nut/clients/.libs/libupsclient.6.dylib' for NUT Client library, loading it
   0.011017     [D1] nutscan_init: succeeded to load the library for NUT Client library
...
````

alexojegu, 2024-07-05:
Waiting for a release :smile:

rstolpe, 2024-07-09:
> Waiting for a release 😄

Me too, wondering when it's going to be released :) Is it possible to install it on ubuntu with sudo apt install nut or do I need to compile it my self?

jimklimov, 2024-07-09:
Currently we do not make snapshot packages (the idea is in backlog though), so your best shot for the latest and greatest features (and mishaps) is https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests

jimklimov, 2024-07-16:
With #2532 in place, it seems all reasonable goals of this issue have been achieved.

closed 2024-07-16 by jimklimov

Majoraslayer, 2024-08-05:
So what was the final verdict on a fix for this?  I just tried installing nut on a new Mint 22 install and got the same error.  Is the above listed commit the fix itself, and the version including the fix just isn't included in Mint's apt repo?  If that's the case, does nut happen to have a ppa that can be enabled for installing newer versions, or is compiling the only option?

jimklimov, 2024-08-05:
Currently, to get recent changes (master branch; the recent change above is not part of any numbered release yet) custom-compiling is the only official option; the https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests page (and docs it references) should help navigate through this.

Another matter is that Debian-like OSes share the recipes and many of those are several NUT releases behind the actual publications. So if your distro serves something older than 2.8.2 (still without this fix, but generally speaking), for current NUT you might be better served by their "experimental" repos or custom builds as well.

hibfamtv, 2024-12-06:
Any updates on this. Just installed today on Ubuntu and got the same issue:

Linux data 6.8.0-49-generic #49-Ubuntu SMP PREEMPT_DYNAMIC Mon Nov  4 02:06:24 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux

"Ubuntu 24.04.1 LTS"
VERSION_ID="24.04"
VERSION="24.04.1 LTS (Noble Numbat)"
VERSION_CODENAME=noble

 nut-scanner 
Cannot load USB library (libusb-1.0.so) : file not found. USB search disabled.
Cannot load SNMP library (libnetsnmp.so) : file not found. SNMP search disabled.
Cannot load XML library (libneon.so) : file not found. XML search disabled.
Cannot load AVAHI library (libavahi-client.so) : file not found. AVAHI search disabled.
Cannot load IPMI library (libfreeipmi.so) : file not found. IPMI search disabled.
Cannot load NUT library (libupsclient.so) : file not found. NUT search disabled.

jimklimov, 2024-12-06:
See above.

edwinbrett, 2024-12-29:
I guess I am also missing where the fix is for this. I didn't have problems a couple months ago when I installed this on a Pi but now that I am trying to do it on a Ubuntu installation I'm getting the same as the above, no matter how I install it.

Can someone in a bit more detail explain what needs to be done to get this to work? The article linked above is a bit vague about how to get this working and seems to be more a general piece about packaging this.

jimklimov, 2024-12-29:
As a quick hotfix, ensure the filenames that your build of `nut-scanner` looks for, like `libXXX.so` or `libXXX.so.N`, do exist. They may be symlinks delivered by `libXXX-dev` packages or can be made manually in same locations your `libXXX.so.N.M.P` actual library files reside.

edmondsiu0, 2025-01-23:
This workaround enabled my nut-scanner to recognise the SNMP library. 

```
ln -s /usr/lib64/libnetsnmp.so.40.1.0 /usr/lib64/libnetsnmp.so
```

karthikrao5, 2025-02-08:
Seeing the same thing on Ubuntu 24.04 server

```
$ sudo nut-scanner -U
Cannot load USB library (/usr/lib/x86_64-linux-gnu/libusb-0.1.so.4.4.4) : /usr/lib/x86_64-linux-gnu/libusb-0.1.so.4.4.4: undefined symbol: libusb_init. USB search disabled.
Cannot load SNMP library (libnetsnmp.so) : file not found. SNMP search disabled.
Cannot load XML library (libneon.so) : file not found. XML search disabled.
Cannot load AVAHI library (libavahi-client.so) : file not found. AVAHI search disabled.
Cannot load IPMI library (libfreeipmi.so) : file not found. IPMI search disabled.
Cannot load NUT library (libupsclient.so) : file not found. NUT search disabled.
```

And the file `/usr/lib/libusb-1.0.so` already exists. I don't fully understand how the symlink is supposed to work here

jimklimov, 2025-02-08:
> `libusb-0.1.so.4.4.4: undefined symbol: libusb_init. USB search disabled.`

This one looks odd - IIRC `libusb*` prefixed names are of libusb-1.0 API. Perhaps your package was built against one generation of the library and you installed it with another? Possibly also the libusb-0.1 you have is actually the "compat" project, recreating the 0.1 API mapped to 1.0 implementation (needs to be installed).

markwalkom, 2025-02-09:
appreciate this is closed, it might help anyone running ubuntu that wants usb scanning;
```
hub@hub:~$ cat /etc/debian_version 
trixie/sid
hub@hub:~$ sudo ln -s /usr/lib/x86_64-linux-gnu/libusb-1.0.so.0.4.0 /usr/lib/x86_64-linux-gnu/libusb.so
hub@hub:~$ sudo nut-scanner -C
Cannot load SNMP library (libnetsnmp.so) : file not found. SNMP search disabled.
Cannot load XML library (libneon.so) : file not found. XML search disabled.
Cannot load AVAHI library (libavahi-client.so) : file not found. AVAHI search disabled.
Cannot load IPMI library (libfreeipmi.so) : file not found. IPMI search disabled.
Cannot load NUT library (libupsclient.so) : file not found. NUT search disabled.
Scanning USB bus.
[nutdev1]
        driver = "usbhid-ups"
        port = "auto"
        vendorid = "0764"
        productid = "0501"
        product = "CP1500EPFCLCD"
        serial = "CTLLN2000041"
        vendor = "CPS"
        bus = "001"
        device = "004"
        busport = "006"
        ###NOTMATCHED-YET###bcdDevice = "0001"
```

you can change the name of the linked file to get any of the others

jimklimov, 2025-02-09:
Out of curiosity, is the problem reproducible with *current* NUT codebase (master branch, eventually to become 2.8.3 release when some other chores are done)?

webash, 2025-02-25:
This fixed helped me on Ubuntu 24.04's included 2.8.1 version of nut to get USB working.

```
pushd /usr/lib/x86_64-linux-gnu/
sudo ln -s libusb-1.0.so.0 libusb-1.0.so
sudo ln -s libnetsnmp.so.40 libnetsnmp.so
sudo ln -s libavahi-client.so.3 libavahi-client.so
sudo ln -s libfreeipmi.so.17 libfreeipmi.so
sudo ln -s libneon-gnutls.so.27 libneon.so
popd
```

From https://bookstack.bluecrow.net/books/linux/page/nut-network-ups-tools

originaltrini0, 2025-04-12:
Also for the list on Ubuntu 24.04:
`usr/lib/x86_64-linux-gnu/libupsclient.so.6`

jimklimov, 2025-05-28:
@SvetlinZarev : agreed. Should be solved for a while. Is the issue reproducible with current NUT? (Current meaning source, not current package in a distro that picked some source years ago)

hanwin, 2025-06-14:
> This fixed helped me on Ubuntu 24.04's included 2.8.1 version of nut to get USB working.
>
> ```
> pushd /usr/lib/x86_64-linux-gnu/
> sudo ln -s libusb-1.0.so.0 libusb-1.0.so
> sudo ln -s libnetsnmp.so.40 [...]

This solved the problem for me

x-MrFloppy-x, 2025-06-18:
@jimklimov : i just did a checkout from git and ran:
```
./autogen.sh
./configure --with-user=nut --with-group=nut
make install-as-root
```

during the configure i see
```
checking for real path to -lusb-1.0 (re-parsed as libusb-1.0.so likely file name / -lusb-1.0 linker arg)... initially '/usr/lib/gcc/x86_64-linux-gnu/12/../../../x86_64-linux-gnu/libusb-1.0.so'
checking for "real path" of '/usr/lib/gcc/x86_64-linux-gnu/12/../../../x86_64-linux-gnu/libusb-1.0.so'... '/usr/lib/x86_64-linux-gnu/libusb-1.0.so.0.3.0' (differs from input)
/usr/lib/x86_64-linux-gnu/libusb-1.0.so.0.3.0
```

nut-scanner shows me this:
```
XXX@YYYY:/tmp/nut# tools/nut-scanner/nut-scanner -U
Network UPS Tools nut-scanner 2.8.3.464-464+g2dce981ef (development iteration after 2.8.3)

NUT utility for detection of available power devices.

OPTIONS:
  -C, --complete_scan: Scan all available devices except serial ports (default).
* Options for USB devices scan not enabled: library not detected.
* Options for SNMP devices scan not enabled: library not detected.
* Options for XML/HTTP devices scan not enabled: library not detected.
  -O, --oldnut_scan: Scan NUT devices (old method via libupsclient).
* Options for NUT devices (new avahi method) scan not enabled: library not detected.
  -n, --nut_simulation_scan: Scan for NUT simulated devices (.dev files in $CONFPATH).
* Options for IPMI devices scan not enabled: library not detected.
  -E, --eaton_serial <serial ports list>: Scan serial Eaton devices (XCP, SHUT and Q1).
  -T, --thread <max number of threads>: Limit the amount of scanning threads running simultaneously (default: 512).
...
```

note:
I'm new to nut and trying to get it to work on my proxmox instance with a new UPS that requires 2.8.3

jimklimov, 2025-06-19:
@x-MrFloppy-x : What do other lines near the libusb detection say? Does the final report of `configure` script confirm it would use libusb? e.g. `* build USB drivers:    yes (libusb-1.0)`?

And just in case, what does/did `ls -la /usr/lib/x86_64-linux-gnu/libusb-1.0.so*` show - just the `/usr/lib/x86_64-linux-gnu/libusb-1.0.so.0.3.0` file and no symlinks (`*.so`, `*.so.0`)?

In the help text it should report the library search paths it would explicitly look into, e.g.
```
Note: many scanning options depend on further loadable libraries.
Run-time loadable library search paths used by this build of NUT:
        Via LD_LIBRARY_PATH:    /usr/src/nut-pve/tools/nut-scanner/.libs
        NOTE: Reporting filtered (existing unique) built-in paths:
        Built-in:       /usr/local/ups/lib
        Built-in:       /usr/lib/x86_64-linux-gnu
        Built-in:       /usr/lib64
        Built-in:       /usr/lib
        Built-in:       /usr/local/lib
        Built-in:       /usr/lib/gcc/x86_64-linux-gnu
```

Does yours include `/usr/lib/x86_64-linux-gnu`?

x-MrFloppy-x, 2025-06-19:
@jimklimov : 
you are right i missed the last occurence in the configure output:
`* build USB drivers:    yes (libusb-1.0)`

```
XXX@YYYY:/tmp/nut# cat configure.log |grep usb
checking for libusb-1.0 version via pkg-config... 1.0.26 found
checking for libusb(-0.1) version via pkg-config... none found
checking for libusb-config... no
checking for libusb preferred version... 1.0.26 (libusb-1.0)
checking for libusb cflags... -I/usr/include/libusb-1.0 
checking for libusb ldflags... -lusb-1.0 
checking for libusb.h... yes
checking for libusb_init... yes
checking for libusb_strerror... yes
checking for libusb_kernel_driver_active... yes
checking for libusb_set_auto_detach_kernel_driver... yes
checking for libusb_detach_kernel_driver... yes
checking for libusb_detach_kernel_driver_np... no
checking for usb_detach_kernel_driver_np... no
checking for libusb_get_port_number... yes
checking for real path to -lusb-1.0 (re-parsed as libusb-1.0.so likely file name / -lusb-1.0 linker arg)... initially '/usr/lib/gcc/x86_64-linux-gnu/12/../../../x86_64-linux-gnu/libusb-1.0.so'
checking for "real path" of '/usr/lib/gcc/x86_64-linux-gnu/12/../../../x86_64-linux-gnu/libusb-1.0.so'... '/usr/lib/x86_64-linux-gnu/libusb-1.0.so.0.3.0' (differs from input)
/usr/lib/x86_64-linux-gnu/libusb-1.0.so.0.3.0
checking whether to build USB drivers... yes (libusb-1.0)
checking whether to generate scripts/udev/nut-usbups.rules... yes
checking whether to generate scripts/devd/nut-usb.conf... yes
config.status: creating scripts/udev/nut-usbups.rules
config.status: creating scripts/devd/nut-usb.conf
* build USB drivers:    yes (libusb-1.0)
```

my /usr/lib/x86_64-linux-gnu/ looks like this:

```
XXX@YYYY:/tmp/nut# ls -la /usr/lib/x86_64-linux-gnu/libusb*       
-rw-r--r-- 1 root root 215282 Apr 10  2022 /usr/lib/x86_64-linux-gnu/libusb-1.0.a
lrwxrwxrwx 1 root root     19 Apr 10  2022 /usr/lib/x86_64-linux-gnu/libusb-1.0.so -> libusb-1.0.so.0.3.0
lrwxrwxrwx 1 root root     19 Apr 10  2022 /usr/lib/x86_64-linux-gnu/libusb-1.0.so.0 -> libusb-1.0.so.0.3.0
-rw-r--r-- 1 root root 121936 Apr 10  2022 /usr/lib/x86_64-linux-gnu/libusb-1.0.so.0.3.0
```

From your comment i understood that it should find, and actualy does find, the librarys in `/usr/lib/x86_64-linux-gnu/`
but i'm also puzzled why the nut-scanner reports `* Options for USB devices scan not enabled: library not detected.`.
 So as i'm running proxmox on intel i might not be related to this issue here.

jimklimov, 2025-06-19:
If you run `nut-scanner` with high debug verbosity it would print many screenfuls of information about the search for libraries before it does anything else (scan, show help, etc.)

You can try e.g. `nut-scanner -DDDDDD -U 2>&1 | less` to keep those messages in the standard viewer, where it would be easier for you to search for mentions of `libusb`. It is possible it looked at valid candidates and discarded them for some reason (would say why), or did not look into the correct directory, etc.

Proxmox should not be the problem, I have NUT running on PVE 7 and 8 - it is just Debian (or similar) as far as programs and packaging in the host-OS are concerned.

jimklimov, 2025-06-19:
Example from a PVE here:
```
root@pve:~# /usr/src/nut-pve/tools/nut-scanner/nut-scanner -DDDDDD -U 2>&1 | grep -B2 -A2 -i libusb
   0.000189     [D1]    Built-in:       /usr/lib/gcc/x86_64-pc-linux-gnu
   0.000190     [D1] nutscan_init: Parallel scan support: max_threads=512
   0.000193     [D3] get_libname('libusb-1.0.so.0.3.0'): Entering method...
   0.000197     [D4] get_libname: Looking for lib libusb-1.0.so.0.3.0 by path-set LD_LIBRARY_PATH_64...
   0.000199     [D3] get_libname_in_pathset('libusb-1.0.so.0.3.0', 19, '(null)', 0): Entering method...
   0.000200     [D4] get_libname: Looking for lib libusb-1.0.so.0.3.0 by path-set LD_LIBRARY_PATH...
   0.000201     [D3] get_libname_in_pathset('libusb-1.0.so.0.3.0', 19, '/usr/src/nut-pve/tools/nut-scanner/.libs', 0): Entering method...
   0.000203     [D4] get_libname_in_pathset: Looking for lib libusb-1.0.so.0.3.0 in a colon-separated path set
   0.000207     [D3] get_libname_in_dir('libusb-1.0.so.0.3.0', 19, '/usr/src/nut-pve/tools/nut-scanner/.libs', 0): Entering method...
   0.000214     [D4] get_libname_in_dir: Looking for lib libusb-1.0.so.0.3.0 in directory #0 : /usr/src/nut-pve/tools/nut-scanner/.libs
...
   0.000388     [D3] get_libname_in_dir('libusb-1.0.so.0.3.0', 19, '/usr/lib/x86_64-linux-gnu', 2): Entering method...
   0.000398     [D4] get_libname_in_dir: Looking for lib libusb-1.0.so.0.3.0 in directory #2 : /usr/lib/x86_64-linux-gnu
   0.000722     [D5] get_libname_in_dir: Comparing lib libusb-1.0.so.0.3.0 with dirpath entry .
   0.000725     [D5] get_libname_in_dir: Comparing lib libusb-1.0.so.0.3.0 with dirpath entry ..
   0.000726     [D5] get_libname_in_dir: Comparing lib libusb-1.0.so.0.3.0 with dirpath entry libpulse-simple.so.0
   0.000727     [D5] get_libname_in_dir: Comparing lib libusb-1.0.so.0.3.0 with dirpath entry libform.so.6
   0.000732     [D5] get_libname_in_dir: Comparing lib libusb-1.0.so.0.3.0 with dirpath entry libprofiler.so.0.5.5
...
   0.003844     [D5] get_libname_in_dir: Comparing lib libusb-1.0.so.0.3.0 with dirpath entry libusb-1.0.so.0.3.0
   0.003856     [D2] Candidate path for lib libusb-1.0.so.0.3.0 is /usr/lib/x86_64-linux-gnu/libusb-1.0.so.0.3.0 (realpath /usr/lib/x86_64-linux-gnu/libusb-1.0.so.0.3.0)
   0.003859     [D1] Looking for lib libusb-1.0.so.0.3.0, found /usr/lib/x86_64-linux-gnu/libusb-1.0.so.0.3.0
   0.003861     [D1] nutscan_init: get_libname() resolved '/usr/lib/x86_64-linux-gnu/libusb-1.0.so.0.3.0' for LibUSB, loading it
   0.004259     [D1] nutscan_init: succeeded to load the library for LibUSB
   0.004263     [D3] get_libname('libnetsnmp.so.40.0.0'): Entering method...
   0.004264     [D4] get_libname: Looking for lib libnetsnmp.so.40.0.0 by path-set LD_LIBRARY_PATH_64...
...
```

so here it had a direct hit for the name resolved during build. Otherwise it would have retried with some other names (e.g. `libusb.so`), etc.

x-MrFloppy-x, 2025-06-19:
@jimklimov 
> However your later post indicates pkg-config data for libusb-1.0 and headers were seen. Or was this after you installed *-dev?

Must be after installed the *-dev!
I just retested 2 times and without *-dev the configure does not find the usblib!

Thanls for your answers and the right questions!