networkupstools/nut #2439: Add S/N to ups.conf and nut breaks

Briandr73, 2024-05-10:
Hi,

I want to add my UPS s/n to ups.conf but when I do I cannot get data when using the upsc command. Remove the s/n line and reboot the Pi and nut is happy. Suggestions please?

[image]

jimklimov, 2024-05-11:
The `serial` is a string matched against what `libusb` says about the device, so possibly your wording is different?

If you start the driver with higher debug verbosity, or run `nut-scanner`, or `lsusb`, etc. you can see the value. Maybe it ends with a stream of spaces? Although matching is tried as exact then regex, so extra chars may not matter (or may need a `.*` depending on regex lib).

Briandr73, 2024-05-11:
Nut detects all my CyberPower UPS's as:

[image]

I for the most have accepted this just being glad to get some info on them to use in Home Assistant. I looked at the bottom of the ups and noted the s/n. Just wondering why lsusb is only seeing what I have as one model

Briandr73, 2024-05-11:
I just noticed one of my UPS's connected to an Orange Pi 800 is detecting a serial number. I am not sure if it is the right one. Would need to go visit the ups in basement. 

[image]

The "problem child"
[image]

the one that is better:
[image]

jimklimov, 2024-05-11:
Check in the Wiki here about https://github.com/networkupstools/nut/wiki/Changing-NUT-daemon-debug-verbosity and perhaps https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests (at least, in the end it says how to run a driver program - after stopping the services that wrap it - to dump discovered information).

Either way, running `usbhid-ups -DDDDDD -a YourUpsName` should reveal what connection metadata it sees from each device it walks.

Also note that in the later screenshots you have NUT v2.7.4 - that's quite old now, about 7 years (although some distros still ship it and not the newer releases).

Briandr73, 2024-05-11:
Each UPS is attached to its own Pi so it is per box. Even with the one that does detect the s/n the HA group responsible for nut did not feel it necessary to create a sensor for the respective s/n. I just checked and nut on the orange pi is running 2.7.4 which is the one showing me the s/n. I think that one is debian based. the other nuts in my home are running 2.8.0 which don't show me s/n. I think for me I would be happy just to get accurate ups info from the Pi's. HA despite its fast and regular release cycle I don't think has touched on nut in a while. Might be wrong on that. 

Do I use the actual name on the ups for this: usbhid-ups -DDDDDD -a YourUpsName

jimklimov, 2024-05-11:
> Do I use the actual name on the ups for this: usbhid-ups -DDDDDD -a YourUpsName

Yes, as the section name from `ups.conf`

Briandr73, 2024-05-11:
I keep getting command not found when I run this command. A search of google is not showing me samples of the code you would like me to run. Can you show screenshot from a terminal window one of your devices?

jimklimov, 2024-05-12:
Assuming you have NUT packaged in a Debian-like system, driver program names would be like `/lib/nut/usbhid-ups`. Otherwise, research your distro's metadata or use `find` to locate the program.

Briandr73, 2024-05-13:
NUT is running on the Rasperry Pi and /lib/nut/usbhid-ups is right where you indicated it was. 

[image]

Apologies but when I try the command from terminal I get an error

[image]

jimklimov, 2024-05-14:
In Unix/Linux for programs not in `PATH` you have to specify the directory to run a program - `./usbhid-ups` when your shell is in `/lib/nut` or `/lib/nut/usbhid-ups` generally.

Also, the section name for the UPS is without the square bracket mark-up :)

So for your case, this should do it:
````
:; /lib/nut/usbhid-ups -DDDDDD -a bedroom-ups
````

Probably you can run it as your own user account to see just what meta-data (serial and other IDs) the driver would consider and why it ruled some out. If it however claims a lot of unknowns, or fails to see config files, etc. you would need to initiate the command as `root` for it to have permissions to the USB device and make sure the driver as a service is stopped, so there are no conflicts in this area.

Briandr73, 2024-05-14:
Network UPS Tools - Generic HID driver 0.41 (2.7.4)
USB communication driver 0.33
   0.000000	send_to_all: SETINFO driver.parameter.port "auto"
   0.000492	send_to_all: SETINFO driver.parameter.vendorid "0764"
   0.000764	send_to_all: SETINFO driver.parameter.productid "0501"
   0.000847	send_to_all: SETINFO driver.parameter.product "ST Series"
   0.000911	send_to_all: SETINFO driver.parameter.vendor "CPS"
   0.000973	send_to_all: SETINFO driver.parameter.bus "001"
   0.001063	debug level is '6'
   0.008500	upsdrv_initups...
   0.039693	Checking device (1D6B/0003) (002/001)
   0.039936	- VendorID: 1d6b
   0.039995	- ProductID: 0003
   0.040040	- Manufacturer: unknown
   0.040084	- Product: unknown
   0.040128	- Serial Number: unknown
   0.040171	- Bus: 002
   0.040213	- Device release number: 0601
   0.040258	Trying to match device
   0.040397	Device does not match - skipping
   0.040523	Checking device (0764/0501) (001/003)
   0.049861	- VendorID: 0764
   0.049987	- ProductID: 0501
   0.050032	- Manufacturer: CPS
   0.050076	- Product: ST Series
   0.050185	- Serial Number: unknown
   0.050233	- Bus: 001
   0.050284	- Device release number: 0001
   0.050329	Trying to match device
   0.050625	Device matches
   0.050721	failed to claim USB device: could not claim interface 0: Device or resource busy
   0.051002	detached kernel driver from USB device...
   0.051130	nut_usb_set_altinterface: skipped usb_set_altinterface(udev, 0)
   0.053075	HID descriptor, method 1: (9 bytes) => 09 21 10 01 21 01 22 57 02
   0.053168	HID descriptor length (method 1) 599
   0.053215	i=0, extra[i]=09, extra[i+1]=21
   0.053266	HID descriptor, method 2: (9 bytes) => 09 21 10 01 21 01 22 57 02
   0.053312	HID descriptor length (method 2) 599
   0.053371	HID descriptor length 599
   0.093120	Report Descriptor size = 599
   0.093222	Report Descriptor: (599 bytes) => 05 84 09 04 a1 01 09 24 a1 00 85 1d 09 fe
   0.093282	 75 08 95 01 15 00 26 ff 00 b1 22 85 03 09 fd b1 22 05 85 09 8f b1 22 85 04
   0.093340	 09 89 b1 22 85 1e 95 02 06 01 ff 09 d0 09 d1 b1 22 85 3d 75 01 95 01 25 02
   0.093398	 09 ac b1 22 75 07 b1 03 85 29 75 08 26 ff 00 09 9e b1 22 09 9e 81 23 85 06
   0.093457	 95 02 25 03 05 85 09 8b 09 2c b1 22 85 07 95 06 25 64 09 83 09 8d 09 8e 09
   0.093514	 8c 09 29 09 67 b1 22 85 20 95 01 09 66 b1 a2 09 66 81 23 75 10 27 ff ff 00
   0.093571	 00 67 21 d1 f0 00 55 06 05 84 09 30 b1 a2 09 30 81 23 85 21 66 01 10 55 00
   0.093628	 05 85 09 68 b1 a2 09 68 81 23 85 82 26 58 02 09 2a 81 23 09 2a b1 a2 85 3f
   0.093687	 27 ff ff 00 00 67 21 d1 f0 00 55 06 05 84 09 40 b1 22 09 02 a1 02 85 22 75
   0.093748	 01 95 06 25 01 65 00 55 00 05 85 09 d0 09 44 09 45 09 46 09 43 09 42 b1 a2
   0.093806	 09 d0 09 44 09 45 09 46 09 43 09 42 81 23 95 01 75 02 b1 01 81 23 c0 85 80
   0.093863	 75 08 15 01 25 03 05 84 09 5a b1 a2 09 5a 81 23 c0 05 84 09 1a a1 00 85 23
   0.093920	 75 10 15 00 26 5e 01 67 21 d1 f0 00 55 07 05 84 09 30 b1 a2 09 30 81 23 85
   0.093977	 83 09 53 15 5b 25 64 b1 22 85 84 09 54 16 87 00 26 90 00 b1 22 85 88 16 00
   0.094035	 00 26 00 00 09 40 b1 a2 09 40 81 23 c0 09 1c a1 00 85 23 15 00 26 2c 01 09
   0.094093	 30 b1 a2 09 30 81 23 85 25 75 08 26 ff 00 65 00 55 00 09 35 b1 a2 09 35 81
   0.094367	 23 85 28 75 01 25 01 95 01 65 00 55 00 09 65 b1 a2 09 65 81 23 75 07 95 01
   0.094433	 b1 01 81 23 85 3e 75 10 27 ff ff 00 00 66 21 d1 55 07 09 44 b1 a2 85 85 75
   0.094491	 08 25 06 65 00 55 00 09 58 b1 a2 09 58 81 23 85 86 75 10 15 ff 26 ff 7f 35
   0.094549	 c4 47 c4 ff 1d 00 66 01 10 55 00 09 57 b1 a2 09 57 81 23 85 87 09 56 b1 a2
   0.094632	 09 56 81 23 c0 06 01 ff 09 e1 a1 01 85 89 75 08 95 07 15 00 26 ff 00 35 00
   0.094691	 46 ff 00 65 00 55 00 09 e0 b2 a2 01 85 1b 95 01 25 0a 45 0a 09 e3 b1 a2 85
   0.094749	 1c 25 01 45 01 09 e2 b1 a2 c0 09 ba a1 02 65 00 55 00 15 00 35 00 45 00 85
   0.094806	 78 75 10 27 ff ff 00 00 09 bb b1 22 75 08 26 ff 00 09 bc b1 22 75 10 27 ff
   0.094855	 ff 00 00 09 bd b2 a2 01 c0 c0
   0.095264	Using subdriver: CyberPower HID 0.4
   0.095583	65 HID objects found
   0.095633	Entering libusb_get_report
   0.172328	Report[get]: (2 bytes) => 1d 01
   0.172389	PhyMax = 0, PhyMin = 0, LogMax = 255, LogMin = 0
[... 493 lines cut ...]
   2.224245	hid_lookup_path: ff0100bc -> not found in lookup table
   2.224267	Path: UPS.ff0100ba.ff0100bc, Type: Feature, ReportID: 0x78, Offset: 16, Size: 8
   2.224290	Entering libusb_get_report
   2.224322	libusb_get_report: error sending control message: Device or resource busy
   2.224347	Can't retrieve Report 78: Device or resource busy
   2.224369	hid_lookup_path: 00840004 -> UPS
   2.224392	hid_lookup_path: ff0100ba -> not found in lookup table
   2.224415	hid_lookup_path: ff0100bd -> not found in lookup table
   2.224438	Path: UPS.ff0100ba.ff0100bd, Type: Feature, ReportID: 0x78, Offset: 24, Size: 16
   2.224468	send_to_all: SETINFO ups.mfr "CPS"
   2.224494	send_to_all: SETINFO ups.model "ST Series"
   2.224520	send_to_all: SETINFO ups.vendorid "0764"
   2.224545	send_to_all: SETINFO ups.productid "0501"
   2.224567	Report descriptor retrieved (Reportlen = 599)
   2.224588	Found HID device
   2.224611	Detected a UPS: CPS/ST Series
   2.224640	hid_lookup_usage: UPS -> 00840004
   2.224664	hid_lookup_usage: PowerSummary -> 00840024
   2.224690	hid_lookup_usage: iDeviceChemistry -> 00850089
   2.224712	string_to_path: depth = 3
   2.224757	Entering libusb_get_report
   2.224799	libusb_get_report: error sending control message: Device or resource busy
   2.224825	Can't retrieve Report 04: Device or resource busy
   2.224848	Got disconnected by another driver: Device or resource busy
   2.224870	Can't initialize data from HID UPS

jimklimov, 2024-05-15:
It seems you did not stop the "driver as a service" first, so your test killed off the service-wrapped instance, and then it got revived and killed this test copy.

Still, for the answer to this question, the report sufficed: `Serial Number: unknown` does not match the specific value you provided, so the driver initialization skipped over this device and went on to look for one that reports all the desired match criteria.