networkupstools/nut #2441: APC Smart-UPS 1000 USB - ups.test.result: No test initiated and missing TIMELEFT

nikolaysu, 2024-05-13:
APC Smart-UPS 1000 when connected via USB,  two very important parameters are missing:

- last test time - LASTSTEST in apcaccess
- battery life - TIMELEFT in apcaccess

```
root@cell06:~/nut# lsusb
Bus 001 Device 003: ID 051d:0002 American Power Conversion Uninterruptible Power Supply
```

<details><summary>upsc UPS_CELL06@127.0.0.1</summary>

```

   0.000000	debug level is '6'
   0.000732	upsdrv_initups...
   0.252814	Checking device (174C/3074) (002/002)
   0.380304	- VendorID: 174c
   0.380321	- ProductID: 3074
   0.380323	- Manufacturer: unknown
   0.380324	- Product: unknown
   0.380325	- Serial Number: unknown
   0.380326	- Bus: 002
   0.380327	- Device release number: 0200
   0.380328	Trying to match device
   0.380332	Device does not match - skipping
   0.380339	Checking device (1D6B/0003) (002/001)
   0.380352	- VendorID: 1d6b
   0.380354	- ProductID: 0003
   0.380355	- Manufacturer: unknown
   0.380356	- Product: unknown
   0.380357	- Serial Number: unknown
   0.380358	- Bus: 002
   0.380359	- Device release number: 0504
   0.380360	Trying to match device
   0.380361	Device does not match - skipping
   0.380364	Checking device (26CE/01A2) (001/004)
   0.380369	- VendorID: 26ce
   0.380371	- ProductID: 01a2
   0.380372	- Manufacturer: unknown
   0.380373	- Product: unknown
   0.380374	- Serial Number: unknown
   0.380375	- Bus: 001
   0.380376	- Device release number: 0000
   0.380377	Trying to match device
   0.380378	Device does not match - skipping
   0.380380	Checking device (051D/0002) (001/003)
   0.389355	- VendorID: 051d
   0.389364	- ProductID: 0002
   0.389365	- Manufacturer: American Power Conversion
   0.389366	- Product: Smart-UPS 1000 FW:600.3.I USB FW:1.5
   0.389367	- Serial Number: AS0451230660
   0.389368	- Bus: 001
   0.389369	- Device release number: 0006
   0.389370	Trying to match device
   0.389390	Device matches
   0.389393	failed to claim USB device: could not claim interface 0: Device or resource busy
   0.389415	detached kernel driver from USB device...
   0.389421	nut_usb_set_altinterface: skipped usb_set_altinterface(udev, 0)
   0.390300	HID descriptor, method 1: (9 bytes) => 09 21 10 01 21 01 22 f3 f1
   0.390306	HID descriptor length (method 1) 61939
   0.390309	i=0, extra[i]=09, extra[i+1]=21
   0.390312	HID descriptor, method 2: (9 bytes) => 09 21 10 01 21 01 22 10 04
   0.390313	HID descriptor length (method 2) 1040
   0.390315	Warning: two different HID descriptors retrieved (Reportlen = 61939 vs. 1040)
   0.390317	HID descriptor length 1040
   0.466932	Report Descriptor size = 1040
   0.466953	Report Descriptor: (1040 bytes) => 05 84 09 04 a1 01 09 24 a1 02 85 01 09
   0.466955	 fe 79 01 75 08 95 01 15 00 26 ff 00 b1 22 85 02 09 ff 79 02 b1 22 85 03 09
   0.466958	 fd 79 03 b1 22 05 85 85 04 09 8f 79 03 b1 22 85 05 09 89 79 04 b1 22 85 06
[... 191 lines cut ...]
</details>

<details><summary>apcaccess status</summary>

```
root@cell06:~# apcaccess status
APC      : 001,045,1117
CABLE    : USB Cable
DRIVER   : USB UPS Driver
MODEL    : Smart-UPS 1000
--skip--
TIMELEFT : 22.0 Minutes
LASTSTEST: 2024-04-25 02:51:11 +0300
FIRMWARE : 600.3.I USB FW:1.5
END APC  : 2024-04-27 13:22:34 +0300
```

</details>

I built the latest version from the master branch, but the result was exactly the same
```
git clone https://github.com/networkupstools/nut.git
./configure --with-user=nut --with-group=nut --with-usb
make
```

jimklimov, 2024-05-13:
For the initial test, NUT is v2.7.4 or older, right?

As for missing values, CC @aquette @clepple @desertwitch : any ideas?

nikolaysu, 2024-05-13:
> For the initial test, NUT is v2.7.4 or older, right?
```
root@cell06:/usr/lib/nut# upsc -V
Network UPS Tools upscmd 2.7.4

root@cell06:/usr/lib/nut# /usr/lib/nut/usbhid-ups -V
Network UPS Tools - Generic HID driver 0.41 (2.7.4)
USB communication driver 0.33
```

desertwitch, 2024-05-13:
apcupsd TIMELEFT : 22.0 Minutes => NUT battery.runtime: 1380 (seconds = 23 minutes)
Not sure about last test, will take a look...

nikolaysu, 2024-05-13:
```
root@cell06:~/nut/clients# ./upsc -V
Network UPS Tools upsc 2.8.2-215-gc342e63a2
```

<details><summary>upsc UPS_CELL06@127.0.0.1</summary>

```
root@cell06:~/nut/clients# ./upsc UPS_CELL06@127.0.0.1
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 50
battery.mfr.date: 2020/03/01
battery.runtime: 1500
battery.runtime.low: 120
battery.temperature: 27.9
battery.type: PbAc
battery.voltage: 27.5
battery.voltage.nominal: 24.0
device.mfr: American Power Conversion
device.model: Smart-UPS 1000
device.serial: AS0451230660
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: 0002
driver.parameter.synchronous: no
driver.parameter.vendorid: 051d
driver.version: 2.7.4
driver.version.data: APC HID 0.96
driver.version.internal: 0.41
input.sensitivity: medium
input.transfer.high: 253
input.transfer.low: 204
input.voltage: 226.0
output.frequency: 50.0
output.voltage: 226.0
output.voltage.nominal: 230.0
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.firmware: 600.3.I
ups.firmware.aux: 1.5
ups.load: 39.0
ups.mfr: American Power Conversion
ups.mfr.date: 2004/12/14
ups.model: Smart-UPS 1000
ups.productid: 0002
ups.serial: AS0451230660
ups.status: OL
ups.test.result: No test initiated
ups.timer.reboot: -1
ups.timer.shutdown: -1
ups.timer.start: -1
ups.vendorid: 051d
root@cell06:~/nut/clients#
```

</details>

desertwitch, 2024-05-13:
> driver.version: 2.7.4

You're still on NUT 2.7.4. on the machine where the UPS is physically connected.
It would be great if you could try it with NUT 2.8.2 or the current master (both driver + upsc for output). 😉

nikolaysu, 2024-05-13:
i need 

> > driver.version: 2.7.4
> 
> You're still on NUT 2.7.4. on the machine where the UPS is physically connected. It would be great if you could try it with NUT 2.8.2 or the current master (both driver + upsc for output). 😉

I need 10-15 minutes

nikolaysu, 2024-05-13:
completely removed the NUT that I installed from the packages
installed from master branch

<details><summary>upsc UPS_CELL06@127.0.0.1</summary>

```
root@cell06:/usr/local/ups/bin# /usr/local/ups/bin/upsc UPS_CELL06@127.0.0.1
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 50
battery.mfr.date: 2020/03/01
battery.runtime: 1320
battery.runtime.low: 120
battery.temperature: 27.9
battery.type: PbAc
battery.voltage: 27.5
battery.voltage.nominal: 24.0
device.mfr: American Power Conversion
device.model: Smart-UPS 1000
device.serial: AS0451230660
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: 0002
driver.parameter.synchronous: auto
driver.parameter.vendorid: 051d
driver.state: quiet
driver.version: 2.8.2-215-gc342e63a2
driver.version.data: APC HID 0.100
driver.version.internal: 0.53
driver.version.usb: libusb-1.0.22 (API: 0x1000106)
input.sensitivity: medium
input.transfer.high: 253
input.transfer.low: 204
input.voltage: 226.0
output.frequency: 50.0
output.voltage: 224.6
output.voltage.nominal: 230.0
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.firmware: 600.3.I
ups.firmware.aux: 1.5
ups.load: 37.7
ups.mfr: American Power Conversion
ups.mfr.date: 2004/12/14
ups.model: Smart-UPS 1000
ups.productid: 0002
ups.serial: AS0451230660
ups.status: OL
ups.test.result: No test initiated
ups.timer.reboot: -1
ups.timer.shutdown: -1
ups.timer.start: -1
ups.vendorid: 051d
```
</details>

desertwitch, 2024-05-13:
I've just looked into this and this is not a value actually reported through the HID (read from the UPS), but rather a date/time variable that APCUPSD calculates and stores itself. So there is some programmatic logic behind it, rather than being read "as is" from the UPS. It seems that when APCUPSD sees the UPS in/or exiting a self-testing state (although that detection does seem a bit fuzzy in the code), it stores that date in a variable and displays it as `LASTSTEST`. Not sure what NUT's stance on such calculated variables, as opposed to actual reported variables, is and if that would be something worth looking into. On the one hand I could see it being useful, but a hit-and-miss `CAL` detection (reporting a successful self test when there was in fact maybe none) may also lead users into a false sense of security... not sure, @jimklimov ?

jimklimov, 2024-05-14:
I believe `upscmd` and `upsrw` by default send a command and return success quickly (if there were no NUT protocol errors). Since NUT v2.8.0 "tracking" is available (the `-w` flag) so these commands send a random-ish cookie and can wait for a driver-dictated result attached to that cookie - and discern that reply to their specific request from whatever other traffic passes on the bus.

Using `-w -t NUM` with a short timeout value (default 10) can be counter-productive if the device interaction takes long. Not sure OTOH what happens for calibration with this driver and hardware, but if *theoretically* the driver accepts the request, passes it to the device, and returns a verdict only when the device is done - the needed `upscmd` timeout may be in hours range to receive the ultimate answer to that original request to run a self-test.

Now that you've dug into the logic and comparisons for all of this, and have a device to test with, you may be in fact better positioned than myself to iterate with a custom driver - see https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests

For starters, try adding different diagnostic print-outs into the driver with `upsdebugx()` to make sense of where it sets or clears the calibration verdict (per your "Done and passed" example that was seen for a while). Maybe it is the driver forgetting it after some loops (e.g. a major vs. minor update... or is that only in `snmp-ups`?..) or the device itself clearing that status in its reports e.g. after some time has passed or after the battery charge or other internal state has changed and the calibration verdict is no longer deemed trustworthy for whatever reasons the vendor thought of.

Regarding the other question, with synthetic variables - they have their place (e.g. when we do not know the battery charge from the device itself, but "guesstimate" it from known voltage and assumed high/low limits and some assumption about the discharge function over time). I don't think the NUT protocol (see `docs/net-protocol.txt`) has a way to discern synthetic values from those reported by the device (I certainly never saw any use of that in driver/client code even if it existed); at best there is a higher-level `TYPE` to tell apart numbers, strings, enums, etc. So as long as there is a suitable name for the concept in `docs/nut-names.txt` (or one gets negotiated on ups-dev mailing list to appear in the document), feel free to populate it meaningfully.

nikolaysu, 2024-05-27:
I would like to continue this discussion

At the moment I keep getting a value _No test initiated_

The first thing I noticed was when running the driver in debug mode
I compare with powercom(works via usbhid) connected on different host,

```
   0.013145	[D3] HID descriptor, method 1: (9 bytes) => 09 21 10 01 21 01 22 f3 f1
   0.013147	[D3] HID descriptor length (method 1) 61939
   0.013148	[D4] i=0, extra[i]=09, extra[i+1]=21
   0.013149	[D3] HID descriptor, method 2: (9 bytes) => 09 21 10 01 21 01 22 10 04
   0.013151	[D3] HID descriptor length (method 2) 1040
   0.013152	[D2] Warning: two different HID descriptors retrieved (Reportlen = 61939 vs. 1040)
```

Perhaps this is some kind of problem?

If you need to do some fine debugging of the code, I’m ready to do it, but I still don’t understand why I need to add a diagnostic output.
For now, if I look at the code, I see that the function receives the value 
```
void upsdrv_updateinfo(void)
....

		if (HIDGetDataValue(udev, event[i], &value, poll_interval) != 1)
			continue;
```

HIDGetDataValue(libhid.c)-> get_item_buffered(libhid.c)->GetValue(hidparser.c)