networkupstools/nut #2443: Are there more readings with usb rather than snmp? Eaton 5P 850

Zuendapp, 2024-05-14:
Hi there,

my colleague uses the same Eaton 5P with Network UPS Tools as I, the only difference is that I'm using it with SNMP and he with a USB connection.
So he's getting about 40 readings and mine "only" 26.

Thank you very much!

BR

jimklimov, 2024-05-15:
This is quite possible, also note that depending on Eaton model, several vendor-extended MIB mappings can apply to the same device and support sometimes not-overlapping data sets and/or represent same information with different precision.

Which one has more - that depends on the information reported by the device and by readiness of the NUT drivers to understand it (protocol-wise and mappings specifically). So this can vary also between versions of firmwares of the UPS and its management card, and what they report over different protocols internally and externally.

For clarity, do both of you use the same NUT version (and a recent 2.8.x one, at that)?

Zuendapp, 2024-05-15:
Hi @jimklimov :),

we use both the latest homeassistant addon in version 0.13.0 which show in the logs following NUT version (nut:all 2.8.0-7).

My config:

```
users:
  - username: admin
    password: P@ssword
    instcmds:
      - all
    actions: []
devices:
  - name: Eaton5P850i
    driver: snmp-ups
    port: 192.168.1.20
    log_level: info
    config:
      - snmp_version = v1
      - community = private
mode: netserver
shutdown_host: "false"
log_level: info
```

The only thing what is different is that I'm using the Tower version of the USV and he the Rack version (both 5P 850).

My installed Firmware versions are:

SNMP-Card (Network M2): 3.1.12
USV Firmware Version: 3.18.0035

Thank you!

BR

Zuendapp, 2024-07-28:
Hi @jimklimov ,

is there any update?

Thank you very much!

BR

jimklimov, 2024-07-28:
As noted before, currently the different drivers and even subdrivers (e.g. MIB mapping tables, several can apply to Eaton devices) are independent in the readings they provide. So depending on what the actual device (its firmware version, controller card, etc.) does serve, they can expose different data points, or same ones with different precision, even when polling the same device with different drivers.

There is a long-backlogged issue #273 and general theme of https://github.com/networkupstools/nut/labels/Data%20multipathing to try some ways of aggregating the information collected by several drivers talking to the same device (over same or different media/protocol) as a means of both communications fallback and information fidelity. But I am not aware if anyone had made progress on this; no PRs were submitted to the NUT project at least.

Zuendapp, 2024-07-29:
Okey, thank you!
Maybe I‘ll connect the Eaton UPS via USB in future, if there are any sensor readings I want to have and are not delivered via SNMP.

BR