networkupstools/nut #2451: Help please poweredge rt2000

FigJam23, 2024-05-22:
I'm wanting to monitor my poweredge rt2000 on my widows pc using likes of a nut server and then monitor this on a rpi on same network. My windows pc has no net access but my rpi can communicate out. But it's not practical to conect the ups to my rpi via USB. Is nut actually able to run on windows I don't need to control ups just get battery and AC stats. Can someone give me some direction on what I need to do as far as getting nut server to work on windows pc. Thank you

jimklimov, 2024-05-22:
Hello, by your question I assume your Windows *does* have network access, as much as is needed to communicate to the RPi. What it does not (normally) have is Internet access. Right?

For NUT, other than getting the files to install (can be via USB stick etc.) no Internet access is required - neither on Win nor on RPi hosts of yours. That said, with precautions like SSL wrapped access you could e.g. publish in the internet a `nut-cgi` driven web-page (or some third party's REST API service for NUT) running on the RPi and acting as a NUT client to the data server on Windows.

Currently there is a slowly progressing effort to provide modern [NUT for Windows](https://github.com/networkupstools/nut/wiki/NUT-for-Windows) - you can build your own or fetch a prepared tarball with binaries and DLLs from the AppVeyor CI server. There are some known incomplete features here and there, including a lack of installer and so a privileged registration of USB driver other than Windows built-in HID or HID Battery device. There is a "Zadig" tool detailed in #1690 (IIRC) to make such registrations manually.

With all that, if you don't stumble upon that not-yet-implemented stuff in your practically significant code path, you should be able to run a NUT driver and data server (`upsd`) configured same way via `etc/...conf` files as on Linux/Unix platforms. The `nut.exe` wrapper should let you register NUT as a Windows service to handle starting/stopping of the daemons involved (with little granularity at the moment, either upsmon alone or everything).

FigJam23, 2024-05-23:
Yes network is accessible on both devices, just restricted WAN on windows pc. Thanks for the info sound like it's complicated lol. But I'll give it a go haven't worked with binnerys on windows before wish me luck 😅

jimklimov, 2024-05-24:
Well, "binary" is a fancy word for an executable program, library, etc. - something with a specific byte-by-byte structure that is not a text file (and in some contexts, picture, movie, etc.)

So surely you've had worked with programs :)