networkupstools/nut #2453: Weird connectivity issues with a generic ViewPower UPS

hmv47, 2024-05-22:
I run Debian 11 with NUT 2.8.0 and connect a [generic ViewPower UPS](https://www.eltrox.pl/p/35383-ups-zasilacz-awaryjny-volt-polska-micro-1000va-600w-1x9ah). It gets identified:
```
Bus 001 Device 002: ID 0665:5161 Cypress Semiconductor USB to Serial
```
I tried running `nut-scanner`:
```
Scanning USB bus.
No start IP, skipping NUT bus (old connect method)
Scanning NUT bus (avahi method).
Failed to create client: Daemon not running
[nutdev1]
        driver = "nutdrv_qx"
        port = "auto"
        vendorid = "0665"
        productid = "5161"
        product = ""
        bus = "001"
```
But it doesn't get detected at all with `nutdrv_qx`. So I tried `blazer_usb` and it _sort of_ worked:
```
[nutdev1]
    driver = blazer_usb
    subdriver = cypress
    protocol = mustek
    port = auto
    desc = "UPS"
    vendorid = 0665
    productid = 5161
```
`upsc` sometimes return status from the UPS, but very often I get the stale data error.

Additionally, after reboots, randomly, USB device is not recognized properly e.g. it's shown as this:
```
Bus 001 Device 002: ID ffff:0000
```
And doesn't work at all with either driver.
I'm pretty positive this is a hardware problem and probably a flaky USB2Serial converter in the UPS, but still I'm opening this issue; I'd like to see if anyone else has had similar problems, or maybe knows a solution.

hmv47, 2024-05-23:
I experimented a bit more, with the latest 2.8.2 version
**/usr/local/ups/bin# nut-scanner**
```
Scanning USB bus.
No start IP, skipping NUT bus (old connect method)
Scanning NUT simulation devices.
[nutdev-usb1]
	driver = "nutdrv_qx"	# alternately: blazer_usb
	port = "auto"
	vendorid = "0665"
	productid = "5161"
	product = ""
	# bus = "001"
	# device = "009"
	# busport = "001"
```
**/usr/local/ups/bin/nutdrv_qx -a nutdev-usb1 -u root -DD**
```
Network UPS Tools - Generic Q* USB/Serial driver 0.36 (2.8.2)
USB communication driver (libusb 1.0) 0.47
   0.000000	[D1] upsdrv_makevartable...
   0.000116	[D1] Using USB implementation: libusb-1.0.21 (API: 0x1000105)
   0.000586	[D1] Built-in default or configured user for drivers 'nobody' was ignored due to 'root' specified on command line
   0.000625	[D1] Network UPS Tools version 2.8.2 (release/snapshot of 2.8.2) built with gcc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0 and configured with flags: --with-usb
   0.000662	[D1] debug level is '2'
   0.002415	[D1] Succeeded to become_user(root): now UID=0 GID=0
   0.002484	[D1] upsdrv_initups.
---- irrelevant data snipped------
   0.271853	[D2] Checking device 5 of 6 (0665/5161)
   0.272680	[D1] nut_libusb_open get iManufacturer failed, retrying...
   0.273499	[D1] nut_libusb_open get iManufacturer failed, retrying...
   0.274462	[D1] nut_libusb_open get iManufacturer failed, retrying...
   0.275727	[D2] - VendorID: 0665
   0.275763	[D2] - ProductID: 5161
   0.275786	[D2] - Manufacturer: unknown
   0.275807	[D2] - Product:           
   0.275828	[D2] - Serial Number: unknown
   0.275850	[D2] - Bus: 001
   0.275873	[D2] - Bus Port: 001
   0.275894	[D2] - Device: 009
   0.275918	[D2] - Device release number: 0002
   0.275939	[D2] Trying to match device
   0.276023	[D2] Device matches
   0.276044	[D2] Reading configuration descriptor 1 of 1
   0.276127	[D2] Claimed interface 0 successfully
  28.512102	[D2] claim: cannot process experimental.series
  29.514398	[D2] claim: cannot process experimental.series
  30.529203	[D2] claim: cannot process experimental.series
  39.586403	Device not supported!
  39.586430	Device not supported!
  39.586445	upsnotify: failed to notify about state 4: no notification tech defined, will not spam more about it

```
**/usr/local/ups/bin/blazer_usb -a nutdev-usb1 -u root -DD**
```
Network UPS Tools - Megatec/Q1 protocol USB driver 0.18 (2.8.2)
   0.000000	[D1] Using USB implementation: libusb-1.0.21 (API: 0x1000105)
   0.000418	[D1] Built-in default or configured user for drivers 'nobody' was ignored due to 'root' specified on command line
   0.000437	[D1] Network UPS Tools version 2.8.2 (release/snapshot of 2.8.2) built with gcc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0 and configured with flags: --with-usb
   0.000453	[D1] debug level is '2'
   0.002098	[D1] Succeeded to become_user(root): now UID=0 GID=0
---- irrelevant data snipped------
   0.265418	[D2] Checking device 5 of 6 (0665/5161)
   0.266363	[D1] nut_libusb_open get iManufacturer failed, retrying...
   0.267379	[D1] nut_libusb_open get iManufacturer failed, retrying...
   0.268371	[D1] nut_libusb_open get iManufacturer failed, retrying...
   0.269661	[D2] - VendorID: 0665
   0.269691	[D2] - ProductID: 5161
   0.269702	[D2] - Manufacturer: unknown
   0.269711	[D2] - Product:           
   0.269720	[D2] - Serial Number: unknown
   0.269729	[D2] - Bus: 001
   0.269742	[D2] - Bus Port: 001
   0.269751	[D2] - Device: 009
   0.269766	[D2] - Device release number: 0002
   0.269779	[D2] Trying to match device
   0.269849	[D2] Device matches
   0.269864	[D2] Reading configuration descriptor 1 of 1
   0.269935	[D2] Claimed interface 0 successfully
   0.269991	Please note that this driver is deprecated and will not receive
new development. If it works for managing your devices - fine,
but if you are running it to try setting up a new device, please
consider the newer nutdrv_qx instead, which should handle all 'Qx'
protocol variants for NUT. (Please also report if your device works
with this driver, but nutdrv_qx would not actually support it with
any subdriver!)

   0.270022	[D2] Trying megatec protocol...
   1.284908	[D2] blazer_status: short reply
   1.284952	[D2] Status read 1 failed
   2.299748	[D2] blazer_status: short reply
   2.299788	[D2] Status read 2 failed
   3.314571	[D2] blazer_status: short reply
   3.314613	[D2] Status read 3 failed
   3.314625	[D2] Trying mustek protocol...
   4.316911	[D2] blazer_status: short reply
   4.316955	[D2] Status read 1 failed
   5.319252	[D2] blazer_status: short reply
   5.319296	[D2] Status read 2 failed
   6.333919	[D2] blazer_status: short reply
   6.333969	[D2] Status read 3 failed
   6.333993	[D2] Trying megatec/old protocol...
   7.336315	[D2] blazer_status: short reply
   7.336361	[D2] Status read 1 failed
   8.338677	[D2] blazer_status: short reply
   8.338722	[D2] Status read 2 failed
   9.341011	[D2] blazer_status: short reply
   9.341057	[D2] Status read 3 failed
   9.341069	[D2] Trying zinto protocol...
  10.355664	[D2] blazer_status: short reply
  10.355709	[D2] Status read 1 failed
  11.358003	[D2] blazer_status: short reply
  11.358048	[D2] Status read 2 failed
  12.360362	[D2] blazer_status: short reply
  12.360409	[D2] Status read 3 failed
  12.360421	No supported UPS detected
  12.360856	upsnotify: failed to notify about state 4: no notification tech defined, will not spam more about it
```
So at this point I'm pretty sure it's a hardware issue. ViewPower on Windows also works only intermittently.

desertwitch, 2024-06-11:
I usually recommend trying with another USB cable and/or another USB port as well to rule out the easiest replacements.
Some UPS do behave better on USB 2.0 ports, for instance, and broken cables have also sometimes led to such problems.

rifkican, 2025-02-10:
I can connect to USB but I cannot enter the SET CONT PARAM field.

rifkican, 2025-02-10:
ViewPowerHTML V1.04-24215 win10 x64