networkupstools/nut #2457: usbhid-ups: Eaton input.current value is always 0

lemeshovich, 2024-05-24:
It is a continuation of the #2380 issue, just noticed that my input.current value is always 0. And as in previous issue that value is present in the official Eaton Intelligent Power software. And for example output.current value is present and OK. 

<details> 
<summary>upsc</summary>

```
$ upsc myups@localhost
battery.capacity: 7.00
battery.charge: 69
battery.charge.low: 20
battery.charge.restart: 0
battery.charger.status: floating
battery.energysave: no
battery.energysave.delay: 300
battery.energysave.load: 20
battery.protection: yes
battery.runtime: 1161
battery.type: PbAc
battery.voltage: 26.8
battery.voltage.nominal: 24
device.mfr: EATON
device.model: Eaton 5SC 750
device.serial: G132H10156
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: auto
driver.state: updateinfo
driver.version: 2.8.2-262-gc8d7e3c61
driver.version.data: MGE HID 1.47
driver.version.internal: 0.54
driver.version.usb: libusb-1.0.24 (API: 0x1000108)
input.current: 0.00
input.frequency: 49.9
input.frequency.nominal: 50
input.sensitivity: high
input.transfer.boost.low: 198
input.transfer.high: 264
input.transfer.low: 176
input.transfer.trim.high: 233
input.voltage: 210.5
input.voltage.nominal: 230
outlet.1.status: on
outlet.desc: Main Outlet
outlet.id: 0
outlet.switchable: no
output.current: 0.60
output.frequency: 49.9
output.frequency.nominal: 50
output.powerfactor: 0.83
output.voltage: 210.5
output.voltage.nominal: 220
ups.beeper.status: muted
ups.delay.shutdown: 20
ups.delay.start: 30
ups.efficiency: 80
ups.firmware: 01.10.0018
ups.load: 20
ups.load.high: 105
ups.mfr: EATON
ups.model: Eaton 5SC 750
ups.power: 126
ups.power.nominal: 750
ups.productid: ffff
ups.realpower: 105
ups.realpower.nominal: 525
ups.serial: G132H10156
ups.shutdown: enabled
ups.start.auto: yes
ups.start.battery: yes
ups.start.reboot: yes
ups.status: OL CHRG
ups.test.interval: 604800
ups.test.result: Done and passed
ups.timer.shutdown: -1
ups.timer.start: -1
ups.type: offline / line interactive
ups.vendorid: 0463
```

</details> 
 Eaton 5SC 750VA model
 
@arnaudquette-eaton @jimklimov

arnaudquette-eaton, 2024-06-20:
@lemeshovich as per your driver debug output in #2380, the device is actually reporting input.current as 0
> Path: UPS.PowerConverter.Input.[1].Current, Type: Feature, ReportID: 0x31, Offset: 0, Size: 16, Value: 0
> send_to_all: SETINFO input.current "0.00"

still, you mention that the official Eaton SW (which one EUC ? or?) is providing it.
If you can 1) get the value in Eaton SW and 2) provide again a driver debug, that could help

lemeshovich, 2024-06-21:
> If you can 1) get the value in Eaton SW 

just noticed that it is also 0 for me in Eaton SW
<details>
  <summary>My UPS</summary>
  <p>
    [image]
  </p>
</details>
<details>
  <summary>Someone's same UPS</summary>
  <p>
    [image]
  </p>
</details>

hmm, strange, maybe you have some ideas?

jimklimov, 2024-06-27:
Just wondering: could an ECO mode come into play here? @arnaudquette-eaton : Is this an input current of the whole UPS, or of the converter to charge the battery (e.g. if the battery is full and awaiting in standby, it would have negligible current needed to keep it charged - possibly in short bursts only after it goes down to some 90-95%)?

lemeshovich, 2024-06-27:
> Is this an input current of the whole UPS, or of the converter to charge the battery

i think it should be current of whole UPS, also here is my UPS history graph over a couple of weeks displaying input current, battery charge and charging state
<details>
  <summary>Home Assistant Graph</summary>
  <p>
    [image]
  </p>
</details>
and as seen from the graphs, im always getting 0 Input Current value regardless of battery charging state @jimklimov