networkupstools/nut #2458: APC SRV2KA support

0xjams, 2024-05-26:
Hi everyone:

This [UPS](https://www.apc.com/mx/es/product/SRV2KA/unidad-easy-ups-online-srv-de-apc-de-2000-va-y-120-v/) is shown like this when I run lsusb:

```
Bus 002 Device 002: ID 04e2:1410 Exar Corp. XR21V1410 USB-UART IC
```

I searched online and I found that the identifier above is used by Huawei models, so I tried to use that driver. But I was told [here](https://github.com/networkupstools/nut/issues/1066#issuecomment-2132227172) that it won't work.

I've also tried with a few other drivers, all of the APC-related ones on [this](https://networkupstools.org/stable-hcl.html) list, but none of them have worked.

Any ideas? I don't mind if it's experimental support.

PS: It's worth noting that there's also a discussion about this in APC's [forum](https://community.se.com/t5/APC-UPS-Data-Center-Enterprise/SRV2KI-Linux-management/td-p/400985).

hertzg, 2024-08-21:
I managed to get my SRV1KI working only with `apcupsd`

```
UPSCABLE smart
UPSTYPE apcsmart
DEVICE /dev/ttyUSB1
```

I would love to get it working with `nut` since `apcupsd` and `nut` packages on ubuntu are mutually exclusive.

jimklimov, 2024-08-21:
@hertzg : Which NUT version? FWIW, did you try a custom build of 2.8.2 or current master with `apc_modbus` driver and tweaked libmodbus to have USB support? Details on wiki (travelling now, hard to link).

hertzg, 2024-08-22:
@jimklimov I managed to get it working with `apcupsd` (`3.14.14-3build1`) not with `nut`. On `ubuntu` those packages are mutually exclusive (installing one removes the other) so I just stayed with `apcuspd`.

Tested with
* apcupsd: `3.14.14-3build1` on Ubuntu 20.04 `focal`
* apcupsd: `3.14.14-3.1build2` on Ubuntu 24.04 `noble`

in both cases the apcupsd.conf was
```
UPSCABLE smart
UPSTYPE apcsmart
```

on `focal`
```
DEVICE /dev/ttyXRUSB1
```

on `noble`
```
DEVICE /dev/ttyUSB1
```

I've also managed to get it working with the RS232 but ⚠️  You can not use both USB and RS232 at the same time, USB takes precedence and the RS232 port becomes unresponsive.

My device is `SRV1KRI`

[image]

itsCARLsanity, 2025-02-24:
Thank you for this, managed to get this working on my SRV1KI