networkupstools/nut #2465: NOTIFYCMD does not work

titithebad, 2024-06-09:
Dear all,

I am on a Rpi with bullseye release up to date. Everything works fine with NUT except NOTIFYCMD script. As I unplugged the USB cable, I have the message on my SSH connection (after doing sudo mesg y) but no mail arrived on my mailbox.

MSMTP is correctly installed and works as I tested it separately.

The notifycmd script is in /etc/nut with root:nut access. I tried when the script is in /usr/local/bin. The access defined like this :
-rwxrwxr--

How can I find why it does not work ?

Best Regards

jimklimov, 2024-06-11:
Well, `NOTIFYCMD` is something that `upsmon` runs, as `nut` probably. Is `nut` allowed to call your `msmtp` to submit the mails locally? What does your script look like in terms of mailing?

For an example from the industry, you can consult with https://github.com/networkupstools/nut/blob/master/scripts/installer/common/ipp-notifier.sh about how different auth etc. mailing options can be used. Generally this depends a lot on the mail servers and program implementations you use.

titithebad, 2024-06-11:
Hello,

Here is my script :
#!/bin/bash
EMAIL='xxx.yyy@gmail.com'
printf "Subject: NUT ALERT UPS EATON FEU  $NOTIFYTYPE\n\nUPS: $UPSNAME\r\nAlert type: $NOTIFYTYPE" | msmtp $EMAIL

with this definition : 
-rwxrwxr--  1 root nut    193 juin   9 15:08 notifycmd.sh
__
How do I know if nut user can access to msmtp ?
An other solution is to change these parameter in upsmon for example ?
 RUN_AS_USER nut

Regards

titithebad, 2024-06-22:
Hello,

It works now.

I changed in upsmon.conf the user. I put the default user and not root. The email is refused by my provider if I tried to send it from root user and not an other user (nut is not an user on my systel Linux).

Regards

closed 2024-06-22 by titithebad