networkupstools/nut #2466: Connectivity Issues with GE VH3000 UPS (Now owned by ABB)

MartinKurtz, 2024-06-10:
Hello, 

I am trying to connect a GE VH3000 UPS via USB to a raspberry pi which is supposed to be serving as a NUT server, but i am not able to find guidance on your website on which driver to use. The UPS presents as

``` Bus 001 Device 006: ID 067b:2303 Prolific Technology, Inc. PL2303 Serial Port / Mobile Phone Data Cable ```

The ups.conf is currently 

```
xretry = 5
pollinterval = 5
[VH3000]

        driver = blazer_ser
        port = /dev/ttyUSB0
        desc = "GE VH3000"
```
but has contained various drivers including nutdrv_qx, which its telling me to use

the nut.conf contains the line
``` MODE=netserver ```
 
 and the upsd.conf contains both 127.0.0.1 and my network in the form 123.123.123.* with port 3493 for both and upsmon.conf contains MONITOR VH3000@localhost 1 with credentials and master.

I have a headache, pls help

jimklimov, 2024-06-11:
I've looked in the code, and this USB vendor is not known to NUT. Typically Prolifics are used in USB-Serial dongles, so if the UPS is internally using a serial protocol - trying drivers like `blazer_ser` vs. `blazer_usb` was a good approach. Note that `nutdrv_qx` is almost unique as it can handle both serial and USB devices; most other NUT drivers are dedicated to one type of media.

Can you please try starting the NUT driver program directly, and with higher verbosity, to see what it discovers about the device or what issues it encounters (e.g. permissions for the `nut` user account to open the device)?

You can even specify all arguments via CLI while testing, to avoid going back and forth to edit the `ups.conf` file (and only write down the successful combo there), e.g. on a typical Debian/Ubuntu derived distro with drivers in `/lib/nut`:

````
:; /lib/nut/nutdrv_qx -DDDDDD -s testups -x port=/dev/ttyUSB0
````

or in case it talks actual USB protocols:

````
:; /lib/nut/nutdrv_qx -DDDDDD -s testups -x port=auto -x vendorid=067b -x productid=2303
````
to check if it talks something similar to Megatec Qx protocols (replies would start with a parenthesis and a series of numbers for voltage etc.), or give a shot to real USB HID:
````
:; /lib/nut/usbhid-ups -DDDDDD -s testups -x port=auto -x vendorid=067b -x productid=2303
````

Note that currently NUT codebase does not know of such vendor, so for USB drivers - you would need to run those drivers with a `--help` option to check which `subdriver` and/or `protocol` options they offer and provide those on command line explicitly. The `usbhid-ups` only offers the `subdriver` tuning in recent releases (NUT v2.8.1 or newer).

Also try adding `-x user=root` while testing (and start the driver tests as `root`, perhaps via `sudo`), to make sure that devfs node ownership is not a problem when the normally running driver drops privileges to `nut` or `ups`. If those permissions are indeed the problem, you can keep `user=root` in `ups.conf` section for the device, or find a way with your OS to make it remember such permissions across reboots and re-plugs of USB devices (e.g. using `udev` or `upower` subsystems on recent Linux).

MartinKurtz, 2024-06-11:
Hi,

none of the 3 options you provided made a successful connection and all 3 complained about lacking permissions.
the results of blazer_ser were that it tried all 4 available protocols, all timed out, for blazer_usb they were similar to the second option you gave.

The NUT compatibility list on the website does know the vendor, that being GE Digital Energy, but it doesnt list the VH series of UPS

Number one without root being explicitly stated to the driver:
```
root@raspberrypi:/home/martin# :; /lib/nut/nutdrv_qx -DDDDDD -s testups -x port=/dev/ttyUSB0
Network UPS Tools - Generic Q* USB/Serial driver 0.32 (2.8.0)
USB communication driver (libusb 1.0) 0.43
   0.000000     [D3] main_arg: var='port' val='/dev/ttyUSB0'
   0.000187     [D5] send_to_all: SETINFO driver.parameter.port "/dev/ttyUSB0"
   0.000281     [D1] debug level is '6'
   0.002461     [D5] send_to_all: SETINFO device.type "ups"
   0.002606     [D1] upsdrv_initups...

Unable to open /dev/ttyUSB0: Permission denied

  Current user id: nut (109)
Serial port owner: root (0)
Serial port group: dialout (20)
     Mode of port: 0660

Things to try:

 - Use another port (with the right permissions)

 - Fix the port owner/group or permissions on this port

 - Run this driver as another user (upsdrvctl -u or 'user=...' in ups.conf).
   See upsdrvctl(8) and ups.conf(5).

   0.003957     Fatal error: unusable configuration

```

and then with root

```
root@raspberrypi:/home/martin# :; /lib/nut/nutdrv_qx -u root -DDDDDD -s testups -x port=/dev/ttyUSB0
Network UPS Tools - Generic Q* USB/Serial driver 0.32 (2.8.0)
USB communication driver (libusb 1.0) 0.43
   0.000000     [D3] main_arg: var='port' val='/dev/ttyUSB0'
   0.000189     [D5] send_to_all: SETINFO driver.parameter.port "/dev/ttyUSB0"
   0.000300     [D1] debug level is '6'
   0.002464     [D5] send_to_all: SETINFO device.type "ups"
   0.002618     [D1] upsdrv_initups...
   0.107894     [D3] send: 'QGS'
   1.109219     [D3] read: timeout (0)
   1.109377     [D2] qx_process_answer: short reply (input.voltage)
   1.109918     [D3] send: 'QGS'
   2.111153     [D3] read: timeout (0)
   2.111344     [D2] qx_process_answer: short reply (input.voltage)
   2.111673     [D3] send: 'QGS'
   3.112941     [D3] read: timeout (0)
   3.113114     [D2] qx_process_answer: short reply (input.voltage)
   3.113467     [D3] send: 'M'
   4.114695     [D3] read: timeout (0)
   4.114841     [D2] qx_process_answer: short reply (ups.firmware.aux)
   4.115338     [D3] send: 'M'
   5.116542     [D3] read: timeout (0)
   5.116690     [D2] qx_process_answer: short reply (ups.firmware.aux)
   5.117121     [D3] send: 'M'
   6.118365     [D3] read: timeout (0)
   6.118503     [D2] qx_process_answer: short reply (ups.firmware.aux)
   6.118953     [D3] send: 'M'
   7.120190     [D3] read: timeout (0)
   7.120341     [D2] qx_process_answer: short reply (ups.firmware.aux)
[... 495 lines cut ...]
   0.054259     [D2] - Bus: 001
   0.054524     [D2] - Device: unknown
   0.054694     [D2] - Device release number: 0bb3
   0.054839     [D2] Trying to match device
   0.054934     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.055085     [D2] match_function_subdriver (non-SHUT mode): failed to match a subdriver to vendor and/or product ID
   0.055492     [D2] Device does not match - skipping
   0.055763     [D2] Checking device 5 of 5 (1D6B/0002)
   0.057593     [D2] - VendorID: 1d6b
   0.057866     [D2] - ProductID: 0002
   0.058063     [D2] - Manufacturer: Linux 6.1.21-v8+ dwc_otg_hcd
   0.058149     [D2] - Product: DWC OTG Controller
   0.058320     [D2] - Serial Number: 3f980000.usb
   0.058767     [D2] - Bus: 001
   0.058924     [D2] - Device: unknown
   0.059008     [D2] - Device release number: 0601
   0.059092     [D2] Trying to match device
   0.059237     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.059396     [D2] match_function_subdriver (non-SHUT mode): failed to match a subdriver to vendor and/or product ID
   0.059495     [D2] Device does not match - skipping
   0.059701     [D2] libusb1: No appropriate HID device found
   0.059960     libusb1: Could not open any HID devices: insufficient permissions on everything
   0.060270     No matching HID UPS found
   
   ```

MartinKurtz, 2024-06-11:
Curiously enough, i did some further attempts with the blazer drivers as well, and after looking at the output, what strikes me, is that these drivers dont seem to attempt to connect to the UPS, no matter with what options ive tried. i suspect this is because the device shows up as Unknown, but i am not to knowledgeable on the topic

```
Network UPS Tools - Megatec/Q1 protocol USB driver 0.14 (2.8.0)
   0.000000     [D3] main_arg: var='port' val='auto'
   0.000237     [D5] send_to_all: SETINFO driver.parameter.port "auto"
   0.000454     [D3] main_arg: var='vendorid' val='067b'
   0.000569     [D5] send_to_all: SETINFO driver.parameter.vendorid "067b"
   0.000823     [D3] main_arg: var='productid' val='2303'
   0.001046     [D5] send_to_all: SETINFO driver.parameter.productid "2303"
   0.001148     [D1] debug level is '6'
   0.003206     [D5] send_to_all: SETINFO device.type "ups"
   0.044866     [D2] Checking device 1 of 5 (067B/2303)
   0.048633     [D2] - VendorID: 067b
   0.048876     [D2] - ProductID: 2303
   0.049988     [D2] - Manufacturer: Prolific Technology Inc.
   0.050714     [D2] - Product: USB-Serial Controller
   0.050816     [D2] - Serial Number: unknown
   0.050913     [D2] - Bus: 001
   0.051008     [D2] - Device: unknown
   0.051119     [D2] - Device release number: 0300
   0.051216     [D2] Trying to match device
   0.051312     [D3] match_function_regex: matching a device...
   0.051898     [D2] Device does not match - skipping
```

jimklimov, 2024-06-11:
Thinking of it, probably the direct USB driver connection fails because the `ttyUSB` provided by the kernel driver took over that device first. Well, was worth trying...

https://library.industrialsolutions.abb.com/publibrary/checkout/OPM_VHU208VUL?TNR=Installation%20and%20Instruction%7COPM_VHU208VUL%7CPDF&filename=GE_UPS_OPM_VHU_208_2K0_3K0_XUS_V010.pdf Chapter 5 details the use of "The RJ11 port is a plug-in contact interface port" and the vague "UPS software required ... For more information please refer to the user manual that comes with the interface software. We strongly recommend to use only original GE software products in combination with the interface port." regarding the USB port.

Probably the device came from a different OEM or team than those talking already known protocols (blazer => megatec)...

MartinKurtz, 2024-06-11:
yeah, ive got that software and also the serial number to use it, but it doesnt run on arm, e.g. a raspi and it also doesnt seem to be able to shut everything down in the was i need it to. the software is this one https://www.generex.de/de/products/software/upsman.

So if possible at all, id like to get nut working with it

jimklimov, 2024-06-11:
Well, at least it is something to try searching for (if anyone deciphered the protocol devices use with UPSMan so it can be used with NUT too), but so far I googled up no hits :(

Still, as a paying customer you may have a better standing to request the protocol documentation, at least.

As for the "Device: unknown" -- it was a problem fixed in I think NUT v2.8.1, so you can try https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests with the current codebase to see if it helps. This is however the part where the manually specified `subdriver` options can come in handy, so the NUT driver would not discard the variants just because they do not have those so-far-unknown identifiers built in.

But the lack of `Q1` etc. responses in one of the tests gives little hope that their protocol is Megatec Qx series compatible :\

jimklimov, 2024-06-11:
I gather that a few drivers were made by listening on the wire for protocol communications between vendor tools and device, and making some sense of that. Unfortunately, I've never gone that deep into the rabbit hole myself, so can't help specifically - but maybe the mailing list members can.

MartinKurtz, 2024-06-11:
Hm, i suppose i could borrow a protocol analyzer from my local computer museum and see what goes, ive got a baud rate and a serial something, but i dont know what the hell i am doing when going further than that, let alone writing drivers

jimklimov, 2024-06-11:
I think it might not even have to be something from a museum. From what I gather, if there's an existing software for Windows or Linux, earlier similar efforts used a variant of WireShark with USB sniffing capability (not sure if they do serial... but then some VM with a virtual serial port as a man in the middle can do). Not sure where a serial-over-USB falls here for sniffing - perhaps serial (if this device is indeed such a serial protocol using an USB plug).

This way they recorded the bits from the wire during communications initialization, readings collection, perhaps some commands (calibration, shutdown...) so there can be educated guesses made for the protocol.

The number one option is getting the protocol spec officially - and maybe as a customer you can convince them you are entitled to that? :)

MartinKurtz, 2024-06-12:
Worth a try, but getting to a human being that knows what the hell i want from them is going to be a challenge, i will keep you updated

nan0desu, 2024-06-22:
I have a GE VH-series too and facing the same problem. Tried fabula subdriver: /lib/nut/nutdrv_qx -DDDDDD -s testups -x port=auto -x vendorid=067b -x productid=2303 -x user=root -x subdriver=fabula [fabula.log](https://github.com/user-attachments/files/15937963/fabula.log)

jimklimov, 2024-06-24:
Posted a question on the mailing list, maybe someone would have a guess about the protocol (or how to implement some).

jimklimov, 2024-07-08:
Great news, thanks! If they could also be enticed into coding a NUT driver, for eternal kudos in `acknowledgements.txt` as well as a high level of device support in NUT HCL (also can be a bonus for sales), that would be great!

MartinKurtz, 2024-07-11:
I am not sure it will get that far